avrdudes/avrdude #1107: linuxspi not available when compiled (aarch64)

seandepagnier, 2022-10-02:
I have compiled avrdude from source because linuxspi is completely missing now from the prebuild binary in raspbian. 

  During configuration of the build it shows ENABLED for both linuxgpio and linuxspi

I get the following output from avrdude -c ?type
`
Valid programmer types are:
  arduino          = Arduino programmer
  avr910           = Serial programmers using protocol described in application note AVR910
  avrftdi          = Interface to the MPSSE Engine of FTDI Chips using libftdi.
  buspirate        = Using the Bus Pirate's SPI interface for programming
  buspirate_bb     = Using the Bus Pirate's bitbang interface for programming
  butterfly        = Atmel Butterfly evaluation board; Atmel AppNotes AVR109, AVR911
  butterfly_mk     = Mikrokopter.de Butterfly
  dragon_dw        = Atmel AVR Dragon in debugWire mode
  dragon_hvsp      = Atmel AVR Dragon in HVSP mode
  dragon_isp       = Atmel AVR Dragon in ISP mode
  dragon_jtag      = Atmel AVR Dragon in JTAG mode
  dragon_pdi       = Atmel AVR Dragon in PDI mode
  dragon_pp        = Atmel AVR Dragon in PP mode
  flip1            = FLIP USB DFU protocol version 1 (doc7618)
  flip2            = FLIP USB DFU protocol version 2 (AVR4023)
  ftdi_syncbb      = FT245R/FT232R Synchronous BitBangMode Programmer
  jtagmki          = Atmel JTAG ICE mkI
  jtagmkii         = Atmel JTAG ICE mkII
  jtagmkii_avr32   = Atmel JTAG ICE mkII in AVR32 mode
  jtagmkii_dw      = Atmel JTAG ICE mkII in debugWire mode
  jtagmkii_isp     = Atmel JTAG ICE mkII in ISP mode
  jtagmkii_pdi     = Atmel JTAG ICE mkII in PDI mode
  jtagmkii_updi    = Atmel JTAG ICE mkII in UPDI mode
  jtagice3         = Atmel JTAGICE3
  jtagice3_pdi     = Atmel JTAGICE3 in PDI mode
  jtagice3_updi    = Atmel JTAGICE3 in UPDI mode
  jtagice3_dw      = Atmel JTAGICE3 in debugWire mode
  jtagice3_isp     = Atmel JTAGICE3 in ISP mode
  linuxgpio        = GPIO bitbanging using the Linux sysfs interface
  linuxspi         = SPI using Linux spidev driver
  micronucleus     = Micronucleus Bootloader
  par              = Parallel port bitbanging
  pickit2          = Microchip's PICkit2 Programmer
  serbb            = Serial port bitbanging
  serialupdi       = Driver for SerialUPDI programmers
  stk500           = Atmel STK500 Version 1.x firmware
  stk500generic    = Atmel STK500, autodetect firmware version
  stk500v2         = Atmel STK500 Version 2.x firmware
  stk500hvsp       = Atmel STK500 V2 in high-voltage serial programming mode
  stk500pp         = Atmel STK500 V2 in parallel programming mode
  stk600           = Atmel STK600
  stk600hvsp       = Atmel STK600 in high-voltage serial programming mode
  stk600pp         = Atmel STK600 in parallel programming mode
  teensy           = Teensy Bootloader
  usbasp           = USBasp programmer, see http://www.fischl.de/usbasp/
  usbtiny          = Driver for "usbtiny"-type programmers
  wiring           = http://wiring.org.co/, Basically STK500v2 protocol, with some glue to trigger the bootloader.
  xbee             = XBee Series 2 Over-The-Air (XBeeBoot)

`

When I execute avrdude -c linuxspi:

`
avrdude: cannot find programmer id linuxspi

Valid programmers are:
  2232HIO            = FT2232H based generic programmer
  4232h              = FT4232H based generic programmer
  adafruit_gemma     = Adafruit Trinket Gemma bootloader disguised as USBtiny
  arduino            = Arduino for bootloader using STK500 v1 protocol
  arduino-ft232r     = Arduino: FT232R connected to ISP
  arduino_gemma      = Arduino Gemma bootloader disguised as USBtiny
[... 86 lines cut ...]
  stk500v1           = Atmel STK500 version 1.x firmware
  stk500v2           = Atmel STK500 version 2.x firmware
  stk600             = Atmel STK600
  stk600hvsp         = Atmel STK600 in high-voltage serial programming mode
  stk600pp           = Atmel STK600 in parallel programming mode
  tc2030             = Tag-Connect TC2030
  teensy             = Teensy for bootloader
  ttl232r            = FTDI TTL232R-5V with ICSP adapter
  tumpa              = TIAO USB Multi-Protocol Adapter
  um232h             = UM232H module from FTDI
  uncompatino        = uncompatino with all pairs of pins shorted
  usbasp             = USBasp, http://www.fischl.de/usbasp/
  usbasp-clone       = Any usbasp clone with correct VID/PID
  usbtiny            = USBtiny simple USB programmer, https://learn.adafruit.com/usbtinyisp
  wiring             = Wiring for bootloader using STK500 v2 protocol
  xbee               = XBee for Series 2 Over-The-Air (XBeeBoot) bootloader using STK500 v1 protocol
  xplainedmini       = Atmel AVR XplainedMini in ISP mode
  xplainedmini_dw    = Atmel AVR XplainedMini in debugWIRE mode
  xplainedmini_updi  = Atmel AVR XplainedMini in UPDI mode
  xplainedpro        = Atmel AVR XplainedPro in JTAG mode
  xplainedpro_updi   = Atmel AVR XplainedPro in UPDI mode

`

A completely different list with many more entrees but no linuxspi.    How can I get linuxspi support?

mcuee, 2022-10-03:
Please check the contents of your `avrdude.conf` to see if linuxspi is included or not. And you need to check the reset pin configuration is correct or not.

Reference:
https://github.com/avrdudes/avrdude/blob/main/src/avrdude.conf.in#L706_L725

seandepagnier, 2022-10-03:
I confirm that linuxspi is not in the avrdude.conf generated.

I built on a raspberry pi.  When running build.sh it reports:
`ENABLED linuxgpio
ENABLED linuxspi
`
grep linux build_linux/src/avrdude.conf*
gives no results

if I do
`
mkdir build
cd build
cmake ..
`
It reports
`
DISABLED linuxgpio
DISABLED linuxspi
`

mcuee, 2022-10-03:
@seandepagnier 

It seems to me there is a bug with latest git. Try 7.0 release and it should work.

```
mcuee@rpi400:~/build $ cd avrdude-7.0/

mcuee@rpi400:~/build/avrdude-7.0 $ ls
AUTHORS  CMakeLists.txt  COPYING  INSTALL  NEWS  README.md  atmel-docs  build.sh  src  tools

mcuee@rpi400:~/build/avrdude-7.0 $ cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -D HAVE_LINUXGPIO=ON -D HAVE_LINUXSPI=ON -B build_linux
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.30.2")
-- Found FLEX: /usr/bin/flex (found version "2.6.4")
-- Found BISON: /usr/bin/bison (found version "3.7.5")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- ENABLED    linuxgpio
-- ENABLED    linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mcuee/build/avrdude-7.0/build_linux
mcuee@rpi400:~/build/avrdude-7.0 $ cmake --build build_linux/
[  1%] [BISON][Parser] Building parser with bison 3.7.5
[  3%] [FLEX][Parser] Building scanner with flex 2.6.4
Scanning dependencies of target libavrdude
[  4%] Building C object src/CMakeFiles/libavrdude.dir/arduino.c.o
[  6%] Building C object src/CMakeFiles/libavrdude.dir/avr.c.o
[  8%] Building C object src/CMakeFiles/libavrdude.dir/avr910.c.o
[  9%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi.c.o
[ 11%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.o
[ 13%] Building C object src/CMakeFiles/libavrdude.dir/avrpart.c.o
[ 14%] Building C object src/CMakeFiles/libavrdude.dir/bitbang.c.o
[ 16%] Building C object src/CMakeFiles/libavrdude.dir/buspirate.c.o
[... 59 lines cut ...]
[ 83%] Building C object src/CMakeFiles/libavrdude.dir/usbtiny.c.o
[ 85%] Building C object src/CMakeFiles/libavrdude.dir/update.c.o
[ 86%] Building C object src/CMakeFiles/libavrdude.dir/wiring.c.o
[ 88%] Building C object src/CMakeFiles/libavrdude.dir/xbee.c.o
[ 90%] Building C object src/CMakeFiles/libavrdude.dir/__/lexer.c.o
[ 91%] Building C object src/CMakeFiles/libavrdude.dir/__/config_gram.c.o
[ 93%] Linking C static library libavrdude.a
[ 93%] Built target libavrdude
Scanning dependencies of target avrdude
[ 95%] Building C object src/CMakeFiles/avrdude.dir/main.c.o
[ 96%] Building C object src/CMakeFiles/avrdude.dir/term.c.o
[ 98%] Building C object src/CMakeFiles/avrdude.dir/whereami.c.o
[100%] Linking C executable avrdude
[100%] Built target avrdude
mcuee@rpi400:~/build/avrdude-7.0 $ grep linux build_linux/src/avrdude.conf*
#To check if your avrdude build has support for the linuxgpio programmer compiled in,
#use -c?type on the command line and look for linuxgpio in the list. If it's not available
#you need pass the --enable-linuxgpio=yes option to configure and recompile avrdude.
#  id    = "linuxgpio";
#  type  = "linuxgpio";
# This programmer uses the built in linux SPI bus devices to program an
   id = "linuxspi";
   type = "linuxspi";

```

mcuee, 2022-10-03:
For 7.0 release, `build.sh` also works fine.

```
mcuee@rpi400:~/build/avrdude-7.0 $ ./build.sh
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.30.2")
-- Found FLEX: /usr/bin/flex (found version "2.6.4")
-- Found BISON: /usr/bin/bison (found version "3.7.5")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- ENABLED    linuxgpio
-- ENABLED    linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mcuee/build/avrdude-7.0/build_linux
[  1%] [BISON][Parser] Building parser with bison 3.7.5
[  3%] [FLEX][Parser] Building scanner with flex 2.6.4
Scanning dependencies of target libavrdude
[  4%] Building C object src/CMakeFiles/libavrdude.dir/arduino.c.o
[  6%] Building C object src/CMakeFiles/libavrdude.dir/avr.c.o
[  8%] Building C object src/CMakeFiles/libavrdude.dir/avr910.c.o
[  9%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi.c.o
[ 11%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.o
[ 13%] Building C object src/CMakeFiles/libavrdude.dir/avrpart.c.o
[ 14%] Building C object src/CMakeFiles/libavrdude.dir/bitbang.c.o
[ 16%] Building C object src/CMakeFiles/libavrdude.dir/buspirate.c.o
[ 18%] Building C object src/CMakeFiles/libavrdude.dir/butterfly.c.o
[ 19%] Building C object src/CMakeFiles/libavrdude.dir/config.c.o
[ 21%] Building C object src/CMakeFiles/libavrdude.dir/confwin.c.o
[ 22%] Building C object src/CMakeFiles/libavrdude.dir/crc16.c.o
[ 24%] Building C object src/CMakeFiles/libavrdude.dir/dfu.c.o
[ 26%] Building C object src/CMakeFiles/libavrdude.dir/fileio.c.o
[ 27%] Building C object src/CMakeFiles/libavrdude.dir/flip1.c.o
[ 29%] Building C object src/CMakeFiles/libavrdude.dir/flip2.c.o
[... 60 lines cut ...]
[ 95%] Building C object src/CMakeFiles/avrdude.dir/main.c.o
[ 96%] Building C object src/CMakeFiles/avrdude.dir/term.c.o
[ 98%] Building C object src/CMakeFiles/avrdude.dir/whereami.c.o
[100%] Linking C executable avrdude
[100%] Built target avrdude

Build succeeded.

Run

sudo cmake --build build_linux --target install

to install.

mcuee@rpi400:~/build/avrdude-7.0 $ grep linux build_linux/src/avrdude.conf*
#To check if your avrdude build has support for the linuxgpio programmer compiled in,
#use -c?type on the command line and look for linuxgpio in the list. If it's not available
#you need pass the --enable-linuxgpio=yes option to configure and recompile avrdude.
#  id    = "linuxgpio";
#  type  = "linuxgpio";
# This programmer uses the built in linux SPI bus devices to program an
   id = "linuxspi";
   type = "linuxspi";

```

mcuee, 2022-10-03:
Now we need to find out which commit is causing the issue.

I can confirm that https://github.com/avrdudes/avrdude/commit/07f1f60020525aa9023cd2c4f17b82f32296dde1 is still okay.

mcuee, 2022-10-03:
@mariusgreuel This points to CMake change problem in https://github.com/avrdudes/avrdude/commit/77430519266f7e7a7f9caae9f2c7a3869c17cdba , please take a look. Thanks.

mariusgreuel, 2022-10-03:
> This points to CMake change problem in https://github.com/avrdudes/avrdude/commit/77430519266f7e7a7f9caae9f2c7a3869c17cdba , please take a look.

You are correct, thank you for figuring that out. Seems like there is a problem with verbal switches like `ON` and `OFF`.

@seandepagnier As a temporary workaround, please use the procedure at https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Linux and replace `ON` with `1`.

seandepagnier, 2022-10-03:
Thank you for taking a look at this!    I managed to get a working version.

 Since this is understood  I trust it will be fixed and can close this.

mcuee, 2022-10-16:
I will keep this open until it is fixed.

seandepagnier, 2022-10-18:
Sorry about (aarch64) in the title, I thought it was relevant but is not.  Not sure how to change the title.

 debian packages include avrdude without linuxspi support now.. this should stay open until linuxspi is built without manually specifying  flags to cmake

mcuee, 2022-10-18:
We can not control how Debian packages avrdude. Usually I build the packages I am interested from the source (libusb, hidapi, libftdi, openocd, avrdude, etc). 

I think the easiest way to build avrdude is to use `build.sh`. CMake flags will always be needed.

seandepagnier, 2022-10-18:
I realize you don't control debian builds, but they simply use the defaults without specifying any special flags which means linuxspi doesn't work anymore.

Of course developers of avrdude will build from source even some dependencies from source, but the majority of users do not.

mcuee, 2022-12-27:
Wiki updated to use `1` instead of `ON`.

mcuee, 2022-12-27:
PR #1240 has been merged as a work-around for this issue. I will still keep this issue open.

mcuee, 2022-12-31:
But this can be a low priority now since the Wiki and `build.sh` have been updated.

mcuee, 2023-01-03:
#1261 may be a similar issue.

mcuee, 2023-01-07:
@mariusgreuel 

Do you still want to work on this or we can close this issue? Thanks.

seandepagnier, 2023-01-11:
I tested with the new build.sh and it works,  but invoking make directly does not.    Wouldn't it be preferable to work without build.sh?

mcuee, 2023-01-12:
> I tested with the new build.sh and it works, but invoking make directly does not. Wouldn't it be preferable to work without build.sh?

I am not so sure what you mean by "invoking directly"?

You can create your own Makefile to replace `build.sh` if you like.

seandepagnier, 2023-01-12:
When I first cloned avrdude I noticed CMakeLists.txt so I run cmake directly subconsciously... 

So my invoking directly I mean just running cmake without passing -D HAVE_LINUXSPI=1 instead making it detect I have /dev/spi automatically?

mcuee, 2023-01-13:
> When I first cloned avrdude I noticed CMakeLists.txt so I run cmake directly subconsciously...
> 
> So my invoking directly I mean just running cmake without passing -D HAVE_LINUXSPI=1 instead making it detect I have /dev/spi automatically?

No it will not detect `/dev/spi` directly and you have to pass `-D HAVE_LINUXSPI=1` option to CMake.

PR #1285 (2023-03-04): CMake: Convert verbal switches into boolean expressions
Fixes #1107
closed 2023-03-04 by stefanrueger