avrdudes/avrdude #1114: Quell mode isn't handled systematically

MCUdude, 2022-10-07:
This is highly related to #1084.

Some messages are not being muted when in quell mode (-qq).

The reason for this is that not all messages take quell mode into account. Here's one that does:
https://github.com/avrdudes/avrdude/blob/3e49f078b3ffc8c0e647fc3600da8ea69e47e487/src/main.c#L1054-L1057

Here's the current `avrdude_message` implementation. Note that it only takes `verbose` into account, not quell.
https://github.com/avrdudes/avrdude/blob/44fe5bec2dd2f016d135f0d2f02173045d7aeef5/src/main.c#L63-L73

IMO, `avrdude_message` should be modified to handle quell directly, so we don't have to "manually add guards" around every avrdude_message function call.

stefanrueger, 2022-10-07:
I'd like to get @dl8dtl's guidance on this.

As far as I can tell from reading the code (and git history) `-q` started out as being only concerned with progress reports (the variable is called `quell_progress`). This is a completely independent option to `-v`. Later, in 2016, `-qq` was introduced as a way to condition (some) outputs with cumbersome `if(quell_progress < 2)` lines. 

I think `-qq` should only stop all _normal_ output, not least because suppressing all output would be a radical departure from past behaviour. From my observations, the `MSG_INFO` level has three functions: a) genuine error messages that the user should really know about when things go wrong; b) warnings, that may or may not be useful for the user and c) info messages, a sort of running commentary, that are probably less vital for the user to know about

Here my proposal: `-qq` to stop the running commentary, `-qqq` to also stop warnings, `-qqqq` to stop all output.

My suggestion to implement this is to split the `MSG_INFO` into three levels: `MSG_ERROR` (new level -2), `MSG_WARNING` (new level -1) and `MSG_INFO` (old level 0) and go through the code to distinguish them. Then add the following line at the beginning of  `avrdude_message()`:
```
    if(quell_progress > 1)
         msglvl += quell_progress-1;
```
Technically, each additional `-q` after the first one reduces the effective verbosity by `1`.

A word of warning, though: avrdude is vvv good at informing the user:
```
$ for i in $(grep MSG_ avrdude.h | cut -d\  -f2); do echo $(grep -r $i | wc -l) $i; done
1562 MSG_INFO
386 MSG_NOTICE
241 MSG_NOTICE2
242 MSG_DEBUG
96 MSG_TRACE
23 MSG_TRACE2
```

If we think this is a good proposal we would need to review 1561 messages! Any volunteers?

I have one more suggestion: How about adding the following macros in `avrdude.h`
```
#define msg_error(...)   avrdude_message(MSG_ERROR, __VA_ARGS__)
#define msg_warning(...) avrdude_message(MSG_WARNING, __VA_ARGS__)
#define msg_info(...)    avrdude_message(MSG_INFO, __VA_ARGS__)
#define msg_notice(...)  avrdude_message(MSG_NOTICE, __VA_ARGS__)
#define msg_notice2(...) avrdude_message(MSG_NOTICE2, __VA_ARGS__)
#define msg_debug(...)   avrdude_message(MSG_DEBUG, __VA_ARGS__)
#define msg_trace(...)   avrdude_message(MSG_TRACE, __VA_ARGS__)
#define msg_trace2(...)  avrdude_message(MSG_TRACE2, __VA_ARGS__)
```

These suggestions taken together allow us to morph the four-line message
https://github.com/avrdudes/avrdude/blob/3e49f078b3ffc8c0e647fc3600da8ea69e47e487/src/main.c#L1054-L1057
into one line
```
  msg_info("%s: AVR device initialized and ready to accept instructions\n",  progname);
```

Replacing the long `avrdude_message(MSG_..., )` lines by 18-byte shorter ones can be achieved painlessly by
```
$  sed -i "s/avrdude_message *( *\(MSG_[A-Z_0-9]*\) *, */\L\1(/g"  *.[chyl]
```

I have made a headstart into this with PR #1117 that still requires the review of the `msg_info()` messages to keep the important ones alive in `-qq`

MCUdude, 2022-10-10:
Thanks for taking your time and looking into it!

> Here my proposal: -qq to stop the running commentary, -qqq to also stop warnings, -qqqq to stop all output.

This sounds like a good solution to me. however, I don't see why we need to add the `msg` macros. I do agree that using these macros instead of `avrdude_message` results in less typing, but is it strictly necessary to achieve what we want to achieve with PR #1117?

stefanrueger, 2022-10-10:
What I'm after is a reclassification of 1500+ msg_info() messages into finer categories. This for two reasons: one for the ability to progressively dampen avdude output and two for use of the code in libavrdude. There, the library user needs provide their own avrdude_message(), which is not part of the library, most likely to set a string that can be examined by the caller. With finer granularity of message levels they can write a better informed error handling module. 

The msg_xyz() macros are a bycatch. Whenever I write code for avrdude, I feel I've almost forgotten what I wanted to output once I have finished typing avrdude_message(MSG_INFO, ...). It takes a quarter of the line and encourages to spend another line, ...

MCUdude, 2022-10-10:
> I feel I've almost forgotten what I wanted to output once I have finished typing avrdude_message(MSG_INFO, ...)

Fair enough, it has happened to me a couple of times too. I'll give the PR a try very soon!

dl8dtl, 2022-10-11:
> IMO, `avrdude_message` should be modified to handle quell directly, so we don't have to "manually add guards" around every avrdude_message function call.

Well, that might be a good idea.

`avrdude_message` has been created during the library separation step in order to catch all these

```
   if (verbose > XXX)
      fprintf(stderr, "...")
```
constructs, and in particular in order to allow library users to perform other things than sending messages of various severity levels just to `stderr`. The idea is that for a potential GUI application, dumping messages to `stderr` makes no sense at all. Instead, they might e.g. present all messages up to `INFO` level in a text widget, but pop up modal windows for `WARNING` or `ERROR` level messages.
`quell_progress` used to be handled quite differently, and indeed, was really only meant to handle progress reporting. This is separate from normal messages as a potential GUI application might want to handle that in a completely different manner, typically like filling up a bar in a status line or something like that.
Maybe it's really a good idea to extend the severity levels into the other (less important) direction to get less information printed. Then, `-q` would just suppress progress indication, and higher `-qq` levels gradually increase the severity limit used by `avrdude_message`.
It's questionable to me though whether the CLI application really needs to handle suppressing *all* messages: on the shell, I'd normally do that with `2> /dev/null` instead.

closed 2022-10-17 by stefanrueger