avrdudes/avrdude #1116: jtag2updi can not write EEPROM in terminal mode

mcuee, 2022-10-07:
Reference:
* https://github.com/avrdudes/avrdude/pull/1113#issuecomment-1271647912

Interestingly for my Arduino Nano ATmega4808, it can not write EEPROM in terminal mode. 

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1113 -qqp atmega4808 -c jtag2updi -P COM13 -b115200 
-U eeprom:w:.\hex2\entest_128B.eep:i && echo OK
avrdude_pr1113.exe: jtagmkII_paged_write(): bad response to write memory command: RSP_ILLEGAL_MEMORY_TYPE
OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1112 -p atmega4808 -c jtag2updi -P COM13 -b115200 -qqt
avrdude> dump eeprom 0x0 0x10
>>> dump eeprom 0x0 0x10
0000  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |

avrdude> write eeprom 0x0 0xff 0xff 0xff 0xff
>>> write eeprom 0x0 0xff 0xff 0xff 0xff
avrdude> quit
>>> quit
avrdude_pr1112.exe: synching cache to device... avrdude_pr1112.exe: jtagmkII_paged_write(): 
bad response to write memory command: RSP_ILLEGAL_MEMORY_TYPE
writeCachePage() eeprom write error at addr 0x0000

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1113 -qqp atmega4808 -c jtag2updi -P COM13 -b115200 
-U eeprom:w:.\hex2\detest_128B.eep:i && echo OK
avrdude_pr1113.exe: jtagmkII_paged_write(): bad response to write memory command: RSP_ILLEGAL_MEMORY_TYPE
OK

mcuee, 2022-10-07:
It is the same for git main (just to make sure).
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -p atmega4808 -c jtag2updi -P COM13 -b115200 -qqt
avrdude> dump eeprom 0x0 0x10
>>> dump eeprom 0x0 0x10
0000  5a 20 6a 61 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |Z jagt im komple|

avrdude> write eeprom 0x0 0xff 0xff 0xff 0xff
>>> write eeprom 0x0 0xff 0xff 0xff 0xff
avrdude> quit
>>> quit
avrdude_git.exe: synching cache to device... avrdude_git.exe: jtagmkII_paged_write(): 
bad response to write memory command: RSP_ILLEGAL_MEMORY_TYPE
writeCachePage() eeprom write error at addr 0x0000

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -qqp atmega4808 -c jtag2updi -P COM13 -b115200
 -U eeprom:w:.\hex2\entest_128B.eep:i && echo OK
avrdude_git.exe: jtagmkII_paged_write(): bad response to write memory command: RSP_ILLEGAL_MEMORY_TYPE
OK

```

stefanrueger, 2022-10-07:
@mcuee Can you write eeprom with `-U`? 
```
echo "The quick brown fox" | avrdude_git -p atmega4808 -c jtag2updi -P COM13 -b115200 -qqU eeprom:w:-r
```

If yes, I might need to tinker with terminal writes (ie, fix the now-merged PR #1106).

mcuee, 2022-10-08:
> @mcuee Can you write eeprom with `-U`?
> 
> ```
> echo "The quick brown fox" | avrdude_git -p atmega4808 -c jtag2updi -P COM13 -b115200 -qqU eeprom:w:-r
> ```
> 
> If yes, I might need to tinker with terminal writes (ie, fix the now-merged PR #1106).

Yes `-U` works fine. 

```
xiaof@LAPTOP-913SIMUD MINGW64 /c/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22000/src
$ echo "The quick brown fox" | ./avrdude -p atmega4808 -c jtag2updi -P COM13 -b115200 -qqU eeprom:w:-:r
avrdude.exe: jtagmkII_paged_write(): bad response to write memory command: RSP_ILLEGAL_MEMORY_TYPE

xiaof@LAPTOP-913SIMUD MINGW64 /c/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22000/src
$ ./avrdude -p atmega4808 -c jtag2updi -P COM13 -b115200 -qq -t
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |

avrdude> quit
>>> quit

xiaof@LAPTOP-913SIMUD MINGW64 /c/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22000/src
$ echo "Hello world!" | ./avrdude -p atmega4808 -c jtag2updi -P COM13 -b115200 -qqU eeprom:w:-:r
avrdude.exe: jtagmkII_paged_write(): bad response to write memory command: RSP_ILLEGAL_MEMORY_TYPE

xiaof@LAPTOP-913SIMUD MINGW64 /c/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22000/src
$ ./avrdude -p atmega4808 -c jtag2updi -P COM13 -b115200 -qq -t
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  48 65 6c 6c 6f 20 77 6f  72 6c 64 21 0a 77 6e 20  |Hello world! wn |

avrdude> quit
>>> quit

```

MCUdude, 2022-10-10:
I believe we discussed this in an earlier thread somewhere. It would be great to implement it, but I'm not sure this is Avrdude's fault. Maybe @eltangas knows?

mcuee, 2022-10-10:
> I believe we discussed this in an earlier thread somewhere. It would be great to implement it, but I'm not sure this is Avrdude's fault. Maybe @ElTangas knows?

We have some discussion in #1069 but the issue is different. The issue #1069 is with jtag2updi problem with AVR128DB28 or similar chips and it is not with the terminal mode.
* https://github.com/avrdudes/avrdude/issues/1069

mcuee, 2022-10-10:
@MCUdude and @stefanrueger 

It seems to me this is a regression. From #1069, it seems to work fine at that time. I will check again.
* https://github.com/avrdudes/avrdude/issues/1069#issuecomment-1217300694

MCUdude, 2022-10-10:
@mcuee I don't see how that should be possible. There have been no significant changes to jtagmkii.c in recent months.

mcuee, 2022-10-10:
> @mcuee I don't see how that should be possible. There have been no significant changes to jtagmkii.c in recent months.

Maybe it is just an error message problem. The result seems to be okay even with the error message.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -p atmega4808 -c jtag2updi -P COM13 -b115200 -qqt
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  ff ff ff ff 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> write eeprom 0x0 0xaa 0xaa 0xaa 0xaa
>>> write eeprom 0x0 0xaa 0xaa 0xaa 0xaa
avrdude> flush
>>> flush
avrdude_git.exe: synching cache to device... avrdude_git.exe: 
jtagmkII_paged_write(): bad response to write memory command: RSP_ILLEGAL_MEMORY_TYPE
writeCachePage() eeprom write error at addr 0x0000
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  aa aa aa aa 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> write eeprom 0x0 0xff 0xff 0xff 0xff
>>> write eeprom 0x0 0xff 0xff 0xff 0xff
avrdude> flush
>>> flush
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  ff ff ff ff 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> quit
>>> quit

```

The FW is known to give error response: `RSP_ILLEGAL_MEMORY_TYPE`.

Maybe the error message `writeCachePage() eeprom write error at addr 0x0000` is because of the above. I am not so sure.

You can see from the following results that avrdude 7.0 release does not have such warning messages.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude7 -C .\avrdude7.conf -p atmega4808 -c jtag2updi -P COM13 -b115200 -qqt

avrdude> dump eeprom 0x0 0x10
>>> dump eeprom 0x0 0x10
0000  ff ff ff ff 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> write eeprom 0x0 0xaa 0xaa 0xaa 0xaa
>>> write eeprom 0x0 0xaa 0xaa 0xaa 0xaa
avrdude> dump eeprom 0x0 0x10
>>> dump eeprom 0x0 0x10
0000  aa aa aa aa 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> write eeprom 0x0 0xff 0xff 0xff 0xff
>>> write eeprom 0x0 0xff 0xff 0xff 0xff
avrdude> dump eeprom 0x0 0x10
>>> dump eeprom 0x0 0x10
0000  ff ff ff ff 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> quit
>>> quit

```

mcuee, 2022-10-10:
It has nothing to do with the changes in jtagmkii.c since last change https://github.com/avrdudes/avrdude/commit/4070ca9012bbaaac92191a75dfade602ad5a7219 is still good.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_4070ca9.exe -C .\avrdude_4070ca9.conf -p atmega4808 -c jtag2updi -P COM13 -b115200 -qqt
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  ff ff ff ff 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> write eeprom 0x0 0xaa 0xaa 0xaa 0xaa
>>> write eeprom 0x0 0xaa 0xaa 0xaa 0xaa
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  aa aa aa aa 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> write eeprom 0x0 0xff 0xff 0xff 0xff
>>> write eeprom 0x0 0xff 0xff 0xff 0xff
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  ff ff ff ff 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> quit
>>> quit

```

MCUdude, 2022-10-10:
> It has nothing to do with the changes in jtagmkii.c since last change https://github.com/avrdudes/avrdude/commit/4070ca9012bbaaac92191a75dfade602ad5a7219 (20 days ago) is still good.

Thanks for looking into it. Are you able to track down which commit that's causing the issue?

mcuee, 2022-10-10:
> > It has nothing to do with the changes in jtagmkii.c since last change [4070ca9](https://github.com/avrdudes/avrdude/commit/4070ca9012bbaaac92191a75dfade602ad5a7219) (20 days ago) is still good.
> 
> Thanks for looking into it. Are you able to track down which commit that's causing the issue?

Yes I am able to track it down quickly as @stefanrueger already suspected #1106 is the potential problem. And indeed https://github.com/avrdudes/avrdude/commit/d74b17b9b4fc37f53084d4210eda4b457a4e3ef4 is the problem.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_d74b17b.exe -C .\avrdude_d74b17b.conf -p atmega4808 -c jtag2updi -P COM13 -b115200 -qqt
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  ff ff ff ff 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> write eeprom 0x0 0xaa 0xaa 0xaa 0xaa
>>> write eeprom 0x0 0xaa 0xaa 0xaa 0xaa
avrdude> flush
>>> flush
avrdude_d74b17b.exe: synching cache to device... avrdude_d74b17b.exe: jtagmkII_paged_write(): bad response to write memory command: RSP_ILLEGAL_MEMORY_TYPE
writeCachePage() eeprom write error at addr 0x0000
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  aa aa aa aa 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> write eeprom 0x0 0xff 0xff 0xff 0xff
>>> write eeprom 0x0 0xff 0xff 0xff 0xff
avrdude> flush
>>> flush
avrdude> quit
>>> quit

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_c4cb242.exe -C .\avrdude_c4cb242.conf -p atmega4808 -c jtag2updi -P COM13 -b115200 -qqt
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  ff ff ff ff 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> write eeprom 0x0 0xaa 0xaa 0xaa 0xaa
>>> write eeprom 0x0 0xaa 0xaa 0xaa 0xaa
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  aa aa aa aa 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> write eeprom 0x0 0xff 0xff 0xff 0xff
>>> write eeprom 0x0 0xff 0xff 0xff 0xff
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  ff ff ff ff 67 74 20 69  6d 20 6b 6f 6d 70 6c 65  |....gt im komple|

avrdude> quit
>>> quit

```

stefanrueger, 2022-10-10:
Yes, I know *exactly* what's been the cause for this regression. I'll merge a few PRs that are ready and then tackle this regression (easier that way and less to test).

mcuee, 2022-10-15:
Yes now PR #1121 has fixed this issue.

closed 2022-10-17 by stefanrueger