avrdudes/avrdude #1124: AVRDUDE not erasing locked UPDI device

avrfreak, 2022-10-12:
I am using avrdude with a Microchip snap debugger. 
Everything works really well with the SNAP programmer with only one exception:
If you program the lock bits, you can no longer access the target device using avrdude.

The same thing happens when using jtag2updi as the programmer. This is avrdude related and not a hardware issue. 

Here is the error log:
`
C:\Test_Snap>avrdude -C avrdude.conf -c snap_updi -p avr64db28 -P usb -B 0.2 -e -U flash:w:test.hex:a -U eeprom:w:test.eep:a
avrdude: usbhid_open(): No device found

         Vtarget                      : 3.11 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 5000 kHz

avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude:
Device is locked! Chip erase required to unlock.
avr_read(): error reading address 0x0000
    read operation failed for memory "signature"
avrdude: error reading signature data for part "AVR64DB28", rc=-68
avrdude: Received FamilyID: "AVR    "
avrdude: Expected FamilyID: ""
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.
`
to program the same device when the target is not locked is working:
`
C:\Test_Snap>avrdude -C avrdude.conf -c snap_updi -p avr64db28 -P usb -B 0.2 -e -U flash:w:test.hex:a -U eeprom:w:test.eep:a
avrdude: usbhid_open(): No device found

         Vtarget                      : 3.02 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 5000 kHz

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.07s

avrdude: Device signature = 0x1e9619 (probably avr64db28)
avrdude: erasing chip
avrdude: reading input file "test.hex"
avrdude: input file test.hex auto detected as Intel Hex
avrdude: writing flash (13036 bytes):

Writing | ################################################## | 100% 0.73s

avrdude: 13036 bytes of flash written
avrdude: verifying flash memory against test.hex:
avrdude: input file test.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.48s

avrdude: 13036 bytes of flash verified
avrdude: reading input file "test.eep"
avrdude: input file test.eep auto detected as Intel Hex
avrdude: writing eeprom (32 bytes):

Writing | ################################################## | 100% 0.16s

avrdude: 32 bytes of eeprom written
avrdude: verifying eeprom memory against test.eep:
avrdude: input file test.eep auto detected as Intel Hex

Reading | ################################################## | 100% 0.07s

avrdude: 32 bytes of eeprom verified

avrdude done.  Thank you.
`

mcuee, 2022-10-13:
Looks like `-e` does not work as per designed. Strange.

MCUdude, 2022-10-13:
Thanks for reporting @avrfreak!

I'm able to reproduce the issue, and it seems to be a problem on all UPDI parts, since, unlike classic AVRs, they don't reveal their device signature when locked.

As a temporary solution, you can add the -F flag to force erasing even though Avrdude. 

```
$ ./avrdude -cpkobn_updi -p avr64dd32 -e

         Vtarget                      : 3.31 V
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude: Device is locked! Chip erase required to unlock.
avr_read_mem(): error reading address 0x0000
    read operation failed for memory signature
avrdude: error reading signature data for part "AVR64DD32", rc=-3
avrdude: Received FamilyID: "AVR    "
avrdude: Expected FamilyID: ""
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

$ ./avrdude -cpkobn_updi -p avr64dd32 -e -F

         Vtarget                      : 3.31 V
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude: Device is locked! Chip erase required to unlock.
avr_read_mem(): error reading address 0x0000
    read operation failed for memory signature
avrdude: error reading signature data for part "AVR64DD32", rc=-3
avrdude: Received FamilyID: "AVR    "
avrdude: Expected FamilyID: ""
avrdude: erasing chip

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e961a (probably avr64dd32)

avrdude done.  Thank you.

```

MCUdude, 2022-10-13:
However, there is a pattern here we can look for to determine that we're dealing with a locked chip.
We do get the `avrdude: AVR device initialized and ready to accept instructions` message which indicated that we in fact can communicate with the chip. We also know that we're dealing with a chip with UPDI. Since no chip can have a signature of 0x000000, it means that the chip _has_ to be locked and avrdude should force a reset command as long as the -e flag is passed.

This would be the logic we'll ve to implement:

* No communication with the target but the programmer is initialized?
  - Target has UPDI?
    - `-e` flag passed?
      - [Force erase command]

(Note that this can _only_ be applied for chips with UPDI. The [Arduino as ISP programmer](https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP) has a "bug" that results in an `avrdude: AVR device initialized and ready to accept instructions` message for classic AVRs, regardless if there is communication with the target or not.

MCUdude, 2022-10-13:
The SerialUPDI programmer has a slightly different output and forcing the erase command doesn't work at all for JTAG2UPDI.

<details>
<summary>SerialUPDI output</summary>

```
$ ./avrdude -cserialupdi -p atmega4808 -e

avrdude: UPDI link initialization OK
avrdude: Device is locked
avrdude: NVM type 0: 16-bit, page oriented write
avrdude: Entering NVM programming mode
avrdude: Timeout waiting for device to unlock
avrdude: Failed to enter NVM programming mode: device is locked
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude: Leaving NVM programming mode

avrdude done.  Thank you.

$ ./avrdude -cserialupdi -p atmega4808 -e -F

avrdude: UPDI link initialization OK
avrdude: Device is locked
avrdude: NVM type 0: 16-bit, page oriented write
avrdude: Entering NVM programming mode
avrdude: Timeout waiting for device to unlock
avrdude: Failed to enter NVM programming mode: device is locked
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
avrdude: Expected signature for ATmega4808 is 1E 96 50
avrdude: erasing chip
avrdude: Device is locked
avrdude: Attempting device erase
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.

$ ./avrdude -cserialupdi -p atmega4808

avrdude: UPDI link initialization OK
avrdude: NVM type 0: 16-bit, page oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9650 (probably m4808)
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.
```

</details>

<details>
<summary>JTAG2UPDI output</summary>

```
$ ./avrdude -cjtag2updi -p atmega4808 -e -F

avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
avrdude: jtagmkII_read_byte(): bad response to read memory command: RSP_ILLEGAL_MCU_STATE
avr_read_mem(): error reading address 0x0000
    read operation not supported for memory signature
avrdude: error reading signature data for part "ATmega4808", rc=-2
avrdude: error reading signature data, rc=-2
avrdude: jtagmkII_program_disable(): bad response to leave progmode command: RSP_ILLEGAL_MCU_STATE

avrdude done.  Thank you.

```
</summary>

avrfreak, 2022-10-13:
When reading the ASI System Status register of a locked UPDI mcu, the response will be 0x83 (Lockstatus = 1).  I believe the erase method & key is the same for all UPDI chips. I checked into this a few months back when I was getting sick and tired of having to erase the device with Studio7 every time. All that needs to be done is to send the erase key, and then a reset instruction. I did this on an ATtiny2313. It just erases and nothing else. I wouldn't know how to modify & compile avrdude.

MCUdude, 2022-10-13:
> All that needs to be done is to send the erase key, and then a reset instruction.

Sounds reasonable. However, when looking at the ASI System Status register, it's not obvious to me why it would return 0x83 when locked. We should probably check the least significant bit instead since this is the bit that indicates if the part is locked or not.

[image]

MCUdude, 2022-10-13:
I've taken a closer look at the root cause. The main problem is that the function `avr_unlock` isn't ran. This is caused by `avr_signature`, since this returns 0 (LIBAVRDUDE_SUCCESS) when reading the "valid" signaure that is 0x000000. This skips the entire unlock logic that's already there. I may be able to come up with a hacky solution, but @stefanrueger usually has clever ideas on how problems like this can be solved in an elegant way.

https://github.com/avrdudes/avrdude/blob/e14e5d2f7488b87d3d3f8308521bac6d2ea5d263/src/main.c#L1186-L1232

MCUdude, 2022-10-13:
OK, I totally fell down the rabbit hole. I'll submit a PR, and then we can discuss the solution I've come up with

avrfreak, 2022-10-14:
> avrdude_message(MSG_INFO, "%s: Received FamilyID: \"%.*s\"\n", progname, AVR_FAMILYIDLEN, sib);

Yes you are right to just check bit 0. I just did a quick erase program to fill my needs. I found that every device I tested either returns 0x82 (10000010) or 0x83 (10000011).  Strange how only bit 1 & bit 7 read as 1 while the other reserved (bit 6) reads 0.

I think if the lockstatus bit = 1, and the -e option is enabled, avrdude should just erase the chip. (not sure if this was already mentioned in the next thread. I will go over there now and have a look)

Thanks MCUdude

mcuee, 2022-10-16:
> The SerialUPDI programmer has a slightly different output and forcing the erase command doesn't work at all for JTAG2UPDI.

Oops, I did not read this properly. Now I need to learn how to use Microchip SNAP with this Nano 4808 board.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1125 -c jtag2updi -P COM13 -p atmega4808 -qq -t
avrdude> dump lock
>>> dump lock
0000  c5                                                |.               |

avrdude> dump signature
>>> dump signature
0000  1e 96 50                                          |..P             |

avrdude> write lock 0 0xc6
>>> write lock 0 0xc6
avrdude> flush
>>> flush
avrdude> dump lock
>>> dump lock
0000  c6                                                |.               |

avrdude> quit
>>> quit
avrdude_pr1125.exe: jtagmkII_reset(): bad response to reset command: RSP_ILLEGAL_MCU_STATE
avrdude_pr1125.exe: jtagmkII_close(): bad response to sign-off command: RSP_ILLEGAL_MCU_STATE

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1125 -c jtag2updi -P COM13 -p atmega4808 -qq -F -e
avrdude_pr1125.exe: jtagmkII_reset(): bad response to reset command: RSP_ILLEGAL_MCU_STATE
avrdude_pr1125.exe: initialization failed, rc=-1
avrdude_pr1125.exe: Yikes!  Invalid device signature.
avrdude_pr1125.exe: Expected signature for ATmega4808 is 1E 96 50
avrdude_pr1125.exe: jtagmkII_close(): bad response to sign-off command: RSP_ILLEGAL_MCU_STATE

```

mcuee, 2022-10-16:
@MCUdude 
Somehow I can not get PR #1125 to work even with Microchip SNAP.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1125.exe -c snap_updi -p atmega4808 -e
avrdude_pr1125.exe: usbhid_open(): No device found

                    Vtarget                      : 3.33 V
                    JTAG clock megaAVR/program   : 1000 kHz
                    JTAG clock megaAVR/debug     : 100 kHz
                    PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude_pr1125.exe: initialization failed, rc=-1
                    Double check connections and try again, or use -F to override
                    this check.

avrdude_pr1125.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1125.exe -c snap_updi -p atmega4808 -e -F
avrdude_pr1125.exe: usbhid_open(): No device found

                    Vtarget                      : 3.33 V
                    JTAG clock megaAVR/program   : 1000 kHz
                    JTAG clock megaAVR/debug     : 100 kHz
                    PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude_pr1125.exe: initialization failed, rc=-1
avrdude_pr1125.exe: AVR device initialized and ready to accept instructions
avrdude_pr1125.exe: Device signature = 0x000000 (retrying)
avrdude_pr1125.exe: Device signature = 0x000000 (retrying)
avrdude_pr1125.exe: Device signature = 0x000000
avrdude_pr1125.exe: Yikes!  Invalid device signature.
avrdude_pr1125.exe: Expected signature for ATmega4808 is 1E 96 50

avrdude_pr1125.exe done.  Thank you.

```

mcuee, 2022-10-16:
@avrfreak Just wondering if you can test PR #1125 with your SNAP as well, thanks.

avrfreak, 2022-10-16:
what is the fix another avrdude version?

MCUdude, 2022-10-16:
Works with my ([modified](https://ww1.microchip.com/downloads/en/DeviceDoc/ETN36_MPLAB%20Snap%20AVR%20Interface%20Modification.pdf)) SNAP:

```
$ ./avrdude -csnap_updi -patmega4808
avrdude: usbhid_open(): No device found

         Vtarget                      : 5.05 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude: Device is locked! Chip erase required to unlock.
avr_read_mem(): error reading address 0x0000
    read operation failed for memory signature
avrdude: error reading signature data for part "ATmega4808", rc=-3
avrdude: Received FamilyID: "megaAVR"
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

$ ./avrdude -csnap_updi -patmega4808 -e
avrdude: usbhid_open(): No device found

         Vtarget                      : 5.06 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude: Device is locked! Chip erase required to unlock.
avr_read_mem(): error reading address 0x0000
    read operation failed for memory signature
avrdude: error reading signature data for part "ATmega4808", rc=-3
avrdude: Received FamilyID: "megaAVR"
avrdude: erasing chip

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9650 (probably m4808)

avrdude done.  Thank you.

```

mcuee, 2022-10-16:
> what is the fix another avrdude version?

With PR #1125, you can get the binary from github action.
https://github.com/avrdudes/avrdude/actions/runs/3242268904

For example, 32bit mingw build here.
https://github.com/avrdudes/avrdude/suites/8756501063/artifacts/396837252
64bit mingw build here.
https://github.com/avrdudes/avrdude/suites/8756501063/artifacts/396837255

avrfreak, 2022-10-17:
does not work at all.
`
C:\avrdude-mingw-x86_64>avrdude -C avrdude.conf -c snap_updi -p avr128db28 -P usb -B 0.2 -e -U flash:w:test.hex:a -U eeprom:w:test.eep:a
avrdude: usbhid_open(): No device found

         Vtarget                      : 4.98 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 5000 kHz
`

MCUdude, 2022-10-17:
@avrfreak the programmer issue is not related to this PR. But if you look at the Avrdude output, you can see that you've set the UPDI clock to 5000kHz. That won't work fore sure! Try using `-B4` or similar instead of `-B0.2`

mcuee, 2022-10-17:
@MCUdude and @stefanrueger 

BTW, the `avrdude: usbhid_open(): No device found` is a false alarm. We should downgrade that.

mcuee, 2022-10-17:
@MCUdude 

Once I modify the SNAP as per Microchip ETN36, yes I can confirm the temp workaround works (`-F -e`).

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c snap_updi -p m4808 -F -e
avrdude_git.exe: usbhid_open(): No device found

                 Vtarget                      : 4.65 V
                 JTAG clock megaAVR/program   : 1000 kHz
                 JTAG clock megaAVR/debug     : 100 kHz
                 PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude_git.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude_git.exe: Device is locked! Chip erase required to unlock.
avr_read_mem(): error reading address 0x0000
    read operation failed for memory signature
avrdude_git.exe: error reading signature data for part "ATmega4808", rc=-3
avrdude_git.exe: Received FamilyID: "megaAVR"
avrdude_git.exe: erasing chip

Reading | ################################################## | 100% 0.02s

avrdude_git.exe: Device signature = 0x1e9650 (probably m4808)

avrdude_git.exe done.  Thank you.

```

mcuee, 2022-10-17:
@MCUdude 

And PR #1125 fixed the issue.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1125 -c snap_updi -p m4808
avrdude_pr1125.exe: usbhid_open(): No device found

                    Vtarget                      : 4.65 V
                    JTAG clock megaAVR/program   : 1000 kHz
                    JTAG clock megaAVR/debug     : 100 kHz
                    PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude_pr1125.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude_pr1125.exe: Device is locked! Chip erase required to unlock.
avr_read_mem(): error reading address 0x0000
    read operation failed for memory signature
avrdude_pr1125.exe: error reading signature data for part "ATmega4808", rc=-3
avrdude_pr1125.exe: Received FamilyID: "megaAVR"
                    Double check chip, or use -F to override this check.

avrdude_pr1125.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1125 -c snap_updi -p m4808 -e
avrdude_pr1125.exe: usbhid_open(): No device found

                    Vtarget                      : 4.65 V
                    JTAG clock megaAVR/program   : 1000 kHz
                    JTAG clock megaAVR/debug     : 100 kHz
                    PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude_pr1125.exe: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude_pr1125.exe: Device is locked! Chip erase required to unlock.
avr_read_mem(): error reading address 0x0000
    read operation failed for memory signature
avrdude_pr1125.exe: error reading signature data for part "ATmega4808", rc=-3
avrdude_pr1125.exe: Received FamilyID: "megaAVR"
avrdude_pr1125.exe: erasing chip

Reading | ################################################## | 100% 0.02s

avrdude_pr1125.exe: Device signature = 0x1e9650 (probably m4808)

avrdude_pr1125.exe done.  Thank you.

```

closed 2022-10-17 by stefanrueger