avrdudes/avrdude #1129: Better Optiboot EEPROM support for classic AVRs in terminal mode

mcuee, 2022-10-16:
Even though some optiboot build configuration supports EEPROM (eg: bigboot) and it works fine in CLI mode, but terminal mode does not seem to work very well (not so stable).

Just wondering if this can be improved or not.

Problems with `-t`.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c usbasp -qqp m328p -U .\hex\bigboot_328.hex && echo OK
OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c arduino -P COM4 -b 115200 -p m328p -qq -t
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
avrdude_git.exe: stk500_recv(): programmer is not responding
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> write eeprom 0 "Hello world"
>>> write eeprom 0 "Hello world"
avrdude> flush
>>> flush
avrdude_git.exe: synching cache to device... done
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  48 65 6c 6c 6f 20 77 6f  72 6c 64 00 ff ff ff ff  |Hello world.....|

avrdude> write eeprom 0 0xff 0xff 0xff
>>> write eeprom 0 0xff 0xff 0xff
avrdude> flush
>>> flush
avrdude_git.exe: synching cache to device... avrdude_git.exe: ser_send(): write error: sorry no info avail
avrdude_git.exe: ser_recv(): read error: Access is denied.

avrdude_git.exe: stk500_recv(): programmer is not responding
avrdude_git.exe: ser_send(): write error: sorry no info avail
avrdude_git.exe: ser_recv(): read error: Access is denied.

avrdude_git.exe: stk500_recv(): programmer is not responding

avrdude_git.exe: writeCachePage() eeprom write error at addr 0x0000
avrdude> quit
>>> quit
avrdude_git.exe: synching cache to device... avrdude_git.exe: ser_send(): write error: sorry no info avail
avrdude_git.exe: ser_recv(): read error: Access is denied.

avrdude_git.exe: stk500_recv(): programmer is not responding
avrdude_git.exe: ser_send(): write error: sorry no info avail
avrdude_git.exe: ser_recv(): read error: Access is denied.

avrdude_git.exe: stk500_recv(): programmer is not responding

avrdude_git.exe: writeCachePage() eeprom write error at addr 0x0000
avrdude_git.exe: ser_send(): write error: sorry no info avail
avrdude_git.exe: ser_recv(): read error: Access is denied.

avrdude_git.exe: stk500_recv(): programmer is not responding
CTRL-C

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c arduino -P COM4 -b 115200 -p m328p -qq -U eeprom:w:.\hex2\entest.eep:i && echo OK
OK
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c arduino -P COM4 -b 115200 -p m328p -qq -t
avrdude> dump eeprom 0 0x20
>>> dump eeprom 0 0x20
avrdude_git.exe: stk500_recv(): programmer is not responding
0000  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
0010  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|

avrdude> write eeprom 0 0xff 0xff 0xff
>>> write eeprom 0 0xff 0xff 0xff
avrdude> flush
>>> flush
avrdude_git.exe: synching cache to device... avrdude_git.exe: stk500_recv(): programmer is not responding
done
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  ff ff ff 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |... quick brown |

avrdude> quit
>>> quit

```

No issues with `-U`.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c usbasp -qqp m328p -U .\hex\bigboot_328.hex && echo OK
OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c arduino -P COM4 -b 115200 -p m328p -qq -U eeprom:w:.\hex2\entest.eep:i && echo OK
OK
```

mcuee, 2022-10-16:
The problem is worse with the hex mentioned by @stefanrueger in #940.

`-U` has no issues.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c usbasp -qqp m328p -U .\hex\atmega328p_sr940.hex 
&& echo OK
OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c arduino -P COM4 -b 115200 -p m328p 
-qq -U eeprom:w:.\hex2\entest.eep:i && echo OK
OK

```

`-t` does not seem to work at all.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c arduino -P COM4 -b 115200 -p m328p -qq -t
avrdude> dump eeprom 0 0x20
>>> dump eeprom 0 0x20
avrdude_git.exe: stk500_recv(): programmer is not responding
avrdude_git.exe: stk500_recv(): programmer is not responding
avrdude_git.exe: loadCachePage() eeprom read error at addr 0x0000
avrdude_git.exe (dump): error reading eeprom address 0x00000 of part ATmega328P
                        read operation not supported on memory type eeprom
avrdude> write eeprom 0 0xff 0xff 0xff
>>> write eeprom 0 0xff 0xff 0xff
avrdude_git.exe: stk500_recv(): programmer is not responding
avrdude_git.exe: stk500_recv(): programmer is not responding
avrdude_git.exe: loadCachePage() eeprom read error at addr 0x0000
avrdude_git.exe (write): error writing 0xff at 0x00000, rc=-1
                         write operation not supported on memory type eeprom
avrdude_git.exe: stk500_recv(): programmer is not responding
avrdude_git.exe: stk500_recv(): programmer is not responding
avrdude_git.exe: loadCachePage() eeprom read error at addr 0x0000
avrdude_git.exe (write): error writing 0xff at 0x00000 cell=0x00
avrdude_git.exe: stk500_recv(): programmer is not responding
avrdude_git.exe: stk500_recv(): programmer is not responding
avrdude_git.exe: loadCachePage() eeprom read error at addr 0x0001
avrdude_git.exe (write): error writing 0xff at 0x00001, rc=-1
                         write operation not supported on memory type eeprom

```

mcuee, 2022-10-16:
But for bigboot enabled optiboot, it does seem to work better.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c usbasp -qqp m328p -U .\hex\bigboot_328.hex && echo OK
OK
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c arduino -P COM4 -b 115200 -p m328p -qq -U eeprom:w:.\hex2\entest.eep:i && echo OK
OK
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1121v1 -c arduino -P COM4 -b 115200 -p m328p -qq -t
avrdude> dump eeprom 0 0x20
>>> dump eeprom 0 0x20
0000  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
0010  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|

avrdude> write eeprom 0 0xff 0xff 0xff
>>> write eeprom 0 0xff 0xff 0xff
avrdude> flush
>>> flush
avrdude_pr1121v1.exe: synching cache to device... done
avrdude> dump eeprom 0 0x20
>>> dump eeprom 0 0x20
0000  ff ff ff 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |... quick brown |
0010  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|

avrdude> quit
>>> quit

```

MCUdude, 2022-10-16:
The Optiboot bootloader is designed to time out quickly if there's no activity. That's why terminal mode doesn't work as expected. Try:

`echo "dump eeprom 0 0x10" | .\avrdude_pr1121v1 -c arduino -P COM4 -b 115200 -t`

mcuee, 2022-10-16:
> The Optiboot bootloader is designed to time out quickly if there's no activity. That's why terminal mode doesn't work as expected. Try:
> 
> `echo "dump eeprom 0 0x10" | .\avrdude_pr1121v1 -c arduino -P COM4 -b 115200 -t`

Great tip. Thanks. 

```
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude_pr1121v1 -c arduino 
-P COM4 -b 115200 -p m328p -qq -t
avrdude> >>> dump eeprom 0 0x10
0000  ff ff ff 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |... quick brown |

avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude_pr1121v1 -c arduino 
-P COM4 -b 115200 -p m328p -qq -t
avrdude> >>> dump eeprom 0 0x10
0000  ff ff ff 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |... quick brown |

avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "write eeprom 0 0xaa 0xaa 0xaa" | .\avrdude_pr1121v1 -c arduino 
-P COM4 -b 115200 -p m328p -qq -t
avrdude> >>> write eeprom 0 0xaa 0xaa 0xaa
avrdude> avrdude_pr1121v1.exe: synching cache to device... done
PS C:\work\avr\avrdude_test\avrdude_bin> echo "write eeprom 0 0x55 0xff 0x55" | .\avrdude_pr1121v1 -c arduino 
-P COM4 -b 115200 -p m328p -qq -t
avrdude> >>> write eeprom 0 0x55 0xff 0x55
avrdude> avrdude_pr1121v1.exe: synching cache to device... done
PS C:\work\avr\avrdude_test\avrdude_bin> echo "write eeprom 0 0x55 0xff 0x55" | .\avrdude_pr1121v1 -c arduino -P COM4 -b
 115200 -p m328p -qq -t
avrdude> >>> write eeprom 0 0x55 0xff 0x55
avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "write eeprom 0 0xaa 0xaa 0xaa" | .\avrdude_pr1121v1 -c arduino 
-P COM4 -b 115200 -p m328p -qq -t
avrdude> >>> write eeprom 0 0xaa 0xaa 0xaa
avrdude> avrdude_pr1121v1.exe: synching cache to device... done
PS C:\work\avr\avrdude_test\avrdude_bin> echo "write eeprom 0 0x55 0xff 0x55" | .\avrdude_pr1121v1 -c arduino 
-P COM4 -b 115200 -p m328p -qq -t
avrdude> >>> write eeprom 0 0x55 0xff 0x55
avrdude> avrdude_pr1121v1.exe: synching cache to device... done

```

mcuee, 2022-10-16:
> The Optiboot bootloader is designed to time out quickly if there's no activity. That's why terminal mode doesn't work as expected. Try:
> 
> `echo "dump eeprom 0 0x10" | .\avrdude_pr1121v1 -c arduino -P COM4 -b 115200 -t`

With this tip, even git main works very well.

```
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude_git -c arduino -P COM4 -b 115200 -p m328p -qq -t
avrdude> >>> dump eeprom 0 0x10
0000  55 ff 55 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |U.U quick brown |

avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "write eeprom 0 0x55 0xff 0x55" | .\avrdude_git -c arduino -P COM4 -b 115200 -p m328p -qq -t
avrdude> >>> write eeprom 0 0x55 0xff 0x55
avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude_git -c arduino -P COM4 -b 115200 -p m328p -qq -t
avrdude> >>> dump eeprom 0 0x10
0000  55 ff 55 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |U.U quick brown |

avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "write eeprom 0 0xaa 0xaa 0xaa" | .\avrdude_git -c arduino -P COM4 -b 115200 -p m328p -qq -t
avrdude> >>> write eeprom 0 0xaa 0xaa 0xaa
avrdude> avrdude_git.exe: synching cache to device... done
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude_git -c arduino -P COM4 -b 115200 -p m328p -qq -t
avrdude> >>> dump eeprom 0 0x10
0000  aa aa aa 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |... quick brown |

avrdude>

```

mcuee, 2022-10-16:
And no more problems with @stefanrueger's hex file as well.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c usbasp -qqp m328p -U .\hex\atmega328p_sr940.hex 
&& echo OK
OK
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude_git -c arduino -P COM4 
-b 115200 -p m328p
 -qq -t
avrdude> >>> dump eeprom 0 0x10
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "write eeprom 0 0xaa 0xaa 0xaa" | .\avrdude_git -c arduino 
-P COM4 -b 115200 -p m328p -qq -t
avrdude> >>> write eeprom 0 0xaa 0xaa 0xaa
avrdude> avrdude_git.exe: synching cache to device... done
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude_git -c arduino -P COM4 
-b 115200 -p m328p -qq -t
avrdude> >>> dump eeprom 0 0x10
0000  aa aa aa ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "write eeprom 0 0x55 0xff 0x55" | .\avrdude_git -c arduino 
-P COM4 -b 115200 -p m328p -qq -t
avrdude> >>> write eeprom 0 0x55 0xff 0x55
avrdude> avrdude_git.exe: synching cache to device... done
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude_git -c arduino -P COM4 
-b 115200 -p m328p -qq -t
avrdude> >>> dump eeprom 0 0x10
0000  55 ff 55 ff ff ff ff ff  ff ff ff ff ff ff ff ff  |U.U.............|

avrdude>

```

mcuee, 2022-10-16:
@MCUdude 

Thanks for the great tip. I will close this issue.

closed 2022-10-16 by mcuee

mcuee, 2022-10-16:
@MCUdude 

This is an important tip for me so I put it here.
https://github.com/mcuee/avrdude/wiki/avrdude-FAQs

MCUdude, 2022-10-17:
@mcuee there are a few hidden tricks like this that Avrdude supports but isn't really properly documented. We should perhaps add it to the manual and the Wiki.

I'm thinking about supporting data from stdin and writing it to _any_ memory by using `-U memtype:w:-:r`. I'm not sure how important this is for real-life scenarios, but it's an excellent feature for testing. And testing is a large part of improving and maintaining the Avrdude project.

mcuee, 2022-10-17:
> @mcuee there are a few hidden tricks like this that Avrdude supports but isn't really properly documented. We should perhaps add it to the manual and the Wiki.

Yes, good idea. We can always use the Wiki first and then discuss whether the tips need to be added to the manual or not.

> I'm thinking about supporting data from stdin and writing it to _any_ memory by using `-U memtype:w:-:r`. I'm not sure how important this is for real-life scenarios, but it's an excellent feature for testing. And testing is a large part of improving and maintaining the Avrdude project.

Yes, this will be a good feature to add.