avrdudes/avrdude #1130: Terminal output uses stdout except for the `parms` command; also errors go to stdout

stefanrueger, 2022-10-17:
The parms command in the terminal has always used the normal AVRDUDE messaging system, printing to stderr, and now following the `-qq` reduction of what is printed. This means that with the command the `parm` command can print nothing in the terminal:
```
$ echo parms | avrdude -p m328p -qqc arduino -t -P /dev/ttyUSB0
avrdude> parms
```

I think Terminal parms command should write to stdout (just like the other commands), and not follow the gradual reduction that `-qq` effects on the `msg_level()` system. 

Trouble is that the library functions avr_display() and programmer_display() share the functions that the terminal `parms` command uses where we *want*  gradual reduction with `-qq`. 

In order to solve this discrepancy, we'd need to change the API for pgm->print_parm() to add a FILE * parameter that can be stdout or stderr.

stefanrueger, 2022-10-17:
I suggest the following solution:
  - Review use of terminal_message() in term.c and make the errors/warnings use the new msg_level() system
  - Change the API of pgm->print_parm() to have an additional FILE * argument
  - Extend the current msg_level() macros to pass on stdout/stderr, the line number and file name of the source
  - Change terminal_message(MSG_INFO, ...) to a macro term_info(...) that utilises the updated messaging system with stdout

stefanrueger, 2022-10-18:
See PR #1132

mcuee, 2022-10-22:
Ref: PR #1132 should fix this. Unfortunately pulling in readline is a bit complicated under macOS and MSYS2 mingw.
https://github.com/avrdudes/avrdude/pull/1132#issuecomment-1285260358

closed 2022-10-23 by stefanrueger