avrdudes/avrdude #1133: Failing to communicate with ATtiny85 with default bitclock

steelman, 2022-10-18:
I've got some brand new ATtiny85 chips I was trying to program with a USBasp programmer. First I tried to read something, anything from them. And I faild. After checking connections and trying `-F` as one of error messages suggest I learnt that the problem may be too fast programming clock as described [here](https://www.avrfreaks.net/forum/usbasp-attiny85-something-going-wrong). Adding `-B 5` (setting SCK frequency to 187500 Hz as reported by avrdude) solved my problem. What puzzled me, was that reading fuses from ATmega328 (Arduino UNO) went without problems.

However, the problem in avrdude IMHO still exists: there is a legitimate use case where default values don't work. First `default_bitclock` in `/etc/avrdude.conf`` is commented out by default. `default_bitclock` and `bitclock` in `main.c` are set to `0.0` which propagates to `usbasp_spi_set_sck_period()`. There the message about setting the value automatically is issued but nothing more happens in this branch and the default value in the firmware seems to be 375 kHz.

I can see several solutions:
1. Extend the [message](https://github.com/avrdudes/avrdude/blob/b864d7e73afbaf2a94a4398913e2972a2984bb73/src/main.c#L1194) and give a hint about adding `-B 5`.
2. Set `default_bitclock` to 5, either in the code in `main.c` or in `avrdude.conf` and uncomment the line.
3. Set bitclock depending on part being programmed.

What in your opinion would be a valid solution to this problem?

mcuee, 2022-10-19:
I do not seem to have issue with default fuse settings (lfuse = 0x62, hfuse = 0xd7, efuse = 0xff):
- 8MHz internal RC oscillator
- Divide clock by 8 internally; [CKDIV8=0]

Ref: https://www.engbedded.com/fusecalc/

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c usbasp -p t85 -qq -t
avrdude> part
>>> part

AVR Part                      : ATtiny85
Chip Erase delay              : 4500 us
RESET disposition             : possible i/o
RETRY pulse                   : SCK
Serial program mode           : yes
Parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

                                  Block Poll               Page                       Polled
  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
  eeprom                 65     6     4    0 no        512    4      0  4000  4500 0xff 0xff
  flash                  65     6    32    0 yes      8192   64    128  4500  4500 0xff 0xff
  lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
  hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
  efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
  lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
  signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
  calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

avrdude> dump lfuse
>>> dump lfuse
0000  f1                                                |.               |

avrdude> dump hfuse
>>> dump hfuse
0000  d7                                                |.               |

avrdude> dump efuse
>>> dump efuse
0000  fe                                                |.               |

avrdude> dump lock
>>> dump lock
0000  ff                                                |.               |

avrdude> write lfuse 0 0x62
>>> write lfuse 0 0x62
avrdude> write hfuse 0 0xdf
>>> write hfuse 0 0xdf
avrdude> write efuse 0 0xff
>>> write efuse 0 0xff
avrdude> flush
>>> flush
avrdude> dump lfuse
>>> dump lfuse
0000  62                                                |b               |

avrdude> dump hfuse
>>> dump hfuse
0000  df                                                |.               |

avrdude> dump efuse
>>> dump efuse
0000  ff                                                |.               |

avrdude> quit
>>> quit

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c usbasp -p t85 -qq -t
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> quit
>>> quit

```

mcuee, 2022-10-19:
> Which seem odd considering what the datasheet [says](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-2586-AVR-8-bit-Microcontroller-ATtiny25-ATtiny45-ATtiny85_Datasheet.pdf#page=151):

>
> Low: > 2 CPU clock cycles for fck < 12 MHz, 3 CPU clock cycles for f ck >= 12 MHz
> High: > 2 CPU clock cycles for f ck < 12 MHz, 3 CPU clock cycles for f ck >= 12 MHz

For the new chip with default fuse, CPU clock will be 1MHz and fck = 8MHz.
2 CPU clock will  be 2ms. 

Low > 2ms; High > 2ms --> Low + High > 4ms --> SCK < 250KHz.

So indeed there may be a problem if the defaul of your usbasp FW is set to 375KHz.

On the other hand, if your usbasp FW is not so old, it should automatically set SCK.

I use the FW here and it is great (it will not work by default with avrdude 7.0 official Windows binary).
https://github.com/dioannidis/usbasp

You can also use the following with the HID interface in the above.
https://github.com/nerdralph/usbasp

Both of these two USBASP FW will set the default SCK to 1.5MHz but slow down  automatically if target does not respond.

steelman, 2022-10-19:
>For the new chip with default fuse, CPU clock will be 1MHz and fck = 8MHz

Indeed! I missed the the CKDIV8 when I was looking for CKSELx. Just a few lines higher.

> 2 CPU clock cycles will be 2ms.

2 *µs*, otherwise your calculations are correct.

> On the other hand, if your usbasp FW is not so old, it should automatically set SCK.

I bought the programmer probably… in 2013, so anything is possible.

Thanks, for the tips regarding FW. I don't use Windows, so the HID stuff doesn't bother me. However, considering the fact that newer firmware is able to handle such situation, it does seem adding hint about `-B` to the error message is good enough (I'll write something). But usbasp isn't the only programmer people may use, albeit probably most popular. Do you consider a more wordy message good enough in general?

@stefanrueger wrote:
> Do you have a suggestion?

How about:
```
avrdude error: program enable: target does not answer (0x01)
avrdude error: initialization failed, rc=-1
        - double check the chip and its connections and try again
        - use -B to set lower ISP clock frequency
        - use -F to override this check
```

mcuee, 2022-10-20:
@steelman 
> 2 µs, otherwise your calculations are correct.

Thanks for pointing that out. I have edited and updated.

The error message proposed looks good to me.
```
avrdude error: program enable: target does not answer (0x01)
avrdude error: initialization failed, rc=-1
        - double check the chip and its connections and try again
        - use -B to set lower ISP clock frequency
        - use -F to override this check
```

avrdude documentation does have some tips in this regard. 
https://avrdudes.github.io/avrdude/7.0/avrdude_39.html#Troubleshooting

_**Problem**: after flashing a firmware that reduces the target’s clock speed (e.g. through the CLKPR register), further ISP connection attempts fail._

_**Solution**: Even though ISP starts with pulling /RESET low, the target continues to run at the internal clock speed as defined by the firmware running before. Therefore, the ISP clock speed must be reduced appropriately (to less than 1/4 of the internal clock speed) using the -B option before the ISP initialization sequence will succeed.

As that slows down the entire subsequent ISP session, it might make sense to just issue a chip erase using the slow ISP clock (option -e), and then start a new session at higher speed. Option -D might be used there, to prevent another unneeded erase cycle._

PR #1139 (2022-10-23): Mention -B in the error message
One of possible problems of failed initialization is too hight frequency of ISP clock. Mention using -B option as a possible solution.

Closes #1133
commit c890ff90 (2022-10-23): Mention -B in the error message

One of possible problems of failed initialization is too hight frequency
of ISP clock. Mention using -B option as a possible solution.

Closes #1133
closed 2022-10-23 by stefanrueger