avrdudes/avrdude #1134: Schematics for avrdude??

stefanrueger, 2022-10-18:
The FILES section of the avrdude man page states
```
           ${PREFIX}/share/doc/avrdude/avrdude.pdf
                         Schematic of programming hardware
```
That's probably wrong (besides that PREFIX is likely to be the directory where avrdude files are installed at install time and not valid at user runtime). However, there is 63-page pdf user manual `avrdude.pdf`. Where is that installed (if it is)?

mcuee, 2022-10-19:
Right now it is installed to `/usr/local/share/doc/avrdude/avrdude.pdf` for the default avrdude build with documentation, under Linux.

```
 2001  cd build/avr/avrdude_main/
 2002  git pull
 2003  cmake -D BUILD_DOC=1 -B build_linux
 2004  cmake --build build_linux
 2005  sudo cmake --build build_linux --target install

mcuee@UbuntuSwift3:~/build/avr/avrdude_main$ sudo cmake --build build_linux --target install
[sudo] password for mcuee: 
[ 55%] Built target libavrdude
[ 60%] Built target avrdude
[ 61%] Built target conf
[ 61%] Built target avrdude_binaries
[ 68%] Built target html
[ 76%] Built target ps
[ 84%] Built target pdf
[ 92%] Built target dvi
[100%] Built target info
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Installing: /usr/local/bin/avrdude
-- Set runtime path of "/usr/local/bin/avrdude" to ""
-- Installing: /usr/local/lib/libavrdude.a
-- Installing: /usr/local/include/libavrdude.h
-- Installing: /usr/local/etc/avrdude.conf
-- Installing: /usr/local/share/man/man1/avrdude.1
-- Installing: /usr/local/share/info/avrdude.info
-- Installing: /usr/local/share/doc/avrdude/avrdude.dvi
-- Installing: /usr/local/share/doc/avrdude/avrdude.pdf
-- Installing: /usr/local/share/doc/avrdude/avrdude.ps
-- Installing: /usr/local/share/doc/avrdude/avrdude-html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_20.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_23.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_33.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_2.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_18.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_21.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_41.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_toc.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_39.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_1.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_32.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_5.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_11.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_19.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_37.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_22.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_31.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_4.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_34.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_35.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_38.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_28.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_15.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_10.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_16.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_7.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_25.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_29.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_40.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_6.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_abt.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_14.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_3.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_30.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_9.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_12.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_36.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_24.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_17.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_13.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_27.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_26.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_8.html

```

mcuee, 2022-10-19:
It is also the same for 7.0 release version using auto-tools.

```
 2013  cd avrdude-7.0
 2014  ls
 2015  make distclean
 2016  ./configure --help
 2017  ./configure --prefix=/usr --enable-doc
 2018  make
 2019  sudo make install

mcuee@UbuntuSwift3:~/build/avr/avrdude_release/avrdude-7.0$ sudo make install
make  install-recursive
make[1]: Entering directory '/home/mcuee/build/avr/avrdude_release/avrdude-7.0'
Making install in .
make[2]: Entering directory '/home/mcuee/build/avr/avrdude_release/avrdude-7.0'
make[3]: Entering directory '/home/mcuee/build/avr/avrdude_release/avrdude-7.0'
 /usr/bin/mkdir -p '/usr/lib'
 /bin/bash ./libtool   --mode=install /usr/bin/install -c   libavrdude.la '/usr/lib'
libtool: install: /usr/bin/install -c .libs/libavrdude.so.1.0.0 /usr/lib/libavrdude.so.1.0.0
libtool: install: (cd /usr/lib && { ln -s -f libavrdude.so.1.0.0 libavrdude.so.1 || { rm -f libavrdude.so.1 && ln -s libavrdude.so.1.0.0 libavrdude.so.1; }; })
libtool: install: (cd /usr/lib && { ln -s -f libavrdude.so.1.0.0 libavrdude.so || { rm -f libavrdude.so && ln -s libavrdude.so.1.0.0 libavrdude.so; }; })
libtool: install: /usr/bin/install -c .libs/libavrdude.lai /usr/lib/libavrdude.la
libtool: install: /usr/bin/install -c .libs/libavrdude.a /usr/lib/libavrdude.a
libtool: install: chmod 644 /usr/lib/libavrdude.a
libtool: install: ranlib /usr/lib/libavrdude.a
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/sbin" ldconfig -n /usr/lib
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/lib

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
 /usr/bin/mkdir -p '/usr/bin'
  /bin/bash ./libtool   --mode=install /usr/bin/install -c avrdude '/usr/bin'
libtool: install: /usr/bin/install -c avrdude /usr/bin/avrdude
Backing up avrdude.conf in /usr/etc
 /usr/bin/mkdir -p '/usr/etc'
 /usr/bin/install -c -m 644 avrdude.conf '/usr/etc'
 /usr/bin/mkdir -p '/usr/include'
 /usr/bin/install -c -m 644 libavrdude.h '/usr/include'
 /usr/bin/mkdir -p '/usr/share/man/man1'
 /usr/bin/install -c -m 644 avrdude.1 '/usr/share/man/man1'
make[3]: Leaving directory '/home/mcuee/build/avr/avrdude_release/avrdude-7.0'
make[2]: Leaving directory '/home/mcuee/build/avr/avrdude_release/avrdude-7.0'
Making install in doc
make[2]: Entering directory '/home/mcuee/build/avr/avrdude_release/avrdude-7.0/doc'
make[3]: Entering directory '/home/mcuee/build/avr/avrdude_release/avrdude-7.0/doc'
make[3]: Nothing to be done for 'install-exec-am'.
/bin/bash /home/mcuee/build/avr/avrdude_release/avrdude-7.0/install-sh -d /usr/share/doc/avrdude-7.0
/usr/bin/install -c -m 644 avrdude.ps /usr/share/doc/avrdude-7.0/avrdude.ps
/usr/bin/install -c -m 644 avrdude.pdf /usr/share/doc/avrdude-7.0/avrdude.pdf
/bin/bash /home/mcuee/build/avr/avrdude_release/avrdude-7.0/install-sh -d /usr/share/doc/avrdude-7.0/avrdude-html
 /usr/bin/mkdir -p '/usr/share/info'
 /usr/bin/install -c -m 644 ./avrdude.info '/usr/share/info'
 install-info --info-dir='/usr/share/info' '/usr/share/info/avrdude.info'
make[3]: Leaving directory '/home/mcuee/build/avr/avrdude_release/avrdude-7.0/doc'
make[2]: Leaving directory '/home/mcuee/build/avr/avrdude_release/avrdude-7.0/doc'
make[1]: Leaving directory '/home/mcuee/build/avr/avrdude_release/avrdude-7.0'

```

stefanrueger, 2022-10-19:
Interesting, on my system I get an error message:
```
$ cmake -D BUILD_DOC=1 -B build_linux
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- ENABLED    doc
-- ENABLED    parport
-- ENABLED    linuxgpio
-- ENABLED    linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/srueger/etc/clock/avrdude/avrdude-main/build_linux

$ cmake --build build_linux
[ 62%] Built target libavrdude
[ 62%] Built target avrdude
[ 62%] Built target conf
[ 62%] Built target avrdude_binaries
[ 62%] Generating avrdude-html/avrdude.html
/bin/sh: 1: TEXI2HTML_EXECUTABLE-NOTFOUND: not found
make[2]: *** [src/doc/CMakeFiles/html.dir/build.make:66: src/doc/avrdude-html/avrdude.html] Error 127
make[1]: *** [CMakeFiles/Makefile2:210: src/doc/CMakeFiles/html.dir/all] Error 2
```

mcuee, 2022-10-19:
@stefanrueger 
You need texi2html utility.
Ref:
* https://github.com/avrdudes/avrdude/pull/1096#issue-1361183166

stefanrueger, 2022-10-19:
@mcuee Brilliant tip. Thanks. Works after `sudo apt install texi2html`.

closed 2022-10-23 by stefanrueger