avrdudes/avrdude #1143: Xplained Pro should be able to support PDI mode

mcuee, 2022-10-22:
Xplained Pro should be able to support PDI mode as per the documentation.

Reference: take note that the following document does not cover UPDI support.
https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-42096-Microcontrollers-Embedded-Debugger_User-Guide.pdf

Reference discussion:
* https://github.com/avrdudes/avrdude/pull/1139#issuecomment-1286760773
* https://github.com/avrdudes/avrdude/discussions/1019#discussioncomment-3569440

<html>
<body>

Programmer | Interface
-- | --
AVRISP | ISP
Xplained Pro | JTAG & PDI & UPDI

</body>
</html>

mcuee, 2022-10-22:
We have already two entries for 
https://github.com/avrdudes/avrdude/blob/main/src/avrdude.conf.in#L2034-L2058

```
#------------------------------------------------------------
# xplainedpro
#------------------------------------------------------------

programmer
    id                     = "xplainedpro";
    desc                   = "Atmel AVR XplainedPro in JTAG mode";
    type                   = "jtagice3";
    prog_modes             = PM_JTAG | PM_XMEGAJTAG | PM_AVR32JTAG;
    connection_type        = usb;
    usbpid                 = 0x2111;
;

#------------------------------------------------------------
# xplainedpro_updi
#------------------------------------------------------------

programmer
    id                     = "xplainedpro_updi";
    desc                   = "Atmel AVR XplainedPro in UPDI mode";
    type                   = "jtagice3_updi";
    prog_modes             = PM_UPDI;
    connection_type        = usb;
    usbpid                 = 0x2111;
    hvupdi_support         = 1;
;
```

I think we need to add one more entry for the PDI mode.

```

#------------------------------------------------------------
# xplainedpro_updi
#------------------------------------------------------------

programmer
    id                     = "xplainedpro_pdi";
    desc                   = "Atmel AVR XplainedPro in PDI mode";
    type                   = "jtagice3_pdi";
    prog_modes             = PM_PDI;
    connection_type        = usb;
    usbpid                 = 0x2111;
    hvupdi_support         = 1;
;
```

mcuee, 2022-10-22:
@MCUdude 
I cannot test this. I am wondering if you have the XplainedPro board for the ATxmega device to check whether the above is correct or not. Thanks.

MCUdude, 2022-10-22:
@mcuee I don't own an Xplained Pro board, so I can't test this.
Ideally, I'd like to get an ATmega324PB Xplained Pro and an Xmega A1U Xplained Pro for testing, but I can't really justify spending the money.

mcuee, 2022-10-22:
> @mcuee I don't own an Xplained Pro board, so I can't test this. Ideally, I'd like to get an ATmega324PB Xplained Pro and an Xmega A1U Xplained Pro for testing, but I can't really justify spending the money.

@MCUdude 
Totally agree. These boards are just too expensive. Maybe someone from Microchip can help to check, probably no need to test either, just some internal design documentations.

mcuee, 2022-10-22:
@MCUdude 

I have documented what I have here. I have no issues spending on tools like the Curiosity Nano or Xplained Mino board or Microchip SNAP. But anything more expensive may not be justified. The only thing I might buy in the future is PICkit 4 as it seems to be the preferred tool down the road.
https://github.com/avrdudes/avrdude/discussions/1021#discussioncomment-3104217

I am not so sure if you can list down the tools and boards you have.

MCUdude, 2022-10-22:
> I am not so sure if you can list down the tools and boards you have. You certainly have more official tools than what I have.

I have acquired quite a lot over the years, mostly because I've been maintaining several Arduino cores. I can probably make a list sometime, but it will take some time to go through everything.

In #1003 we discussed how one can support the Avrdude project. Personally, I'm not interested I'm money, but hardware for testing is always helpful.

mcuee, 2022-10-23:
Pull request #1147 has been created. Even though we can not test it, I think it is correct based on the documentation.

mcuee, 2022-10-28:
> Pull request #1147 has been created. Even though we can not test it, I think it is correct based on the documentation.

It is better to test that before merging. Hopefully we can find someone who can test #1147.

I managed to borrow the PICkit 4 and ATtiny817 Xplained Pro evaluation kit from local Microchip office but unfortunately they do not have the  XMEGA A1U Xplained Pro evaluation kit to test #1147.

mcuee, 2022-10-28:
BTW,  xplainedpro_updi works fine.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedpro_updi -p t817 -t

         Vtarget                      : 3.34 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 1000 kHz
         JTAG clock Xmega             : 1000 kHz
         PDI/UPDI clock Xmega/megaAVR : 1000 kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9320 (probably t817)
avrdude> dump flash 0 0x10
>>> dump flash 0 0x10

Reading | ################################################## | 100% 0.05 s

0000  55 c8 18 95 18 95 18 95  18 95 18 95 ca c7 18 95  |U...............|

avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10

Reading | ################################################## | 100% 0.05 s

0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> write eeprom 0 0xaa 0x55 0xaa
>>> write eeprom 0 0xaa 0x55 0xaa

Caching | ################################################## | 100% 0.01 s

avrdude> flush
>>> flush
avrdude: synching cache to device ...
Writing | ################################################## | 100% 0.01 s
avrdude> dump eeprom 0 0x10
>>> dump eeprom 0 0x10

Reading | ################################################## | 100% 0.04 s

0000  aa 55 aa ff ff ff ff ff  ff ff ff ff ff ff ff ff  |.U..............|

avrdude> quit
>>> quit
avrdude>
avrdude done.  Thank you.

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedpro_updi -p t817 -U flash:w:t817.hex:i

         Vtarget                      : 3.34 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 1000 kHz
         JTAG clock Xmega             : 1000 kHz
         PDI/UPDI clock Xmega/megaAVR : 1000 kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9320 (probably t817)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
erasing chip
avrdude: reading input file t817.hex for flash
         with 8192 bytes in 1 section within [0, 0x1fff]
         using 128 pages and 0 pad bytes
avrdude: writing 8192 bytes flash ...

Writing | ################################################## | 100% 1.34 s

avrdude: 8192 bytes of flash written
avrdude: verifying flash memory against t817.hex

Reading | ################################################## | 100% 0.65 s

avrdude: 8192 bytes of flash verified

avrdude done.  Thank you.
```

closed 2022-11-22 by stefanrueger