avrdudes/avrdude #1144: Disable unsupported command for `-arduino`

mcuee, 2022-10-22:
Now that we can differentiate `-arduino` and `-stk500v1`, I think we should disable the following commands for `-arduino`.

```
  send    : send a raw command: send <b1> <b2> <b3> <b4>
  parms   : display adjustable parameters
  vtarg   : set <V[target]>
  varef   : set <V[aref]>
  fosc    : set <oscillator frequency>
```

```
avrdude_git: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_git: device signature = 0x1e9801 (probably m2560)
avrdude> help
>>> help
Valid commands:
  dump    : dump <memory> [<addr> <len> | <addr> ... | <addr> | ...]
  read    : alias for dump
  write   : write <memory> <addr> <data>[,] {<data>[,]}
          : write <memory> <addr> <len> <data>[,] {<data>[,]} ...
  flush   : synchronise flash & EEPROM writes with the device
  abort   : abort flash & EEPROM writes (reset the r/w cache)
  erase   : perform a chip erase
  sig     : display device signature bytes
  part    : display the current part information
  send    : send a raw command: send <b1> <b2> <b3> <b4>
  parms   : display adjustable parameters
  vtarg   : set <V[target]>
  varef   : set <V[aref]>
  fosc    : set <oscillator frequency>
  sck     : set <SCK period>
  verbose : change verbosity
  quell   : set quell level for progress bars
  help    : show help message
  ?       : same as help
  quit    : quit after writing out cache for flash & EEPROM

Note that not all programmer derivatives support all commands. Flash and
EEPROM type memories are normally read and written using a cache via paged
read and write access; the cache is synchronised on quit. Use the part
command to display valid memory types for use with dump and write.
```

mcuee, 2022-10-22:
Same for `-c wiring` as it is a bootloader and not physical STK500v2.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c wiring -P COM5 -b 115200 -p m2560 -t

avrdude_git: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude_git: device signature = 0x1e9801 (probably m2560)
avrdude> help
>>> help
Valid commands:
  dump    : dump <memory> [<addr> <len> | <addr> ... | <addr> | ...]
  read    : alias for dump
  write   : write <memory> <addr> <data>[,] {<data>[,]}
          : write <memory> <addr> <len> <data>[,] {<data>[,]} ...
  flush   : synchronise flash & EEPROM writes with the device
  abort   : abort flash & EEPROM writes (reset the r/w cache)
  erase   : perform a chip erase
  pgerase : pgerase <memory> <addr>
  sig     : display device signature bytes
  part    : display the current part information
  send    : send a raw command: send <b1> <b2> <b3> <b4>
  parms   : display adjustable parameters
  vtarg   : set <V[target]>
  varef   : set <V[aref]>
  fosc    : set <oscillator frequency>
  sck     : set <SCK period>
  verbose : change verbosity
  quell   : set quell level for progress bars
  help    : show help message
  ?       : same as help
  quit    : quit after writing out cache for flash & EEPROM

Note that not all programmer derivatives support all commands. Flash and
EEPROM type memories are normally read and written using a cache via paged
read and write access; the cache is synchronised on quit. Use the part
command to display valid memory types for use with dump and write.

avrdude> quit
>>> quit

avrdude_git done.  Thank you.

```

mcuee, 2022-10-22:
This should be low priority low as we already mentioned that `Note that not all programmer derivatives support all commands.`

stefanrueger, 2022-10-22:
Particularly with bootloaders, we don't have control over what they *actually* offer and/or do. AVRDUDE simply cannot know whether a bootloader implements the STK_[SG]ET_PARAMETER calls or not. I realise optiboot doesn't implement these, *but* other bootloaders might. So I would argue to leave these commands enabled.

mcuee, 2022-10-23:
> Particularly with bootloaders, we don't have control over what they _actually_ offer and/or do. AVRDUDE simply cannot know whether a bootloader implements the STK_[SG]ET_PARAMETER calls or not. I [...]

Fair enough. I will close this issue then.

closed 2022-10-23 by mcuee