avrdudes/avrdude #1150: Compile warnings from github action runs

mcuee, 2022-10-23:
1) macOS git main
https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795

```
/Users/runner/work/avrdude/avrdude/src/flip2.c:126:12: warning: unused function 'flip2_open' [-Wunused-function]
static int flip2_open(PROGRAMMER *pgm, const char *port_spec);
           ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:127:12: warning: unused function 'flip2_initialize' [-Wunused-function]
static int flip2_initialize(const PROGRAMMER *pgm, const AVRPART *part);
           ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[28](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:29):13: warning: unused function 'flip2_close' [-Wunused-function]
static void flip2_close(PROGRAMMER* pgm);
            ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[29](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:30):13: warning: unused function 'flip2_enable' [-Wunused-function]
static void flip2_enable(PROGRAMMER *pgm, const AVRPART *p);
            ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[30](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:31):13: warning: unused function 'flip2_disable' [-Wunused-function]
static void flip2_disable(const PROGRAMMER *pgm);
            ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[31](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:32):13: warning: unused function 'flip2_display' [-Wunused-function]
static void flip2_display(const PROGRAMMER *pgm, const char *prefix);
            ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[32](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:33):12: warning: unused function 'flip2_program_enable' [-Wunused-function]
static int flip2_program_enable(const PROGRAMMER *pgm, const AVRPART *part);
           ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[33](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:34):12: warning: unused function 'flip2_chip_erase' [-Wunused-function]
static int flip2_chip_erase(const PROGRAMMER *pgm, const AVRPART *part);
           ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[34](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:35):12: warning: unused function 'flip2_start_app' [-Wunused-function]
static int flip2_start_app(const PROGRAMMER *pgm);
           ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[35](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:36):12: warning: unused function 'flip2_read_byte' [-Wunused-function]
static int flip2_read_byte(const PROGRAMMER *pgm, const AVRPART *part, const AVRMEM *mem,
           ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[37](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:38):12: warning: unused function 'flip2_write_byte' [-Wunused-function]
static int flip2_write_byte(const PROGRAMMER *pgm, const AVRPART *part, const AVRMEM *mem,
           ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[39](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:40):12: warning: unused function 'flip2_paged_load' [-Wunused-function]
static int flip2_paged_load(const PROGRAMMER *pgm, const AVRPART *part, const AVRMEM *mem,
           ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[41](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:42):12: warning: unused function 'flip2_paged_write' [-Wunused-function]
static int flip2_paged_write(const PROGRAMMER *pgm, const AVRPART *part, const AVRMEM *mem,
           ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[43](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:44):12: warning: unused function 'flip2_read_sig_bytes' [-Wunused-function]
static int flip2_read_sig_bytes(const PROGRAMMER *pgm, const AVRPART *part, const AVRMEM *mem);
           ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[44](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:45):12: warning: unused function 'flip2_parseexitspecs' [-Wunused-function]
static int flip2_parseexitspecs(PROGRAMMER* pgm, const char *s);
           ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[45](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:46):13: warning: unused function 'flip2_setup' [-Wunused-function]
static void flip2_setup(PROGRAMMER * pgm);
            ^
/Users/runner/work/avrdude/avrdude/src/flip2.c:1[46](https://github.com/avrdudes/avrdude/actions/runs/3266550590/jobs/5370443795#step:5:47):13: warning: unused function 'flip2_teardown' [-Wunused-function]
static void flip2_teardown(PROGRAMMER * pgm);
            ^
17 warnings generated.
```

mcuee, 2022-10-23:
2. MSYS2 mingw32/ming64 build.

```
In function 'tokenize',
    inlined from 'terminal_mode' at D:/a/avrdude/avrdude/src/term.c:1259:12:
D:/a/avrdude/avrdude/src/term.c:1149:17: warning: pointer 'buf' may be used after 'realloc' [-Wuse-after-free]
 1149 |       ptrdiff_t k = buf - bufp;
      |                 ^
D:/a/avrdude/avrdude/src/term.c:1133:18: note: call to 'realloc' here
 1133 |       buf_tmp  = realloc(buf, bufsize);
      |                  ^~~~~~~~~~~~~~~~~~~~~

```

stefanrueger, 2022-10-23:
> flip2. macOS git main

I noticed that and solved it en passant in one of the PRs - should no longer happen.

> 2. MSYS2 mingw32/ming64 build.
```
In function 'tokenize',
    inlined from 'terminal_mode' at D:/a/avrdude/avrdude/src/term.c:1259:12:
D:/a/avrdude/avrdude/src/term.c:1149:17: warning: pointer 'buf' may be used after 'realloc' [-Wuse-after-free]
 1149 |       ptrdiff_t k = buf - bufp;
      |                 ^
D:/a/avrdude/avrdude/src/term.c:1133:18: note: call to 'realloc' here
 1133 |       buf_tmp  = realloc(buf, bufsize);
      |                  ^~~~~~~~~~~~~~~~~~~~~

```

This one is a false alarm warning that's kind of hard to get rid off (we discussed that before)

So, all things condidered this can be closed.

closed 2022-10-23 by stefanrueger