avrdudes/avrdude #1152: Terminal write is adding a trailing 0x00 when writing base 10 numbers

MCUdude, 2022-10-23:
First reported by @dbuchwald  in #1111.

There seems to be a bug in terminal write mode when base 10 number greater than 127 is written. For some reason, a trailing 0x00 is also written to memory. See output below. Tested on ATmega4808 and AVR128DA32 using a SerialUPDI programmer.

```
$ ./avrdude -cserialupdi -patmega4808 -t

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: device signature = 0x1e9650 (probably m4808)
avrdude> read eeprom
>>> read eeprom 

Reading | ################################################## | 100% 0.07s

0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> write eeprom 0x00 0
>>> write eeprom 0x00 0 

Caching | ################################################## | 100% 0.00s

avrdude> write eeprom 0x10 1
>>> write eeprom 0x10 1 

Caching | ################################################## | 100% 0.00s

avrdude> write eeprom 0x20 2
>>> write eeprom 0x20 2 

Caching | ################################################## | 100% 0.00s

avrdude> write eeprom 0x30 4
>>> write eeprom 0x30 4 

Caching | ################################################## | 100% 0.00s

avrdude> write eeprom 0x40 8
>>> write eeprom 0x40 8 

Caching | ################################################## | 100% 0.00s

avrdude> write eeprom 0x50 16
>>> write eeprom 0x50 16 

Caching | ################################################## | 100% 0.00s

avrdude> write eeprom 0x60 32
>>> write eeprom 0x60 32 

Caching | ################################################## | 100% 0.00s

avrdude> write eeprom 0x70 64
>>> write eeprom 0x70 64 

Caching | ################################################## | 100% 0.00s

avrdude> write eeprom 0x80 127
>>> write eeprom 0x80 127 

Caching | ################################################## | 100% 0.00s

avrdude> write eeprom 0x90 128
>>> write eeprom 0x90 128 

Caching | ################################################## | 100% 0.00s

avrdude> write eeprom 0xA0 255
>>> write eeprom 0xA0 255 

Caching | ################################################## | 100% 0.00s

avrdude> write eeprom 0xB0 256
>>> write eeprom 0xB0 256 

Caching | ################################################## | 100% 0.00s

avrdude> flush
>>> flush 
avrdude: synching cache to device ... 
Writing | ################################################## | 100% 0.85s
avrdude> read eeprom
>>> read eeprom 

Reading | ################################################## | 100% 0.00s

0000  00 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  01 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  02 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  04 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  08 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  10 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  20 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  | ...............|
0070  40 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |@...............|
0080  7f ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  80 00 ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff 00 ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  00 01 ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> quit
>>> quit 

avrdude done.  Thank you.
```

MCUdude, 2022-10-23:
It may be that this line is what's causing all this. It may not be a "bug", but instead a "feature". But IMO, it breaks the principle of least surprise.

https://github.com/avrdudes/avrdude/blob/b864d7e73afbaf2a94a4398913e2972a2984bb73/src/term.c#L516

stefanrueger, 2022-10-23:
> when base 10 number greater than 127 is written. For some reason, a trailing 0x00 is also written to memory

This is correct behaviour. The principle is that, in AVRDUDE, non-hex numbers occupy the smallest 1, 2, 4 or 8 byte space that is necessary to entertain the value.

128 is a signed integer. You need to add a `u` suffix to make it unsigned. Signed 128 doesn't fit into [-128, 127] for a signed byte.

In contrast to this, hex numbers have their own kind of "psychology" in AVRDUDE. A) In absence of an explicit sign hex numbers in AVRDUDE are treated as unsigned (so, 0x80 is unsigned but +0x80 is signed). B) Leading  zeros of hex numbers occupy space, ie, 0x0001 occupies 2 bytes as does 0x8000 b/c ordinary hex numbers are interpreted as unsigned.

If you are unsure about the size you can always add HH, H/S, L or LL.

stefanrueger, 2022-10-23:
PS: The special treatment of hex numbers in AVRDUDE is so that one can copy and paste hex lists printed with, eg, 0x%08x format strings.

stefanrueger, 2022-10-24:
> it breaks the principle of least surprise

I agree. I wonder whether this is the price the AVRDUDE project has to pay for switching from `write` lists of bytes to `write` lists of virtually anything.  On the other hand, maybe we find a slightly creative way of avoiding surprises?

Treating all numbers as unsigned and assigning smallest type does *not* work: then, -1 would occupy 8 bytes (it's the same as 18446744073709551615 and 0xffFFffFFffFFffFF, after all, being unsigned), but that will be surprising. We are looking for something that maps numbers to a size that does the right thing *and* that is easy to document *and* easy to understand.

Perhaps the size occupied by a signed integer $k$ should be the smallest $n \in \{1, 2, 4, 8\}$  for which $k \in [-2^{8n-1}, 2^{8n}-1]$? These are asymmetric boundaries of the type [INTn_MIN, **U**INTn_MAX], eg, [-127, 255], [-32768, 65535], ... for which a third of the interval would appear twice: $-1 \equiv 255$, $-2 \equiv 254$, but also $-1 \equiv 65535$, $-2 \equiv 65534$, etc. Once I've thought a little more about this and other ideas, I might suggest something. Or maybe it's simply too hard to surprise nobody...

mcuee, 2022-10-25:
> Or maybe it's simply too hard to surprise nobody...

I think that is okay. Can we treat everything as `unsigned char` by default? At least then 128 as 128. And for things like `-1`, can we treat it as "signed char" to treat it as 255?

stefanrueger, 2022-10-25:
> Can we treat everything as `unsigned char` by default?

Interesting idea. If the data item fits into an unsigned char, by hook or crook, then write it as unsigned char. The next surprise lurks for those who think 32768 (as signed) fits into 2 bytes and are surprised that it actually needs 4 bytes (as signed).

mcuee, 2022-10-25:
> Interesting idea. If the data item fits into an unsigned char, by hook or crook, then write it as unsigned char. The next surprise lurks for those who think 32768 (as signed) fits into 2 bytes and [...]

As mentioned, we can not satisfy everyone. 

I think we can still treat 32768 or other decimal literals without the `-` sign as unsigned (byte/word/double-word/quad-word/etc). I think that people will have less supprise in this case.

MCUdude, 2022-11-23:
Resolved in #1175

closed 2022-11-23 by MCUdude