avrdudes/avrdude #1159: Enhance CMake to be able to use static link for Linux and macOS

mcuee, 2022-10-25:
Ref:
* https://github.com/avrdudes/avrdude/issues/955#issuecomment-1288964378

> #962 has been merged and the option to use static lib or dynamic lib was added. But it has not been implemented yet.
> 
> * [CMake Enhancements #962 (comment)](https://github.com/avrdudes/avrdude/pull/962#issuecomment-1150075472)
> 
> @mariusgreuel wrote in the pull request comments.
> 
> > Note that the CMake files now include the option USE_STATIC_LIBS, which people can use to select either static or dynamic linking. Note, however, that I leave the implementation for somebody else in a separate PR. The current implementation will not work for non-defaults, i.e. statically linking for Linux or dynamically on Windows.

Indeed git main static link build for Linux and macOS failed, probably due to libreadline. This may pose some challenges for Arduino people. Maybe @umbynos wants to take a look. I think the fix may not be too complicated -- need to add the dependancy to libreadline.a (eg: libtinfo.a for Linux).

Failed build log:
https://github.com/mcuee/avrdude/actions/runs/3312632411

mcuee, 2022-10-25:
We can see that Arduino is using the following patch on top of avrdude 7.0 release to link to static libraries. But it does not seem to work with git main as of now.
https://github.com/arduino/avrdude-packing/blob/main/patches/0005-Cmake-always-prefer-static-libraries.patch

danapplegate, 2022-10-25:
I can try to do some digging into this tonight for MacOS. I don't know much about cmake, but I did have a lot of trouble building on MacOS (using homebrew) recently when trying to solve another issue.

mcuee, 2022-10-25:
> I can try to do some digging into this tonight for MacOS. I don't know much about cmake, but I did have a lot of trouble building on MacOS (using homebrew) recently when trying to solve another issue.

@danapplegate 
Thanks a lot for the help. 

Please also give the current git main and the following PR a try. Thanks.
https://github.com/avrdudes/avrdude/pull/1156

mcuee, 2022-10-26:
Just an example under Linux.

```
 $ cmake -D DEBUG_CMAKE=1 -D CMAKE_BUILD_TYPE=RelWithDebInfo -D USE_STATIC_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON  -B build_static
 $ cmake --build build_static
...
[ 98%] Linking C executable avrdude
cd /home/mcuee/build/avr/avrdude_main/build_static/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/avrdude.dir/link.txt --verbose=1
/usr/bin/cc -O2 -g -DNDEBUG   CMakeFiles/avrdude.dir/main.c.o CMakeFiles/avrdude.dir/term.c.o CMakeFiles/avrdude.dir/avrintel.c.o CMakeFiles/avrdude.dir/developer_opts.c.o CMakeFiles/avrdude.dir/whereami.c.o  -o avrdude  libavrdude.a -lm /usr/lib/x86_64-linux-gnu/libelf.a /usr/local/lib/libusb.a /usr/local/lib/libusb-1.0.a /usr/local/lib/libhidapi-libusb.a /usr/lib/x86_64-linux-gnu/libftdi.a /usr/local/lib/libftdi1.a /usr/lib/x86_64-linux-gnu/libreadline.a 
...
```

We can see that the dependancies are not pulled in, like libudev, pthread, etc. Please also take note that pkg-config is also not very reliable for static build.

```
$ pkg-config --static --libs libusb
-L/usr/local/lib -lusb -lusb-1.0 -ludev -lpthread

$ pkg-config --static --libs libftdi1
-L/usr/local/lib -lftdi1 -lusb-1.0 -ludev -lpthread

$ pkg-config --static --libs readline
-lreadline -Wl,-Bsymbolic-functions -ltinfo

$ pkg-config --static --libs hidapi-libusb 
-L/usr/local/lib -lhidapi-libusb

$ pkg-config --static --libs hidapi-hidraw 
-L/usr/local/lib -lhidapi-hidraw

$ pkg-config --static --libs libelf
-lelf -lz

```

danapplegate, 2022-10-26:
@mcuee Ok, calling it quits for tonight, but I was able to get through compilation using USE_STATIC_LIBS=ON. Unfortunately, I'm now getting some gnarly failures in the linker step.

First, I noticed that it appeared that CMake was not properly finding the header files installed by Homebrew in `/usr/local/include`.

<details><summary>Example output from cmake configure step</summary>
<p>

```
applegatedt@Daniels-MBP-2 avrdude % cmake -D DEBUG_CMAKE=1 -D CMAKE_BUILD_TYPE=RelWithDebInfo -D USE_STATIC_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON -B build_static
-- The C compiler identification is AppleClang 13.0.0.13000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/local/bin/git (found version "2.35.1")
-- Found FLEX: /usr/bin/flex (found version "2.6.4")
-- Found BISON: /usr/bin/bison (found version "2.3")
-- Looking for libelf.h
-- Looking for libelf.h - not found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - not found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - not found
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Darwin-20.6.0
-- CMAKE_SYSTEM: Darwin-20.6.0
-- CMAKE_FIND_ROOT_PATH:
-- CMAKE_C_COMPILER: /Library/Developer/CommandLineTools/usr/bin/cc
-- CONFIG_DIR: /usr/local/etc
-- AVRDUDE_FULL_VERSION: 7.0-20221024 (5862a5b)
-- USE_EXTERNAL: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: /usr/local/lib/libelf.a
-- HAVE_LIBUSB: /usr/local/lib/libusb.a
-- HAVE_LIBUSB_1_0: /usr/local/lib/libusb-1.0.a
-- HAVE_LIBUSB_WIN32: HAVE_LIBUSB_WIN32-NOTFOUND
-- HAVE_LIBHID: /usr/local/lib/libhid.dylib
-- HAVE_LIBHIDAPI: /usr/local/lib/libhidapi.dylib
-- HAVE_LIBFTDI: /usr/local/lib/libftdi.a
-- HAVE_LIBFTDI1: HAVE_LIBFTDI1-NOTFOUND
-- HAVE_LIBREADLINE: HAVE_LIBREADLINE-NOTFOUND
-- HAVE_LIBELF_H:
-- HAVE_LIBELF_LIBELF_H:
-- HAVE_USB_H:
-- HAVE_LUSB0_USB_H:
-- HAVE_LIBUSB_H:
-- HAVE_LIBUSB_1_0_LIBUSB_H:
-- HAVE_HIDAPI_HIDAPI_H:
-- LIBUSB_COMPAT_DIR:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi
-- DON'T HAVE libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
[... 218 lines cut ...]
      _darwin_get_cached_device in libusb-1.0.a(darwin_usb.o)
      _get_ioregistry_value_number in libusb-1.0.a(darwin_usb.o)
      _usb_find_interface_matching_location in libusb-1.0.a(darwin_usb.o)
  "_kCFAllocatorSystemDefault", referenced from:
      _darwin_device_from_service in libusb-1.0.a(darwin_usb.o)
      _darwin_claim_interface in libusb-1.0.a(darwin_usb.o)
  "_kCFRunLoopCommonModes", referenced from:
      _darwin_open in libusb-1.0.a(darwin_usb.o)
  "_kCFRunLoopDefaultMode", referenced from:
      _darwin_close in libusb-1.0.a(darwin_usb.o)
      _darwin_event_thread_main in libusb-1.0.a(darwin_usb.o)
      _darwin_release_interface in libusb-1.0.a(darwin_usb.o)
      _darwin_claim_interface in libusb-1.0.a(darwin_usb.o)
  "_kCFTypeDictionaryKeyCallBacks", referenced from:
      _usb_find_interface_matching_location in libusb-1.0.a(darwin_usb.o)
  "_kCFTypeDictionaryValueCallBacks", referenced from:
      _usb_find_interface_matching_location in libusb-1.0.a(darwin_usb.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/avrdude] Error 1
make[1]: *** [src/CMakeFiles/avrdude.dir/all] Error 2
make: *** [all] Error 2
```

If anyone has any suggestions, I'm all ears.

mcuee, 2022-10-26:
@danapplegate You need to specify the CMake option as per here.

Default dynamic link option:
https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-macOS
```
cmake -D CMAKE_C_FLAGS=-I/usr/local/include 
-D CMAKE_EXE_LINKER_FLAGS=-L/usr/local/Cellar 
-D CMAKE_BUILD_TYPE=RelWithDebInfo 
-B build_macos
```

For static link option, and for debugging, I will suggest you use the following option (it will fail in the last step).
```
cmake -D CMAKE_C_FLAGS=-I/usr/local/include 
-D CMAKE_EXE_LINKER_FLAGS=-L/usr/local/Cellar 
-D CMAKE_BUILD_TYPE=RelWithDebInfo 
-D DEBUG_CMAKE=1 
-D USE_STATIC_LIBS=ON 
-D CMAKE_VERBOSE_MAKEFILE=ON  
-B build_macos
```

danapplegate, 2022-10-26:
@mcuee Yeah, even with `CMAKE_EXE_LINKER_FLAGS` provided, I'm still getting the "Undefined symbols" messages.

```
[100%] Linking C executable avrdude
cd /Users/applegatedt/workspace/avrdude/build_static/src && /usr/local/Cellar/cmake/3.24.2/bin/cmake -E cmake_link_script CMakeFiles/avrdude.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/cc -I/usr/local/include -O2 -g -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk -mmacosx-version-min=11.6 -Wl,-search_paths_first -Wl,-headerpad_max_install_names -L/usr/local/Cellar CMakeFiles/avrdude.dir/main.c.o CMakeFiles/avrdude.dir/term.c.o CMakeFiles/avrdude.dir/avrintel.c.o CMakeFiles/avrdude.dir/developer_opts.c.o CMakeFiles/avrdude.dir/whereami.c.o -o avrdude  libavrdude.a -lm /usr/local/lib/libelf.a /usr/local/lib/libusb.a /usr/local/lib/libusb-1.0.a /usr/local/lib/libhid.dylib /usr/local/lib/libhidapi.dylib /usr/local/lib/libftdi.a
Undefined symbols for architecture x86_64:
  "_CFBooleanGetTypeID", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
  "_CFBooleanGetValue", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
  "_CFDataGetBytes", referenced from:
      _darwin_get_cached_device in libusb-1.0.a(darwin_usb.o)
  ...
  ```

mcuee, 2022-10-26:
For Arduino, they have two patches for static linking for avrdude 7.0 release.

1) prefer static library, this should not be necessary now -- we can use `-D USE_STATIC_LIBS=ON`
https://github.com/arduino/avrdude-packing/blob/main/patches/0005-Cmake-always-prefer-static-libraries.patch

2) add libtinfo and libncurses as dependancy, this is still needed.
https://github.com/arduino/avrdude-packing/blob/main/patches/0006-Cmake-add-libtinfo-and-libncurses-as-dependencies.patch

mcuee, 2022-10-26:
Following the approach from Arduino, the following patch seems to be good now for Linux.

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 400da29..5ec1a34 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -225,6 +225,31 @@ if(HAVE_LIBREADLINE)
     set(LIB_LIBREADLINE ${HAVE_LIBREADLINE})
 endif()
 
+find_library(HAVE_LIBNCURSES NAMES ncurses)
+if(HAVE_LIBNCURSES)
+    set(LIB_LIBNCURSES ${HAVE_LIBNCURSES})
+endif()
+
+find_library(HAVE_LIBTINFO NAMES tinfo)
+if(HAVE_LIBTINFO)
+    set(LIB_LIBTINFO ${HAVE_LIBTINFO})
+endif()
+
+find_library(HAVE_LIBZ NAMES z)
+if(HAVE_LIBZ)
+    set(LIB_LIBZ ${HAVE_LIBZ})
+endif()
+
+find_library(HAVE_LIBUDEV NAMES udev)
+if(HAVE_LIBZ)
+    set(LIB_LIBUDEV ${HAVE_LIBUDEV})
+endif()
+
+find_library(HAVE_LIBPTHREAD NAMES pthread)
+if(HAVE_LIBZ)
+    set(LIB_LIBPTHREAD ${HAVE_LIBPTHREAD})
+endif()
+
 # =====================================
 # Use external libraries if requested
 # =====================================
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index ca6cb64..47322f9 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -268,6 +268,10 @@ target_link_libraries(libavrdude
     ${LIB_LIBFTDI}
     ${LIB_LIBFTDI1}
     ${LIB_LIBREADLINE}
mcuee@UbuntuSwift3:~/build/avr/avrdude_main$ git diff >diff.txt
mcuee@UbuntuSwift3:~/build/avr/avrdude_main$ cat diff.txt 
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 400da29..5ec1a34 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -225,6 +225,31 @@ if(HAVE_LIBREADLINE)
     set(LIB_LIBREADLINE ${HAVE_LIBREADLINE})
 endif()
 
+find_library(HAVE_LIBNCURSES NAMES ncurses)
+if(HAVE_LIBNCURSES)
+    set(LIB_LIBNCURSES ${HAVE_LIBNCURSES})
+endif()
+
+find_library(HAVE_LIBTINFO NAMES tinfo)
+if(HAVE_LIBTINFO)
+    set(LIB_LIBTINFO ${HAVE_LIBTINFO})
+endif()
+
+find_library(HAVE_LIBZ NAMES z)
+if(HAVE_LIBZ)
+    set(LIB_LIBZ ${HAVE_LIBZ})
+endif()
+
+find_library(HAVE_LIBUDEV NAMES udev)
+if(HAVE_LIBZ)
+    set(LIB_LIBUDEV ${HAVE_LIBUDEV})
+endif()
+
+find_library(HAVE_LIBPTHREAD NAMES pthread)
+if(HAVE_LIBZ)
+    set(LIB_LIBPTHREAD ${HAVE_LIBPTHREAD})
+endif()
+
 # =====================================
 # Use external libraries if requested
 # =====================================
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index ca6cb64..47322f9 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -268,6 +268,10 @@ target_link_libraries(libavrdude
     ${LIB_LIBFTDI}
     ${LIB_LIBFTDI1}
     ${LIB_LIBREADLINE}
+    ${LIB_LIBNCURSES}
+    ${LIB_LIBZ}
+    ${LIB_LIBUDEV}
+    ${LIB_LIBPTHREAD}
     ${EXTRA_WINDOWS_LIBRARIES}
     )

$ rm -rf build_linux_static/
$ cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=ON 
-D CMAKE_VERBOSE_MAKEFILE=ON -B build_linux_static
...
$ cmake --build build_linux_static
...
mcuee@UbuntuSwift3:~/build/avr/avrdude_main$ ldd ./build_linux_static/src/avrdude
	linux-vdso.so.1 (0x00007ffdc9d08000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f7ed6e98000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f7ed6e68000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f7ed6e4c000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f7ed6e1f000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f7ed6dfc000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f7ed6c0a000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f7ed7160000)
```

mcuee, 2022-10-26:
@mariusgreuel and @umbynos
Please check if the above approach is the right approach or not. Thanks.

mcuee, 2022-10-27:
Info for macOS using pkg-config and Homebrew.

```

mcuee@mcuees-Mac-mini avrdude_main % pkg-config --static --libs libusb
-L/opt/homebrew/Cellar/libusb-compat/0.1.5_1/lib -L/opt/homebrew/Cellar/libusb/HEAD-ba69847/lib -lusb -lusb-1.0 -lobjc -Wl,-framework,IOKit -Wl,-framework,CoreFoundation -Wl,-framework,Security
mcuee@mcuees-Mac-mini avrdude_main % pkg-config --static --libs libftdi1
-L/opt/homebrew/Cellar/libftdi/1.5_2/lib -L/opt/homebrew/Cellar/libusb/HEAD-ba69847/lib -lftdi1 -lusb-1.0 -lobjc -Wl,-framework,IOKit -Wl,-framework,CoreFoundation -Wl,-framework,Security
mcuee@mcuees-Mac-mini avrdude_main % pkg-config --static --libs hidapi  
-L/opt/homebrew/Cellar/hidapi/0.12.0/lib -lhidapi
mcuee@mcuees-Mac-mini avrdude_main % pkg-config --static --libs readline
-L/opt/homebrew/Cellar/readline/8.2.1/lib -lreadline -lncurses
mcuee@mcuees-Mac-mini avrdude_main % pkg-config --static --libs libelf
-L/opt/homebrew/Cellar/libelf/0.8.13_1/lib -lelf
```

mcuee, 2022-10-27:
The same approach can be done for MSYS2 mingw32/mingw64.

```
$ git diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 400da29..77d757f 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -125,6 +125,7 @@ if(USE_STATIC_LIBS)
     set(PREFERRED_LIBFTDI libftdi.a ftdi)
     set(PREFERRED_LIBFTDI1 libftdi1.a ftdi1)
     set(PREFERRED_LIBREADLINE libreadline.a)
+       set(PREFERRED_LIBTERMCAP libtermcap.a)
 else()
     set(PREFERRED_LIBELF elf)
     set(PREFERRED_LIBUSB usb)
@@ -133,6 +134,7 @@ else()
     set(PREFERRED_LIBFTDI ftdi)
     set(PREFERRED_LIBFTDI1 ftdi1)
     set(PREFERRED_LIBREADLINE readline)
+       set(PREFERRED_LIBTERMCAP termcap)
 endif()

 # -------------------------------------
@@ -225,6 +227,11 @@ if(HAVE_LIBREADLINE)
     set(LIB_LIBREADLINE ${HAVE_LIBREADLINE})
 endif()

+find_library(HAVE_LIBTERMCAP NAMES ${PREFERRED_LIBTERMCAP})
+if(HAVE_LIBTERMCAP)
+    set(LIB_LIBTERMCAP ${HAVE_LIBTERMCAP})
+endif()
+
 # =====================================
 # Use external libraries if requested
 # =====================================
@@ -310,6 +317,7 @@ if (DEBUG_CMAKE)
     message(STATUS "HAVE_LIBFTDI: ${HAVE_LIBFTDI}")
     message(STATUS "HAVE_LIBFTDI1: ${HAVE_LIBFTDI1}")
     message(STATUS "HAVE_LIBREADLINE: ${HAVE_LIBREADLINE}")
+       message(STATUS "HAVE_LIBTERMCAP: ${HAVE_LIBTERMCAP}")
     message(STATUS "HAVE_LIBELF_H: ${HAVE_LIBELF_H}")
     message(STATUS "HAVE_LIBELF_LIBELF_H: ${HAVE_LIBELF_LIBELF_H}")
     message(STATUS "HAVE_USB_H: ${HAVE_USB_H}")
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index ca6cb64..d42c166 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -268,7 +268,8 @@ target_link_libraries(libavrdude
     ${LIB_LIBFTDI}
     ${LIB_LIBFTDI1}
     ${LIB_LIBREADLINE}
-    ${EXTRA_WINDOWS_LIBRARIES}
+       ${LIB_LIBTERMCAP}
+       ${EXTRA_WINDOWS_LIBRARIES}
     )

 add_executable(avrdude

```

Of course I will still need to rebuild termcap as per the workaround mentioned here.
* https://github.com/msys2/MINGW-packages/issues/8105#issuecomment-1229270428

```
$ cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON -B build_mingw_static
-- Building for: Ninja
-- The C compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
[... 63 lines cut ...]
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude_main/build_mingw_static

$ cmake --build build_mingw_static
...
[66/66] cmd.exe /C "cd . && C:\msys64\mingw64\bin\cc.exe -O2 -g -DNDEBUG  src/CMakeFiles/avrdude.dir/main.c.obj src/CMakeFiles/avrdude.dir/term.c.obj src/CMakeFiles/avrdude.dir/avrintel.c.obj src/CMakeFiles/avrdude.dir/developer_opts.c.obj src/CMakeFiles/avrdude.dir/whereami.c.obj src/CMakeFiles/avrdude.dir/windows.rc.obj -o src\avrdude.exe -Wl,--out-implib,src\libavrdude.dll.a -Wl,--major-image-version,0,--minor-image-version,0  src/libavrdude.a  -lm  C:/msys64/mingw64/lib/libelf.a  C:/msys64/mingw64/lib/libusb.a  C:/msys64/mingw64/lib/libusb-1.0.a  C:/msys64/mingw64/lib/libhid.a  C:/msys64/mingw64/lib/libhidapi.a  C:/msys64/mingw64/lib/libftdi1.a  C:/msys64/mingw64/lib/libreadline.a  C:/msys64/mingw64/lib/libtermcap.a  -lsetupapi  -lws2_32  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."

$ ldd ./build_mingw_static/src/avrdude.exe
        ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ffd14cf0000)
        KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ffd137d0000)
        KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ffd12970000)
        msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ffd13210000)
        SETUPAPI.dll => /c/WINDOWS/System32/SETUPAPI.dll (0x7ffd14560000)
        cfgmgr32.dll => /c/WINDOWS/System32/cfgmgr32.dll (0x7ffd126d0000)
        ucrtbase.dll => /c/WINDOWS/System32/ucrtbase.dll (0x7ffd12470000)
        RPCRT4.dll => /c/WINDOWS/System32/RPCRT4.dll (0x7ffd136a0000)
        bcrypt.dll => /c/WINDOWS/System32/bcrypt.dll (0x7ffd12410000)
        USER32.dll => /c/WINDOWS/System32/USER32.dll (0x7ffd149d0000)
        win32u.dll => /c/WINDOWS/System32/win32u.dll (0x7ffd12440000)
        GDI32.dll => /c/WINDOWS/System32/GDI32.dll (0x7ffd130f0000)
        gdi32full.dll => /c/WINDOWS/System32/gdi32full.dll (0x7ffd12c40000)
        msvcp_win.dll => /c/WINDOWS/System32/msvcp_win.dll (0x7ffd128d0000)
        WS2_32.dll => /c/WINDOWS/System32/WS2_32.dll (0x7ffd14050000)
        HID.DLL => /c/WINDOWS/SYSTEM32/HID.DLL (0x7ffd0fec0000)
```

mcuee, 2022-10-27:
@umbynos 

I tend to think some parts of the Arduino avrdude workflow file here seem to be good to be adopted by the official avrdude project as well. Good work!

https://github.com/arduino/avrdude-packing/blob/7295e15feca9d40122a61ce7ca2842b9ab3da00e/.github/workflows/release.yml

mcuee, 2022-10-27:
My trial and errors to use static linking for Linux and macOS  are here. It is still using dynamic linking for hidapi, strange.
https://github.com/mcuee/avrdude/commits/cmake_static_linux_macOS

mcuee, 2022-10-28:
> we use this to build macos: https://github.com/arduino/avrdude-packing/blob/7295e15feca9d40122a61ce7ca2842b9ab3da00e/.github/workflows/release.yml#L91 and what @mcuee said [here](https://github.com/avrdudes/avrdude/issues/1159#issuecomment-1291949821) is true. We will drop the first patch when the new version of avrdude comes out [arduino/avrdude-packing#22](https://github.com/arduino/avrdude-packing/pull/22)

@umbynos 

A minor improvement to supress many warnings is here for the Arduino script. The frameworks are supposed to be for the linker option and not compiler option.
https://github.com/mcuee/avrdude/commit/3f3f739025de8154f5c1ef30d2bc62daf46acb9c

I tested it in the github action and it works.
https://github.com/mcuee/avrdude/actions/runs/3341739395/jobs/5533237235

mcuee, 2022-10-28:
For those who want to try out the Arduino crossbuild container, it is here, thanks to @umbynos.
https://github.com/arduino/crossbuild/pkgs/container/crossbuild

mcuee, 2022-10-28:
Just tried the container and it works well -- only tested v7.0 release and Linux. It may need minor tuning for git main -- I will try that tomorrow.

```
$ git clone https://github.com/arduino/avrdude-packing.git
$ git clone https://github.com/avrdudes/avrdude.git
$ cd avrdude
$ git checkout tags/v7.0
$ cd ..
$ cd avrdude-packaging
$ cd patches
$ perl -pi -e "s/ARDUINO_VERSION_PLACEHOLDER/7.0/g" 0008-Append-arduino-to-version-string.patch
$ cd ../../avrdude
$ git apply -v ../avrdude-packing/patches/*.patch
$ cmake -DCMAKE_C_COMPILER=x86_64-ubuntu16.04-linux-gnu-gcc -DCMAKE_CXX_COMPILER=x86_64-ubuntu16.04-linux-gnu-g++ -DCMAKE_EXE_LINKER_FLAGS="-static-libgcc -static-libstdc++" -DCMAKE_C_FLAGS="-I/opt/lib/x86_64-ubuntu16.04-linux-gnu/include/libusb-1.0/ -I/opt/lib/x86_64-ubuntu16.04-linux-gnu/include -pthread" -DCMAKE_PREFIX_PATH=/opt/lib/x86_64-ubuntu16.04-linux-gnu/ -DHAVE_LIBFTDI="NO" -B build/
$ cmake --build build/ -v

root@1e76cc636a1d:/workdir/avrdude# ldd ./build/src/avrdude
	linux-vdso.so.1 (0x00007ffd8f3e6000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fdd0bf43000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fdd0bf20000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fdd0bd2e000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fdd0c099000)

```

mcuee, 2022-10-29:
> Just tried the container and it works well -- only tested v7.0 release and Linux. It may need minor tuning for git main -- I will try that tomorrow.

And it works well for git main as well. I only need one patch as I do not need to use Arduino branding.

```
$ git clone https://github.com/arduino/avrdude-packing.git
$ git clone https://github.com/avrdudes/avrdude.git
$ cd avrdude
$ git apply ../avrdude-packing/patches/0006-Cmake-add-libtinfo-and-libncurses-as-dependencies.patch 

$ cmake -DCMAKE_C_COMPILER=x86_64-ubuntu16.04-linux-gnu-gcc 
-DCMAKE_CXX_COMPILER=x86_64-ubuntu16.04-linux-gnu-g++ 
-DCMAKE_EXE_LINKER_FLAGS="-static-libgcc -static-libstdc++" 
-DCMAKE_C_FLAGS="-I/opt/lib/x86_64-ubuntu16.04-linux-gnu/include/libusb-1.0/ -I/opt/lib/x86_64-ubuntu16.04-linux-gnu/include -pthread" 
-DCMAKE_PREFIX_PATH=/opt/lib/x86_64-ubuntu16.04-linux-gnu/ 
-DHAVE_LIBFTDI="NO" 
-DCMAKE_BUILD_TYPE=RelWithDebInfo 
-DUSE_STATIC_LIBS=ON 
-DCMAKE_VERBOSE_MAKEFILE=ON 
-B build/ 

$ cmake --build build/

root@1e76cc636a1d:/workdir/avrdude# ldd ./build/src/avrdude
	linux-vdso.so.1 (0x00007fff20bbf000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f49fcbb0000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f49fcb8d000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f49fc99b000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f49fcd06000)
root@1e76cc636a1d:/workdir/avrdude# ./build/src/avrdude 
Usage: avrdude [options]
Options:
  -p <partno>                Specify AVR device
  -b <baudrate>              Override RS-232 baud rate
  -B <bitclock>              Specify bit clock period (us)
  -C <config-file>           Specify location of configuration file
  -c <programmer>            Specify programmer type
  -A                         Disable trailing-0xff removal from file and AVR read
  -D                         Disable auto erase for flash memory; implies -A
  -i <delay>                 ISP Clock Delay [in microseconds]
  -P <port>                  Specify connection port
  -F                         Override invalid signature or initialisation check
  -e                         Perform a chip erase
  -O                         Perform RC oscillator calibration (see AVR053)
  -U <memtype>:r|w|v:<filename>[:format]
                             Memory operation specification
                             Multiple -U options are allowed, each request
                             is performed in the order specified
  -n                         Do not write anything to the device
  -V                         Do not verify
  -t                         Enter terminal mode
  -E <exitspec>[,<exitspec>] List programmer exit specifications
  -x <extended_param>        Pass <extended_param> to programmer
  -v                         Verbose output; -v -v for more
  -q                         Quell progress output; -q -q for less
  -l logfile                 Use logfile rather than stderr for diagnostics
  -?                         Display this usage

avrdude version 7.0-20221026 (a101899), URL: <https://github.com/avrdudes/avrdude>

```

mcuee, 2022-10-29:
For macOS, it is also okay with git main.

```
$  mv /usr/bin/ranlib /usr/bin/ranlib.bk
$  ln -s /opt/osxcross/target/bin/x86_64-apple-darwin13-ranlib /usr/bin/ranlib
$ cmake -DCMAKE_C_COMPILER=o64-clang 
-DCMAKE_CXX_COMPILER=o64-clang++ 
-DCMAKE_AR=/opt/osxcross/target/bin/x86_64-apple-darwin13-ar 
-DCMAKE_LINKER=/opt/osxcross/target/bin/x86_64-apple-darwin13-ld 
-DCMAKE_EXE_LINKER_FLAGS="-L/opt/lib/x86_64-apple-darwin13/lib/" 
-DCMAKE_C_FLAGS="-I/opt/lib/x86_64-apple-darwin13/include -pthread -framework Foundation -framework IOKit -framework Cocoa -framework Security -DHAVE_USB_H" 
-DCMAKE_PREFIX_PATH=/opt/lib/x86_64-apple-darwin13/ 
-DHAVE_LIBFTDI="NO" 
-DCMAKE_BUILD_TYPE=RelWithDebInfo 
-DUSE_STATIC_LIBS=ON 
-DCMAKE_VERBOSE_MAKEFILE=ON 
-B build_macos 
$cmake --build build_macos/

root@1e76cc636a1d:/workdir/avrdude# /opt/osxcross/target/bin/x86_64-apple-darwin13-otool -L ./build_macos/src/avrdude
./build_macos/src/avrdude:
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1673.126.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 23.0.0)
	/System/Library/Frameworks/Security.framework/Versions/A/Security (compatibility version 1.0.0, current version 59306.41.2)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.0.0)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 1894.10.126)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1673.126.0)

```

mcuee, 2022-10-29:
@umbynos 

But the above generates one warning in configure stage (C++ compiler is not necessary) and too many warnings building the compiling stage (frameworks should be linker script and not compiler script).

I also think AppKit is necessary. But I am not so sure if older version of macOS supports it or not.

The following may be better.

```
$ cmake -DCMAKE_C_COMPILER=o64-clang 
-DCMAKE_AR=/opt/osxcross/target/bin/x86_64-apple-darwin13-ar 
-DCMAKE_LINKER=/opt/osxcross/target/bin/x86_64-apple-darwin13-ld 
-DCMAKE_EXE_LINKER_FLAGS="-L/opt/lib/x86_64-apple-darwin13/lib/ -lobjc -Wl,-framework,IOKit -Wl,-framework,AppKit -Wl,-framework,CoreFoundation -Wl,-framework,Security" 
-DCMAKE_C_FLAGS="-I/opt/lib/x86_64-apple-darwin13/include -pthread 
-DHAVE_USB_H" -DCMAKE_PREFIX_PATH=/opt/lib/x86_64-apple-darwin13/ 
-DHAVE_LIBFTDI="NO" -DCMAKE_BUILD_TYPE=RelWithDebInfo 
-DUSE_STATIC_LIBS=ON 
-DCMAKE_VERBOSE_MAKEFILE=ON 
-B build_macos_alt 
  
$ cmake --build build_macos_alt/

root@1e76cc636a1d:/workdir/avrdude# /opt/osxcross/target/bin/x86_64-apple-darwin13-otool -L ./build_macos_alt/src/avrdude
./build_macos_alt/src/avrdude:
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 1894.10.126)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1673.126.0)
	/System/Library/Frameworks/Security.framework/Versions/A/Security (compatibility version 1.0.0, current version 59306.41.2)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.0.0)
```
[avrdude_macos_intel64bit_alt.zip](https://github.com/avrdudes/avrdude/files/9892664/avrdude_macos_intel64bit_alt.zip)

mcuee, 2022-10-29:
@umbynos 
I have issues for mingw32 build though. The patch "0007-Cmake-fix-build-not-working-with-mingw-toolchain.patch" does not apply to git main properly. So I need to manually apply the patch.

Then the build will still fail. Not so sure what is wrong.

<details>
<summary> detail build log for mingw32 </summary>

```
root@1e76cc636a1d:/workdir/avrdude# cmake -DCMAKE_C_COMPILER=i686-w64-mingw32-gcc -DCMAKE_EXE_LINKER_FLAGS="-static-libgcc" -DCMAKE_C_FLAGS="-I/opt/lib/i686-w64-mingw32/include/libusb-1.0/ -I/opt/lib/i686-w64-mingw32/include -pthread" -DCMAKE_PREFIX_PATH=/opt/lib/i686-w64-mingw32 -DHAVE_LIBFTDI="NO" -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUSE_STATIC_LIBS=ON -DCMAKE_VERBOSE_MAKEFILE=ON -B build_mingw32 
-- The C compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/i686-w64-mingw32-gcc
-- Check for working C compiler: /usr/bin/i686-w64-mingw32-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "3.5.1")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - found
-- Looking for libusb.h
-- Looking for libusb.h - found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - not found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /workdir/avrdude/build_mingw32
root@1e76cc636a1d:/workdir/avrdude# cmake --build build_mingw32/
/usr/bin/cmake -S/workdir/avrdude -B/workdir/avrdude/build_mingw32 --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /workdir/avrdude/build_mingw32/CMakeFiles /workdir/avrdude/build_mingw32/CMakeFiles/progress.marks
/usr/bin/make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/workdir/avrdude/build_mingw32'
/usr/bin/make -f src/CMakeFiles/libavrdude.dir/build.make src/CMakeFiles/libavrdude.dir/depend
make[2]: Entering directory '/workdir/avrdude/build_mingw32'
[  1%] [BISON][Parser] Building parser with bison 3.5.1
cd /workdir/avrdude/src && /usr/bin/bison --defines=/workdir/avrdude/build_mingw32/config_gram.h -o /workdir/avrdude/build_mingw32/config_gram.c config_gram.y
[  3%] [FLEX][Parser] Building scanner with flex 2.6.4
cd /workdir/avrdude/src && /usr/bin/flex -o/workdir/avrdude/build_mingw32/lexer.c lexer.l
cd /workdir/avrdude/build_mingw32 && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /workdir/avrdude /workdir/avrdude/src /workdir/avrdude/build_mingw32 /workdir/avrdude/build_mingw32/src /workdir/avrdude/build_mingw32/src/CMakeFiles/libavrdude.dir/DependInfo.cmake --color=
Dependee "/workdir/avrdude/build_mingw32/src/CMakeFiles/libavrdude.dir/DependInfo.cmake" is newer than depender "/workdir/avrdude/build_mingw32/src/CMakeFiles/libavrdude.dir/depend.internal".
Dependee "/workdir/avrdude/build_mingw32/src/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/workdir/avrdude/build_mingw32/src/CMakeFiles/libavrdude.dir/depend.internal".
Scanning dependencies of target libavrdude
make[2]: Leaving directory '/workdir/avrdude/build_mingw32'
/usr/bin/make -f src/CMakeFiles/libavrdude.dir/build.make src/CMakeFiles/libavrdude.dir/build
make[2]: Entering directory '/workdir/avrdude/build_mingw32'
[... 493 lines cut ...]
/workdir/avrdude/src/developer_opts.c:1200: undefined reference to `lfirst'
/usr/bin/i686-w64-mingw32-ld: CMakeFiles/avrdude.dir/developer_opts.c.o: in function `dev_output_pgm_part':
/workdir/avrdude/src/developer_opts.c:806: undefined reference to `cfg_get_prologue'
/usr/bin/i686-w64-mingw32-ld: /workdir/avrdude/src/developer_opts.c:808: undefined reference to `default_programmer'
/usr/bin/i686-w64-mingw32-ld: /workdir/avrdude/src/developer_opts.c:808: undefined reference to `cfg_escape'
/usr/bin/i686-w64-mingw32-ld: /workdir/avrdude/src/developer_opts.c:809: undefined reference to `default_parallel'
/usr/bin/i686-w64-mingw32-ld: /workdir/avrdude/src/developer_opts.c:809: undefined reference to `cfg_escape'
/usr/bin/i686-w64-mingw32-ld: /workdir/avrdude/src/developer_opts.c:810: undefined reference to `default_serial'
/usr/bin/i686-w64-mingw32-ld: /workdir/avrdude/src/developer_opts.c:810: undefined reference to `cfg_escape'
/usr/bin/i686-w64-mingw32-ld: /workdir/avrdude/src/developer_opts.c:811: undefined reference to `default_spi'
/usr/bin/i686-w64-mingw32-ld: /workdir/avrdude/src/developer_opts.c:811: undefined reference to `cfg_escape'
/usr/bin/i686-w64-mingw32-ld: /workdir/avrdude/src/developer_opts.c:817: undefined reference to `cfg_strdup'
/usr/bin/i686-w64-mingw32-ld: /workdir/avrdude/src/developer_opts.c:825: undefined reference to `cfg_strdup'
[... cut ...]
/usr/bin/i686-w64-mingw32-ld: /opt/lib/i686-w64-mingw32/lib/libtinfo.a(lib_ttyflags.o):lib_ttyflags.c:(.text+0xeb): undefined reference to `_nc_mingw_ioctl'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/avrdude.dir/build.make:156: src/avrdude] Error 1
make[2]: Leaving directory '/workdir/avrdude/build_mingw32'
make[1]: *** [CMakeFiles/Makefile2:101: src/CMakeFiles/avrdude.dir/all] Error 2
make[1]: Leaving directory '/workdir/avrdude/build_mingw32'
make: *** [Makefile:133: all] Error 2
```

</details>

mcuee, 2022-10-30:
May latest efforts mentioned here, still got problem with Windows.
* https://github.com/avrdudes/avrdude/issues/1162#issuecomment-1296055989

mcuee, 2022-10-30:
Let me close this issue as well. The thing is whether this patch is acceptable or not to avrdude project.
https://github.com/arduino/avrdude-packing/blob/main/patches/0006-Cmake-add-libtinfo-and-libncurses-as-dependencies.patch

I will continue the discussion in #1162.

mcuee, 2022-10-30:
> 1. Need to move back the ranlib after buiding for macOS
> 2. Need to upgrade ncurses version as found by @umbynos.

Once this is done, I can build the mingw32 version using the Docker container.

```
$ ls -la /usr/bin/ranlib
$ rm /usr/bin/ranlib
$ mv /usr/bin/ranlib.bk /usr/bin/ranlib
$ rm -rf build_mingw
$ cmake -DCMAKE_C_COMPILER=i686-w64-mingw32-gcc -DCMAKE_EXE_LINKER_FLAGS="-static-libgcc -static-libstdc++" -DCMAKE_C_FLAGS="-I/opt/lib/i686-w64-mingw32/include/libusb-1.0/ -I/opt/lib/i686-w64-mingw32/include -pthread -Wall -std=gnu11" -DCMAKE_PREFIX_PATH=/opt/lib/i686-w64-mingw32/ -DHAVE_LIBFTDI="NO" -B build_mingw
$ cmake --build build_mingw -v
...

[ 96%] Linking C executable avrdude
cd /workdir/avrdude/build_mingw/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/avrdude.dir/link.txt --verbose=1
/usr/bin/i686-w64-mingw32-gcc -I/opt/lib/i686-w64-mingw32/include/libusb-1.0/ -I/opt/lib/i686-w64-mingw32/include -pthread -Wall -std=gnu11  -static-libgcc -static-libstdc++ CMakeFiles/avrdude.dir/main.c.o CMakeFiles/avrdude.dir/term.c.o CMakeFiles/avrdude.dir/avrintel.c.o CMakeFiles/avrdude.dir/developer_opts.c.o CMakeFiles/avrdude.dir/whereami.c.o  -o avrdude  libavrdude.a -lm /opt/lib/i686-w64-mingw32/lib/libelf.a /opt/lib/i686-w64-mingw32/lib/libusb.a /opt/lib/i686-w64-mingw32/lib/libusb-1.0.a /opt/lib/i686-w64-mingw32/lib/libhidapi.a /opt/lib/i686-w64-mingw32/lib/libftdi1.a /opt/lib/i686-w64-mingw32/lib/libreadline.a /opt/lib/i686-w64-mingw32/lib/libncurses.a /opt/lib/i686-w64-mingw32/lib/libtinfo.a -lsetupapi -lws2_32 
make[2]: Leaving directory '/workdir/avrdude/build_mingw'
[ 98%] Built target avrdude
/usr/bin/make -f src/CMakeFiles/conf.dir/build.make src/CMakeFiles/conf.dir/depend
make[2]: Entering directory '/workdir/avrdude/build_mingw'
cd /workdir/avrdude/build_mingw && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /workdir/avrdude /workdir/avrdude/src /workdir/avrdude/build_mingw /workdir/avrdude/build_mingw/src /workdir/avrdude/build_mingw/src/CMakeFiles/conf.dir/DependInfo.cmake --color=
Dependee "/workdir/avrdude/build_mingw/src/CMakeFiles/conf.dir/DependInfo.cmake" is newer than depender "/workdir/avrdude/build_mingw/src/CMakeFiles/conf.dir/depend.internal".
Dependee "/workdir/avrdude/build_mingw/src/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/workdir/avrdude/build_mingw/src/CMakeFiles/conf.dir/depend.internal".
Scanning dependencies of target conf
make[2]: Leaving directory '/workdir/avrdude/build_mingw'
/usr/bin/make -f src/CMakeFiles/conf.dir/build.make src/CMakeFiles/conf.dir/build
make[2]: Entering directory '/workdir/avrdude/build_mingw'
[100%] Generating avrdude.conf
cd /workdir/avrdude/build_mingw/src && /usr/bin/cmake -E copy /workdir/avrdude/src/avrdude.conf.in avrdude.conf.in
cd /workdir/avrdude/build_mingw/src && /usr/bin/cmake -D HAVE_PARPORT=OFF -D HAVE_LINUXSPI=OFF -D HAVE_LINUXGPIO=OFF -D DEFAULT_PAR_PORT=/dev/parport0 -D DEFAULT_SER_PORT=/dev/ttyS0 -P /workdir/avrdude/src/configure.cmake
make[2]: Leaving directory '/workdir/avrdude/build_mingw'
[100%] Built target conf
make[1]: Leaving directory '/workdir/avrdude/build_mingw'
/usr/bin/cmake -E cmake_progress_start /workdir/avrdude/build_mingw/CMakeFiles 0

root@1e76cc636a1d:/workdir/avrdude# ls -la ./build_mingw/src/
total 6168
drwxr-xr-x 3 root root    4096 Oct 30 10:34 .
drwxr-xr-x 4 root root    4096 Oct 30 10:33 ..
drwxr-xr-x 6 root root    4096 Oct 30 10:33 CMakeFiles
-rw-r--r-- 1 root root   56711 Oct 30 10:33 Makefile
-rw-r--r-- 1 root root    2953 Oct 30 10:33 ac_cfg.h
-rw-r--r-- 1 root root  523196 Oct 30 10:34 avrdude.conf
-rw-r--r-- 1 root root  523227 Oct 30 10:34 avrdude.conf.in
-rwxr-xr-x 1 root root 4236398 Oct 30 10:34 avrdude.exe
-rw-r--r-- 1 root root    2652 Oct 30 10:33 avrdude.spec
-rw-r--r-- 1 root root    2529 Oct 30 10:33 cmake_install.cmake
-rw-r--r-- 1 root root  939346 Oct 30 10:34 libavrdude.a
-rw-r--r-- 1 root root    3768 Oct 30 10:33 windows.rc
```

mcuee, 2022-10-30:
More binaries here.
https://github.com/mcuee/avrdude-packing/releases/tag/snapshot_30Oct2022

mcuee, 2022-10-30:
Re-open this issue since there may be other ways to sort out this issue.

mcuee, 2022-11-02:
Latest binary (using Arduino Crossbuild latest 0.2.2 version), no manually intervention this time (last one I have to do it manually for Windows).
https://github.com/mcuee/avrdude-packing/releases/tag/snapshot_2Nov2022

mcuee, 2022-11-02:
Now I will close this issue.

mcuee, 2022-11-12:
Reopen this issue so that people will be able to find it.

mcuee, 2022-12-14:
Even though #1162 can be a potential solution, right now at least two patches used by @umbynos are not really acceptable for the avrdude project.

Patch 06 may need some work. Patch 07 is just a hack.
We need CMake experts like @mariusgreuel to come out with the real solution.

https://github.com/arduino/avrdude-packing/tree/main/patches
Patch 06: https://github.com/arduino/avrdude-packing/blob/main/patches/0006-Cmake-add-libtinfo-and-libncurses-as-dependencies.patch
Patch 07: https://github.com/arduino/avrdude-packing/blob/main/patches/0007-Cmake-fix-build-not-working-with-mingw-toolchain.patch

(Patch 08 is just Arduino branding)

mcuee, 2023-01-05:
@mariusgreuel 

Please help to review the above two patches when you got some time. Thanks.

mariusgreuel, 2023-01-05:
@mcuee I have not tried building static executables, but for patch 06, I can see that there are some extra dependencies that need to be resolved. I think they should be added conditionally when static builds are requested, but other than that I am fine with the additions.

As for the 07 patch, I do not understand this. Why would you need to remove the version resource from the MINGW build?

I think it is best to submit PRs for our avrdude main branch. The PRs are great, because the CI provides me with a lot of context: I can see the build system being used, the toolchain versions, the exact build procedure, the logs, everything, which saves me from guessing.

mcuee, 2023-01-05:
@mariusgreuel 

I agree with your patch 06 makes sense. However, there is a complication for MSYS2 build that MSYS2 libreadline depends on libtermcap. Then there is a problem with static linking due to the problems of MSYS2 packages.

I've submited a PR previously but then it has to be reverted due to the problem.
https://github.com/avrdudes/avrdude/actions/runs/3308684165/jobs/5461248027

Arduino-pack project builds their own libreadline, not the same as MSYS2.

mcuee, 2023-01-05:
@umbynos 
Just wondering why you need patch 07? I agree with @mariusgreuel that patch 07 is kind of strange.

mcuee, 2023-01-11:
@mariusgreuel 

I just tried it and patch 007, even though not really correct, is still needed.

If not, the build will fail due to missing ws2_32.
https://github.com/mcuee/avrdude-packing/actions/runs/3889057820/jobs/6636976036

umbynos, 2023-01-11:
>  Just wondering why you need patch 07? I agree with @mariusgreuel that patch 07 is kind of strange.

IIRC it was required because mingw do not define `WIN32` env var, thus not defining `EXTRA_WINDOWS_RESOURCES` and `EXTRA_WINDOWS_LIBRARIES`.

By the way I updated the patch in https://github.com/arduino/avrdude-packing/commit/033caf3f0ff5bb3bb734c63196792c55ae7d5244

mariusgreuel, 2023-01-11:
> IIRC it was required because mingw do not define WIN32 env var

If you are targeting Windows, CMake must define WIN32. If not, there is something wrong with your cross-build setup.

> If not, the build will fail due to missing ws2_32.

In avrdude-packing that is, not in our CI pipeline.

mcuee, 2023-01-12:
> > IIRC it was required because mingw do not define WIN32 env var
> 
> If you are targeting Windows, CMake must define WIN32. If not, there is something wrong with your cross-build setup.

I think this is a fair statement based on my experiences with other projects.

mcuee, 2023-01-14:
Actually I can build git main under macOS using static link without any patches.

```
mcuee@mcuees-Mac-mini avrdude_main % cmake -D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS="-L/opt/homebrew/Cellar -Wl,-framework,IOKit -Wl,-framework,CoreFoundation -Wl,-framework,Security"  -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON -B build_macos                         
-- The C compiler identification is AppleClang 14.0.0.14000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /opt/homebrew/bin/git (found version "2.39.0") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "2.3")
-- Looking for libelf.h
-- Looking for libelf.h - not found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Darwin-22.2.0
-- CMAKE_SYSTEM: Darwin-22.2.0
-- CMAKE_FIND_ROOT_PATH: 
-- CMAKE_C_COMPILER: /Library/Developer/CommandLineTools/usr/bin/cc
-- CONFIG_DIR: /usr/local/etc
-- AVRDUDE_FULL_VERSION: 7.1-20230113 (9cf4f55)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: /opt/homebrew/lib/libelf.a
-- HAVE_LIBUSB: /opt/homebrew/lib/libusb.a
-- HAVE_LIBUSB_1_0: /opt/homebrew/lib/libusb-1.0.a
-- HAVE_LIBUSB_WIN32: HAVE_LIBUSB_WIN32-NOTFOUND
-- HAVE_LIBHID: HAVE_LIBHID-NOTFOUND
-- HAVE_LIBHIDAPI: /opt/homebrew/lib/libhidapi.dylib
-- HAVE_LIBFTDI: HAVE_LIBFTDI-NOTFOUND
-- HAVE_LIBFTDI1: /opt/homebrew/lib/libftdi1.a
-- HAVE_LIBREADLINE: HAVE_LIBREADLINE-NOTFOUND
-- HAVE_LIBELF_H: 
-- HAVE_LIBELF_LIBELF_H: 1
-- HAVE_USB_H: 1
-- HAVE_LUSB0_USB_H: 
-- HAVE_LIBUSB_H: 
-- HAVE_LIBUSB_1_0_LIBUSB_H: 1
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR: 
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DON'T HAVE libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/mcuee/build/avr/avrdude_test/avrdude_main/build_macos
mcuee@mcuees-Mac-mini avrdude_main % cmake --build build_macos
...

[100%] Linking C executable avrdude
cd /Users/mcuee/build/avr/avrdude_test/avrdude_main/build_macos/src && /opt/homebrew/Cellar/cmake/3.25.1/bin/cmake -E cmake_link_script CMakeFiles/avrdude.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/cc -I/opt/homebrew/include -O2 -g -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names -L/opt/homebrew/Cellar -Wl,-framework,IOKit -Wl,-framework,CoreFoundation -Wl,-framework,Security CMakeFiles/avrdude.dir/main.c.o CMakeFiles/avrdude.dir/term.c.o CMakeFiles/avrdude.dir/avrintel.c.o CMakeFiles/avrdude.dir/developer_opts.c.o CMakeFiles/avrdude.dir/whereami.c.o -o avrdude  libavrdude.a -lm /opt/homebrew/lib/libelf.a /opt/homebrew/lib/libusb.a /opt/homebrew/lib/libusb-1.0.a /opt/homebrew/lib/libhidapi.dylib /opt/homebrew/lib/libftdi1.a 
[100%] Built target avrdude
/opt/homebrew/Cellar/cmake/3.25.1/bin/cmake -E cmake_progress_start /Users/mcuee/build/avr/avrdude_test/avrdude_main/build_macos/CMakeFiles 0
mcuee@mcuees-Mac-mini avrdude_main % otool -L ./build_macos/src/avrdude
./build_macos/src/avrdude:
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1953.255.0)
	/System/Library/Frameworks/Security.framework/Versions/A/Security (compatibility version 1.0.0, current version 60420.60.24)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1319.0.0)
	/opt/homebrew/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.13.1)

```

mcuee, 2023-01-14:
However, I have not figured a way to do that for Linux.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_main$ cmake -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS="-L/usr/local/lib -pthread -ludev -lz" -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON -B build_static
-- The C compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "3.5.1")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - not found
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Linux-5.15.10-051510-generic
-- CMAKE_SYSTEM: Linux-5.15.10-051510-generic
-- CMAKE_FIND_ROOT_PATH: 
-- CMAKE_C_COMPILER: /usr/bin/cc
-- CONFIG_DIR: /usr/local/etc
-- AVRDUDE_FULL_VERSION: 7.1-20230113 (9cf4f55)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: /usr/lib/x86_64-linux-gnu/libelf.a
-- HAVE_LIBUSB: /usr/local/lib/libusb.a
-- HAVE_LIBUSB_1_0: /usr/local/lib/libusb-1.0.a
-- HAVE_LIBUSB_WIN32: HAVE_LIBUSB_WIN32-NOTFOUND
-- HAVE_LIBHID: HAVE_LIBHID-NOTFOUND
-- HAVE_LIBHIDAPI: /usr/local/lib/libhidapi-libusb.a
-- HAVE_LIBFTDI: HAVE_LIBFTDI-NOTFOUND
-- HAVE_LIBFTDI1: /usr/local/lib/libftdi1.a
-- HAVE_LIBREADLINE: /usr/lib/x86_64-linux-gnu/libreadline.a
-- HAVE_LIBELF_H: 1
-- HAVE_LIBELF_LIBELF_H: 
-- HAVE_USB_H: 1
-- HAVE_LUSB0_USB_H: 
-- HAVE_LIBUSB_H: 
-- HAVE_LIBUSB_1_0_LIBUSB_H: 1
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR: 
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
[... 39 lines cut ...]
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:125: undefined reference to `udev_unref'
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `linux_udev_stop_event_monitor':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:150: undefined reference to `udev_monitor_unref'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:155: undefined reference to `udev_unref'
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `linux_udev_scan_devices':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:275: undefined reference to `udev_enumerate_new'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:281: undefined reference to `udev_enumerate_add_match_subsystem'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:282: undefined reference to `udev_enumerate_add_match_property'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:283: undefined reference to `udev_enumerate_scan_devices'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:284: undefined reference to `udev_enumerate_get_list_entry'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:288: undefined reference to `udev_list_entry_get_name'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:291: undefined reference to `udev_device_new_from_syspath'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:300: undefined reference to `udev_device_unref'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:287: undefined reference to `udev_list_entry_get_next'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:303: undefined reference to `udev_enumerate_unref'
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `linux_udev_hotplug_poll':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:314: undefined reference to `udev_monitor_receive_device'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/avrdude.dir/build.make:154: src/avrdude] Error 1
make[2]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_static'
make[1]: *** [CMakeFiles/Makefile2:101: src/CMakeFiles/avrdude.dir/all] Error 2
make[1]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_static'
make: *** [Makefile:133: all] Error 2

```

mcuee, 2023-01-14:
@mariusgreuel 

It seems to me git main is not too bad to address this issue. Please take a look at the Linux build issue when you got some time. Thanks.

mcuee, 2023-01-14:
Working macOS build scripts.

```
mcuee@mcuees-Mac-mini avrdude_main % cat build_brew.sh 
cmake -S . -B build/shared -DCMAKE_INSTALL_SYSCONFDIR=/opt/homebrew/etc -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_RPATH=@loader_path/../lib -DCMAKE_SHARED_LINKER_FLAGS=-Wl,-undefined,dynamic_lookup   -D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/homebrew/Cellar -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1  -D CMAKE_VERBOSE_MAKEFILE=ON
cmake --build build/shared
#cmake --install build/shared
cmake -S . -B build/static -DCMAKE_INSTALL_SYSCONFDIR=/opt/homebrew/etc cmake -D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS="-L/opt/homebrew/Cellar -Wl,-framework,IOKit -Wl,-framework,CoreFoundation -Wl,-framework,Security"  -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON
cmake --build build/static

mcuee@mcuees-Mac-mini avrdude_main % otool -L build/shared/src/avrdude
build/shared/src/avrdude:
	@rpath/libavrdude.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1319.0.0)
	/opt/homebrew/opt/libusb-compat/lib/libusb-0.1.4.dylib (compatibility version 9.0.0, current version 9.4.0)
	/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/opt/homebrew/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.13.1)
	/opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib (compatibility version 2.0.0, current version 2.5.0)
	/usr/lib/libedit.3.dylib (compatibility version 2.0.0, current version 3.0.0)
mcuee@mcuees-Mac-mini avrdude_main % otool -L build/shared/src/libavrdude.dylib 
build/shared/src/libavrdude.dylib:
	@rpath/libavrdude.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1319.0.0)
	/opt/homebrew/opt/libusb-compat/lib/libusb-0.1.4.dylib (compatibility version 9.0.0, current version 9.4.0)
	/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/opt/homebrew/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.13.1)
	/opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib (compatibility version 2.0.0, current version 2.5.0)
	/usr/lib/libedit.3.dylib (compatibility version 2.0.0, current version 3.0.0)
mcuee@mcuees-Mac-mini avrdude_main % otool -L build/static/src/avrdude         
build/static/src/avrdude:
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1953.255.0)
	/System/Library/Frameworks/Security.framework/Versions/A/Security (compatibility version 1.0.0, current version 60420.60.24)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1319.0.0)
	/opt/homebrew/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.13.1)
mcuee@mcuees-Mac-mini avrdude_main % file -L build/static/src/libavrdude.a
build/static/src/libavrdude.a: current ar archive random library
```

mcuee, 2023-10-18:
@umbynos 

It seems to me now you only need a small patch to get avrdude 7.2 build using your build script.

The patch is a bit strange though. Looks like `WIN32` is not defined for your cross-compiler, is that the case? Thanks.

https://github.com/arduino/avrdude-packing/blob/main/patches/0007-Cmake-fix-build-not-working-with-mingw-toolchain.patch

```
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index c10a77eb..b7769c5b 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -50,7 +50,7 @@ endif()
 include_directories(BEFORE ${CMAKE_CURRENT_BINARY_DIR})
 add_compile_definitions(CONFIG_DIR=\"${CONFIG_DIR}\")
 
-if(WIN32)
+if(WIN32 OR MINGW)
     set(EXTRA_WINDOWS_RESOURCES "${PROJECT_BINARY_DIR}/src/windows.rc")
     set(EXTRA_WINDOWS_LIBRARIES setupapi hid ws2_32)
 endif()
@@ -112,7 +112,7 @@ endif()
 
 configure_file(cmake_config.h.in ac_cfg.h)
 configure_file(avrdude.spec.in avrdude.spec)
-if(WIN32)
+if(WIN32 OR MINGW)
     configure_file(windows.rc.in windows.rc)
 endif()
```

mcuee, 2023-10-19:
@umbynos 

BTW, please help to check the following feature request for the Arduino crossbuild container. Thanks.
* https://github.com/arduino/crossbuild/issues/13

mcuee, 2023-10-22:
The Linux issue is still there. 

Need to figure out which CMAKE_EXE_LINKER_FLAGS to use.

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_main (main)$ cmake -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS="-L/usr/local/lib -pthread -ludev -lz" -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON -B build_static
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "3.5.1")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - not found
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Linux-5.15.0-84-generic
-- CMAKE_SYSTEM: Linux-5.15.0-84-generic
-- CMAKE_FIND_ROOT_PATH: 
-- CMAKE_C_COMPILER: /usr/bin/cc
-- CONFIG_DIR: /usr/local/etc
-- AVRDUDE_FULL_VERSION: 7.2-20231018 (481a91dd)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: /usr/lib/x86_64-linux-gnu/libelf.a
-- HAVE_LIBUSB: /usr/local/lib/libusb.a
-- HAVE_LIBUSB_1_0: /usr/local/lib/libusb-1.0.a
-- HAVE_LIBUSB_WIN32: HAVE_LIBUSB_WIN32-NOTFOUND
-- HAVE_LIBHIDAPI: /usr/local/lib/libhidapi-libusb.a
-- HAVE_LIBFTDI: HAVE_LIBFTDI-NOTFOUND
-- HAVE_LIBFTDI1: /usr/local/lib/libftdi1.a
-- HAVE_LIBREADLINE: /usr/lib/x86_64-linux-gnu/libreadline.a
-- HAVE_LIBSERIALPORT: 1
-- HAVE_LIBELF_H: 1
-- HAVE_LIBELF_LIBELF_H: 
-- HAVE_USB_H: 1
-- HAVE_LUSB0_USB_H: 
-- HAVE_LIBUSB_H: 
-- HAVE_LIBUSB_1_0_LIBUSB_H: 1
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR: 
-- HAVE_LIBGPIOD: 
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DO HAVE    libserialport
-- DISABLED   doc
-- DISABLED   parport
[... 242 lines cut ...]
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `linux_udev_stop_event_monitor':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:150: undefined reference to `udev_monitor_unref'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:155: undefined reference to `udev_unref'
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `linux_udev_scan_devices':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:275: undefined reference to `udev_enumerate_new'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:281: undefined reference to `udev_enumerate_add_match_subsystem'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:282: undefined reference to `udev_enumerate_add_match_property'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:283: undefined reference to `udev_enumerate_scan_devices'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:284: undefined reference to `udev_enumerate_get_list_entry'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:288: undefined reference to `udev_list_entry_get_name'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:291: undefined reference to `udev_device_new_from_syspath'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:300: undefined reference to `udev_device_unref'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:287: undefined reference to `udev_list_entry_get_next'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:303: undefined reference to `udev_enumerate_unref'
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `linux_udev_hotplug_poll':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:314: undefined reference to `udev_monitor_receive_device'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/avrdude.dir/build.make:141: src/avrdude] Error 1
make[2]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_static'
make[1]: *** [CMakeFiles/Makefile2:157: src/CMakeFiles/avrdude.dir/all] Error 2
make[1]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_static'
make: *** [Makefile:139: all] Error 2
```

</details>

mcuee, 2023-10-22:
@Youw

Sorry to disturb you here. Just wondring if you have some ideas here. Thanks.

mcuee, 2023-10-22:
This one seems to work. I am not so sure if it is the right way though. I think it is just a hack.

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_main (main)$ cmake -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS="-L/usr/local/lib -pthread -lz -Wl,--no-as-needed -ldl -lz -ludev" -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON -B build_static
...
...
mcuee@UbuntuSwift3 ~/build/avr/avrdude_main (main)$ cmake --build build_static/
...
...
[100%] Linking C executable avrdude
cd /home/mcuee/build/avr/avrdude_main/build_static/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/avrdude.dir/link.txt --verbose=1
/usr/bin/cc -I/usr/local/include -O2 -g -DNDEBUG -L/usr/local/lib -pthread -lz -Wl,--no-as-needed -ldl -lz -ludev CMakeFiles/avrdude.dir/main.c.o CMakeFiles/avrdude.dir/developer_opts.c.o CMakeFiles/avrdude.dir/whereami.c.o -o avrdude  libavrdude.a -lm /usr/lib/x86_64-linux-gnu/libelf.a /usr/local/lib/libusb.a /usr/local/lib/libusb-1.0.a /usr/local/lib/libhidapi-libusb.a /usr/local/lib/libftdi1.a /usr/lib/x86_64-linux-gnu/libreadline.a /usr/local/lib/libserialport.a -lncurses 
make[2]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_static'
[100%] Built target avrdude
make[1]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_static'
/usr/bin/cmake -E cmake_progress_start /home/mcuee/build/avr/avrdude_main/build_static/CMakeFiles 0

mcuee@UbuntuSwift3 ~/build/avr/avrdude_main (main)$ ldd ./build_static/src/avrdude
	linux-vdso.so.1 (0x00007ffc22f66000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f3f5d43c000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f3f5d420000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f3f5d3f3000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f3f5d2a4000)
	libncurses.so.6 => /lib/x86_64-linux-gnu/libncurses.so.6 (0x00007f3f5d27b000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f3f5d24b000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f3f5d226000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f3f5d034000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f3f5d674000)
```

Youw, 2023-10-23:
> -D USE_STATIC_LIBS=ON

This is fine, should be the only required parameter.

> -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS="-L/usr/local/lib -pthread -lz -Wl,--no-as-needed -ldl -lz -ludev"

Needing to pass these from command line is not fine.
By all the best practices the build script of a library/executable should know its own dependencies and add them explicitly w/o needing to do so manually. (Unless for some rare corner cases, which is not the case.)

mcuee, 2023-11-04:
PR https://github.com/avrdudes/avrdude/pull/1540 has been merged. 
* https://github.com/avrdudes/avrdude/pull/1540

This issue is still valid, therefore, I will keep it open.

mcuee, 2023-11-05:
@Youw 

It is a bit strange that cross compile works fine with the script in PR #1540 (now merged). But it does not work if I adopt similar approach with native build under Linux.

Just wondering if you have some ideas or not.

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_main (main)$ cat build_static1.sh 
cmake -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS="-pthread -static-libgcc -static-libstdc++" -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON -B build_static1
cmake --build build_static1/
```

https://github.com/avrdudes/avrdude/blob/main/.github/workflows/arduino_packing.yml

For example, the following is from latest git main github avrdude-packing build.
https://github.com/avrdudes/avrdude/actions/runs/6746001868

```
cmake -DCMAKE_C_COMPILER=${{ matrix.config.cross_compile }}-gcc -DCMAKE_CXX_COMPILER=${{ matrix.config.cross_compile }}-g++ -DCMAKE_EXE_LINKER_FLAGS="-static-libgcc -static-libstdc++" -DCMAKE_C_FLAGS="-I/opt/lib/${{ matrix.config.cross_compile }}/include/libusb-1.0/ -I/opt/lib/${{ matrix.config.cross_compile }}/include -pthread" -DCMAKE_PREFIX_PATH=/opt/lib/${{ matrix.config.cross_compile }}/ -DHAVE_LIBFTDI="NO" -B build/
        fi
```

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin/dist/avrdude_Linux_64bit $ ldd ./bin/avrdude 
	linux-vdso.so.1 (0x00007ffd11be7000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f0a124e6000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f0a124c3000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f0a122d1000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f0a12663000)
```

Youw, 2023-11-07:
I'm not sure what is not working?

mcuee, 2023-11-07:
> I'm not sure what is not working?

@Youw 

The linking step failed.

<details>
<summary> failed build log with verbose output </summary>

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_main (main)$ cat build_static1.sh 
cmake -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS="-pthread -static-libgcc -static-libstdc++" -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON -B build_static1
cmake --build build_static1/

mcuee@UbuntuSwift3 ~/build/avr/avrdude_main (main)$ ./build_static1.sh 
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Linux-5.15.0-87-generic
-- CMAKE_SYSTEM: Linux-5.15.0-87-generic
-- CMAKE_FIND_ROOT_PATH: 
-- CMAKE_C_COMPILER: /usr/bin/cc
-- CONFIG_DIR: /usr/local/etc
-- AVRDUDE_FULL_VERSION: 7.2-20231103 (08b82ea6)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: /usr/lib/x86_64-linux-gnu/libelf.a
-- HAVE_LIBUSB: /usr/local/lib/libusb.a
-- HAVE_LIBUSB_1_0: /usr/local/lib/libusb-1.0.a
-- HAVE_LIBUSB_WIN32: HAVE_LIBUSB_WIN32-NOTFOUND
-- HAVE_LIBHIDAPI: /usr/local/lib/libhidapi-libusb.a
-- HAVE_LIBFTDI: HAVE_LIBFTDI-NOTFOUND
-- HAVE_LIBFTDI1: /usr/local/lib/libftdi1.a
-- HAVE_LIBREADLINE: /usr/lib/x86_64-linux-gnu/libreadline.a
-- HAVE_LIBSERIALPORT: 1
-- HAVE_LIBELF_H: 1
-- HAVE_LIBELF_LIBELF_H: 
-- HAVE_USB_H: 1
-- HAVE_LUSB0_USB_H: 
-- HAVE_LIBUSB_H: 
-- HAVE_LIBUSB_1_0_LIBUSB_H: 1
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR: 
-- HAVE_LIBGPIOD: 
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DO HAVE    libserialport
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /home/mcuee/build/avr/avrdude_main/build_static1
Change Dir: '/home/mcuee/build/avr/avrdude_main/build_static1'

Run Build Command(s): /usr/bin/cmake -E env VERBOSE=1 /usr/bin/make -f Makefile
/usr/bin/cmake -S/home/mcuee/build/avr/avrdude_main -B/home/mcuee/build/avr/avrdude_main/build_static1 --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/mcuee/build/avr/avrdude_main/build_static1/CMakeFiles /home/mcuee/build/avr/avrdude_main/build_static1//CMakeFiles/progress.marks
/usr/bin/make  -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/mcuee/build/avr/avrdude_main/build_static1'
/usr/bin/make  -f src/CMakeFiles/conf.dir/build.make src/CMakeFiles/conf.dir/depend
make[2]: Entering directory '/home/mcuee/build/avr/avrdude_main/build_static1'
cd /home/mcuee/build/avr/avrdude_main/build_static1 && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/mcuee/build/avr/avrdude_main /home/mcuee/build/avr/avrdude_main/src /home/mcuee/build/avr/avrdude_main/build_static1 /home/mcuee/build/avr/avrdude_main/build_static1/src /home/mcuee/build/avr/avrdude_main/build_static1/src/CMakeFiles/conf.dir/DependInfo.cmake "--color="
[... 136 lines cut ...]
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `linux_udev_stop_event_monitor':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:150: undefined reference to `udev_monitor_unref'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:155: undefined reference to `udev_unref'
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `linux_udev_scan_devices':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:275: undefined reference to `udev_enumerate_new'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:281: undefined reference to `udev_enumerate_add_match_subsystem'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:282: undefined reference to `udev_enumerate_add_match_property'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:283: undefined reference to `udev_enumerate_scan_devices'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:284: undefined reference to `udev_enumerate_get_list_entry'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:288: undefined reference to `udev_list_entry_get_name'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:291: undefined reference to `udev_device_new_from_syspath'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:300: undefined reference to `udev_device_unref'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:287: undefined reference to `udev_list_entry_get_next'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:303: undefined reference to `udev_enumerate_unref'
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `linux_udev_hotplug_poll':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:314: undefined reference to `udev_monitor_receive_device'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/avrdude.dir/build.make:141: src/avrdude] Error 1
make[2]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_static1'
make[1]: *** [CMakeFiles/Makefile2:157: src/CMakeFiles/avrdude.dir/all] Error 2
make[1]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_static1'
make: *** [Makefile:139: all] Error 2
```

</details>

mcuee, 2023-11-07:
@Youw 

Basically the issue is that the linking step is missing `-ludev -ldl -lz`.

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_main/build_static1/src (main)$ /usr/bin/cc -I/usr/local/include -O2 -g -DNDEBUG -pthread -static-libgcc -static-libstdc++ CMakeFiles/avrdude.dir/main.c.o CMakeFiles/avrdude.dir/developer_opts.c.o CMakeFiles/avrdude.dir/whereami.c.o -o avrdude  libavrdude.a -lm /usr/lib/x86_64-linux-gnu/libelf.a /usr/local/lib/libusb.a /usr/local/lib/libusb-1.0.a /usr/local/lib/libhidapi-libusb.a /usr/local/lib/libftdi1.a /usr/lib/x86_64-linux-gnu/libreadline.a /usr/local/lib/libserialport.a -lncurses 
/usr/bin/ld: /usr/local/lib/libusb.a(libusb_la-core.o): in function `libusb_dl_init':
/home/mcuee/build/libusb-compat-0.1/libusb/libusb-dload.h:179: undefined reference to `dlopen'
/usr/bin/ld: /home/mcuee/build/libusb-compat-0.1/libusb/libusb-dload.h:181: undefined reference to `dlsym'
/usr/bin/ld: /home/mcuee/build/libusb-compat-0.1/libusb/libusb-dload.h:182: undefined reference to `dlsym'
/usr/bin/ld: /home/mcuee/build/libusb-compat-0.1/libusb/libusb-dload.h:183: undefined reference to `dlsym'
/usr/bin/ld: /home/mcuee/build/libusb-compat-0.1/libusb/libusb-dload.h:184: undefined reference to `dlsym'
/usr/bin/ld: /home/mcuee/build/libusb-compat-0.1/libusb/libusb-dload.h:185: undefined reference to `dlsym'
/usr/bin/ld: /usr/local/lib/libusb.a(libusb_la-core.o):/home/mcuee/build/libusb-compat-0.1/libusb/libusb-dload.h:186: more undefined references to `dlsym' follow
/usr/bin/ld: /usr/local/lib/libusb.a(libusb_la-core.o): in function `libusb_dl_init':
/home/mcuee/build/libusb-compat-0.1/libusb/libusb-dload.h:210: undefined reference to `dlerror'
/usr/bin/ld: /usr/local/lib/libusb.a(libusb_la-core.o): in function `libusb_dl_exit':
/home/mcuee/build/libusb-compat-0.1/libusb/libusb-dload.h:220: undefined reference to `dlclose'
/usr/bin/ld: /home/mcuee/build/libusb-compat-0.1/libusb/libusb-dload.h:220: undefined reference to `dlclose'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libelf.a(elf_compress.o): in function `__libelf_compress':
(.text+0x10a): undefined reference to `deflateInit_'
/usr/bin/ld: (.text+0x1ce): undefined reference to `deflate'
/usr/bin/ld: (.text+0x23c): undefined reference to `deflateEnd'
/usr/bin/ld: (.text+0x25e): undefined reference to `deflateEnd'
/usr/bin/ld: (.text+0x2fc): undefined reference to `deflateEnd'
/usr/bin/ld: (.text+0x324): undefined reference to `deflateEnd'
/usr/bin/ld: (.text+0x380): undefined reference to `deflate'
/usr/bin/ld: (.text+0x426): undefined reference to `deflateEnd'
/usr/bin/ld: (.text+0x456): undefined reference to `deflateEnd'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libelf.a(elf_compress.o): in function `__libelf_decompress':
(.text+0x52e): undefined reference to `inflateInit_'
/usr/bin/ld: (.text+0x55c): undefined reference to `inflate'
/usr/bin/ld: (.text+0x569): undefined reference to `inflateReset'
/usr/bin/ld: (.text+0x57d): undefined reference to `inflateEnd'
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `udev_device_info':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:214: undefined reference to `udev_device_get_devnode'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:219: undefined reference to `udev_device_get_sysname'
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `udev_hotplug_event':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:237: undefined reference to `udev_device_get_action'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:262: undefined reference to `udev_device_unref'
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `linux_udev_event_thread_main':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:197: undefined reference to `udev_monitor_receive_device'
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `linux_udev_start_event_monitor':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:49: undefined reference to `udev_new'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:55: undefined reference to `udev_monitor_new_from_netlink'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:61: undefined reference to `udev_monitor_filter_add_match_subsystem_devtype'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:67: undefined reference to `udev_monitor_enable_receiving'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:72: undefined reference to `udev_monitor_get_fd'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:121: undefined reference to `udev_monitor_unref'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:125: undefined reference to `udev_unref'
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `linux_udev_stop_event_monitor':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:150: undefined reference to `udev_monitor_unref'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:155: undefined reference to `udev_unref'
/usr/bin/ld: /usr/local/lib/libusb-1.0.a(linux_udev.o): in function `linux_udev_scan_devices':
/home/mcuee/build/libusb/libusb/os/linux_udev.c:275: undefined reference to `udev_enumerate_new'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:281: undefined reference to `udev_enumerate_add_match_subsystem'
/usr/bin/ld: /home/mcuee/build/libusb/libusb/os/linux_udev.c:282: undefined reference to `udev_enumerate_add_match_property'
[... cut ...]
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f71d4f04000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f71d4ed4000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f71d4ea7000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f71d4ea1000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f71d4e85000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f71d4e62000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f71d4c6e000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f71d5284000)
```

Youw, 2023-11-07:
> Basically the issue is that the linking step is missing -ludev -ldl -lz.

Right, so those should be explicitly added in CMake for platforms that requires it.

mcuee, 2023-11-07:
> > Basically the issue is that the linking step is missing -ludev -ldl -lz.
> 
> Right, so those should be explicitly added in CMake for platforms that requires it.

Take note this is only required for static linking under Linux/macOS/BSDs (the default is using dynamic linking. So probably this kind of extra options can be added for them.

For Windows by default avrdude is using static linking and not really allowing dynamic linking as of now (maybe no point to add the support for dynamic linking either).

mcuee, 2024-03-02:
@ndim 

I am not so sure if you have some interests in helping here as well. I believe the fix is not that difficult but unfortunately I do not know much about CMake.

BTW, we may extend the subject for autotools build as well.

ndim, 2024-03-02:
@mcuee Yes, I am interested. I am using avrdude to learn cmake with, I want to build a good avrdude package for Fedora, and I want avrdude to have a good Autotools build. Thanks for mentioning me here.

TLDR: A large number of different builds are possible, and we need to decide which ones make sense for us to support, and determine how to.

I will connect the following general thoughts with the existing build system some time in the next few days.

There are two build products: The libavrdude library and the `avrdude` tool executable. Each can be linked statically or dynamically to any subset of its dependencies (libelf, libusb, libftdi, libhidapi, etc). In the case of the `avrdude` executable, libavrdude is one of its dependencies. And libavrdude itself can be built as a static or as a dynamic library. And we are building for different platforms: static by default like Windows and dynamic by default like BSD, Linux. Not sure whether OSX is 100% a part of the dynamic by default camp.

Without precisely counting them all, that results in a huge number of combinations, and not all of them make sense.

Here are a few of those combinations which make sense, starting with the most extreme ones:

  * An "all-dynamic" build producing a shared libavrdude (dynamically linked to all dependencies) and an `avrdude` executable dynamically linked to libavrdude and all other libraries. That is the ideal build for Linux distro packages, BSD ports, homebrew etc. in a world with a stable libavrdude API and ABI.

  * A "static tool-only" build of the `avrdude` executable statically linked to all libraries, including libavrdude and libc. That is the ideal build for producing self contained `avrdude` tool binaries for probably all operating systems. This build does not even need to install libavrdude alongside the `avrdude` executable.

  *  For Windows and its typical static by default linking, building libavrdude as a static library makes sense. Probably statically linked for most to all of its dependencies. 

  * Until the libavrdude API and ABI have stabilized, building libavrdude as a static library for library users to develop their own tools with also makes sense on dynamic linking by default platforms (BSD, Linux, MacOS). This static lib could then be dynamically linked to all its dependencies.

  * The current Fedora package ships an avrdude tool executable which is dynamically linked to all of its and libavrdude's dependencies, but has been statically linked to libavrdude. Libavrdude by itself is not shipped at all, due to the state of API/ABI stability. This could be called a "dynamic tool-only" build.

So someone building the code might want to select:

  * build products: libavrdude library, avrdude tool, or both
  * if building libavrdude: as static library or as dynamic library or both
  * How to link libavrdude library and avrdude tool against their dependencies: statically only, dynamically only, dynamically if possible and statically else, statically if possible and dynamically else. Specifying per library dependency probably makes no sense.

Defaults could differ depending on the target OS, e.g. mostly static for Windows and mostly dynamic for BSD, Linux, OSX.

mcuee, 2024-03-02:
@ndim 

Thanks a lot for the detailed analysis and I agree with you.

> Defaults could differ depending on the target OS, e.g. mostly static for Windows and mostly dynamic for BSD, Linux, OSX.

For avrdude binary, I think this is the case already for the CMake build (MSVC and MSYS2 mingw). 

> A "static tool-only" build of the avrdude executable statically linked to all libraries, including libavrdude and libc. That is the ideal build for producing self contained avrdude tool binaries for [...]

Yes there are pretty strong requests to have this types of static binaries for Linux and macOS. Arduino avrdude-packing project is distributing such binaries for Linux, macOS and Windows. We have the CI script based on Arduino avrdude-packing project as well but needs some improvement (eg: no libserialport support).
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/arduino_packing.yml
https://github.com/arduino/avrdude-packing (only up to 7.2 release).

For libavrdude library, currently it is not possible to build dynamic library for Windows and there may be such use case if we want to allow 3rd party apps developer who do not want to open source their implementation. There may be such use case (dynamic library) for other OS as well. This is a moot point as of now there are no real GUI application which leverage libarvdude library.
* #1414 
* #774 (I may be wrong but this could be a prerequisite to fix #1414)

mcuee, 2024-03-02:
Example working static build under Linux.

```
cmake -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS="-L/usr/local/lib -pthread -Wl,--no-as-needed -ldl -lz -ludev" -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON -B build_static
cmake --build build_static/
```

<details>
<summary> detail build log under Ubuntu Linux 20.04 </summary>

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_main (main)$ ./build_static.sh 
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "3.5.1")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Linux-5.15.0-94-generic
-- CMAKE_SYSTEM: Linux-5.15.0-94-generic
-- CMAKE_FIND_ROOT_PATH: 
-- CMAKE_C_COMPILER: /usr/bin/cc
-- CONFIG_DIR: /usr/local/etc
-- AVRDUDE_FULL_VERSION: 7.3-20240229 (34366a6b)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: /usr/lib/x86_64-linux-gnu/libelf.a
-- HAVE_LIBUSB: /usr/local/lib/libusb.a
-- HAVE_LIBUSB_1_0: /usr/local/lib/libusb-1.0.a
-- HAVE_LIBUSB_WIN32: HAVE_LIBUSB_WIN32-NOTFOUND
-- HAVE_LIBHIDAPI: /usr/local/lib/libhidapi-libusb.a
-- HAVE_LIBFTDI: HAVE_LIBFTDI-NOTFOUND
-- HAVE_LIBFTDI1: /usr/local/lib/libftdi1.a
-- HAVE_LIBREADLINE: /usr/lib/x86_64-linux-gnu/libreadline.a
-- HAVE_LIBSERIALPORT: 1
-- HAVE_LIBELF_H: 1
-- HAVE_LIBELF_LIBELF_H: 
-- HAVE_USB_H: 1
-- HAVE_LUSB0_USB_H: 
-- HAVE_LIBUSB_H: 
-- HAVE_LIBUSB_1_0_LIBUSB_H: 1
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR: 
-- HAVE_LIBGPIOD: 
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
[... 199 lines cut ...]
[100%] Built target avrdude
make[1]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_static'
/usr/bin/cmake -E cmake_progress_start /home/mcuee/build/avr/avrdude_main/build_static/CMakeFiles 0
```

</details>

Results:
```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_main (main)$ ls ./build_static/src/
ac_cfg.h  avrdude.conf     avrdude.spec  cmake_install.cmake  Makefile
avrdude   avrdude.conf.in  CMakeFiles    libavrdude.a

mcuee@UbuntuSwift3 ~/build/avr/avrdude_main (main)$ ldd ./build_static/src/avrdude
	linux-vdso.so.1 (0x00007ffe44b9c000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fafe4f71000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fafe4f55000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fafe4f28000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fafe4dd9000)
	libncurses.so.6 => /lib/x86_64-linux-gnu/libncurses.so.6 (0x00007fafe4db0000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007fafe4d80000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fafe4d5b000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fafe4b69000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fafe54d2000)
```

stefanrueger, 2024-03-03:
@ndim Good analysis. 

> "all-dynamic" build [...] ideal [...] in a world with a stable libavrdude API and ABI

Isn't the critical part here that avrdude is linked with the libavrdude dynamic library that is shipped at the same time?Could that be enforced by correct versioning? I agree, linking avrdude statically against libavrdude guards against this type of error.

Anyway, I have too little experience in building projects for different platforms to have a preference for which options the AVRDUDE project should have.

[15 more replies]

closed 2025-05-22 by mcuee