avrdudes/avrdude #1160: Changing verbosity and quell level in terminal mode is not effective

mcuee, 2022-10-26:
It seems to me changing changing verbosity and quell level in terminal mode is not effective.

```
C:\work\avr\avrdude_test\avrdude_bin> ./avrdude_git -c serialupdi -P COM16 -p m4808 -qq -t
avrdude> verbose
>>> verbose
Verbosity level: 0
avrdude> quell
>>> quell
Quell level: 2
avrdude> verbose 3
>>> verbose 3
New verbosity level: 3
avrdude> quell 3
>>> quell 3
New quell level: 3
avrdude> write flash 0 0x55
>>> write flash 0 0x55
avrdude> flush
>>> flush
avrdude> dump flash 0 0x10
>>> dump flash 0 0x10
0000  55 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |U...............|

avrdude> write flash 0 0xaa
>>> write flash 0 0xaa
avrdude> flush
>>> flush
avrdude_git: synching cache to device ... avrdude_git serialupdi_page_erase() [serialupdi.c:884] error: page erase not implemented yet
reading/chip erase/writing cycle needed ... done
avrdude> quit
>>> quit
avrdude> avrdude_git: leaving NVM programming mode

avrdude_git done.  Thank you.

```

stefanrueger, 2022-10-26:
Perhaps try `quell 0` and `verbosity 3` if you want the same as `-vvv` (without any `-q`)?

mcuee, 2022-10-26:
I misunderstand the relationship between `quell` and `verbose` command. I will close this issue.

closed 2022-10-26 by mcuee