avrdudes/avrdude #1161: Linux avrdude stk500_recv: programmer is not responding

maxmbed, 2022-10-28:
Avrdude fails to receive data with timeout on Linux host machine. At least on two machines running both Debian 11, 5.10.0-19-amd64.
The board I am using is [Adafruit Feather328p](https://learn.adafruit.com/adafruit-feather-328p-atmega328-atmega328p) and I don't have other board to test but I have tried that board and same command on Windows machine is it works with success.

I have checked DTR line and it is working properly to reset MCU in bootloader mode.

This issue seems to happens later, after serial port is opened and avrdude get sync byte successfully.
Below is log when avrdude try to initialize using stk500 programmer (`stk500_initialize`)

The issued command is following:

```sh
avrdude -c arduino -p m328p -P /dev/ttyUSB0 -b 57600 -U flash:w:../avr/board-reference/feather-328p/build/led.hex:i
```
Then failure:

```sh
avrdude: Send: A [41] . [81]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: A [41] . [82]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: B [42] . [86] . [00] . [00] . [01] . [01] . [01] . [01] . [03] . [ff] . [ff] . [ff] . [ff] . [00] . [80] . [04] . [00] . [00] . [00] . [80] . [00]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: initialization failed, rc=-1
```

I attach the full log with max verbose mode enabled (-vvvv) for reference [avrdude-programmer-not-respond-log.txt](https://github.com/avrdudes/avrdude/files/9889094/avrdude-programmer-not-respond-log.txt)

Can someone make a try and tell wehther avrdude is working on Linux ? (If so, please specify your distrubtion info)

maxmbed, 2022-10-28:
I recompile avrdude with build_type set to Debug and place break point at location of failure `ser_recv(): programmer is not responding`.

It seems the system call `select()` always return 0, which mean serial port never become ready for read.
If rebuild with select() section commented, the system call read() returns `Resource temporarily unavailable`

What bugged me too is after first trial of read failure, the bootlodaer seem to have exited already.

mcuee, 2022-10-28:
Which version of avrdude are you using? Can you try latest git main? Do you encounter the issues with other Linux distros or other machines?

maxmbed, 2022-10-28:
@mcuee. 
> Which version of avrdude are you using?

Debian 11 ships avrdude in version 6.3-20171130

> Can you try latest git main?
Yes, I did try with latest avrdude changes (`~a1018999`)
Same result.

I don't have other AVR board, nor ISP programemr but I do have Teensy LC.
Maybe I can make a try with (?)

maxmbed, 2022-10-28:
> Do you encounter the issues with other Linux distros or other machines?

I only have Debian machines and the issue were reproduced on both ones.

Regarding bootloader, since it worked fine on Windows, I will not go reflash feather328 for now. I mean, I don't heave "easy to go" hardware and nor the time.

I would prefer to get feedback of other people first. Then going further with bootloader reflash if necessary.

mcuee, 2022-10-29:
I can only telll you that arduino is working well under Linux for me.

stefanrueger, 2022-10-29:
> What bugged me too is after first trial of read failure, the bootlodaer seem to have exited already.

If the bootloader has a vvv short WDT timeout then subtle different wait times under windows and Linux whilst draining the input *might* lead to what you describe. I cannot reliably get `-c arduino` to work for WDT of 256 ms or less. 512 ms works for me under Linux (Ubuntu 20.04), but even that might be that I'm lucky.

maxmbed, 2022-10-31:
Hi,
thanks for your feedback, guys.

@mcuee, can it be possible that the issue is board specific ? If I can find an Arduino Nano or any with 16MHz crystal, I will make a try.

@stefanrueger, I am not so sure. If the Watchdog Timer is triggering, I should see the feather red led blink again. Or maybe the WDT interrupt performs a jump to application ? I am not even sure which bootloader feather328 is using to verify. Could it be Optiboot ? If so, WDT is 1 sec by default.

mcuee, 2022-10-31:
> Hi, thanks for your feedback, guys.
> 
> @mcuee, can it be possible that the issue is board specific ? If I can find an Arduino Nano or any with 16MHz crystal, I will make a try.

Yes it could be board specific. But I am more worried about bad bootloader firmware. I've encountered that myself from the cheap Uno/Nano clones -- I always burn the good bootloader first. 

If you want to get a programmer, usbasp is a good one and cheap enough.
Ref: 
* https://github.com/avrdudes/avrdude/discussions/1021#discussioncomment-4011988

mcuee, 2022-11-06:
I will close this issue for now. Please re-open if you still encounter the issue with avrdude. You can also try to ask in Adafruit as it is very much specific to the board.

closed 2022-11-06 by mcuee