avrdudes/avrdude #1165: To double check whether optiboot_dx and optiboot_x are working with avrdude

mcuee, 2022-11-01:
Continuation of the following two.
* https://github.com/avrdudes/avrdude/pull/1140
* https://github.com/avrdudes/avrdude/issues/1120

mcuee, 2022-11-01:
As of now both @MCUdude and I have no <=64KB AVR DA/DB parts for testing. We have the AVR64DD32 but unfortunately DxCore and Optiboot_dx do not work with AVR DD parts yet.

mcuee, 2022-11-04:
BTW, I have tested optiboot_x for ATmega4808/4809 (MegaCoreX) and ATtiny1627 (megaTinyCore).
https://github.com/avrdudes/avrdude/issues/1166

MCUdude, 2022-11-17:
I can confirm that optiboot_dx works with the AVR64DD32 and the upstream version of Avrdude:

```
$ ./avrdude -carduino -pavr64db32 -D -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_534228/Blink.ino.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9618 (probably avr64db32)
avrdude: reading input file /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_534228/Blink.ino.hex for flash
         with 866 bytes in 1 section within [0x200, 0x561]
         using 2 pages and 158 pad bytes
avrdude: writing 866 bytes flash ...

Writing | ################################################## | 100% 0.08 s 

avrdude: 866 bytes of flash written
avrdude: verifying flash memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_534228/Blink.ino.hex

Reading | ################################################## | 100% 0.05 s 

avrdude: 866 bytes of flash verified

avrdude done.  Thank you.

$ ./avrdude -carduino -pavr64db32 -D -Uflash:w:chaucer64k_0x200_offset_AVR-DB.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9618 (probably avr64db32)
avrdude: reading input file chaucer64k_0x200_offset_AVR-DB.hex for flash
         with 62540 bytes in 1 section within [0x200, 0xf64b]
         using 123 pages and 436 pad bytes
avrdude: writing 62540 bytes flash ...

Writing | ################################################## | 100% 9.75 s 

avrdude: 62540 bytes of flash written
avrdude: verifying flash memory against chaucer64k_0x200_offset_AVR-DB.hex

Reading | ################################################## | 100% 6.10 s 

avrdude: 62540 bytes of flash verified

avrdude done.  Thank you.

```

EEPROM write does not work, but this is to be expected since the bootloader is built without EEPROM support to fit within 512 bytes:

```
echo "Hello World" | ./avrdude -carduino -pavr64db32 -Ueeprom:w:-:r

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9618 (probably avr64db32)
avrdude: reading input file <stdin> for eeprom
         with 12 bytes in 1 section within [0, 11]
avrdude: writing 12 bytes eeprom ...

Writing | ################################################## | 100% 0.08 s 

avrdude: 12 bytes of eeprom written
avrdude: verifying eeprom memory against <stdin>

Reading | ################################################## | 100% 0.07 s 

avrdude error: verification mismatch, first encountered at addr 0x0000
        device 0xff != input 0x48
avrdude error: verification mismatch

avrdude done.  Thank you.
```

mcuee, 2022-11-18:
> I can confirm that optiboot_dx works with the AVR64DD32 and the upstream version of Avrdude.

Wonderful. Just wondering which bootloader hex file you are using. I am getting another AVR64DD32 Curiosity Nano after the first one died.

MCUdude, 2022-11-19:
I flashed the `DxCore/megaavr/bootloaders/hex/optiboot_dx64_ser2.hex` file

These are the fuse settings:

```
$ ./avrdude -cpickit4_updi -pavr64db32 -Ufuse0:r:-:h -Ufuse0:r:-:h -Ufuse1:r:-:h -Ufuse2:r:-:h -Ufuse4:r:-:h -Ufuse5:r:-:h -Ufuse6:r:-:h -Ufuse7:r:-:h -Ufuse8:r:-:h

         Vtarget                      : 5.07 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9618 (probably avr64db32)
avrdude: reading fuse0/wdtcfg memory ...
avrdude: writing output file <stdout>
0x0
avrdude: reading fuse0/wdtcfg memory ...
avrdude: writing output file <stdout>
0x0
avrdude: reading fuse1/bodcfg memory ...
avrdude: writing output file <stdout>
0x0
avrdude: reading fuse2/osccfg memory ...
avrdude: writing output file <stdout>
0x0
avrdude: reading fuse4/tcd0cfg memory ...
avrdude: writing output file <stdout>
0xff
avrdude: reading fuse5/syscfg0 memory ...
avrdude: writing output file <stdout>
0xc8
avrdude: reading fuse6/syscfg1 memory ...
avrdude: writing output file <stdout>
0xc
avrdude: reading fuse7/codesize memory ...
avrdude: writing output file <stdout>
0x0
avrdude: reading fuse8/bootsize memory ...
avrdude: writing output file <stdout>
0x1

avrdude done.  Thank you.

```

>  I am getting another AVR64DD32 Curiosity Nano after the first one died.

Wouldn't it be cheaper to just replace the AVR-DD chip? They're fairly easy to solder

mcuee, 2022-11-20:
> Wouldn't it be cheaper to just replace the AVR-DD chip? They're fairly easy to solder

The shipment cost is killing factor to get chips here in Singapore. But I will try to get the raw chips as well.

mcuee, 2022-11-20:
Now I will close this issue since it is confirmed that git main now works well after the merge of PR #1140.
* #1140

closed 2022-11-20 by mcuee

mcuee, 2022-12-25:
I got the new AVR64DD32 Curiosity Nano and now it works (DxCore 1.5.1 beta).

* https://github.com/SpenceKonde/DxCore/issues/354#issuecomment-1364672746

```
C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude -CC:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10/avrdude.conf -v -pavr64dd32 -cpkobn_updi -Pusb -e -Ufuse0:w:0b00000000:m -Ufuse1:w:0x00:m -Ufuse5:w:0b11010000:m -Ufuse6:w:0b00001100:m -Ufuse8:w:0x01:m -Ufuse7:w:0x00:m -Uflash:w:C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10/bootloaders/hex/optiboot_64dd_ser0_alt3_all_8sec.hex:i 

avrdude: Version 7.0-20221101 (4c92030)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\avrdude.conf

         Using Port                    : usb
         Using Programmer              : pkobn_updi
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : AVR64DD32
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       codesize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootsize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   16      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no        125  125      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         16    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256    1      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      65536  512      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Curiosity nano (nEDBG) in UPDI mode
         ICE HW version  : 0
         ICE FW version  : 1.25 (rel. 116)
         Serial number   : MC020019502HIP000639
         Vtarget         : 3.31 V
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: partial Family_ID returned: "AVR "
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e961a (probably avr64dd32)
erasing chip
avrdude: reading input file 0b00000000 for fuse0/wdtcfg
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse0/wdtcfg ...
avrdude: 1 byte of fuse0/wdtcfg written
avrdude: verifying fuse0/wdtcfg memory against 0b00000000
avrdude: 1 byte of fuse0/wdtcfg verified
avrdude: reading input file 0x00 for fuse1/bodcfg
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse1/bodcfg ...
avrdude: 1 byte of fuse1/bodcfg written
avrdude: verifying fuse1/bodcfg memory against 0x00
avrdude: 1 byte of fuse1/bodcfg verified
avrdude: reading input file 0b11010000 for fuse5/syscfg0
         with 1 byte in 1 section within [0, 0]
[... 43 lines cut ...]
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink_64dd32.ino.avr64dd32opti.24c0.m{build.highestcb}.wO.v151.hex for flash
         with 816 bytes in 1 section within [0x200, 0x52f]
         using 2 pages and 208 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 816 bytes flash ...

Writing | ################################################## | 100% 0.16 s

avrdude: 816 bytes of flash written
avrdude: verifying flash memory against .\Blink_64dd32.ino.avr64dd32opti.24c0.m{build.highestcb}.wO.v151.hex

Reading | ################################################## | 100% 0.11 s

avrdude: 816 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM5 -p avr64dd32 -xshowall -xbootsize=512 -xstrict

avrdude: AVR device initialized and ready to accept instructions
0 2022-12-25 20.13 Blink_64dd32.ino.avr64dd32opti.24c0.m{build.highestcb}.wO.v151.hex 1328 store 64130 meta 78 boot 512 x0.0 ......... vector 0 (RESET) AVR64DD32

```