avrdudes/avrdude #1166: megaTinyCore optiboot_x does not work with avrdude git main

mcuee, 2022-11-01:
I can not get avrdude git main to work with megaTinyCore optiboot_x using ATtiny1627 Curiosity Nano board.

More details here.
* https://github.com/avrdudes/avrdude/pull/1140#issuecomment-1294834366
* https://github.com/SpenceKonde/megaTinyCore/issues/852#issuecomment-1298152676

mcuee, 2022-11-01:
@MCUdude 

Just wondering if you can help checking on megaTinyCore and MegaCoreX. Thanks.

MCUdude, 2022-11-01:
@mcuee sure! I'm currently on vacation, and didn't brought the right hardware with me.
I'm _pretty sure_ optiboot_x for MegaCoreX works without any problems, but I can double-check when I'm back on Monday next week.

@mcuee meanwhile, can you try to change `a_div` back to `2` on line 770 and re-compile?

https://github.com/avrdudes/avrdude/blob/b5699663c4ebdb185c21f2f1ca02610cd078ed77/src/stk500.c#L762-L772

It is strange that it doesn't work with Avrdude 7.0, since no major changes to stk500.c was done after the project was migrated over to Github. Can you try with Avrdude 6.4? The follwing commit is present in Avrdude 6.4 and later, but not in 6.3. Can you try to revert this as well?

If this doesn't help, please try to roll even further back to see if you can narrow it down to a single commit:
https://github.com/avrdudes/avrdude/commits/main/src/stk500.c

MCUdude, 2022-11-01:
@mcuee from the output you provided, It doesn't look like you actually tested `-carduino` with Avrdude 6.3:

```
C:\Users\xfchen\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude 
-CC:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1/avrdude.conf 
-pattiny1627 -ccuriosity_updi -Pusb -D -U .\Blink.ino.t1627opti.20c0.mB1.v261.hex -qq && echo OK
OK
```

Also, you shouldn't use the `-D` flag with non-bootloader based programmers, since the `-D` flag prevents a page/chip erase, and may leave the previous flash content intact.

mcuee, 2022-11-01:
Looks like a megaTinyCore issue to me now. 

I use Arduino IDE 2.0.1 and now it okay to burn the bootloader (no need to manually using avrdude) and upload the sketch. Take note I am using avrdude git master and not avrdude 6.3.

```
"C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude" "-CC:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1/avrdude.conf" -v -pattiny1627 -cpkobn_updi -Pusb -e "-Ufuse0:w:0x00:m" "-Ufuse1:w:0b00000101:m" "-Ufuse5:w:0b11110110:m" "-Ufuse6:w:0x04:m" "-Ufuse8:w:0x00:m" "-Ufuse7:w:0x00:m" "-Uflash:w:C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1/bootloaders/hex/optiboot_Curiosity1627.hex:i"

avrdude: Version avrdude-main-test-libusb-compat
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1\avrdude.conf

         Using Port                    : usb
         Using Programmer              : pkobn_updi
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATtiny1627
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   32      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      16384   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Curiosity nano (nEDBG) in UPDI mode
         ICE HW version  : 0
         ICE FW version  : 1.21 (rel. 37)
         Serial number   : MCHP3258021800000382
         Vtarget         : 3.31 V
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: partial Family_ID returned: "tiny"
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9428 (probably t1627)
erasing chip
avrdude: reading input file 0x00 for fuse0/wdtcfg
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse0/wdtcfg ...
avrdude: 1 byte of fuse0/wdtcfg written
avrdude: verifying fuse0/wdtcfg memory against 0x00
avrdude: 1 byte of fuse0/wdtcfg verified
avrdude: reading input file 0b00000101 for fuse1/bodcfg
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse1/bodcfg ...
avrdude: 1 byte of fuse1/bodcfg written
[... 17 lines cut ...]
avrdude: 1 byte of fuse8/bootend written
avrdude: verifying fuse8/bootend memory against 0x00
avrdude: 1 byte of fuse8/bootend verified
avrdude: reading input file 0x00 for fuse7/append
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse7/append ...
avrdude: 1 byte of fuse7/append written
avrdude: verifying fuse7/append memory against 0x00
avrdude: 1 byte of fuse7/append verified
[... cut ...]
         using 8 pages and 46 pad bytes
avrdude: writing 466 bytes flash ...

Writing | ################################################## | 100% 0.24s

avrdude: 466 bytes of flash written
avrdude: verifying flash memory against C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1/bootloaders/hex/optiboot_Curiosity1627.hex

Reading | ################################################## | 100% 0.11s

avrdude: 466 bytes of flash verified

avrdude done.  Thank you.
```

mcuee, 2022-11-01:
The binary compiled by Arduino 2.0.1 seems to be okay using latest avrdude git main, as well as avrdude 7.0 release.

```
PS C:\work\avr\avrdude_test\avrdude_bin> cp C:\Users\xiaof\AppData\Local\Temp\arduino-sketch-CFD7186EB0E645E355FF22678D2A1984/Blink.ino.hex blink_attiny1627_optiboot.hex
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM20 -b 115200 -p attiny1627 -D -U .\blink_attiny1627_optiboot.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9428 (probably t1627)
avrdude: reading input file .\blink_attiny1627_optiboot.hex for flash
         with 722 bytes in 1 section within [0x200, 0x4d1]
         using 12 pages and 46 pad bytes
avrdude: writing 722 bytes flash ...

Writing | ################################################## | 100% 0.11 s

avrdude: 722 bytes of flash written
avrdude: verifying flash memory against .\blink_attiny1627_optiboot.hex

Reading | ################################################## | 100% 0.11 s

avrdude: 722 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude7 -C .\avrdude7.conf -c arduino -P COM20 -b 115200 -p attiny1627 -D -U .\blink_attiny1627_optiboot.hex

avrdude7.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude7.exe: Device signature = 0x1e9428 (probably t1627)
avrdude7.exe: reading input file ".\blink_attiny1627_optiboot.hex"
avrdude7.exe: input file .\blink_attiny1627_optiboot.hex auto detected as Intel Hex
avrdude7.exe: writing flash (1234 bytes):

Writing | ################################################## | 100% 0.11s

avrdude7.exe: 1234 bytes of flash written
avrdude7.exe: verifying flash memory against .\blink_attiny1627_optiboot.hex:
avrdude7.exe: input file .\blink_attiny1627_optiboot.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.11s

avrdude7.exe: 1234 bytes of flash verified

avrdude7.exe done.  Thank you.
```

mcuee, 2022-11-01:
I am going to close this one as invalid for now as the bootloader does not seem to work at all.

MCUdude, 2022-11-01:
I find it hard to believe that Optiboot doesn't work on the ATtiny1627. I believe Spence would have tested this.
To narrow things down, we should also leave out as much "noise" as possible and only focus on Avrdude and the optiboot_x bootloader. Arduino IDE/megaTinyCore may have bugs, but we don't care about that since what we really care about is if we can upload a binary (with start address 0x200) using optiboot_x and Avrdude.

Since I don't have the correct hardware at hand, can you test this @mcuee?

* If possible, use a dedicated USB to serial adapter to interact with optiboot_x. This is how the bootloader is intended to be used.
* Flash a different bootloader that uses different physical pins, for instance, `optiboot_tx2z_alt.hex` which uses PA1/PA2, UART0 and has an 8-second timeout so you don't need to disable the UPDI pin. Use Avrdude directly, not Arduino IDE
* Make sure the fuses are correct. Nothing fancy, but make sure the following fuses are correct
  - SYSCFG0 (I think 0xC4 is suitable)
  - APPEND (0x00 is correct)
  - BOOTEND (0x02 is correct)
  - OSCCFG (0x01 is correct)
* Upload a "syntetic hex file" that starts from address 0x200 with known content to flash and EEPROM. Ause Avrdude directly. Please try with Avrdude 6.3, 7.0 and upstream
  - Since a bootloader with an 8 second timeout is used, you have 8 seconds to run Avrdude after you've powered the board.

Here's an `srec_cat` command for generating 256 byte large hex file that starts from 0x200. If `head -c256` doesn't work, try `head -256c` instead:
```
yes "hello, world" | head -c256 | srec_cat - -bin -offset 0x200 -o hello_world_256B_offset_0x200.hex -Intel
```

mcuee, 2022-11-02:
@MCUdude 
Good advice. I will try later. I have multiple issues with megaTinyCore and I think the issues may have already been fixed in megaTinyCore git master. But you are right, I should focus on avrdude and optiboot_x.

The Tiny2 series is supposed to be easier with optiboot_x and it indeed seem to be so (to configure PB4 as the alternative reset pin). I do not really need to use the 8s trick. The only problem is that it does not seem to write to the flash.
https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/Ref_Reset.md#the-reset-pin

mcuee, 2022-11-03:
@MCUdude 

Thanks for the advice, I think the issue is with `optiboot_curiosity1627.hex`. I choose to go with `/optiboot_tx2z_rst.hex` (default UART0 pins, UPDI on RA0, Reset on the alternative PB4 pin) so that I can still use the on-board USB to Serial port.

And it works fine. I verified with the on-board pkobn_updi as well.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -qq -pattiny1627 -cpkobn_updi -Pusb -e 
-Ufuse0:w:0x00:m -Ufuse1:w:0x00:m -Ufuse5:w:0b11111110:m -Ufuse6:w:0x04:m 
-Ufuse8:w:0x02:m -Ufuse7:w:0x00:m -Uflash:w:optiboot_tx2z_rst.hex:i && echo OK
OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM20 -b 115200 -p t1627 -D 
-U .\hello_world_256B_offset_0x200.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9428 (probably t1627)
avrdude: reading input file .\hello_world_256B_offset_0x200.hex for flash
         with 256 bytes in 1 section within [0x200, 0x2ff]
         using 4 pages and 0 pad bytes
avrdude: writing 256 bytes flash ...

Writing | ################################################## | 100% 0.01 s

avrdude: 256 bytes of flash written
avrdude: verifying flash memory against .\hello_world_256B_offset_0x200.hex

Reading | ################################################## | 100% 0.01 s

avrdude: 256 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c pkobn_updi -p t1627 
-U flash:v:.\hello_world_256B_offset_0x200.hex:i -qq && echo OK
OK

```

closed 2022-11-03 by mcuee

mcuee, 2022-11-03:
@MCUdude 

I'd like to see the possibility of using optiboot_x on the ATmega4808 based Nano 4808 board. I understand that you do not support it with optiboot_x in MegaCoreX. Just wondering what is the reason. Thanks. 

If I am willing to use PICKit 4 HV UPDI with the board (I have two boards now and another on-order, so I can sacrifice one just in case), is it possible to use optiboot_x?

mcuee, 2022-11-04:
@MCUdude 
I agree with your rational not to offer the bootloader to the general MegaCoreX users. Just one minor correction, Nano 4808 does not require the 1200bps touch (only Arduino Nano Every needs the 1200bps touch).

Thanks a lot for the bootloaders. The one for Arduino Nano Every seems to work fine.

```
PS C:\work\avr\avrdude_test\avrdude_bin> mode COM12 baud=12 dtr=on > nul && mode COM12 baud=12 dtr=off > nul 
&& .\avrdude -c jtag2updi -P COM12 -p m4809 -U .\Optiboot_mega0_UART3_ALT_115200_E2.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
erasing chip
avrdude: reading input file .\Optiboot_mega0_UART3_ALT_115200_E2.hex for flash
         with 468 bytes in 2 sections within [0, 0x1ff]
         using 4 pages and 44 pad bytes
avrdude: writing 468 bytes flash ...

Writing | ################################################## | 100% 0.40 s

avrdude: 468 bytes of flash written
avrdude: verifying flash memory against .\Optiboot_mega0_UART3_ALT_115200_E2.hex

Reading | ################################################## | 100% 0.10 s

avrdude: 468 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM12 -b 115200 -p m4809 -D 
-U .\hello_world_256B_offset_0x200.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)
avrdude: reading input file .\hello_world_256B_offset_0x200.hex for flash
         with 256 bytes in 1 section within [0x200, 0x2ff]
         using 2 pages and 0 pad bytes
avrdude: writing 256 bytes flash ...

Writing | ################################################## | 100% 0.00 s

avrdude: 256 bytes of flash written
avrdude: verifying flash memory against .\hello_world_256B_offset_0x200.hex

Reading | ################################################## | 100% 0.00 s

avrdude: 256 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-11-04:
@MCUdude 

But the one for Mega4808 does not seem to work.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM13 -b 115200 -p m4808 -D -U .\hello_world_256B_offset_0x200.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)
avrdude: reading input file .\hello_world_256B_offset_0x200.hex for flash
         with 256 bytes in 1 section within [0x200, 0x2ff]
         using 2 pages and 0 pad bytes
avrdude: writing 256 bytes flash ...

Writing | ################################################## | 100% 0.00 s

avrdude: 256 bytes of flash written
avrdude: verifying flash memory against .\hello_world_256B_offset_0x200.hex

Reading | ################################################## | 100% 0.00 s

avrdude error: verification mismatch, first encountered at addr 0x0200
        device 0xff != input 0x68
avrdude error: verification mismatch

avrdude done.  Thank you.

```

mcuee, 2022-11-04:
> @MCUdude
> 
> But the one for Mega4808 does not seem to work. I will need to double check if the fuse settings play a part or not.

In the end, it is due to the wrong fuse8 setting. Once I fixed the fuse8 setting, it works fine.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c jtag2updi -P COM6 -p 115200 -p m4808 -t

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)
avrdude> dump fuses
>>> dump fuses

Reading | ################################################## | 100% 0.08 s

0000  00 54 01 ff 00 c9 06 00  00                       |.T.......       |

avrdude> dump fuse8
>>> dump fuse8

Reading | ################################################## | 100% 0.01 s

0000  00

avrdude> write fuse8 0x0 0x02
>>> write fuse8 0x0 0x02

Writing | ################################################## | 100% 0.01 s

avrdude> flush
>>> flush
avrdude> dump fuses
>>> dump fuses

Reading | ################################################## | 100% 0.08 s

0000  00 54 01 ff 00 c9 06 00  02                       |.T.......       |

avrdude> quit
>>> quit
avrdude>
avrdude done.  Thank you.

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM6 -p 115200 -p m4808 -D -U .\hello_world_256B_offset_0x200.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)
avrdude: reading input file .\hello_world_256B_offset_0x200.hex for flash
         with 256 bytes in 1 section within [0x200, 0x2ff]
         using 2 pages and 0 pad bytes
avrdude: writing 256 bytes flash ...

Writing | ################################################## | 100% 0.01 s

avrdude: 256 bytes of flash written
avrdude: verifying flash memory against .\hello_world_256B_offset_0x200.hex

Reading | ################################################## | 100% 0.01 s

avrdude: 256 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-11-04:
I have learned to choose the right hex file (`optiboot_txyz.hex`, UART pins as PB2/PB3, UPDI/RESET pin as UPDI, 8 seconds delay) for ATtiny817 Xplained Mini and ATtiny817 XPlained Pro from megaTinyCore, and I can confirm that avrdude git main works with optiboot_x for ATtiny817.

* https://github.com/SpenceKonde/megaTinyCore/issues/842

mcuee, 2022-11-28:
Interestingly I got a Nano4808 clone where the on-board jtag2updi does not work (only work once to upload the optiboot hex file and then can not talk the board). Luckily I used the PICkit4 to change the fuse settings and now it works okay.

mcuee, 2022-12-25:
@MCUdude 

Could you also tell me how to build the hex file for ATmega4809 Curiosity Nano? I just got the board.

The following command generated a bootloader (8s delay, UART3, LED is B5) which works only one time. After uploading the blink hex file it no longer works. The fuse should be correct.

```
make mega0 UARTTX=B0 TIMEOUT=8 LED=B5 BAUD_RATE=115200 SKIP_BOOTLOADER_ON_POR=1
```

Run log:
```
PS C:\Users\xiaof> cd c:\work\avr\avrdude_test\avrdude_bin\
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c pkobn_updi -p m4809 -qqt
avrdude> dump fuses
0000  00 54 01 ff 00 c9 06 00  02                       |.T.......       |

avrdude> quit
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c pkobn_updi -p m4809 -U .\Optiboot_mega0_UART3_DEF_115200_B5_8s.hex

         Vtarget         : 3.31 V
         PDI/UPDI clk    : 100 kHz

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Optiboot_mega0_UART3_DEF_115200_B5_8s.hex for flash
         with 466 bytes in 2 sections within [0, 0x1ff]
         using 4 pages and 46 pad bytes
avrdude: writing 466 bytes flash ...

Writing | ################################################## | 100% 0.24 s

avrdude: 466 bytes of flash written
avrdude: verifying flash memory against .\Optiboot_mega0_UART3_DEF_115200_B5_8s.hex

Reading | ################################################## | 100% 0.11 s

avrdude: 466 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM25 -p m4809 -xshowall -xbootsize=512

avrdude: AVR device initialized and ready to accept instructions
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 x0.0 ......... vector 0 (RESET) ATmega4809
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM25 -p m4809 -U .\BlinkCu.ino_atmega4809_16000000L.hex -xbootsize=512

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\BlinkCu.ino_atmega4809_16000000L.hex for flash
         with 764 bytes in 1 section within [0x200, 0x4fb]
         using 6 pages and 4 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 764 bytes flash ...

Writing | ################################################## | 100% 0.12 s

avrdude: 764 bytes of flash written
avrdude: verifying flash memory against .\BlinkCu.ino_atmega4809_16000000L.hex

Reading | ################################################## | 100% 0.08 s

avrdude: 764 bytes of flash verified

avrdude done.  Thank you.

(unplug and plug)

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM25 -p m4809 -xshowall -xbootsize=512
avrdude warning: attempt 1 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 2 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 3 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 4 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 5 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 6 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 7 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 8 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 9 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 10 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude error: unable to open programmer urclock on port COM25
```