avrdudes/avrdude #1167: Arduino UNO, jtag2isp: flash verification error when the default chip erase operation is performed

cdealti, 2022-11-01:
## Environment

- macOS Monterey v12.6
- libusb v1.0.26 (installed with Homebrew)
- avrdude v7.0 (installed with Homebrew)
- avrdude v6.3 (distributed with Arduino IDE v2.0.1) 
- Arduino UNO R.3 (ATmega328p)
- Atmel JTAGICE mkII with fw v7.39 (upgraded with Microchip Studio a few months ago)
- [Optiboot bootloader for the Arduino UNO](https://github.com/Optiboot/optiboot/tree/master/optiboot/bootloaders/optiboot)

## Failure description
Programming the flash, performing the default chip erase operation, fails with a verification error, as follows:

```
$ avrdude -pm328p -cjtag2isp -v -v -Uflash:w:/Users/cdealti/Downloads/optiboot_atmega328.hex.txt:a

avrdude: Version 7.0
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/usr/local/etc/avrdude.conf"
         User configuration file is "/Users/cdealti/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : jtag2isp
avrdude: stk500v2_jtagmkII_open()
avrdude: usbdev_open(): Found JTAGICE mkII, serno: 0700000025B4
avrdude: jtagmkII_getsync() attempt 1 of 10: Sending sign-on command: 0x86 (28 bytes msg)
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              0
S_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              1
Serial number:                   07:00:00:00:25:b4
Device ID:                       JTAGICEmkII
avrdude: jtagmkII_setparm()
avrdude: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes): 0x80 (1 bytes msg)
avrdude: jtagmkII_getsync(): Sending get sync command: 0x80 (1 bytes msg)
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
[... 529 lines cut ...]
avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as DE
avrdude: safemode: efuse reads as FD
avrdude: erasing chip

avrdude: safemode: lfuse reads as FE
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as FE
avrdude: safemode: lfuse changed! Was ff, and is now fe
Would you like this fuse to be changed back? [y/n] n
avrdude: safemode: hfuse changed! Was de, and is now 0
Would you like this fuse to be changed back? [y/n] n
avrdude: safemode: efuse changed! Was fd, and is now fe
Would you like this fuse to be changed back? [y/n] n
avrdude: safemode: Fuses OK (E:FD, H:DE, L:FF)
avrdude: stk500v2_jtagmkII_close()
avrdude: jtagmkII_close()
avrdude: jtagmkII_close(): Sending sign-off command: 0x80 (1 bytes msg)

avrdude done.  Thank you.
```

Ad then, programming the flash with _-D_ succeeds as with avrdude v7.0.

In a summary, the workaround for me is to erase the chip and then program the flash with the _-D_ option but, unfortunately,  IDEs do not support this.

mcuee, 2022-11-01:
I believe this is the same as #425.
* https://github.com/avrdudes/avrdude/issues/425#issuecomment-1221545618

mcuee, 2022-11-01:
If possible, please check if 5.11.1 release works for you or not. Issue #425 was traced to a mega commit in September 2011.

mcuee, 2022-11-02:
@cdealti 
Can you try latest avrdude as well? You can do that with brew command. Or you can use the `build.sh` command.
``` 
brew uninstall avrdude
brew install avrdude --HEAD
```

After that you can use the following command to always follow the git main. 
```
brew reinstall avrdude 
```

cdealti, 2022-11-02:
> Can you try latest avrdude as well?

It behaves the same as v7.0.

> It is strange that erase command is not used from your run log.

What do you mean? What command line are you referring to?

mcuee, 2022-11-03:
> > Can you try latest avrdude as well?
> 
> It behaves the same as v7.0.

Thanks for the confirmation.
 
> > It is strange that erase command is not used from your run log.
> 
> What do you mean? What command line are you referring to?

Sorry I need to be more specific. I was suspecting a few possibilities.
1) that chip erase command was not really executed somehow in your run log. 
2) chip erasing was not really successful
3) not enough delay between chip erasing and the flash writing 

```
avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/Users/cdealti/Downloads/optiboot_atmega328.hex.txt"
avrdude: input file /Users/cdealti/Downloads/optiboot_atmega328.hex.txt auto detected as Intel Hex
avrdude: writing flash (32768 bytes)

```

Can you try the following command instead?
```
avrdude -pm328p -cjtag2isp -v -v -e -Uflash:w:/Users/cdealti/Downloads/optiboot_atmega328.hex.txt:i
```

cdealti, 2022-11-03:
Thanks for your time. Please note that the first command below has been executed with the DWEN fuse bit set, as I was using _avarice_ before this test. Avrdude correctly copes with disabling debugWire and enters ISP mode until the next power on of the MCU. I can clear DWEN to match the fuse value of the previous test if you think it is necessary. Anyway, the failure is the same as without _-e_. I believe this is expected as _-e_ is implied unless _-D_ is specified. I've also read the content of the flash after the command and verified it has been actually erased.

```
$ avrdude -pm328p -cjtag2isp -v -v -e -Uflash:w:/Users/cdealti/Downloads/optiboot_atmega328.hex.txt:i

avrdude: Version 7.0-20221101 (4c92030)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is /usr/local/etc/avrdude.conf
         User configuration file is /Users/cdealti/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : jtag2isp
avrdude: stk500v2_jtagmkII_open()
avrdude: usbdev_open(): found JTAGICE mkII, serno: 0700000025B4
avrdude: jtagmkII_getsync() attempt 1 of 10: sending sign-on command: 0x86 (28 bytes msg)
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              0
S_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              1
Serial number:                   07:00:00:00:25:b4
Device ID:                       JTAGICEmkII
avrdude: jtagmkII_setparm()
avrdude: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes): 0xa0 (1 bytes msg)
avrdude jtagmkII_setparm() error: bad response to set parameter command: RSP_FAILED
avrdude jtagmkII_getsync() warning: ISP activation failed, trying debugWire
avrdude: jtagmkII_setparm()
avrdude: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes): 0x80 (1 bytes msg)
avrdude: jtagmkII_reset(): sending reset command: 0x80 (1 bytes msg)
avrdude jtagmkII_getsync() warning: target prepared for ISP, signed off
        now retrying without power-cycling the target
avrdude: jtagmkII_setparm()
avrdude: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes): 0x80 (1 bytes msg)
avrdude: jtagmkII_getsync(): sending get sync command: 0x80 (1 bytes msg)
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

[... 56 lines cut ...]

Compare this with a chip erase-only execution:

```
$ avrdude -pm328p -cjtag2isp -e

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
erasing chip

avrdude done.  Thank you.

$ avrdude -pm328p -cjtag2isp -Uflash:r:-:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: reading flash memory ...

Reading | ################################################## | 100% 12.17 s

avrdude: writing output file <stdout>
:00000001FF

avrdude done.  Thank you.
```

mcuee, 2022-11-04:
So the issue may be the delay between chip erase and the flash programming.

cdealti, 2022-11-04:
> But please try without the DWEN fuse bit being set to see if it makes any difference. Thanks.

I'll do but I'm sure it will not change anything as I've done all kinds of tests with and without DWEN before opening the bug.

> Sorry I do not have the JATGICE mkii myself.

No worries.

> So the issue may be the delay between chip erase and the flash programming. Maybe there is a way to increase the delay.

As the source code is readable, I'm currently looking to:

https://github.com/avrdudes/avrdude/blob/4c92030e3a486cfbaeb36e298f6f2929e8e031eb/src/stk500v2.c#L945-L952

1. The chip erase delay is a parameter of the command sent to the ICE which should honor it and respond after the delay has elapsed. Why do we have to sleep the same time interval?
2. `pgm->initialize` will eventually send the ICE another `CMD_ENTER_PROGMODE_ISP`. I don't see why this is necessary and maybe harmful to the mkII.

I'll make some changes and report the results.

mcuee, 2022-11-05:
> I've removed both 1 and 2 and it works like a charm. I'll submit a PR. I hope you can test if it works for the AVRISPmkii and STK500 v2 as well.

@cdealti 
Wow, those codes are from the very begining in 2006 (first version of stk500v2.c).
https://github.com/avrdudes/avrdude/blob/04831af970a277832721e0395baef2949725a364/stk500v2.c

I will test it out with my programmers using the STK500v2 ISP protocol.

@dl8dtl 
As this is very much legacy codes, please take a look as well. Thanks.

mcuee, 2022-11-05:
Quick test results using AVRISP mkii clone.

1) git main is OK (same results for avrdude 7.0 release)
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c avrispmkii -p m328p -U .\hex\optiboot_atmega328_big.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
erasing chip
avrdude: reading input file .\hex\optiboot_atmega328_big.hex for flash
         with 754 bytes in 2 sections within [0x7c00, 0x7fff]
         using 7 pages and 142 pad bytes
avrdude: writing 754 bytes flash ...

Writing | ################################################## | 100% 0.02 s

avrdude: 754 bytes of flash written
avrdude: verifying flash memory against .\hex\optiboot_atmega328_big.hex

Reading | ################################################## | 100% 0.02 s

avrdude: 754 bytes of flash verified

avrdude done.  Thank you.

```

2) Remove the `pgm->initialize(pgm, p); ` line. It is still OK.
```
diff --git a/src/stk500v2.c b/src/stk500v2.c
index edf26ca..673f220 100644
--- a/src/stk500v2.c
+++ b/src/stk500v2.c
@@ -949,7 +949,7 @@ static int stk500v2_chip_erase(const PROGRAMMER *pgm, const AVRPART *p) {
   avr_set_bits(p->op[AVR_OP_CHIP_ERASE], buf+3);
   result = stk500v2_command(pgm, buf, 7, sizeof(buf));
   usleep(p->chip_erase_delay);
-  pgm->initialize(pgm, p);
+  //pgm->initialize(pgm, p);

   pgm->pgm_led(pgm, OFF);

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_stk500v2_mod -c avrispmkii -p m328p -U .\hex\optiboot_atmega328_big.hex

avrdude_stk500v2_mod: AVR device initialized and ready to accept instructions
avrdude_stk500v2_mod: device signature = 0x1e950f (probably m328p)
avrdude_stk500v2_mod: Note: flash memory has been specified, an erase cycle will be performed.
                      To disable this feature, specify the -D option.
erasing chip
avrdude_stk500v2_mod: reading input file .\hex\optiboot_atmega328_big.hex for flash
                      with 754 bytes in 2 sections within [0x7c00, 0x7fff]
                      using 7 pages and 142 pad bytes
avrdude_stk500v2_mod: writing 754 bytes flash ...

Writing | ################################################## | 100% 0.02 s

avrdude_stk500v2_mod: 754 bytes of flash written
avrdude_stk500v2_mod: verifying flash memory against .\hex\optiboot_atmega328_big.hex

Reading | ################################################## | 100% 0.02 s

avrdude_stk500v2_mod: 754 bytes of flash verified

avrdude_stk500v2_mod done.  Thank you.

```

3. Remove both lines. It is still OK.

```
diff --git a/src/stk500v2.c b/src/stk500v2.c
index edf26ca..21b5138 100644
--- a/src/stk500v2.c
+++ b/src/stk500v2.c
@@ -948,8 +948,8 @@ static int stk500v2_chip_erase(const PROGRAMMER *pgm, const AVRPART *p) {
   memset(buf+3, 0, 4);
   avr_set_bits(p->op[AVR_OP_CHIP_ERASE], buf+3);
   result = stk500v2_command(pgm, buf, 7, sizeof(buf));
-  usleep(p->chip_erase_delay);
-  pgm->initialize(pgm, p);
+  //usleep(p->chip_erase_delay);
+  //pgm->initialize(pgm, p);

   pgm->pgm_led(pgm, OFF);

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_stk500v2_mod1 -c avrispmkii -p m328p -U .\hex\optiboot_atmega328_big.hex

avrdude_stk500v2_mod1: AVR device initialized and ready to accept instructions
avrdude_stk500v2_mod1: device signature = 0x1e950f (probably m328p)
avrdude_stk500v2_mod1: Note: flash memory has been specified, an erase cycle will be performed.
                       To disable this feature, specify the -D option.
erasing chip
avrdude_stk500v2_mod1: reading input file .\hex\optiboot_atmega328_big.hex for flash
                       with 754 bytes in 2 sections within [0x7c00, 0x7fff]
                       using 7 pages and 142 pad bytes
avrdude_stk500v2_mod1: writing 754 bytes flash ...

Writing | ################################################## | 100% 0.02 s

avrdude_stk500v2_mod1: 754 bytes of flash written
avrdude_stk500v2_mod1: verifying flash memory against .\hex\optiboot_atmega328_big.hex

Reading | ################################################## | 100% 0.02 s

avrdude_stk500v2_mod1: 754 bytes of flash verified

avrdude_stk500v2_mod1 done.  Thank you.
```

mcuee, 2022-11-06:
AVR Dragon is usually similar to JTAGICE mkii in terms of behavior. But this time I can not reproduce the issue with my AVR Dragon.

I have tried a few optiboot hex files (Arduino original, Mini Core) and another Arduino compatible hex file from #940. I have no issues with them.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c dragon_isp -p m328p -U .\optiboot_flash_atmega328p_UART0_115200_16000000L_B5.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
erasing chip
avrdude: reading input file .\optiboot_flash_atmega328p_UART0_115200_16000000L_B5.hex for flash
         with 484 bytes in 2 sections within [0x7e00, 0x7fff]
         using 4 pages and 28 pad bytes
avrdude: writing 484 bytes flash ...

Writing | ################################################## | 100% 0.01 s

avrdude: 484 bytes of flash written
avrdude: verifying flash memory against .\optiboot_flash_atmega328p_UART0_115200_16000000L_B5.hex

Reading | ################################################## | 100% 0.01 s

avrdude: 484 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM3 -b 115200 -p m328p

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)

avrdude done.  Thank you.

```

cdealti, 2022-11-06:
> However, I think the chip erase dealy may be necessary, it is common in the code base.

Looking at the other occurrences of `usleep(p->chip_erase_delay)`, it is necessary for programmers that do not implement the  AVRISP mkII protocol (Atmel application note AVR069). The _CMD_CHIP_ERASE_ISP_ command (section 3.2.3, page 7) waits for the value of the _eraseDelay_ field if the field _pollMethod_ equals 0. Sleeping is not the end of the world but it is misleading (for the AVRISP mkII compliant programmers).

It is possible that the two lines are there due to a copy and paste of older code, for example:

https://github.com/avrdudes/avrdude/blob/baaad71aa5eb447470f768b5842c76caea1e7c4b/src/stk500.c#L196-L197

In my opinion they should both go.

If I'm not wrong @dl8dtl has access the these AVRISP mkII compatible programmers (including the JTAGICE mkII) and he can maybe test if they all work removing the two lines.

mcuee, 2022-11-07:
@cdealti 
Please help to create a PR so that more people can take a look and review/test. Thanks.

mcuee, 2022-11-27:
Fixed by #1172

closed 2022-11-27 by mcuee