avrdudes/avrdude #1168: debugWire really working -- dragon_dw and jtag2dw

mcuee, 2022-11-05:
As per the documentation both should work.

But it seems to me I can not get dragon_dw to work properly.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c dragon_dw -p m328p -v

avrdude: Version 7.0-20221101 (4c92030)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

         Using Port                    : usb
         Using Programmer              : dragon_dw
avrdude: usbdev_open(): found AVRDRAGON, serno: 00A20000xxxx
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              1
S_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              2
Serial number:                   00:a2:00:00:xx:xx
Device ID:                       AVRDRAGON
avrdude jtagmkII_setparm() error: bad response to set parameter command: RSP_DEBUGWIRE_SYNC_FAILED
avrdude main() error: unable to open programmer dragon_dw on port usb

avrdude done.  Thank you.
```

mcuee, 2022-11-05:
Then there are other JTAGICE3 based debugWire capable debuggers like jtag3dw, atmelice_dw, powerdebugger_dw and xplainedmini_dw. I do not have access to those tools so I am not so sure if they work or not.

mcuee, 2022-11-05:
@cdealti
From the other issue, I think you can program the ATmega328P with DW bits enabled using jtag2isp, which is in line with avrdude documentation FAQ section (other than the issue you mentioned which you seem to find the fix and the issue has nothing to do with dW anyway).
* https://github.com/avrdudes/avrdude/issues/1167

Just wondering if programming works if you use `-c jtag2dw`.

mcuee, 2022-11-05:
Just get another Uno clone to detail the steps to reproduce the issue. I will try to see if I can rescue the two Uno clones using stk500v2 HV programmer later.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c dragon_isp -p m328p -q -t

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude> dump lfuse
>>> dump lfuse
0000  ff                                                |.               |

avrdude> dump hfuse
>>> dump hfuse
0000  dc                                                |.               |

avrdude> dump efuse
>>> dump efuse
0000  fd                                                |.               |

avrdude> dump lock
>>> dump lock
0000  ff      
                                          |.               |
avrdude> write hfuse 0 0x9c
>>> write hfuse 0 0x9c
avrdude> dump hfuse
>>> dump hfuse
0000  9c                                                |.               |

avrdude> quit
>>> quit
avrdude>
avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c dragon_isp -p m328p -q -t
avrdude error: bad response to set parameter command: RSP_FAILED
avrdude warning: ISP activation failed, trying debugWire
avrdude error: bad response to set parameter command: RSP_DEBUGWIRE_SYNC_FAILED
avrdude error: unable to sync with the AVR Dragon in ISP mode
avrdude error: unable to open programmer dragon_isp on port usb

avrdude done.  Thank you.

```

mcuee, 2022-11-05:
AVR Dragon is supposed to be able to disable DW and enable ISP again using Atmel Studio as per Page 19 of the user guide. However, it does not seem to work. I can not get AVR Dragon to communicate with the ATmega328P using the connection mentioned in the user guide (using the 6-pin SPI connector, RESET pin becomes DW pin).
https://ww1.microchip.com/downloads/en/devicedoc/atmel-42723-avr-dragon_userguide.pdf
[image]
[image]

cdealti, 2022-11-05:
>AVR Dragon is supposed to be able to disable DW and enable ISP

Did you cut the _RESET EN_ PCB trace on you Arduino UNO? If not you will not be able to talk debugWire.

>Just wondering if programming still works if you use -c jtag2dw with the ATmega328P with DWEN fuse activated.

Unfortunately it does not work. Shall I create a separate bug or use this one for the details? And, yes, on my UNO R.3 I've cut the _RESET EN_ trace and can debug over debugWire with avarice.

mcuee, 2022-11-05:
@cdealti 
Thanks a lot for the info, I will try your tip. 
You can continue use this issue as you have issues using avrdude with jtag2dw. Please post your run log as well.

mcuee, 2022-11-05:
> Then there are other JTAGICE3 based debugWire capable debuggers like jtag3dw, atmelice_dw, powerdebugger_dw and xplainedmini_dw. I do not have access to those tools so I am not so sure if they work or not.

@MCUdude 

As you have access to the JTAGICE 3, Atmel ICE and Power Debugger, please help to check one of them for debugWire to see if they work or not.

dl8dtl, 2022-11-05:
To me, all these messages look like hardware that is not debugWIRE-capable – not the MCU itself, but the circuitry attached to the /RESET pin.
Keep in mind:
* there must be *no* capacitor on /RESET
* the cable length should not exceed about 20 cm
* sometimes, attaching a 4.7 or 10 kΩ pullup might improve the situation

dl8dtl, 2022-11-05:
I don't think attempting to use different debugger hardware will change anything. They are more or less all the same in that respect.

dl8dtl, 2022-11-05:
For a test, I put an ATmega328P into my STK500, and used the elderly Dragon to handle it.
It works as expected.
```
$ ./build_freebsd/src/avrdude -c dragon_isp -p m328p -B 100khz -t

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude> d hf

Reading | ################################################## | 100% 0.05 s 

0000  d9                                                |.               |

avrdude> w hf 0 0x99

Writing | ################################################## | 100% 0.11 s 

avrdude> quit
avrdude done.  Thank you.

$ ./build_freebsd/src/avrdude -c dragon_isp -p m328p -B 100khz -t
avrdude error: bad response to set parameter command: RSP_FAILED
avrdude warning: ISP activation failed, trying debugWire
avrdude error: bad response to set parameter command: RSP_DEBUGWIRE_SYNC_FAILED
avrdude error: unable to sync with the AVR Dragon in ISP mode
avrdude error: unable to open programmer dragon_isp on port usb

avrdude done.  Thank you.

$ ./build_freebsd/src/avrdude -c dragon_isp -p m328p -B 100khz -t
avrdude error: bad response to set parameter command: RSP_FAILED
avrdude warning: ISP activation failed, trying debugWire
avrdude warning: target prepared for ISP, signed off
        now retrying without power-cycling the target

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude> d hf

Reading | ################################################## | 100% 0.05 s 

0000  99                                                |.               |

avrdude> w hf 0 0xd9

Writing | ################################################## | 100% 0.11 s 

avrdude> quit
avrdude done.  Thank you.
```
Note that in the first connection attempt after setting the DWEN fuse, I still had the RESET jumper from the STK500 attached. The symptoms then are the same as described above: it can easily still use ISP, but debugWIRE fails with anything attached to the /RESET line. I then removed the jumper on the STK500, and tried everything again – it works.

mcuee, 2022-11-05:
@dl8dtl 
Can you try dragon_dw? Or we must use dragon_isp? 

Same for JTAGICE mkii. Is it a must to use jtag2isp and not jtag2dw?

What is the use of dragon_dw and jtag2dw?

dl8dtl, 2022-11-05:
If dragon_isp is able to revert to debugWIRE mode (to reactivate ISP), dragon_dw would work as well. But I could give that a try – basically, it's then limited to what debugWIRE can handle, in particular, it cannot change fuses. But you could reprogram flash or EEPROM in these modi.

Atmel Studio never even allowed to program using debugWIRE itself (afaict), for programming, they always revert to ISP mode. They use debugWIRE solely for debugging itself (which AVRDUDE doesn't do, but AVaRICE can).

The entire matter is pretty twisted: ISP is an "outside" programming mode, works without CPU (but needs clock for the SPI unit), and it can access everything.

debugWIRE is a monitor ROM program. It can only access what the CPU can access. Once enabled, it disables the normal /RESET functionality that is needed for ISP. So the way return is a specific command ("reset with MonCon disabled" or something like that – "MonCon" being the initial internal name before they released it as "debugWIRE", you might note the reference to a "monitor" there) that temporarily re-enables /RESET, until the next power cycle, when the fuses are re-evaluated. If your ISP session disabled DWEN, you are then in permanent ISP mode (Studio always uses it that way).

The crux is, if you activate DWEN (thus kicking /RESET off), but then have a circuitry that cannot handle debugWIRE (due to external cicuitry on /RESET), you have shot into your foot. There is no way back to ISP then.

dl8dtl, 2022-11-05:
OK, so here's all that with the JTAGICEmkII.
```
$ ./build_freebsd/src/avrdude -c jtag2isp -p m328p -B 100khz -t

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude> d hf

Reading | ################################################## | 100% 0.01 s 

0000  d9                                                |.               |

avrdude> w hf 0 0x99

Writing | ################################################## | 100% 0.02 s 

avrdude> quit
avrdude> 
avrdude done.  Thank you.

$ ./build_freebsd/src/avrdude -c jtag2dw -p m328p -t

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude> d flash

Reading | ################################################## | 100% 0.04 s 

0000  0c 94 94 01 0c 94 bc 01  0c 94 bc 01 0c 94 bc 01  | ... ... ... ...|
0010  0c 94 bc 01 0c 94 c0 1e  0c 94 bc 01 0c 94 0b 1f  | ... ... ... . .|
0020  0c 94 bc 01 0c 94 bc 01  0c 94 bc 01 0c 94 bc 01  | ... ... ... ...|
0030  0c 94 bc 01 0c 94 bc 01  0c 94 bc 01 0c 94 bc 01  | ... ... ... ...|
0040  0c 94 32 1f 0c 94 bc 01  0c 94 bc 01 0c 94 bc 01  | .2. ... ... ...|
0050  0c 94 bc 01 0c 94 bc 01  0c 94 bc 01 0c 94 bc 01  | ... ... ... ...|
0060  0c 94 bc 01 0c 94 bc 01  08 00 00 00 be 92 24 49  | ... .........$I|
0070  12 3e ab aa aa 2a be cd  cc cc 4c 3e 00 00 00 80  |.>...*....L>....|
0080  be ab aa aa aa 3e 00 00  00 00 bf 00 00 00 80 3f  |.....>.........?|
0090  00 00 00 00 00 08 41 78  d3 bb 43 87 d1 13 3d 19  |......Ax..C...=.|
00a0  0e 3c c3 bd 42 82 ad 2b  3e 68 ec 82 76 be d9 8f  |.<..B..+>h..v...|
00b0  e1 a9 3e 4c 80 ef ff be  01 c4 ff 7f 3f 00 00 00  |..>L........?...|
00c0  00 00 43 68 65 63 6b 20  50 54 54 2f 6b 65 79 00  |..Check PTT/key.|
00d0  52 65 73 65 74 20 73 65  74 74 69 6e 67 73 2e 2e  |Reset settings..|
00e0  00 31 2e 30 32 77 00 20  52 00 08 04 0a 05 0a 04  |.1.02w. R... . .|
00f0  08 00 00 00 00 00 00 00  00 00 10 10 10 10 10 10  |................|

avrdude> d hf

Reading | -------------------------------------------------- | 0% 0.00 s 

avrdude error: (dump) error reading hfuse address 0x00000 of part ATmega328P
               read operation not supported on memory type hfuse
avrdude> quit
avrdude> 
avrdude done.  Thank you.

$ ./build_freebsd/src/avrdude -c jtag2isp -p m328p -B 100khz -t
avrdude error: bad response to set parameter command: RSP_FAILED
avrdude warning: ISP activation failed, trying debugWire
avrdude warning: target prepared for ISP, signed off
        now retrying without power-cycling the target

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude> w hf 0 0xd9

Writing | ################################################## | 100% 0.02 s 

avrdude> quit
avrdude> 
avrdude done.  Thank you.

$ ./build_freebsd/src/avrdude -c jtag2isp -p m328p -B 100khz -t

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude> d hf

Reading | ################################################## | 100% 0.01 s 

0000  d9                                                |.               |

avrdude> quit
avrdude> 
avrdude done.  Thank you.
```

dl8dtl, 2022-11-05:
I hope it is convincing there's no bug that generally prevents debugWIRE from working with AVRDUDE. ;-)

mcuee, 2022-11-06:
Thanks. I will close this issue then.

I will need to figure out how to isolate the RESET trace on my two Uno Clones (quite different designs) to recover them.

mcuee, 2022-11-06:
@dl8dtl and @MCUdude 

As for PICKit 4, it is not clear to me whether it can support DW or not under avrdude. We can continue the discussion in the other thread. By right it should also be able to support DW.
* https://github.com/avrdudes/avrdude/issues/1052

mcuee, 2022-11-06:
> Unfortunately it does not work. Shall I create a separate bug or use this one for the details? And, yes, on my UNO R.3 I've cut the RESET EN trace and can debug over debugWire with avarice.

@cdealti 
Please double check. You can reopen this issue if you still find that jtag2isp works but jtag2dw does not work. Thanks.

dl8dtl, 2022-11-06:
Right now, I don't own a PICKit, so I cannot test that part.

mcuee, 2022-11-09:
I have not got any success on the Uno CH340 clone. I have remove two components linked to the RESET line but no success yet. I will need to check whether there are anything else connected.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c dragon_isp -p m328p -B 100khz -t
avrdude error: bad response to set parameter command: RSP_FAILED
avrdude warning: ISP activation failed, trying debugWire
avrdude error: bad response to set parameter command: RSP_DEBUGWIRE_SYNC_FAILED
avrdude error: unable to sync with the AVR Dragon in ISP mode
avrdude error: unable to open programmer dragon_isp on port usb

avrdude done.  Thank you.

```

dl8dtl, 2022-11-09:
Try a 4.7 or 10 kΩ pullup on /RESET. Sometimes, it helps.

cdealti, 2022-11-09:
@mcuee do you still want me to followup here about failing to program the flash with JTAGICE mkII and _-c jtag2dw_?
AVRDUDE fails with the usual "Verification mismatch" message. There is nothing relevant in the verbose output. The flash is not erased and the old program still runs after a reset. Tested against and original Arduino UNO R.3, with _DWEN_ enabled and the _RESET_EN_ trace cut. With the same setup I can debug with AVaRICE. However, I need to double-check as chip erase is not supported in debugWire and *only* the pages of the flash spanned by the hex file being programmed should be automatically erased before writing the flash.

mcuee, 2022-11-10:
@cdealti 
Yes please help to provide the details. I have re-opened the issue.

mcuee, 2022-11-10:
> Try a 4.7 or 10 kΩ pullup on /RESET. Sometimes, it helps.

Add 10 kΩ pullup on /RESET, still no go. Looks like the 

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c dragon_isp -p m328p
avrdude error: bad response to set parameter command: RSP_FAILED
avrdude warning: ISP activation failed, trying debugWire
avrdude error: bad response to set parameter command: RSP_DEBUGWIRE_SYNC_FAILED
avrdude error: unable to sync with the AVR Dragon in ISP mode
avrdude error: unable to open programmer dragon_isp on port usb

avrdude done.  Thank you.

```

mcuee, 2022-11-11:
Reference:
https://github.com/avrdudes/avrdude/pull/1171#issuecomment-1311733481

@nerdralph as a serial DW implementation here. It will be good to have DW support using USB-TTL adapter, `-c serial2dw` will be a great addition to avrdude.
https://github.com/nerdralph/dwprog

cdealti, 2022-11-12:
> @mcuee do you still want me to followup here about failing to program the flash with JTAGICE mkII and _-c jtag2dw_? AVRDUDE fails with the usual "Verification mismatch" message. There is nothing relevant in the verbose output. The flash is not erased and the old program still runs after a reset. Tested against an original Arduino UNO R.3, with _DWEN_ enabled and the _RESET_EN_ trace cut. With the same setup I can debug with AVaRICE. However, I need to double-check as chip erase is not supported in debugWire and **only** the pages of the flash spanned by the hex file being programmed should be automatically erased before writing the flash.

> @cdealti Yes please help to provide the details. I have re-opened the issue.

## Environament
macOS Monterey v12.6
libusb v1.0.26 (installed with Homebrew)
AVRDUDE main branch
Arduino UNO R.3 (ATmega328p) having _DWEN_ enabled and the _RESET_EN_ PCB trace cut. I can debug with _AVaRICE_ with the _--debugwire_ option.
Atmel JTAGICE mkII with fw v7.39 (upgraded with Microchip Studio a few months ago)

## Failure Description
Writing the flash with the _-c jtag2dw_ programmer option fails as follows:

```
$ ./build_darwin/src/avrdude -C ./build_darwin/src/avrdude.conf  -pm328p -c jtag2dw -v -v -v -t
sys_config = ./build_darwin/src/avrdude.conf
sys_config_found = false

avrdude: Version 7.0-20221101 (4c92030)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is /Users/cristiano.dealti/dev/git/avrdude/build_darwin/src/avrdude.conf
         User configuration file is /Users/cristiano.dealti/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : jtag2dw
avrdude: jtagmkII_open_dw()
avrdude: usbdev_open(): found JTAGICE mkII, serno: 0700000025B4
avrdude: jtagmkII_getsync()
avrdude: jtagmkII_getsync() attempt 1 of 10: sending sign-on command:
avrdude: jtagmkII_send(): sending 1 bytes

avrdude: jtagmkII_recv(): got message seqno 0 (command_sequence == 0)

Sign-on succeeded

JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              0
S_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              1
Serial number:                   07:00:00:00:25:b4
Device ID:                       JTAGICEmkII
avrdude: jtagmkII_getsync(): using a 298-byte device descriptor
avrdude: jtagmkII_setparm()
avrdude: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes):
avrdude: jtagmkII_send(): sending 3 bytes

avrdude: jtagmkII_recv(): got message seqno 1 (command_sequence == 1)

OK

avrdude: jtagmkII_getsync(): sending get sync command:
avrdude: jtagmkII_send(): sending 1 bytes

avrdude: jtagmkII_recv(): got message seqno 2 (command_sequence == 2)

OK

         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
[... 171 lines cut ...]

avrdude: jtagmkII_paged_load(.., flash, 128, 128)
avrdude: jtagmkII_paged_load(): block_size at addr 28672 is 128
avrdude: jtagmkII_paged_load(): sending read memory command:
avrdude: jtagmkII_send(): sending 10 bytes

avrdude: jtagmkII_recv(): got message seqno 13 (command_sequence == 13)

memory contents:
0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff
0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff
0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff
0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff
0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff
0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff
0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff
0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff  0xff

Writing | -------------------------------------------------- | 0% 0.16 s

avrdude avr_flush_cache() [avrcache.c:538] error: verification mismatch at flash page addr 0x7000
```

A more catastrophic behavior if you attempt to write locations of flash that have been previously written (i.e. non 0xff).
I've found a temporary fix for the m328p comparing the device descriptor with the one of AVaRICE.

mcuee, 2022-11-12:
@dl8dtl 
Just wondering if your JTAGICE mkII has the latest FW version as @cdealti or not. That may play a part.

My issue is different -- I do not have the good HW setup. But @cdealti has a good setup and seems to have a genuine problem with `-c jtag2dw`

closed 2022-11-25 by stefanrueger