avrdudes/avrdude #1169: `-B0.5` command line option does not work but `-B 0.5` works.

mcuee, 2022-11-05:
This is not expected. Tested under Windows but I will try other OS as well.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v2 -P COM5 -B0.25 -p m32a -U .\zeros32KB.hex -v
avrdude error: invalid bit clock period specified '0'
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v2 -P COM5 -B0.5 -p m32a -U .\zeros32KB.hex -v
avrdude error: invalid bit clock period specified '0'

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v2 -P COM5 -B 0.25 -p m32a -U .\zeros32KB.hex -v

avrdude: Version 7.0-20221101 (4c92030)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

         Using Port                    : COM5
         Using Programmer              : stk500v2
         Setting bit clk period        : 0.2
avrdude: input file .\zeros32KB.hex auto detected as Intel Hex
         AVR Part                      : ATmega32A
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                  4    10    64    0 no       1024    4      0  9000  9000 0xff 0xff
           flash                  33     6    64    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel STK500 version 2.x firmware
         Programmer Model: STK500
         Hardware Version: 2
         Firmware Version Master : 2.10
         Topcard         : Unknown
         Vtarget         : 5.1 V
         SCK period      : 0.5 us
         Varef           : 5.0 V
         Oscillator      : 3.686 MHz

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9502 (probably m32a)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
erasing chip
avrdude: reading input file .\zeros32KB.hex for flash
         with 32767 bytes in 1 section within [0, 0x7ffe]
         using 256 pages and 1 pad byte
avrdude: writing 32767 bytes flash ...

Writing | ################################################## | 100% 8.03 s

avrdude: 32767 bytes of flash written
avrdude: verifying flash memory against .\zeros32KB.hex

Reading | ################################################## | 100% 5.33 s

avrdude: 32767 bytes of flash verified

avrdude done.  Thank you.

```

MCUdude, 2022-11-05:
I can not reproduce the issue with MacOS using the USBasp programmer:

```
$ ./avrdude -cusbasp -patmega8535 -B0.5

avrdude: set SCK frequency to 1500000 Hz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9308 (probably m8535)

avrdude done.  Thank you.
```

dl8dtl, 2022-11-05:
Windows has no native `getopt()`. There's a replacement in the `msvc/` subdir, maybe its behaviour does not exactly match the Unix one?

mcuee, 2022-11-06:
> Windows has no native `getopt()`. There's a replacement in the `msvc/` subdir, maybe its behaviour does not exactly match the Unix one?

Take note that `msvc` subdir only covers MSVC but not MinGW. I am using mingw here (usually I do not use Visual Studio) and it does have `getopt`. I will check out MSVC version.

Strange thing is that `-B1` and `-B1.5` works.

mcuee, 2022-11-06:
Same problem for MSVC version. I checked with official 7.0 release binary and github action binary.
```
PS C:\work\avr\avrdude_test\avrdude_bin\bin_misc\avrdude-msvc-x64> .\avrdude -c stk500v2 -P COM5
 -B0.5 -p m32a -U .\zeros32KB.hex -v
avrdude error: invalid bit clock period specified '0'

PS C:\work\avr\avrdude_test\avrdude_bin\bin_misc\avrdude-v7.0-windows-x64> .\avrdude -c stk500v2 -P COM5
 -B0.5 -p m32a -U .\zeros32KB.hex -v
avrdude.exe: invalid bit clock period specified '0'
```

mcuee, 2022-11-06:
@mariusgreuel 
For MSVC maybe you want to check if this is a limitation of the `getopt` replacement or not. Thanks.

mcuee, 2022-11-06:
> I can not reproduce the issue with MacOS using the USBasp programmer:
> 

No issues under Linux either.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c stk500v2 -P /dev/ttyUSB0 -B0.5 -p m128a

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9702 (probably m128a)

avrdude done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c stk500v2 -P /dev/ttyUSB0 -B0.5 -p m128a
 -qq -U flash:w:m128a_blinking.hex:i -&& echo OK
OK

```

dl8dtl, 2022-11-06:
Just a wild guess: is it possible that whatever shell you are using under Windows manipulates the commandline before passing it to the program?

You could try inserting:
```
    for (int i = 0; i < args; i++) printf("arg %d: \"%s\"\n", i, arg[i]);
```
early in `main()`.

mcuee, 2022-11-06:
> Just a wild guess: is it possible that whatever shell you are using under Windows manipulates the commandline before passing it to the program?
> 
> You could try inserting:
> 
> ```
>     for (int i = 0; i < args; i++) printf("arg %d: \"%s\"\n", i, arg[i]);
> ```
> 
> early in `main()`.

Great guess and it is indeed true.
1) No issue with Windows command prompt
```
c:\work\avr\avrdude_test\avrdude_bin>.\avrdude -c stk500v2 -P COM5 -B0.5 -p m8a

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9307 (probably m8a)

avrdude done.  Thank you.
```

2) No issue with MSYS2 shell
```
MINGW64 /c/work/avr/avrdude_test/avrdude_bin
$ ./avrdude -c stk500v2 -P COM5 -B0.5 -p m8a

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9307 (probably m8a)

avrdude done.  Thank you.
```

3) Issue only happens under PowerShell.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v2 -P COM5 -B0.5 -p m8a
avrdude error: invalid bit clock period specified '0'
```

dl8dtl, 2022-11-06:
After having seen that PowerShell corrupts traffic of a pipeline (I've been searchin a day for that), it doesn't really surprise me.

PowerShell only appears to be good for one purpose: handling .NET objects.
I'm done with it, and back to plain cmd.exe for everything I do under Windows.

mcuee, 2022-11-06:
> After having seen that PowerShell corrupts traffic of a pipeline (I've been searchin a day for that), it doesn't really surprise me.

Indeed I will say this is a PowerShell bug.

```
diff --git a/src/main.c b/src/main.c
index 23ef5ae..b69a402 100644
--- a/src/main.c
+++ b/src/main.c
@@ -530,6 +530,7 @@ int main(int argc, char * argv [])
     progname[strlen(progname)-4] = 0;
   }

+  for (int i = 0; i < args; i++) printf("arg %d: \"%s\"\n", i, arg[i]);
   default_programmer = "";
   default_parallel   = "";
   default_serial     = "";

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_arg -c stk500v2 -P COM5 -B0.5 -p m8a
arg 0: "C:\work\avr\avrdude_test\avrdude_bin\avrdude_arg.exe"
arg 1: "-c"
arg 2: "stk500v2"
arg 3: "-P"
arg 4: "COM5"
arg 5: "-B0"
arg 6: ".5"
arg 7: "-p"
arg 8: "m8a"
avrdude_arg error: invalid bit clock period specified '0'
```

mcuee, 2022-11-06:
@dl8dtl
Would you consider this issue report to be invalid? In that case, I will close the issue.

Or would you prefer to fix it by sanitizing the command line first by adding a space after the command?

PowerShell is supposed to be the replacement for Windows Command Prompt.

dl8dtl, 2022-11-06:
I'm afraid, at the point where AVRDUDE is called, the commandline arguments are already damaged, so there's no chance inside AVRDUDE to fix the breakage PowerShell provided. So yes, this bug report is invalid wrt. AVRDUDE.

I've also heard that it was Microsoft's intention to replace cmd.exe by PowerShell, but given what they are doing, there's simply no chance. It might be good for shuffling .NET objects around, but it is a complete failure for anything else. In my case, I had a small tool that extracted packet sniffer data from a hardware gadget, formatted them as PCAP, with the intention to pass it on to Wireshark for displaying it. The data that arrived at Wireshark were pretty much corrupted in multiple ways.

As I take it, all this wouldn't be considered PowerShell bugs by Microsoft but features. My conclusion was to simply no longer use it. (After all, it's scriptlet syntax is, well, pretty much off from anything that exists elsewhere.)

mcuee, 2022-11-06:
> I'm afraid, at the point where AVRDUDE is called, the commandline arguments are already damaged, so there's no chance inside AVRDUDE to fix the breakage PowerShell provided. So yes, this bug report is invalid wrt. AVRDUDE.

You are right. I will close this as invalid.

> As I take it, all this wouldn't be considered PowerShell bugs by Microsoft but features. My conclusion was to simply no longer use it. (After all, it's scriptlet syntax is, well, pretty much off from anything that exists elsewhere.)

I agree with you that the syntax is pretty much off from anything that exists elsewhere. I like PowerShell better than CMD since at least they accept things like `ls` by emulation. Since I switch between three major OS very often (Linux, macOS and Windows) and mostly use three apps (terminal, Chrome browser and txt editor). I like it better than CMD.

closed 2022-11-06 by mcuee

mcuee, 2022-11-06:
BTW, I use Windows Terminal and it can use various shell like CMD, PowerShell and MSYS2 shell.

Somehow Alacrity only use PowerShell (the built-in old version).

mcuee, 2022-11-06:
I think this PowerShell bug is the same as the following.
* https://github.com/PowerShell/PowerShell/issues/12975

mcuee, 2022-11-06:
Work-around is to use `-B"0.5"`.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_arg -c stk500v2 -P COM5 -B"0.5" -p m8a
arg 0: "C:\work\avr\avrdude_test\avrdude_bin\avrdude_arg.exe"
arg 1: "-c"
arg 2: "stk500v2"
arg 3: "-P"
arg 4: "COM5"
arg 5: "-B0.5"
arg 6: "-p"
arg 7: "m8a"

avrdude_arg: AVR device initialized and ready to accept instructions
avrdude_arg: device signature = 0x1e9307 (probably m8a)

avrdude_arg done.  Thank you.

```

dl8dtl, 2022-11-06:
> Work-around is to use `-B"0.5"` or `-B'0.5'`.

As I said: pretty much off from what any other shell is doing.

mcuee, 2022-11-06:
In the end it is a known bug.
* https://github.com/PowerShell/PowerShell/issues/6291

dl8dtl, 2022-11-06:
But even after 4.5 years, not fixed. :-(