avrdudes/avrdude #1173: Avrdude terminal mode on macOS isn't right

MCUdude, 2022-11-08:
The issue is caused by #1132.

I'm building using make, and after #1132 was merged, I have to hit enter after every command in order for `avrdude>` to appear. When I start to type, the text appears on the line below.

Shouldn't this "just work"?

```
$ ./avrdude -cpickit4_isp -p atmega1284p -t

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9705 (probably m1284p)
avrdude> ?
>>> ? 
Valid commands:
  dump    : dump <memory> [<addr> <len> | <addr> ... | <addr> | ...]
  read    : alias for dump
  write   : write <memory> <addr> <data>[,] {<data>[,]}
          : write <memory> <addr> <len> <data>[,] {<data>[,]} ...
  flush   : synchronise flash & EEPROM writes with the device
  abort   : abort flash & EEPROM writes (reset the r/w cache)
  erase   : perform a chip erase
  pgerase : pgerase <memory> <addr>
  sig     : display device signature bytes
  part    : display the current part information
  send    : send a raw command: send <b1> <b2> <b3> <b4>
  parms   : display adjustable parameters
  sck     : set <SCK period>
  verbose : change verbosity
  quell   : set quell level for progress bars
  help    : show help message
  ?       : same as help
  quit    : quit after writing out cache for flash & EEPROM

Note that not all programmer derivatives support all commands. Flash and
EEPROM type memories are normally read and written using a cache via paged
read and write access; the cache is synchronised on quit or flush commands.
The part command displays valid memory types for use with dump and write.

avrdude> 
read eeprom
avrdude> read eeprom
>>> read eeprom 

Reading | ################################################## | 100% 0.34 s 

0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> 
sig
avrdude> sig
>>> sig 
Device signature = 0x1e9705

avrdude> 
read flash 0 0x100
avrdude> read flash 0 0x100
>>> read flash 0 0x100 

Reading | ################################################## | 100% 0.09 s 

0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> 
quit
avrdude> quit
>>> quit 

avrdude done.  Thank you.
```

MCUdude, 2022-11-09:
The question is why did we have to break terminal mode for macOS? It worked perfectly fine before.
Can't we just use the "legacy way of doing things" if Avrdude was compiled without libreadline?

> BTW, you may want to use build.sh to simplify the workflow. It will take care of the readline issue.

Yes, but the classic `./configure` + `make` should also work. At the moment, there is no indication that libreadline is required when building.

The main reason why I'm still using make is that the commands are _much_ shorter and easier to remember. I might adopt Stefan's makefile:

```
all:
	cmake --build build_linux

install:
	cmake --build build_linux; sudo cmake --build build_linux --target install

new:
	sudo rm -rf build_linux
	cmake -D BUILD_DOC=1 -D CMAKE_BUILD_TYPE=RelWithDebInfo -D HAVE_LINUXGPIO=1 -D HAVE_LINUXSPI=1 -D HAVE_PARPORT=1 -B build_linux
	cmake --build build_linux
```

mcuee, 2022-11-09:
> The question is why did we have to break terminal mode for macOS? It worked perfectly fine before.
> Can't we just use the "legacy way of doing things" if Avrdude was compiled without libreadline?

The problem is that macOS Homebrew reported that they have libreadline without libreadline being installed, since it treats libedit as the replacement as libreadline. Unfortunately libedit behavior is not the same as libreadline. Maybe there is a simple fix but I do not know.

If we need to differentiate libreadline and libedit, probably we need to have some CMake magic from @mariusgreuel; and we need some code change in `terminal.c` from @stefanrueger.

MCUdude, 2022-11-09:
> The problem is that macOS Homebrew reports that it has libreadline without libreadline being installed, since it treats libedit as the replacement as libreadline. Unfortunately libedit behavior is not the same as libreadline.

OK, fair enough. But is libreadline statically or dynamically linked? Is this something every user has to install before using an Avrdude binary?

It might be possible to report an issue over at the Homebrew repo to fix the issue where Homebrew treats libedit as a replacement.

mcuee, 2022-11-09:
> OK, fair enough. But is libreadline statically or dynamically linked? Is this something every user has to install before using an Avrdude binary?

By default it is dynamically linked. That was why I created issue #1159.
* https://github.com/avrdudes/avrdude/issues/1159

After extensive testing there, finally I came to the conclusion that me the solution is to adopt arduino-packing way of building the static link binaries. Unfortunately it needs two patches which as is will not be accepted by the avrdude project. 
* https://github.com/avrdudes/avrdude/issues/1162

> It might be possible to report an issue over at the Homebrew repo to fix the issue where Homebrew treats libedit as a replacement.

mcuee, 2022-11-09:
There are many hits about libedit not a real replacement for libreadline.

References:
https://superuser.com/questions/1399622/trouble-with-readline-library-on-mac
https://pewpewthespells.com/blog/osx_readline.html

MCUdude, 2022-11-09:
The problem is this line:
https://github.com/avrdudes/avrdude/blob/4c92030e3a486cfbaeb36e298f6f2929e8e031eb/src/term.c#L1233

If I replace it with `#if 0`, terminal mode works perfectly fine again. Do you know _where_ LIBREADLINE is defined? maybe there's something else we can look for to detect if it's actually using libedit? This way, we could make it work on macOS with and without libreadline

But how do I install libreadline using homebrew?

When I run `brew search libreadline`, there's no obvious one to install:

```
$ brew search readline
==> Formulae
libreadline-java                          readline ✔                                with-readline                             redland

==> Casks
readmoreading
```

mcuee, 2022-11-09:
@MCUdude 
You can install libreadline by using `brew install readline`. 

As for `(HAVE_LIBREADLINE)`, unfortunately CMake will set it under macOS no matter you install `readline` or not, because of the presence of libedit in macOS. So we need some CMake magic from @mariusgreuel to differentiate libedit and libreadline. I will also try but I do not know so much about CMake.

stefanrueger, 2022-11-10:
> If I replace it with #if 0, terminal mode works *perfectly* fine again.

You will lose the ability to connect to bootloaders in the terminal (they will time out).

I feel we should solve this by getting CMake to do the right thing for MacOS before the 7.1 release.

mcuee, 2022-11-10:
> > If I replace it with #if 0, terminal mode works _perfectly_ fine again.
> 
> You will lose the ability to connect to bootloaders in the terminal (they will time out).
> I feel we should solve this by getting CMake to do the right thing for MacOS before the 7.1 release.

I think we can detect the presence of Homebrew libreadline by checking the presence of `/opt/homebrew/opt/readline/lib/libreadline.dylib` or `/usr/local/opt/readline/lib/libreadline.dylib`. And we can detect the presence of MacPorts libreadline by checking the presence of `/usr/local/lib/libreadline.dylib`. 

If the above is not present, then it will use system libedit. We can set `HAVE_LIBREADLINE` to be worse in that case.

@mariusgreuel 
Please jump in as well and I believe you can find a quick solution for this in CMake.

mcuee, 2022-11-12:
> > If I replace it with #if 0, terminal mode works _perfectly_ fine again.
> 
> You will lose the ability to connect to bootloaders in the terminal (they will time out).
> 
> I feel we should solve this by getting CMake to do the right thing for MacOS before the 7.1 release.

There are three issues here.
1) Enhance CMake to differentiate libedit and libreadline. Right now you have to use `build.sh` which is hardcoded to use libreadline. This may not be a big issue -- we can just to modify the Wiki so that user will install the libreadline dependancy. But it is still good to get CMake to do the job automatically. We can even warn the user certain function may not work without libreadline.

2) @MCUdude would like to get the autotools build working with and without libreadline (to differentiate libedit and libreadline).

3) Staitc linking to libreadline if it is present. This is more difficult since currently it is using dynamic linking by default. I think it is out of scope for this ticket. We can discuss that in the following ticket.
* https://github.com/avrdudes/avrdude/issues/1162

mcuee, 2022-11-12:
Interestingly Homebrew git main build is actually able to use libreadline and not libedit for my system.

The formula does not seem to do anything special.
https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/avrdude.rb

```
mcuee@mcuees-Mac-mini avrdude_main % brew reinstall avrdude
==> Cloning https://github.com/avrdudes/avrdude.git
Updating /Users/mcuee/Library/Caches/Homebrew/avrdude--git
From https://github.com/avrdudes/avrdude
   3e49f07..4c92030  main       -> origin/main
==> Checking out branch main
Already on 'main'
Your branch is behind 'origin/main' by 145 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
HEAD is now at 4c92030 Update NEWS
==> Reinstalling avrdude 
==> cmake -S . -B build/shared -DCMAKE_INSTALL_SYSCONFDIR=/opt/homebrew/etc -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTAL
==> cmake --build build/shared
==> cmake --install build/shared
==> cmake -S . -B build/static -DCMAKE_INSTALL_SYSCONFDIR=/opt/homebrew/etc
==> cmake --build build/static
🍺  /opt/homebrew/Cellar/avrdude/HEAD-4c92030_2: 13 files, 1.9MB, built in 8 seconds
==> Running `brew cleanup avrdude`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
mcuee@mcuees-Mac-mini avrdude_main % otool -L /opt/homebrew/bin/avrdude 
/opt/homebrew/bin/avrdude:
	@rpath/libavrdude.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1319.0.0)
	/opt/homebrew/opt/libusb-compat/lib/libusb-0.1.4.dylib (compatibility version 9.0.0, current version 9.4.0)
	/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/opt/homebrew/opt/libhid/lib/libhid.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/opt/homebrew/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.12.0)
	/opt/homebrew/opt/libftdi0/lib/libftdi.1.dylib (compatibility version 22.0.0, current version 22.0.0)
	/opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib (compatibility version 2.0.0, current version 2.5.0)
	/opt/homebrew/opt/readline/lib/libreadline.8.dylib (compatibility version 8.2.0, current version 8.2.0)
```

mcuee, 2022-11-12:
@MCUdude 
I just tried the auto-tool build under macOS, as long as you have readline installed, you should be fine.

There is a warning about the `ld: warning: -undefined dynamic_lookup may not work with chained fixups` and I am not so sure if the Homebrew patch is relevant or not.
https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/avrdude.rb

```
mcuee@mcuees-Mac-mini src % ./bootstrap 
+ rm -rf autom4te.cache
+ LIBTOOLIZE=libtoolize
+ glibtoolize --version
+ LIBTOOLIZE=glibtoolize
+ glibtoolize
glibtoolize: putting auxiliary files in '.'.
glibtoolize: linking file './ltmain.sh'
glibtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
glibtoolize: linking file 'm4/libtool.m4'
glibtoolize: linking file 'm4/ltoptions.m4'
glibtoolize: linking file 'm4/ltsugar.m4'
glibtoolize: linking file 'm4/ltversion.m4'
glibtoolize: linking file 'm4/lt~obsolete.m4'
+ aclocal
+ autoheader
+ autoconf
configure.ac:46: warning: AC_PROG_LEX without either yywrap or noyywrap is obsolete
./lib/autoconf/programs.m4:716: _AC_PROG_LEX is expanded from...
./lib/autoconf/programs.m4:709: AC_PROG_LEX is expanded from...
configure.ac:46: the top level
configure.ac:120: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:120: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:120: the top level
configure.ac:227: warning: The macro `AC_HEADER_TIME' is obsolete.
configure.ac:227: You should run autoupdate.
./lib/autoconf/headers.m4:743: AC_HEADER_TIME is expanded from...
configure.ac:227: the top level
configure.ac:249: warning: The macro `AC_TRY_LINK' is obsolete.
configure.ac:249: You should run autoupdate.
./lib/autoconf/general.m4:2920: AC_TRY_LINK is expanded from...
configure.ac:249: the top level
configure.ac:281: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:281: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:281: the top level
configure.ac:299: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:299: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:299: the top level
configure.ac:311: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:311: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:311: the top level
configure.ac:323: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:323: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:323: the top level
configure.ac:335: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:335: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:335: the top level
configure.ac:440: warning: The macro `AC_TRY_COMPILE' is obsolete.
configure.ac:440: You should run autoupdate.
./lib/autoconf/general.m4:2847: AC_TRY_COMPILE is expanded from...
configure.ac:440: the top level
configure.ac:465: warning: The macro `AC_TRY_COMPILE' is obsolete.
configure.ac:465: You should run autoupdate.
./lib/autoconf/general.m4:2847: AC_TRY_COMPILE is expanded from...
configure.ac:465: the top level
configure.ac:489: warning: The macro `AC_TRY_LINK' is obsolete.
configure.ac:489: You should run autoupdate.
[... 236 lines cut ...]
  CC       libavrdude_la-serialupdi.lo
  CC       libavrdude_la-updi_link.lo
  CC       libavrdude_la-updi_state.lo
  CC       libavrdude_la-updi_readwrite.lo
  CC       libavrdude_la-updi_nvm.lo
  CC       libavrdude_la-usb_hidapi.lo
  CC       libavrdude_la-usb_libusb.lo
  CC       libavrdude_la-usbtiny.lo
  CC       libavrdude_la-update.lo
  CC       libavrdude_la-wiring.lo
  CC       libavrdude_la-xbee.lo
  CCLD     libavrdude.la
ld: warning: -undefined dynamic_lookup may not work with chained fixups
mcuee@mcuees-Mac-mini src % otool -L ./avrdude
./avrdude:
	/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/opt/homebrew/opt/libusb-compat/lib/libusb-0.1.4.dylib (compatibility version 9.0.0, current version 9.4.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1953.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1319.0.0)
	/opt/homebrew/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.12.0)
	/opt/homebrew/opt/readline/lib/libreadline.8.dylib (compatibility version 8.2.0, current version 8.2.0)
	/usr/lib/libncurses.5.4.dylib (compatibility version 5.4.0, current version 5.4.0)
	/opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib (compatibility version 2.0.0, current version 2.5.0)
```

mcuee, 2022-11-12:
@MCUdude 
You do not need to force link readline either, just follow the instruction from `brew info readline` and you will be fine.

```
mcuee@mcuees-Mac-mini src % brew info readline
==> readline: stable 8.2.1 (bottled) [keg-only]
Library for command-line editing
https://tiswww.case.edu/php/chet/readline/rltop.html
/opt/homebrew/Cellar/readline/8.2.1 (50 files, 1.7MB)
  Poured from bottle on 2022-10-20 at 19:12:26
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/readline.rb
License: GPL-3.0-or-later
==> Caveats
readline is keg-only, which means it was not symlinked into /opt/homebrew,
because macOS provides BSD libedit.

For compilers to find readline you may need to set:
  export LDFLAGS="-L/opt/homebrew/opt/readline/lib"
  export CPPFLAGS="-I/opt/homebrew/opt/readline/include"

For pkg-config to find readline you may need to set:
  export PKG_CONFIG_PATH="/opt/homebrew/opt/readline/lib/pkgconfig"

==> Analytics
install: 968,971 (30 days), 1,939,945 (90 days), 7,279,339 (365 days)
install-on-request: 33,853 (30 days), 72,830 (90 days), 245,882 (365 days)
build-error: 11 (30 days)

mcuee@mcuees-Mac-mini src % CFLAGS="-I/opt/homebrew/include -I/opt/homebrew/opt/readline/include" LDFLAGS="-L/opt/homebrew/lib -L/opt/homebrew/opt/readline/lib" ./configure
...
mcuee@mcuees-Mac-mini src % make
...
mcuee@mcuees-Mac-mini src % otool -L avrdude  
avrdude:
	/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/opt/homebrew/opt/libusb-compat/lib/libusb-0.1.4.dylib (compatibility version 9.0.0, current version 9.4.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1953.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1319.0.0)
	/opt/homebrew/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.12.0)
	/opt/homebrew/opt/readline/lib/libreadline.8.dylib (compatibility version 8.2.0, current version 8.2.0)
	/usr/lib/libncurses.5.4.dylib (compatibility version 5.4.0, current version 5.4.0)
	/opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib (compatibility version 2.0.0, current version 2.5.0)
```

mcuee, 2022-11-12:
There is a discussion here for reference -- to me it is good to mandate the use of readline.
* https://github.com/PostgresApp/PostgresApp/issues/144#issuecomment-840665036

mcuee, 2022-11-12:
@mariusgreuel 

Would you please look at the script here for a more comprehensive way of finding readline so that libedit will not be detected as readline? Thanks.
https://github.com/monero-project/monero/blob/master/cmake/FindReadline.cmake

The project has similar more complex scripts for libusb and hidapi, but I think those are not necessary.
https://github.com/monero-project/monero/blob/master/cmake/FindLibUSB.cmake
https://github.com/monero-project/monero/blob/master/cmake/FindHIDAPI.cmake

mcuee, 2022-11-12:
For auto-tools build `configure.ac` needs to be changed to detect libedit, right now the following lines will not do the job.
https://github.com/avrdudes/avrdude/blob/main/src/configure.ac#L80-L83
```
# Checks for libraries.
AC_CHECK_LIB([termcap], [tputs])
AC_CHECK_LIB([ncurses], [tputs])
AC_CHECK_LIB([readline], [readline])
```

Something for reference from Postgres.
https://github.com/postgres/postgres/blob/master/configure.ac#L1503-L1521

mcuee, 2022-11-12:
@MCUdude and @stefanrueger 

> > If I replace it with #if 0, terminal mode works _perfectly_ fine again.
> 
> You will lose the ability to connect to bootloaders in the terminal (they will time out).
> 
> I feel we should solve this by getting CMake to do the right thing for MacOS before the 7.1 release.

Yet the other quick solution is to follow @MCUdude's suggestion to sort out the immediate issue and then forego the benefits of the terminal time out as of now. In this case, everything will work out of the box under macOS except the terminal mode timeout improvement (no more quick flashing LEDs if you want some visual hints).
```
mcuee@mcuees-Mac-mini avrdude_sr % git diff
diff --git a/src/term.c b/src/term.c
index c3f643f..7cbaf8f 100644
--- a/src/term.c
+++ b/src/term.c
@@ -1230,7 +1230,7 @@ static int process_line(char *cmdbuf, PROGRAMMER *pgm, struct avrpart *p) {
 
 
 
-#if defined(HAVE_LIBREADLINE)
+#if defined(HAVE_LIBREADLINE) && !defined(__APPLE__)
 
 static PROGRAMMER *term_pgm;
 static struct avrpart *term_p;
```

Take note libedit (or libreadline) is still needed for other parts of `term.c`. For example, if I disable `HAVE_LIBREADLINE` totoally it will fail to build.

```

mcuee@mcuees-Mac-mini avrdude_main % cmake -D CMAKE_C_FLAGS=-I/opt/homebrew/include
 -D CMAKE_EXE_LINKER_FLAGS=-L/opt/homebrew/Cellar -D HAVE_LIBREADLINE='' 
-D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_macos
...
mcuee@mcuees-Mac-mini avrdude_main % cmake --build build_macos -v
...
[100%] Linking C executable avrdude
cd /Users/mcuee/build/avr/avrdude_test/avrdude_main/build_macos/src && /opt/homebrew/Cellar/cmake/3.24.2/bin/cmake -E cmake_link_script CMakeFiles/avrdude.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/cc -I/opt/homebrew/include -O2 -g -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names -L/opt/homebrew/Cellar CMakeFiles/avrdude.dir/main.c.o CMakeFiles/avrdude.dir/term.c.o CMakeFiles/avrdude.dir/avrintel.c.o CMakeFiles/avrdude.dir/developer_opts.c.o CMakeFiles/avrdude.dir/whereami.c.o -o avrdude  libavrdude.a -lm /opt/homebrew/lib/libelf.a /opt/homebrew/lib/libusb.dylib /opt/homebrew/lib/libusb-1.0.dylib /opt/homebrew/lib/libhid.dylib /opt/homebrew/lib/libhidapi.dylib /opt/homebrew/lib/libftdi.dylib /opt/homebrew/lib/libftdi1.dylib 
Undefined symbols for architecture arm64:
  "_add_history", referenced from:
      _term_gotline in term.c.o
  "_rl_callback_handler_install", referenced from:
      _terminal_mode in term.c.o
  "_rl_callback_handler_remove", referenced from:
      _terminal_mode in term.c.o
  "_rl_callback_read_char", referenced from:
      _terminal_mode in term.c.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/avrdude] Error 1
make[1]: *** [src/CMakeFiles/avrdude.dir/all] Error 2
```

mcuee, 2022-11-12:
BTW, it is not a big problem to require readline under macOS. You can see common software like Python depends on readline. Most likely it is because libedit is not able to do the job.

```
mcuee@mcuees-Mac-mini ~ % brew uninstall readline
Error: Refusing to uninstall /usr/local/Cellar/readline/8.2.1
because it is required by amtk, atk, cairo, enchant, glib-networking, gnutls, gobject-introspection, gtk+, gtk+3, gtk-mac-integration, gtksourceview4, gts, guile, itstool, liblqr, libpeas, libsoup, libxml2, neofetch, py3cairo, pygobject3, python-tk@3.10, python-tk@3.9, python@3.10, python@3.9, ranger, ruby, shared-mime-info and sqlite, which are currently installed.
You can override this and force removal with:
  brew uninstall --ignore-dependencies readline
```

mcuee, 2022-11-12:
So the following is my recommendation.
1) Require readline under macOS but no force link
2) Ask the user to use `build.sh` or use CMake option which hard code readline location
3) For auto-tool users who use Homebrew, ask them to use the follow the configure option recommendation from readline keg.
```
CFLAGS="-I/opt/homebrew/include -I/opt/homebrew/opt/readline/include" LDFLAGS="-L/opt/homebrew/lib -L/opt/homebrew/opt/readline/lib" ./configure
```
or
```
CFLAGS="-I/usr/local/include -I/usr/local/opt/readline/include" LDFLAGS="-L/usr/local/lib -L/usr/local/opt/readline/lib" ./configure
```

mcuee, 2022-11-12:
But if @MCUdude prefers not to require readline, I am also okay with the following one line change. (Line number is for PR #1171 as I expect it to be merged soon).

```
mcuee@mcuees-Mac-mini avrdude_sr % git diff
diff --git a/src/term.c b/src/term.c
index c3f643f..7cbaf8f 100644
--- a/src/term.c
+++ b/src/term.c
@@ -1230,7 +1230,7 @@ static int process_line(char *cmdbuf, PROGRAMMER *pgm, struct avrpart *p) {
 
 
 
-#if defined(HAVE_LIBREADLINE)
+#if defined(HAVE_LIBREADLINE) && !defined(__APPLE__)
 
 static PROGRAMMER *term_pgm;
 static struct avrpart *term_p;
```

mcuee, 2022-11-12:
Future improvement when @mariusgreuel or other CMake experts are available -- to differentiate libedit vs libreadline and set that line accordingly. I am not so sure if @dl8dtl or @mariusgreuel want to fix the `configure.ac` file.

MCUdude, 2022-11-16:
> Please install libreadline and test using both CMake and auto-tools. You will need to force link readline as well by running brew link --force readline. If both are fine, we have a very simple work-around, ask the user to install libreadline and force link it. Without the force link step, avrdude will use libedit and the issue will be present.
```
mcuee@mcuees-Mac-mini avrdude_bin % brew link readline  
Warning: readline is keg-only and must be linked with `--force`.
mcuee@mcuees-Mac-mini avrdude_bin % brew link readline --force
Linking /opt/homebrew/Cellar/readline/8.2.1... 18 symlinks created.
mcuee@mcuees-Mac-mini avrdude_bin % pkg-config --libs readline
-L/opt/homebrew/Cellar/readline/8.2.1/lib -lreadline
mcuee@mcuees-Mac-mini avrdude_bin % pkg-config --cflags readline
-I/opt/homebrew/Cellar/readline/8.2.1/include
```

I'm really having a hard time getting the Avrdude build system to pick the correct readline library.
libreadline are for some reason installed under `/usr/local/opt/readline/lib/libreadline.dylib`
```
$ brew install readline

Warning: readline 8.2.1 is already installed and up-to-date.
To reinstall 8.2.1, run:
  brew reinstall readline

$ brew link readline --force
Warning: Refusing to link macOS provided/shadowed software: readline
For compilers to find readline you may need to set:
  export LDFLAGS="-L/usr/local/opt/readline/lib"
  export CPPFLAGS="-I/usr/local/opt/readline/include"

For pkg-config to find readline you may need to set:
  export PKG_CONFIG_PATH="/usr/local/opt/readline/lib/pkgconfig"
```

When I build using
```
$ cmake -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/local/Cellar -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_macos

$ cmake --build build_macos
```

Terminal mode is still broken, just like when I build with `make`

I've also tried

```
$ cmake -D CMAKE_C_FLAGS="-I/usr/local/include -I/usr/local/opt/readline/include" -D CMAKE_EXE_LINKER_FLAGS=-L/usr/local/Cellar -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_macos
```
but I'm getting the exact same result

mcuee, 2022-11-17:
@MCUdude 
I have already added the necessary change to `build.sh` in PR #1156 which has been merged.
https://github.com/avrdudes/avrdude/blob/main/build.sh

You need to use something like this.
```
cmake -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/local/Cellar 
-D HAVE_LIBREADLINE:FILEPATH=/usr/local/opt/readline/lib/libreadline.dylib" 
-D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_macos

```

MCUdude, 2022-11-17:
Thanks @mcuee! Using build.sh did work. I'm totally OK with Avrdude requiring libreadline in order to work, but I think that the precense/absense of libreadline should be detected and printed on the screen when building with build.sh, cmake, or make.

Currently, there's no mention of libreadline anywhere:

```
 ./build.sh 
-- The C compiler identification is AppleClang 10.0.1.10010046
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/local/bin/git (found version "2.30.0") 
-- Found FLEX: /usr/bin/flex (found version "2.5.35") 
-- Found BISON: /usr/bin/bison (found version "2.3")
-- Looking for libelf.h
-- Looking for libelf.h - not found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/hans/Downloads/avrdude-terminal/build_darwin
```

To me, it's important that building Avrdude "just works", on any system, as long as the user follows the build instructions. The instructions are yet to be updated (for macOS at least).

mcuee, 2022-11-17:
@MCUdude 
I agree with you. I will see what I can do to improve CMake to printout readline library.

As for the build instruction, do you have access to the Wiki? If yes please help to update. 

@dl8dtl 
Sorry to ping again on the Wiki access. As of now I do not have Wiki access. I'd like to update the build instruction.

MCUdude, 2022-11-18:
> Sorry to ping again on the Wiki access. As of now I do not have Wiki access. I'd like to update the build instruction.

@dl8dtl would be great if you could consider this. It would be beneficial if @mcuee were allowed to edit the wiki

mcuee, 2022-11-19:
@dl8dtl 

Thanks. Now I am able to edit the Wiki.

mcuee, 2022-11-19:
@MCUdude 
It is easy to print out the presence of libreadline with the following small patch.
```
PS C:\work\avr\avrdude_test\avrdude_main> git diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 400da29..1269650 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -361,6 +361,12 @@ else()
     message(STATUS "DON'T HAVE libftdi1")
 endif()

+if(HAVE_LIBREADLINE)
+    message(STATUS "DO HAVE    libreadline")
+else()
+    message(STATUS "DON'T HAVE libreadline")
+endif()
+
 if(BUILD_DOC)
        message(STATUS "ENABLED    doc")
 else()
```

But that does not sort out the issue under macOS since it will still detect the system libedit as readline if the user does not explicitly install libreadline and point CMake to use the installed libreadline.

MCUdude, 2022-11-19:
> But that does not sort out the issue under macOS yet since it will still detect the system libedit as readline if the user does not explicitly install libreadline and point CMake to use the installed libreadline.

Yes, this is the root problem. I hope there is a way for the build system to determine that libreadline isn't really installed, and it's using libedit instead. Is it perhaps possible to check the install path or libreadline version number to determine what kind of library we're dealing with?

The updated Wiki look OK, but is there a typo in the build command for Apple silicon? `CMAKE_C_FLAGS` appears twice.

```
cmake -D CMAKE_C_FLAGS=-D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/homebrew/Cellar -D HAVE_LIBREADLINE:FILEPATH=/opt/homebrew/opt/readline/lib/libreadline.dylib -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_macos

```

dl8dtl, 2022-11-19:
> This is because you use the default libedit and not libreadline.

Curious: do you know what causes that behaviour with libedit?

If it's possible, I think it would be worth the while to fix the root problem rather than insisting on 3rd-party software.

MCUdude, 2022-11-19:
I'm not sure what's _causing it_, but if you try to build Avrdude without libreadline installed, terminal mode is a real pain to use. And I'm sure it will take years from the libedit issue being fixed to Apple updating the libedit binary they provide.

I'm totally fine with Avrdude depending on libreadline (from brew or macports). What bothers me is the readline.dylib path has to be "manually" passed to Cmake since macOS instead will use libedit, which causes problems for terminal mode. Another thing; when running the Cmake build command for macOS, it will print `DO HAVE    libreadline` regardless of libreadline is installed or not. I hope that this is something that can be resolved before the 7.1 release

mcuee, 2022-11-20:
> The updated Wiki look OK, but is there a typo in the build command for Apple silicon? CMAKE_C_FLAGS appears twice.
Fixed. Thanks.

mcuee, 2022-11-20:
> > This is because you use the default libedit and not libreadline.
> 
> Curious: do you know what causes that behaviour with libedit?
> 
> If it's possible, I think it would be worth the while to fix the root problem rather than insisting on 3rd-party software.

Yes it would be great to be able to use libedit without readline under macOS. But I think this may be challenging. I can see many applications under macOS still requires readline as libedit behaves differently in some cases.

As @MCUdude found out, the issue is with the following codes for the newly added feature of Terminal Mode timeout prevention.
https://github.com/avrdudes/avrdude/blob/main/src/term.c#L1240-L1343

As mentioned in earlier comments, we can use one line change to get libedit to work well under macOS but without the new feature.
* https://github.com/avrdudes/avrdude/issues/1173#issuecomment-1312423630

mcuee, 2022-11-20:
@stefanrueger @MCUdude and @dl8dtl 

If we want to try to get libedit to work, I am thinking we may need workaround like the following somewhere in the above readline callback codes.
https://github.com/avrdudes/avrdude/blob/main/src/term.c#L1217-L1222

``
#if !defined(HAVE_LIBREADLINE) || defined(WIN32) || defined(__APPLE__)
    term_out(">>> ");
    for (int i=0; i<argc; i++)
      term_out("%s ", argv[i]);
    term_out("\n");
#endif
```

dl8dtl, 2022-11-20:
So it seems the issue with the way the callbacks are being handled. Maybe we can come up with a solution that "just works" for both libraries.
Btw., since libedit comes from NetBSD, it's likely we would encounter similar issues there.

mcuee, 2022-11-20:
> So it seems the issue with the way the callbacks are being handled. Maybe we can come up with a solution that "just works" for both libraries. Btw., since libedit comes from NetBSD, it's likely we would encounter similar issues there.

That will be good. As for NetBSD, I am not able to get it working yet so I can not test it.

MCUdude, 2022-11-20:
> Maybe we can come up with a solution that "just works" for both libraries.

If possible, that would be the absolute best solution. Alternatively, a way that results in a working terminal mode for macOS regardless if libreadline is installed or not. I know libreadline is _required_ in order to use urboot in terminal mode, but some users may not have this requirement.

dl8dtl, 2022-11-20:
I'm currently fetching the NetBSD source, to have a look into their implementation.
Strange enough, they still use CVS …

mcuee, 2022-11-20:
> I'm currently fetching the NetBSD source, to have a look into their implementation. Strange enough, they still use CVS …

You can use their github mirror.
https://github.com/NetBSD/src/tree/trunk/lib/libedit

mcuee, 2022-11-20:
> Alternatively, a way that results in a working terminal mode for macOS regardless if libreadline is installed or not. I know libreadline is _required_ in order to use urboot in terminal mode, but some users may not have this requirement.

@MCUdude 

This should be pretty easy. 
Can you try one line change here? Thanks.
https://github.com/avrdudes/avrdude/blob/main/src/term.c#L1233

Change the line from
```
#if defined(HAVE_LIBREADLINE)
```
to
```
#if defined(HAVE_LIBREADLINE) && !defined(__APPLE__)
```

stefanrueger, 2022-11-20:
> So it seems the issue with the way the callbacks are being handled. Maybe we can come up with a solution that "just works" for both libraries.

Issue #1028 had a go at a small stand-alone program that exhibits the necessary callback functionality. @mariusgreuel then [ported this to Windows](https://github.com/mariusgreuel/termreadline) and, essentially this is what I then put in term.c. So I suggest trying
```
$ wget https://raw.githubusercontent.com/mariusgreuel/termreadline/main/termreadline.c
$ cc termreadline.c -lreadline
$ a.out
avrdude> ..te.st
working hard on test
avrdude> .fe.st
working hard on fest
avrdude> .q..uit
quit: bye
```

If you see the avrdude> prompt, can type a command and see periodically some dots appearing (these are background actions, which in avrdude will be replaced to keep the terminal alive), then all is good.

This to facilitate `-t` with bootloaders (optiboot, stk500v1-type, urboot). Currently only with `-c urclock` but once that's successful, this technique could be rolled out to other bootloader programmers.

MCUdude, 2022-11-20:
> Change the line from [...]

If I do this, terminal mode works as expected for all programmers _except_ for urclock/urboot, since urboot times out. This is because terminal mode isn't keeping the bootloader alive by periodically sending data to it.

libedit is actually keeping the bootloader alive, but the issue is how it deals with user input. It _works_ but is really weird to use.

stefanrueger, 2022-11-20:
> really weird to use

What's the undesired behaviour?

Is it a doubling up of input lines? this may be caused by
```
#if !defined(HAVE_LIBREADLINE) || defined(WIN32) || defined(__APPLE__)
    term_out(">>> ");
    for (int i=0; i<argc; i++)
      term_out("%s ", argv[i]);
    term_out("\n");
#endif
```

This code is actually only meant to be used in absence of `HAVE_LIBREADLINE`. Maybe removing the `|| defined(__APPLE__)` helps? Other than that libreadline (or libedit for that matter) needs to switch the terminal into raw mode as opposed to cooked mode. All that works well in Linux. Don't know how you can get this in MacOS. I recommend tinkering with the standalone example above https://github.com/avrdudes/avrdude/issues/1173#issuecomment-1321108999

MCUdude, 2022-11-20:
> What's the undesired behaviour?

When type a command and hit enter, the command is run, but the `avrdude>` text doesn't show up until I press enter again.
When I hit enter again and start typing, the text I'm typing appears on the line _below_ `avrdude>`.
And when I type (before hitting enter), I can not move the cursor left or right, I can only use backspace.

BUT! Urcock doesn't time out in terminal mode!
```
$ ./avrdude -cusbasp -patmega16 -t

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9403 (probably m16)
avrdude> read flash
>>> read flash 

Reading | ################################################## | 100% 0.04 s 

0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> 
read flash
avrdude> read flash
>>> read flash 

Reading | ################################################## | 100% 0.05 s 

0100  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0110  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0120  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0130  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0140  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0150  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0160  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0170  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0180  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0190  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
01a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
01b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
01c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
01d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
01e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
01f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

read flash^[[D^[[D^[[D^[[D^[[D
avrdude> read flash
>>> read flash 

Reading | ################################################## | 100% 0.05 s 

0200  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0210  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0220  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0230  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0240  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0250  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0260  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0270  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0280  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0290  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

quit
avrdude> quit
>>> quit 

avrdude done.  Thank you.
```

MCUdude, 2022-11-20:
> This code is actually only meant to be used in absence of HAVE_LIBREADLINE. Maybe removing the || defined(__APPLE__) helps?

Correct. And the artifacts I'm seeing are because the "libreadline code" is using the libedit library on macOS. Removing `|| defined(__APPLE__) ` does not help. The problem is that `HAVE_LIBREADLINE` gets defined because macOS thinks that libedit is a compatible alternative to libreadline

stefanrueger, 2022-11-20:
Forgive me if I am stating the obvious: So, probably best to install libreadline and link with this and not libedit?

MCUdude, 2022-11-20:
> Forgive me if I am stating the obvious: So, probably best to install libreadline and link with this and not libedit? [edit I mean irrespective of whether HAVE_LIBREADLINE is defined or not - install it and link with it]

Requiring libreadline is totally fine, and now the wiki has been updated so that the Cmake build command work.
When I opened this issue, the docs were't updated, and libreadline was not mentioned in the build output log.

But the "classic" ./configure + make should also be updated to properly deal with libreadline on macOS

dl8dtl, 2022-11-20:
> So, probably best to install libreadline and link with this and not libedit?

As libedit is supposed to be a replacement for GNU readline (it comes from NetBSD), if we can get both to work, that would be my preferred solution.

dl8dtl, 2022-11-20:
I compiled NetBSD's libedit, and linked it to Marius' test program you mentioned.
It experiences the same behaviour as Hans described.

dl8dtl, 2022-11-20:
Jumping up and down, I found a crude hack that works around the issue.
My MacOS environment does have GNU libreadline so I could not reproduce the original issue there, but using NetBSD's source and the test program, I can see the effect is no longer present when using the workaround from PR #1185.

Please give it a try.

Of course, someone might file the bug report with NetBSD as well. The original test program from the GNU libreadline documentation doesn't work there either. However, even if they eventually fix the bug, it will likely take a substantial amount of time until it propagates to MacOS.

mcuee, 2022-11-20:
> My MacOS environment does have GNU libreadline so I could not reproduce the original issue there

This is because you use MacPorts where GNU readline will replace system provided libedit completely. Thanks for the confirmation -- now I can confirm that Wiki is still correct (no need to update for MacPorts).

For Homebrew, GNU readline is a `keg` and by default it does not replace system provided libedit unless the user forces link it. That is why the `hackish` solution to point CMake to the `readline keg`.

dl8dtl, 2022-11-20:
It doesn't really replace the system provided library, but the Macports path is searched first, so it will be found there.

(I could have temporarily removed the port for a test.)

Let's see whether Hans can test it.

mcuee, 2022-11-20:
> It doesn't really replace the system provided library, but the Macports path is searched first, so it will be found there.

Yes this is the right statement.

> (I could have temporarily removed the port for a test.)
> 
> Let's see whether Hans can test it.

I can confirm that PR #1185 fixed the issue for me.

I actually use the urclock PR #1171 to test this.
```
mcuee@mcuees-Mac-mini avrdude_sr % git diff
diff --git a/src/term.c b/src/term.c
index 78f3721..ba4fa7c 100644
--- a/src/term.c
+++ b/src/term.c
@@ -1331,7 +1331,16 @@ void term_gotline(char *cmdstr) {
         term_running = 0;
     }
     free(cmdstr);
-  } else {
+     /*
+     * This is a workaround for a bug apparently present in the
+     * readline compat layer of libedit which is natively present in
+     * NetBSD and MacOS.
+     *
+     * see https://github.com/avrdudes/avrdude/issues/1173
+     */
+    rl_callback_handler_remove();
+    rl_callback_handler_install("avrdude> ", term_gotline);
+    } else {
     // call quit at end of file or terminal ^D
     term_out("\n");
     cmd_quit(term_pgm, term_p, 0, NULL);
```

mcuee, 2022-11-21:
@MCUdude 
Please PR #1185 a try to see if it works for you. You need to revert the build.sh change from PR #1156.

@dl8dtl 
I will also test it under Linux/Windows to see if there is any side-effect for GNU readline.

If all works, we can revert #1156 changes to `build.sh` and github action for macOS. I will change the Wiki macOS build instruction as well after the confirmations.

mcuee, 2022-11-21:
@stefanrueger 

Hopefully #1185 will finally sort out the issue under macOS. 

As for Windows, we can deal with the Readline issues in the following two tickets.
* #1155 (MinGW)
* #1186 (MSVC, we may have to wait for @mariusgreuel)

At least I can build the mingw32/mingw64 binaries for people who want to try `-c urclock` in terminal mode.

dl8dtl, 2022-11-21:
> I will also test it under Linux/Windows to see if there is any side-effect for GNU readline.

I tested it under both MacOS and FreeBSD with GNU readline.

MCUdude, 2022-11-21:
> I tested it under both MacOS and FreeBSD with GNU deadline.

Excellent! Would there be a way for #1185 to print something like "USING LIBEDIT" or "USING LIBREADLINE"? How can the build system detect what's being used on macOS?

mcuee, 2022-11-21:
> > I tested it under both MacOS and FreeBSD with GNU deadline.
> 
> Excellent! Would there be a way for #1185 to print something like "USING LIBEDIT" or "USING LIBREADLINE"? How can the build system detect what's being used on macOS?

That is the challenging part -- we need a CMake expert here.
https://github.com/avrdudes/avrdude/issues/1173#issuecomment-1312370924

Maybe something like the following.
https://github.com/monero-project/monero/blob/master/cmake/FindReadline.cmake

For `configure.ac`, maybe something like the following.
https://github.com/postgres/postgres/blob/master/configure.ac#L1503-L1521

mcuee, 2022-11-21:
> I will also test it under Linux/Windows to see if there is any side-effect for GNU readline.

No issues under Windows with GNU readline. Tested with PR1171 + PR1185 with `-c urclock`.

mcuee, 2022-11-21:
> * [Readline support for MSVC #1186](https://github.com/avrdudes/avrdude/issues/1186) (MSVC, we may have to wait for @mariusgreuel)

There is also another work-around using the tip from @MCUdude 
```
C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude_pr1171v8mod_80ms_pr1185_msvc64
 -C .\avrdude_pr1171v8.conf -c urclock -P COM4 -b 7800 -p m328p -qqt
avrdude_pr1171v8mod_80ms_pr1185_msvc64 warning: compiled without readline library, cannot use avrdude -t -c urclockavrdude> >>> dump eeprom 0 0x10
0000  55 55 55 55 55 55 55 55  55 55 55 55 55 55 55 55  |UUUUUUUUUUUUUUUU|

avrdude>

```

[46 more replies]

closed 2022-12-14 by stefanrueger