avrdudes/avrdude #1177: To support NetBSD

mcuee, 2022-11-12:
As of now NetBSD/pkgsrc only support avrdude 6.4.
https://pkgsrc.se/cross/avrdude

It will be good to support avrdude git main for NetBSD.

This may not be high priority though.

dl8dtl, 2022-11-12:
I think it's more important to have recent releases there rather than git main.
It's a matter of someone taking over the job at NetBSD (and OpenBSD).

mcuee, 2022-11-12:
Good point. Edit issue text to say `avrdude 7.0 release and git main`.

mcuee, 2022-11-12:
I actally tried git main and it chocked at finding `libusb.h` in `avrftdi_private.h`.

mcuee, 2022-11-12:
> It's a matter of someone taking over the job at NetBSD (and OpenBSD).

Indeed. For example, for libusb project, OpenBSD is only up to 1.0.23 release whereas NetBSD/pkgsrc is at the latest 1.0.26 release.

However, for libusb project, I will at least do some sanity check before the release for NetBSD/OpenBSD. For example, the following is for libusb-1.0.26 (I did the same thing for 1.0.25).
* https://github.com/libusb/libusb/issues/1099#issuecomment-1080067439

dl8dtl, 2022-11-12:
> I actally tried git main and it chocked at finding `libusb.h` in `avrftdi_private.h`.

I guess they differ in that respect from FreeBSD. FreeBSD provides its own libusb implementation that covers the 0.1 and 1.0 APIs (plus one they call 2.0).

What do NetBSD and OpenBSD use? Something that ships as add-on package, or did they reuse the FreeBSD implementation?

mcuee, 2022-11-12:
OpenBSD and NetBSD kind of use upstream libusb with some patches.

OpenBSD's Martin Pieuchot contributed the OpenBSD backend back in 2011.
https://github.com/libusb/libusb/commits?author=mpieuchot

Then I found that it also supports NetBSD and Pete added the commit to add NetBSD support here.
https://github.com/libusb/libusb/commit/968df1244673e1e2b0e1acfffa9375d092d2c26d

We had to split NetBSD backend from OpenBSD backend here since no one from NetBSD jumped in to match further development from Martin.
https://github.com/libusb/libusb/commit/eb4e7bee4488cefcf13929d9c4fe784650fef7c7

OpenBSD libusb is now at 1.0.23 release but with some patches.
http://cvsweb.openbsd.org/cgi-bin/cvsweb/ports/devel/libusb1/

NetBSD/pkgsrc is already at the latest 1.0.26 release. It seems to me they do not add any patches.
http://cvsweb.netbsd.org/bsdweb.cgi/pkgsrc/devel/libusb1/

mcuee, 2022-11-13:
Just tried under NetBSD and I have no problems build avrdude git main under NetBSD once I use the right build command.
```
cmake -D CMAKE_C_FLAGS=-I/usr/pkg/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/pkg/lib
 -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_netbsd

cmake --build build_netbsd
```

mcuee, 2022-11-13:
Under OpenBSD it has some compiling issues.
```
cmake -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/local/lib
 -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_openbsd

cmake --build build_netbsd
```

dl8dtl, 2022-11-13:
The first error means OpenBSD never updated their ELF header files.

The respective VCS log entry in FreeBSD dates back to 2008:

```
commit 926297fefb4c31902f11714276694d217f8e82c0
Author: Kai Wang <kaiw@FreeBSD.org>
Date:   Fri Dec 19 09:32:31 2008 +0000

    Update elf_common.h with new EM_ types found in the latest ELF gABI.
    
    This commit is slightly different from the original patch in the PR:
    1. EM_ALPHA keeps the old value for compatibility reason.
    2. Non-standard SHT_NUM is not added.
    3. Style.
    
    PR:                     kern/118540
    Submitted by:           "Pedro F. Giffuni" <giffunip[at]tutopia.com>
```

Well, we could hack around that with some #ifndef magic, but I'd prefer they adapt to the real world theirselves.

2 and 3 means they appear to no longer support the historical select(2) system interface. We could change the code to use poll(2) instead. If you want, you can open me a ticket for that, it's not going to be a major piece of work (only affects a few lines of code).

mcuee, 2022-11-13:
I have created #1179.

I can see that `term.c` uses `select`.

What about `ser_posix.c`?

mcuee, 2022-11-13:
> The first error means OpenBSD never updated their ELF header files.
> The respective VCS log entry in FreeBSD dates back to 2008:
> Well, we could hack around that with some #ifndef magic, but I'd prefer they adapt to the real world theirselves.

The patch from OpenBSD is here. Looks like we have to use #ifndef magic.
http://cvsweb.openbsd.org/cgi-bin/cvsweb/ports/devel/avrdude/patches/patch-fileio_c?rev=1.2&content-type=text/x-cvsweb-markup

```
Define EM_AVR, since it isn't included in our sys/exec_elf.h.

Index: fileio.c
--- fileio.c.orig
+++ fileio.c
@@ -34,6 +34,9 @@
 #elif defined(HAVE_LIBELF_LIBELF_H)
 #include <libelf/libelf.h>
 #endif
+#ifndef EM_AVR
+#define EM_AVR 83
+#endif
 #define EM_AVR32 0x18ad         /* inofficial */
 #endif

```

mcuee, 2022-11-13:
@dl8dtl 
There is also a patch for `stk500v2.c`, not so sure why they need the patch.
http://cvsweb.openbsd.org/cgi-bin/cvsweb/ports/devel/avrdude/patches/patch-stk500v2_c?rev=1.7&content-type=text/x-cvsweb-markup&hideattic=1

dl8dtl, 2022-11-13:
> Looks like we have to use #ifndef magic.

Too bad they still don't include that after 14 years … but we could do that.

dl8dtl, 2022-11-13:
> There is also a patch for `stk500v2.c`, not so sure why they need the patch.

Might make sense to include that one as well. It ensures the pointer don't dangle with old values.
Not really an issue for the current CLI (because after a teardown, it will exit anway), but good idea for the library context.

dl8dtl, 2022-11-13:
> I have created #1179.
> 
> I can see that `term.c` uses `select`.
> 
> What about `ser_posix.c`?

The same.

Interesting that one compiles under OpenBSD, the other one doesn't. But then, neither includes `<sys/select.h>` which is the documented requirement.

(Even our Windows implementation uses select().)

mcuee, 2022-11-14:
> (Even our Windows implementation uses select().)

Indeed Windows provides the support from `winsock2.h`.
https://learn.microsoft.com/en-us/windows/win32/api/winsock2/

mcuee, 2022-11-15:
> > I have created #1179.
> > I can see that `term.c` uses `select`.
> > What about `ser_posix.c`?
> 
> The same.
> 
> Interesting that one compiles under OpenBSD, the other one doesn't. But then, neither includes `<sys/select.h>` which is the documented requirement.

@dl8dtl 

Great, adding the `<sys/select.h>` sorted out the issue for OpenBSD. 

I will try to create a PR for OpenBSD support.

mcuee, 2022-11-16:
CLI mode seems to work fine under OpenBSD.

[image]

dl8dtl, 2022-11-16:
> I will try to create a PR for OpenBSD support.

No, we are supposed to add this header to our sources that want to use select().

mcuee, 2022-11-16:
> > I will try to create a PR for OpenBSD support.
> 
> No, we are supposed to add this header to our sources that want to use select().

Yes I agree, we need to add `<sys/select.h>` header file for non-Win32 platform, for both `term.c` and `ser_posix.c`. This is not really for OpenBSD per se.

For `fileio.c` we will have to adopt the patch from OpenBSD as they are still using the very old version.

I am pretty bad at git, could you please create the PR? Thanks.

dl8dtl, 2022-11-16:
https://github.com/avrdudes/avrdude/pull/1181

mcuee, 2022-11-16:
> #1181

Thanks. This will fix `fileio.c`. 

What about the `<sys/select.h>` includ file changes to `term.c` and `ser_posix.c`? 

One more change could be `build.sh` to be able to build under NetBSD/OpenBSD out of box.

dl8dtl, 2022-11-17:
I updated the PR to handle `<sys/select.h>`. I don't think it makes sense to have a separate PR just for that.

Regarding build.sh, the question is which -I and -L flags would be required for them.

mcuee, 2022-11-17:
> I updated the PR to handle `<sys/select.h>`. I don't think it makes sense to have a separate PR just for that.

I agree.

> Regarding build.sh, the question is which -I and -L flags would be required for them.

OpenBSD -- same as FreeBSD.
```
build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/local/lib"
```

NetBSD (pkgsrc) -- using the following.
```
build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/usr/pkg/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/pkg/lib"
```

mcuee, 2022-11-18:
> OK, added as well.

Thanks. Now OpenBSD should work out of the box. For NetBSD, it is still not working out of the box as the result avrdude binary will not be able to find the libraries-- maybe some further tuning of `build.sh` is required.

1) Adding `-D CMAKE_INSTALL_PREFIX:PATH=/usr/pkg` helps the installation 
2) But the above is still not correct.

mcuee, 2022-11-18:
@dl8dtl 
Still I think you can merget #1181 as it is so that people who care about NetBSD will be able to jump in and contribute the change. Or you may want to remove the change to `build.sh` for NetBSD for now.

mcuee, 2022-11-21:
PR #1181 has been merged. Openbsd should now work out of the box. 
The remaining issue is with NetBSD. As mentioned above, `build.sh` is not totally correct for NetBSD. Hopefully someone can help to fix the build issue under NetBSD.

mcuee, 2022-12-03:
@dl8dtl 

I have figured out how to build avrdude under NetBSD properly. We just need to add `-R` option for the linker.
```
cmake -D CMAKE_C_FLAGS=-I/usr/pkg/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/pkg/lib -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_freebsd
cmake --build build_freebsd
```

Ref:
https://github.com/ocaml/opam/issues/3482#issuecomment-408066622

[image]

dl8dtl, 2022-12-03:
> We just need to add `-R` option for the linker.

That surprises me quite a bit. It records the LD_RUNPATH attribute.
However, on a system using pkgsrc, I'd expect that `/usr/pkg/lib` is already handled by `ldconfig` properly – at least, that's the way all other operating systems usually handle this.

mcuee, 2022-12-04:
> That surprises me quite a bit. It records the LD_RUNPATH attribute. However, on a system using pkgsrc, I'd expect that `/usr/pkg/lib` is already handled by `ldconfig` properly – at least, that's the [...]

You are right. I checked on a different NetBSD installation under PVE vitual platform and the above `-R` solutiion does not seem to work out there.

mcuee, 2022-12-04:
Failed run log:

```
localhost$ uname -a
NetBSD localhost 9.3 NetBSD 9.3 (GENERIC) #0: Thu Aug  4 15:30:37 UTC 2022  mkrepro@mkrepro.NetBSD.org:/usr/src/sys/arch/amd64/compile/GENERIC amd64
localhost$ cmake -D CMAKE_C_FLAGS=-I/usr/pkg/include -D CMAKE_EXE_LINKER_FLAGS="-R -L/usr/pkg/lib" -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_netbsd
cmake --build build_netbsd-- The C compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/pkg/bin/git (found version "2.37.1")
-- Found FLEX: /usr/bin/flex (found version "2.6.4")
-- Found BISON: /usr/pkg/bin/bison (found version "3.8.2")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DON'T HAVE libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mcuee/avrdude/build_netbsd
localhost$ cmake --build build_netbsd
[  1%] Generating avrdude.conf
[  1%] Built target conf
[  3%] [BISON][Parser] Building parser with bison 3.8.2
[  4%] [FLEX][Parser] Building scanner with flex 2.6.4
[  6%] Building C object src/CMakeFiles/libavrdude.dir/arduino.c.o
[  7%] Building C object src/CMakeFiles/libavrdude.dir/avr.c.o
[  9%] Building C object src/CMakeFiles/libavrdude.dir/avr910.c.o
[ 10%] Building C object src/CMakeFiles/libavrdude.dir/avrcache.c.o
[ 12%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi.c.o
[ 13%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.o
[ 15%] Building C object src/CMakeFiles/libavrdude.dir/avrpart.c.o
[ 16%] Building C object src/CMakeFiles/libavrdude.dir/bitbang.c.o
[ 18%] Building C object src/CMakeFiles/libavrdude.dir/buspirate.c.o
[ 19%] Building C object src/CMakeFiles/libavrdude.dir/butterfly.c.o
[ 21%] Building C object src/CMakeFiles/libavrdude.dir/config.c.o
[ 22%] Building C object src/CMakeFiles/libavrdude.dir/confwin.c.o
[ 24%] Building C object src/CMakeFiles/libavrdude.dir/crc16.c.o
[ 25%] Building C object src/CMakeFiles/libavrdude.dir/dfu.c.o
[ 27%] Building C object src/CMakeFiles/libavrdude.dir/fileio.c.o
[ 28%] Building C object src/CMakeFiles/libavrdude.dir/flip1.c.o
[ 30%] Building C object src/CMakeFiles/libavrdude.dir/flip2.c.o
[ 31%] Building C object src/CMakeFiles/libavrdude.dir/ft245r.c.o
[ 33%] Building C object src/CMakeFiles/libavrdude.dir/jtagmkI.c.o
[... 34 lines cut ...]
[ 86%] Building C object src/CMakeFiles/libavrdude.dir/xbee.c.o
[ 87%] Building C object src/CMakeFiles/libavrdude.dir/__/lexer.c.o
[ 89%] Building C object src/CMakeFiles/libavrdude.dir/__/config_gram.c.o
[ 90%] Linking C static library libavrdude.a
[ 90%] Built target libavrdude
[ 92%] Building C object src/CMakeFiles/avrdude.dir/main.c.o
[ 93%] Building C object src/CMakeFiles/avrdude.dir/term.c.o
[ 95%] Building C object src/CMakeFiles/avrdude.dir/avrintel.c.o
[ 96%] Building C object src/CMakeFiles/avrdude.dir/developer_opts.c.o
[ 98%] Building C object src/CMakeFiles/avrdude.dir/whereami.c.o
[100%] Linking C executable avrdude
[100%] Built target avrdude
localhost$ ldd ./build_netbsd/src/avrdude
./build_netbsd/src/avrdude:
        -lm.0 => /usr/lib/libm.so.0
        -lc.12 => /usr/lib/libc.so.12
        -lelf.0 => not found
        -lusb-0.1.4 => not found
        -lusb-1.0.0 => not found
        -lhid.0 => not found
        -lftdi.1 => not found
        -lftdi1.2 => not found
        -lreadline.8 => not found
        -lgcc_s.1 => /usr/lib/libgcc_s.so.1
```

mcuee, 2022-12-04:
@dl8dtl 

I think I find the reason, `LD_LIBRARY_PATH` is not set and that is the reason.

```
localhost$ pwd
/home/mcuee/avrdude/build_netbsd/src
localhost$ LD_LIBRARY_PATH=/usr/pkg/lib ./avrdude
Usage: avrdude [options]
Options:
  -p <partno>                Specify AVR device
  -b <baudrate>              Override RS-232 baud rate
  -B <bitclock>              Specify bit clock period (us)
  -C <config-file>           Specify location of configuration file
  -c <programmer>            Specify programmer type
  -A                         Disable trailing-0xff removal from file and AVR read
  -D                         Disable auto erase for flash memory; implies -A
  -i <delay>                 ISP Clock Delay [in microseconds]
  -P <port>                  Specify connection port
  -F                         Override invalid signature or initialisation check
  -e                         Perform a chip erase
  -O                         Perform RC oscillator calibration (see AVR053)
  -U <memtype>:r|w|v:<filename>[:format]
                             Memory operation specification
                             Multiple -U options are allowed, each request
                             is performed in the order specified
  -n                         Do not write anything to the device
  -V                         Do not verify
  -t                         Enter terminal mode
  -E <exitspec>[,<exitspec>] List programmer exit specifications
  -x <extended_param>        Pass <extended_param> to programmer
  -v                         Verbose output; -v -v for more
  -q                         Quell progress output; -q -q for less
  -l logfile                 Use logfile rather than stderr for diagnostics
  -?                         Display this usage

avrdude version 7.0-20221129 (ea2d7ee), URL: <https://github.com/avrdudes/avrdude>

```

mcuee, 2022-12-04:
Full build log using default `build.sh`.

```
localhost$ LD_LIBRARY_PATH=/usr/pkg/lib

localhost$ export  LD_LIBRARY_PATH

localhost$ echo $LD_LIBRARY_PATH
/usr/pkg/lib

localhost$ git clone https://github.com/avrdudes/avrdude
Cloning into 'avrdude'...
remote: Enumerating objects: 14691, done.
remote: Counting objects: 100% (597/597), done.
remote: Compressing objects: 100% (249/249), done.
remote: Total 14691 (delta 407), reused 514 (delta 344), pack-reused 14094
Receiving objects: 100% (14691/14691), 7.93 MiB | 4.10 MiB/s, done.
Resolving deltas: 100% (10286/10286), done.

localhost$ cd avrdude/

localhost$ ./build.sh
-- The C compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/pkg/bin/git (found version "2.37.1")
-- Found FLEX: /usr/bin/flex (found version "2.6.4")
-- Found BISON: /usr/pkg/bin/bison (found version "3.8.2")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DON'T HAVE libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mcuee/avrdude/build_netbsd
[  1%] Generating avrdude.conf
[  1%] Built target conf
[  3%] [BISON][Parser] Building parser with bison 3.8.2
[  4%] [FLEX][Parser] Building scanner with flex 2.6.4
[  6%] Building C object src/CMakeFiles/libavrdude.dir/arduino.c.o
[  7%] Building C object src/CMakeFiles/libavrdude.dir/avr.c.o
[  9%] Building C object src/CMakeFiles/libavrdude.dir/avr910.c.o
[ 10%] Building C object src/CMakeFiles/libavrdude.dir/avrcache.c.o
[... 61 lines cut ...]

Build succeeded.

Run

sudo cmake --build build_netbsd --target install

to install.

localhost$ ldd ./build_netbsd/src/avrdude
./build_netbsd/src/avrdude:
        -lm.0 => /usr/lib/libm.so.0
        -lc.12 => /usr/lib/libc.so.12
        -lelf.0 => /usr/pkg/lib/libelf.so.0
        -lusb-0.1.4 => /usr/pkg/lib/libusb-0.1.so.4
        -lusb-1.0.0 => /usr/pkg/lib/libusb-1.0.so.0
        -lpthread.1 => /usr/lib/libpthread.so.1
        -lhid.0 => /usr/pkg/lib/libhid.so.0
        -lftdi.1 => /usr/pkg/lib/libftdi.so.1
        -lftdi1.2 => /usr/pkg/lib/libftdi1.so.2
        -lreadline.8 => /usr/pkg/lib/libreadline.so.8
        -lterminfo.1 => /usr/lib/libterminfo.so.1
        -lgcc_s.1 => /usr/lib/libgcc_s.so.1

```

mcuee, 2022-12-04:
I can not attach USB devices to NetBSD 9.2 VM under Windows Virtual box. But I can attach COM port.

And avrdude seems to work fine.

[image]

mcuee, 2022-12-04:
I think I can close this issue now.

dl8dtl, 2022-12-04:
> `LD_LIBRARY_PATH` is not set

But again, on a system that uses pkgsrc, I'd expect that `/usr/pkg/lib` is pre-configured system-wide using `ldconfig`. That has the additional advantage that e.g. setuid binaries will work using libraries from that location (they do not honor `LD_LIBRARY_PATH` or any other user-provided location for security reasons).

I tried looking up their pkgsrc guide, but could not find any statement about it.

Well, it seems the NetBSD folks don't really like `ldconfig` and `ld.so.conf`, and prefer recording the library path in the binary:

https://www.netbsd.org/docs/elf.html

Consequently, your solution with adding `-R` to the linker options is the right thing to do.

mcuee, 2022-12-04:
@dl8dtl 

Great. Now we have a simple solution. I tested the following simple patch and it works fine.

```
localhost$ git diff
diff --git a/build.sh b/build.sh
index d5685a0..bf9a9bc 100755
--- a/build.sh
+++ b/build.sh
@@ -61,7 +61,7 @@ case "${ostype}" in
        ;;

     netbsd)
-       build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/usr/pkg/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/pkg/lib"
+       build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/usr/pkg/include -D CMAKE_EXE_LINKER_FLAGS=-R/usr/pkg/lib"
        ;;

     *bsd)
localhost$ ldd ./build_netbsd/src/avrdude
./build_netbsd/src/avrdude:
        -lm.0 => /usr/lib/libm.so.0
        -lc.12 => /usr/lib/libc.so.12
        -lelf.0 => /usr/pkg/lib/libelf.so.0
        -lusb-0.1.4 => /usr/pkg/lib/libusb-0.1.so.4
        -lusb-1.0.0 => /usr/pkg/lib/libusb-1.0.so.0
        -lpthread.1 => /usr/lib/libpthread.so.1
        -lhid.0 => /usr/pkg/lib/libhid.so.0
        -lftdi.1 => /usr/pkg/lib/libftdi.so.1
        -lftdi1.2 => /usr/pkg/lib/libftdi1.so.2
        -lreadline.8 => /usr/pkg/lib/libreadline.so.8
        -lterminfo.1 => /usr/lib/libterminfo.so.1
        -lgcc_s.1 => /usr/lib/libgcc_s.so.1
```

mcuee, 2022-12-04:
Full runlog.

```
localhost$ git clone https://github.com/avrdudes/avrdude avrdude_git
Cloning into 'avrdude_git'...
remote: Enumerating objects: 14691, done.
remote: Counting objects: 100% (597/597), done.
remote: Compressing objects: 100% (253/253), done.
remote: Total 14691 (delta 406), reused 511 (delta 340), pack-reused 14094
Receiving objects: 100% (14691/14691), 7.94 MiB | 5.04 MiB/s, done.
Resolving deltas: 100% (10286/10286), done.

localhost$ cd avrdude_git/

localhost$ cp build.sh build_org.sh

localhost$ nano build.sh

localhost$ git diff
diff --git a/build.sh b/build.sh
index d5685a0..69ad0ae 100755
--- a/build.sh
+++ b/build.sh
@@ -61,7 +61,7 @@ case "${ostype}" in
        ;;

     netbsd)
-       build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/usr/pkg/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/pkg/lib"
+       build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/usr/pkg/include -D CMAKE_EXE_LINKER_FLAGS=-R/usr/pkg/lib -D CMAKE_INSTALL_PREFIX:PATH=/usr/pkg"
        ;;

     *bsd)

localhost$ ./build.sh
-- The C compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/pkg/bin/git (found version "2.37.1")
-- Found FLEX: /usr/bin/flex (found version "2.6.4")
-- Found BISON: /usr/pkg/bin/bison (found version "3.8.2")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DON'T HAVE libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
[... 120 lines cut ...]
  -c <programmer>            Specify programmer type
  -A                         Disable trailing-0xff removal from file and AVR read
  -D                         Disable auto erase for flash memory; implies -A
  -i <delay>                 ISP Clock Delay [in microseconds]
  -P <port>                  Specify connection port
  -F                         Override invalid signature or initialisation check
  -e                         Perform a chip erase
  -O                         Perform RC oscillator calibration (see AVR053)
  -U <memtype>:r|w|v:<filename>[:format]
                             Memory operation specification
                             Multiple -U options are allowed, each request
                             is performed in the order specified
  -n                         Do not write anything to the device
  -V                         Do not verify
  -t                         Enter terminal mode
  -E <exitspec>[,<exitspec>] List programmer exit specifications
  -x <extended_param>        Pass <extended_param> to programmer
  -v                         Verbose output; -v -v for more
  -q                         Quell progress output; -q -q for less
  -l logfile                 Use logfile rather than stderr for diagnostics
  -?                         Display this usage

avrdude version 7.0-20221129 (ea2d7ee), URL: <https://github.com/avrdudes/avrdude>

```

mcuee, 2022-12-04:
@dl8dtl 

Please help to update `build.sh` accordingly for NetBSD when you have the time to work on it. Thanks. 

```
localhost$ git diff
diff --git a/build.sh b/build.sh
index d5685a0..69ad0ae 100755
--- a/build.sh
+++ b/build.sh
@@ -61,7 +61,7 @@ case "${ostype}" in
        ;;

     netbsd)
-       build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/usr/pkg/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/pkg/lib"
+       build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/usr/pkg/include -D CMAKE_EXE_LINKER_FLAGS=-R/usr/pkg/lib -D CMAKE_INSTALL_PREFIX:PATH=/usr/pkg"
        ;;

     *bsd)

```

dl8dtl, 2022-12-04:
Reopening to handle `build.sh`

mcuee, 2022-12-04:
@dl8dtl 

As of now. `build.sh` is not correct for HomeBrew either when it comes to installation path. I am not exactly sure how to handle that. It is supposed to install the the Cellar (with the version) and then the user needs to use `brew link`.

I think we can ignore that for now.

The easy way is to use the following commands if one wants to follow git main.
```
brew install avrdude --head (one time)
brew reinstall avrude (whenever the users want to upgrade to latest git main)
```

If you want to look at the details, here is the Homebrew avrdude formula.
https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/avrdude.rb

mcuee, 2022-12-09:
I have installed a new NetBSD 9.3 Virtualbox VM under Windows and now USB works fine.

[image]

mcuee, 2022-12-23:
@dl8dtl 

The basic fix to NetBSD build.sh is good to have before 7.1 release. Thanks.

mcuee, 2022-12-27:
I have merged #1241 and this issue can be closed.

closed 2022-12-27 by mcuee