avrdudes/avrdude #1180: Terminal mode: to support command similar to U in CLI mode

mcuee, 2022-11-14:
Right now terminal mode is very powerful. Still I miss one feature in the terminal mode, that is, to use things similar to `-U` command in CLI mode.

Eg:

`write flash 0 entest.hex` --> 0 is probably not necessary for hex file, but may be easier to require for easier handling
`write eeprom start_addr entest.bin`

MCUdude, 2022-11-14:
This would be a really neat feature for sure!

> `write memory addr file:entest.bin` --> start address is required for bin file.

Alternatively, no start address provided could mean the start address is 0.

We should think this one through. 

* Should binary files be treated like other arguments, like this for instance? 
  - `write eeprom 0 0xcafe 'a' 'v' 'r' /path/to/testfile.bin 0x1234 "Hello World\n"`
* Should ellipsis be supported?
  - `write eeprom 0 0x1000 /path/to/testfile.bin ...`
* Should it be possible to override the start address in a hex file (convenient if experimenting with avr8x's and bootloaders)
  - `write flash 0x200 /path/to/userprogram.hex`

mcuee, 2022-12-04:
Comment from @stefanrueger 
* https://github.com/avrdudes/avrdude/issues/1201#issuecomment-1336283936
 
> Yes, now that one can write to flash from the terminal #1180 is a natural extension. Absolutely makes sense. And it may be less work than it looks. After all, there are a lot of useful functions in fileio.c and update.c. Would be a cool thing for v8.0. As to `ls`, these days it's easy to switch windows on the desktop and do the `ls` elsewhere. The 90s way would have been to implement a command `!` in term.c that calls `system()`:
> 
> ```
> avrdude> !ls *hex
> blink-mega2560+lext-test.hex  rsketch-ending-in-ff.hex	sketch-ending-in-ff.hex  zeros16kib.hex 
> ```
> 
> That's probably only a few lines code and gives you access to all that your shell can muster (but depending how you implement it might time out the bootloader)
> 
> Maybe it is a good exercise before attempting something like `write flash file.hex` or `load file.hex` (where load is an alias to write flash) in terminal is to look closely at `-U file.elf` that we discussed in another PR and learn from our experience how to implement _that_.
>

PR #1358 (2023-05-20): Extend <data> notion to files in terminal write
Fixes #1180 

```
$ echo "The quick brown fox jumps over the lazy dog" >fox

$ avrdude -qqt
avrdude> w
Usage: write <memory> <addr> <data>[,] {<data>[,]}
       write <memory> <addr> <len> <data>[,] {<data>[,]} ...
       write <memory> <datum> # only for 1-byte memories
       write <memory> <file>  # only for memories with more than 1 byte

Ellipsis ... writes <len> bytes padded by repeating the last <data> item.

Both the <addr> and <len> can be negative numbers; a negative <addr> starts
an interval from that many bytes below the memory size;  [... cut]
closed 2023-05-20 by stefanrueger