avrdudes/avrdude #1186: Readline support for MSVC

mcuee, 2022-11-21:
As of now MSVC build does not support libreadline and then `-c urclock` may time out in terminal mode.

@mariusgreuel may have the solution from the discussion here.
* https://github.com/avrdudes/avrdude/issues/1028

mcuee, 2022-11-21:
The workaround for now is to use MinGW build below.
* https://github.com/avrdudes/avrdude/issues/1162

Once #1171 is merged, I will publish avrdude snapshot binaries for Linux/Windows/macOS.
* #1171

mcuee, 2022-11-25:
BTW, it is not too bad as there is a work around for the terminal mode using pipes.

Examples using the latest github action MSVC binary.
https://github.com/avrdudes/avrdude/actions/runs/3539649818

```
C:\work\avr\avrdude_test\avrdude_bin\github_action\avrdude-msvc-x64> 
echo "dump flash 0 0x10" | .\avrdude -c urclock -P COM4 -p m328p  -t -xbootsize=2048
avrdude warning: compiled without readline library, cannot use avrdude -t -c urclock
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude> >>> dump flash 0 0x10

Reading | ################################################## | 100% 0.02 s

0000  0c 94 5c 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .\. .n. .n. .n.|

avrdude>
avrdude done.  Thank you.

```

mcuee, 2022-11-25:
@stefanrueger 

Would you please change the warning message to a bit like this? You can change the wording as you see fit.
```
warning: compiled without readline library, you may not be able to use "avrdude -t -c urclock" with full functionality.
```

mariusgreuel, 2022-11-25:
@mcuee , @stefanrueger : I have not been able to keep up the pace you guys are working, but I am planning on fixing some Windows related issues around the upcoming holidays. This includes patching up the old readline library for Windows. I am assigning this issue to me.

stefanrueger, 2022-11-25:
> I am assigning this issue to me.

@mariusgreuel Thank you. BTW, you can try `avrdude -t -curclock` with a STK500v1 bootloader, eg. This should keep the bootloader alive. Current git main requires `-xdelay=350` for old optiboots.

Without libreadline one needs to pipe into avrdude:
```
echo "d fl 0 32; quit" | tr \; \\n | avrdude -t -curclock
```

mcuee, 2022-11-25:
> @mcuee , @stefanrueger : I have not been able to keep up the pace you guys are working, but I am planning on fixing some Windows related issues around the upcoming holidays. This includes patching up the old readline library for Windows. I am assigning this issue to me.

Thanks, Readline for MSVC build will be wonderful. The other thing you may help is how to better detect Readline under Windows. Now it wrongly detect Strberry Perl's libreadline.a as having Readline for the MSVC build. I have to explicitly disable it in githhub action MSVC build with PR #1196.
* https://github.com/avrdudes/avrdude/pull/1194#issuecomment-1325824150

mariusgreuel, 2022-11-27:
> Now it wrongly detect Strberry Perl's libreadline.a

Yes, when you install Strawberry Perl, a lot of things break. They put their entire MinGW toolchain in the systems PATH variable, having no clue about the breakage this causes: https://github.com/StrawberryPerl/Perl-Dist-Strawberry/issues/11

It surprises me to see this on a GitHub runner enabled by default. I dropped GitHub a message:
https://github.com/actions/runner-images/issues/5459

mcuee, 2022-12-06:
Build log using Readline-win32.

```
1> CMake generation started for configuration: 'x64-Release'.
1> Command line: "C:\WINDOWS\system32\cmd.exe" /c "%SYSTEMROOT%\System32\chcp.com 65001 >NUL && "C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2019\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\CMake\bin\cmake.exe"  -G "Ninja"  -DCMAKE_BUILD_TYPE:STRING="RelWithDebInfo" -DCMAKE_INSTALL_PREFIX:PATH="C:\work\avr\avrdude_test\avrdude_jw\out\install\x64-Release" -DDEBUG_CMAKE:BOOL="True" -DUSE_EXTERNAL:BOOL="True" -DHAVE_LIBREADLINE:FILEPATH="C:/work/avr/avrdude_test/avrdude_bin/readline_msvc/src/readline/5.0/readline-5.0/vs/x64/Release/readline.lib" -DCMAKE_C_COMPILER:FILEPATH="C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe"  -DCMAKE_MAKE_PROGRAM="C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2019\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\Ninja\ninja.exe" "C:\work\avr\avrdude_test\avrdude_jw" 2>&1"
1> Working directory: C:\work\avr\avrdude_test\avrdude_jw\out\build\x64-Release
1> [CMake] -- Fetching external libraries, please wait...
1> [CMake] -- Using external library 'libelf'
1> [CMake] -- Using external library 'libusb'
1> [CMake] -- Using external library 'libhidapi'
1> [CMake] -- Using external library 'libftdi'
1> [CMake] -- Configuration summary:
1> [CMake] -- ----------------------
1> [CMake] -- CMAKE_HOST_SYSTEM: Windows-10.0.19044
1> [CMake] -- CMAKE_SYSTEM: Windows-10.0.19044
1> [CMake] -- CMAKE_FIND_ROOT_PATH: 
1> [CMake] -- CMAKE_C_COMPILER: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
1> [CMake] -- CONFIG_DIR: C:/work/avr/avrdude_test/avrdude_jw/out/install/x64-Release/etc
1> [CMake] -- AVRDUDE_FULL_VERSION: 7.0-20221201 (30bffc3)
1> [CMake] -- USE_EXTERNAL: True
1> [CMake] -- USE_LIBUSBWIN32: OFF
1> [CMake] -- HAVE_LIBELF: 1
1> [CMake] -- HAVE_LIBUSB: 1
1> [CMake] -- HAVE_LIBUSB_1_0: HAVE_LIBUSB_1_0-NOTFOUND
1> [CMake] -- HAVE_LIBUSB_WIN32: HAVE_LIBUSB_WIN32-NOTFOUND
1> [CMake] -- HAVE_LIBHID: C:/Program Files (x86)/Windows Kits/10/Lib/10.0.19041.0/um/x64/hid.lib
1> [CMake] -- HAVE_LIBHIDAPI: 1
1> [CMake] -- HAVE_LIBFTDI: 1
1> [CMake] -- HAVE_LIBFTDI1: HAVE_LIBFTDI1-NOTFOUND
1> [CMake] -- HAVE_LIBREADLINE: C:/work/avr/avrdude_test/avrdude_bin/readline_msvc/src/readline/5.0/readline-5.0/vs/x64/Release/readline.lib
1> [CMake] -- HAVE_LIBELF_H: 1
1> [CMake] -- HAVE_LIBELF_LIBELF_H: 
1> [CMake] -- HAVE_USB_H: 
1> [CMake] -- HAVE_LUSB0_USB_H: 1
1> [CMake] -- HAVE_LIBUSB_H: 
1> [CMake] -- HAVE_LIBUSB_1_0_LIBUSB_H: 
1> [CMake] -- HAVE_HIDAPI_HIDAPI_H: 1
1> [CMake] -- LIBUSB_COMPAT_DIR: 
1> [CMake] -- ----------------------
1> [CMake] -- DO HAVE    libelf
1> [CMake] -- DO HAVE    libusb
1> [CMake] -- DON'T HAVE libusb_1_0
1> [CMake] -- DO HAVE    libhidapi
1> [CMake] -- DO HAVE    libftdi
1> [CMake] -- DON'T HAVE libftdi1
1> [CMake] -- DO HAVE    libreadline
1> [CMake] -- DISABLED   doc
1> [CMake] -- DISABLED   parport
1> [CMake] -- DISABLED   linuxgpio
1> [CMake] -- DISABLED   linuxspi
1> [CMake] -- ----------------------
1> [CMake] -- Configuring done
1> [CMake] -- Generating done
1> [CMake] -- Build files have been written to: C:/work/avr/avrdude_test/avrdude_jw/out/build/x64-Release
1> Extracted CMake variables.
1> Extracted source files and headers.
1> Extracted code model.
1> Extracted toolchain configurations.
1> Extracted includes paths.
1> CMake generation finished.

```

mcuee, 2022-12-06:
One more try.

```
1> CMake generation started for configuration: 'x64-Release'.
1> Command line: "C:\WINDOWS\system32\cmd.exe" /c "%SYSTEMROOT%\System32\chcp.com 65001 >NUL && "C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2019\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\CMake\bin\cmake.exe"  -G "Ninja"  -DCMAKE_BUILD_TYPE:STRING="RelWithDebInfo" -DCMAKE_INSTALL_PREFIX:PATH="C:\work\avr\avrdude_test\avrdude_main\out\install\x64-Release" -DUSE_EXTERNAL:BOOL="True" -DHAVE_LIBREADLINE:FILEPATH="C:/work/avr/avrdude_test/avrdude_main/src/msvc/readline_bin/readline.lib" -DUSE_STATIC_LIBS:BOOL="true" -DCMAKE_C_COMPILER:FILEPATH="C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe"  -DCMAKE_MAKE_PROGRAM="C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2019\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\Ninja\ninja.exe" "C:\work\avr\avrdude_test\avrdude_main" 2>&1"
1> Working directory: C:\work\avr\avrdude_test\avrdude_main\out\build\x64-Release
1> [CMake] -- The C compiler identification is MSVC 19.29.30146.0
1> [CMake] -- Detecting C compiler ABI info
1> [CMake] -- Detecting C compiler ABI info - done
1> [CMake] -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
1> [CMake] -- Detecting C compile features
1> [CMake] -- Detecting C compile features - done
1> [CMake] -- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.35.3.windows.1") 
1> [CMake] -- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.4") 
1> [CMake] -- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "3.7.4")
1> [CMake] -- Fetching external libraries, please wait...
1> [CMake] -- The CXX compiler identification is MSVC 19.29.30146.0
1> [CMake] -- Detecting CXX compiler ABI info
1> [CMake] -- Detecting CXX compiler ABI info - done
1> [CMake] -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
1> [CMake] -- Detecting CXX compile features
1> [CMake] -- Detecting CXX compile features - done
1> [CMake] -- Using external library 'libelf'
1> [CMake] -- Using external library 'libusb'
1> [CMake] -- Using external library 'libhidapi'
1> [CMake] -- Using external library 'libftdi'
1> [CMake] -- Configuration summary:
1> [CMake] -- ----------------------
1> [CMake] -- DO HAVE    libelf
1> [CMake] -- DO HAVE    libusb
1> [CMake] -- DON'T HAVE libusb_1_0
1> [CMake] -- DO HAVE    libhidapi
1> [CMake] -- DO HAVE    libftdi
1> [CMake] -- DON'T HAVE libftdi1
1> [CMake] -- DO HAVE    libreadline
1> [CMake] -- DISABLED   doc
1> [CMake] -- DISABLED   parport
1> [CMake] -- DISABLED   linuxgpio
1> [CMake] -- DISABLED   linuxspi
1> [CMake] -- ----------------------
1> [CMake] -- Configuring done
1> [CMake] -- Generating done
1> [CMake] -- Build files have been written to: C:/work/avr/avrdude_test/avrdude_main/out/build/x64-Release
1> Extracted CMake variables.
1> Extracted source files and headers.
1> Extracted code model.
1> Extracted toolchain configurations.
1> Extracted includes paths.
1> CMake generation finished.

```

mcuee, 2022-12-06:
Binary for reference (also included in the post above). Dynamic link to `readline.dll` as of now.
[avrdude_msvc_readline_x64.zip](https://github.com/avrdudes/avrdude/files/10163423/avrdude_msvc_readline_x64.zip)

mcuee, 2022-12-06:
Yet the other is to get the Readline binary from vcpkg, and copy the necessary into a directory named `readline`, then move the file under `src/msvc directory`.

```
PS C:\work\avr\avrdude_test\avrdude_pr1207> & 'C:\Program Files\CMake\bin\cmake.exe' -D HAVE_LIBREADLINE:FILEPATH=C:\work\avr\avrdude_test\avrdude_pr1207\src\msvc\readline\readline.lib  -D USE_EXTERNAL=TRUE -B build
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22621.
-- The C compiler identification is MSVC 19.31.31105.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.33.0.windows.2")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "3.7.4")
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.31.31105.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using external library 'libelf'
-- Using external library 'libusb'
-- Using external library 'libhidapi'
-- Using external library 'libftdi'
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DON'T HAVE libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi
-- DON'T HAVE libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude_pr1207/build

```

VS2022 build log.
```
Rebuild started...
1>------ Rebuild All started: Project: ZERO_CHECK, Configuration: RelWithDebInfo x64 ------
1>Checking Build System
2>------ Rebuild All started: Project: libusb, Configuration: RelWithDebInfo x64 ------
3>------ Rebuild All started: Project: libhidapi, Configuration: RelWithDebInfo x64 ------
4>------ Rebuild All started: Project: libftdi, Configuration: RelWithDebInfo x64 ------
5>------ Rebuild All started: Project: libelf, Configuration: RelWithDebInfo x64 ------
6>------ Rebuild All started: Project: conf, Configuration: RelWithDebInfo x64 ------
2>Building Custom Rule C:/work/avr/avrdude_test/avrdude_pr1207/build/_deps/libusb-src/CMakeLists.txt
3>Building Custom Rule C:/work/avr/avrdude_test/avrdude_pr1207/build/_deps/libhidapi-src/CMakeLists.txt
2>descriptors.c
4>Building Custom Rule C:/work/avr/avrdude_test/avrdude_pr1207/build/_deps/libftdi-src/CMakeLists.txt
3>hid.c
5>Building Custom Rule C:/work/avr/avrdude_test/avrdude_pr1207/build/_deps/libelf-src/CMakeLists.txt
4>ftdi.cpp
6>Generating avrdude.conf
2>error.c
5>32.fsize.c
6>Building Custom Rule C:/work/avr/avrdude_test/avrdude_pr1207/src/CMakeLists.txt
5>32.getehdr.c
5>32.getphdr.c
5>32.getshdr.c
[... 125 lines cut ...]
7>xbee.c
7>lexer.c
7>config_gram.c
7>getopt.c
7>gettimeofday.c
7>Generating Code...
7>usleep.cpp
7>libavrdude.vcxproj -> C:\work\avr\avrdude_test\avrdude_pr1207\build\src\RelWithDebInfo\libavrdude.lib
8>------ Rebuild All started: Project: avrdude, Configuration: RelWithDebInfo x64 ------
8>Building Custom Rule C:/work/avr/avrdude_test/avrdude_pr1207/src/CMakeLists.txt
8>main.c
8>term.c
8>avrintel.c
8>developer_opts.c
8>whereami.c
8>Generating Code...
8>   Creating library C:/work/avr/avrdude_test/avrdude_pr1207/build/src/RelWithDebInfo/avrdude.lib and object C:/work/avr/avrdude_test/avrdude_pr1207/build/src/RelWithDebInfo/avrdude.exp
8>avrdude.vcxproj -> C:\work\avr\avrdude_test\avrdude_pr1207\build\src\RelWithDebInfo\avrdude.exe
9>------ Rebuild All started: Project: ALL_BUILD, Configuration: RelWithDebInfo x64 ------
9>Building Custom Rule C:/work/avr/avrdude_test/avrdude_pr1207/CMakeLists.txt
10>------ Skipped Rebuild All: Project: INSTALL, Configuration: RelWithDebInfo x64 ------
10>Project not selected to build for this solution configuration 
========== Rebuild All: 9 succeeded, 0 failed, 1 skipped ==========

```

mcuee, 2022-12-06:
Result binary seems to work fine.

```
PS C:\work\avr\avrdude_test\avrdude_pr1207\build\src\RelWithDebInfo> .\avrdude -c urclock -P COM1 -p m328p -xbootsize=512 -qqt
avrdude> help
>>> help
Valid commands:
  dump    : dump <memory> [<addr> <len> | <addr> ... | <addr> | ...]
  read    : alias for dump
  write   : write <memory> <addr> <data>[,] {<data>[,]}
          : write <memory> <addr> <len> <data>[,] {<data>[,]} ...
  flush   : synchronise flash & EEPROM writes with the device
  abort   : abort flash & EEPROM writes (reset the r/w cache)
  erase   : perform a chip erase
  sig     : display device signature bytes
  part    : display the current part information
  send    : send a raw command: send <b1> <b2> <b3> <b4>
  verbose : change verbosity
  quell   : set quell level for progress bars
  help    : show help message
  ?       : same as help
  quit    : quit after writing out cache for flash & EEPROM

Note that not all programmer derivatives support all commands. Flash and
EEPROM type memories are normally read and written using a cache via paged
read and write access; the cache is synchronised on quit or flush commands.
The part command displays valid memory types for use with dump and write.

avrdude> dump flash 0 0x20
>>> dump flash 0 0x20
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> quit
>>> quit
avrdude>
PS C:\work\avr\avrdude_test\avrdude_pr1207\build\src\RelWithDebInfo> echo "dump flash 0 0x20" | .\avrdude -c urclock -P COM1 -p m328p -xbootsize=512 -qqt
avrdude> >>> dump flash 0 0x20
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>

```

[readline_msvc_x64.zip](https://github.com/avrdudes/avrdude/files/10166217/readline_msvc_x64.zip)

mcuee, 2022-12-06:
Just to illustrate the issue again with readline-win32. PR #1207 is good for this case.

```
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump flash 0 0x20" | .\avrdude_git_msvc_readline
 -c urclock -P COM1 -p m328p -xbootsize=512 -qqt
avrdude> dump flash 0 0x20
>>> dump flash 0 0x20
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude>(need to hit CTRL-C to get out)

PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump flash 0 0x20" | .\avrdude_pr1207_msvc_readline
 -c urclock -P COM1 -p m328p -xbootsize=512 -qqt
avrdude> >>> dump flash 0 0x20
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>

```

stefanrueger, 2022-12-06:
> avrdude>(need to hit CTRL-C to get out)

I think @mariusgreuel is aware of this. Somehow reading from a pipe/file does not generate EOF he told us.

mcuee, 2022-12-07:
> > avrdude>(need to hit CTRL-C to get out)
> 
> I think @mariusgreuel is aware of this. Somehow reading from a pipe/file does not generate EOF he told us.

Good to know that. No idea why PR #1207 helps in this case.

mcuee, 2023-01-02:
@mariusgreuel mentioned that he would like to include this in the 7.1 release.

https://github.com/avrdudes/avrdude/pull/1242#issuecomment-1365980848

> I was planning on going through the issues assigned to me. At least I would like to include #1186 in the next release.

mcuee, 2023-01-03:
@mariusgreuel, @stefanrueger @dl8dtl and @MCUdude 

I have added this to 7.1 milestone. I think PR #1259 is almost there. Hopefully the fix for the remaining issues will not take too much time.

stefanrueger, 2023-01-05:
Closing as PR #1264 addresses this.

closed 2023-01-05 by stefanrueger