avrdudes/avrdude #1192: `&& echo OK` does not work after `-c urclock`

mcuee, 2022-11-23:
Reference:
* https://github.com/avrdudes/avrdude/pull/1171#issuecomment-1325151678

It is strange that `&& echo OK` does not work after `-c urclock`. No pronlems with `-c arduino`.

It is the same under Linux.
```

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c arduino -P /dev/ttyUSB0 -p m328p -U ./Blink.ino.standard.hex
 -qq && echo OK
OK

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c urclock -P /dev/ttyUSB0 -p m328p -U ./Blink.ino.standard.hex
 -xbootsize=512 -qq && echo OK

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ 
```

stefanrueger, 2022-11-23:
`-c arduino` pretends chip erase works (but does nothing), so no error is being shown.

`-c urclock` correctly assumes the non-native bootloader won't have CE, so delayes CE to be dealt with by filling the remaining space with 0xff on the next -U flash upload, but marks the shell `exitrc` value provisionally as 1 (error, no echo `OK`) as at this point it does not know whether there will be a correct `-U` flash upload later

The bug is that this then never reset to 0 on successful `-U` to flash.

Will repair

mcuee, 2022-11-24:
@stefanrueger 
Thanks for the detailed explanation why the issue happened. I can confirm PR #1194 fixed the issue.

PR #1194 (2022-11-25): Fix shell exit value when chip erase is delayed to next flash write
Fixes #1192
closed 2022-11-25 by stefanrueger