avrdudes/avrdude #1193: Serail drain timeout and `-c urclock`

mcuee, 2022-11-23:
From here:
* https://github.com/avrdudes/avrdude/pull/1171#issuecomment-1324966504

Testing results with the default Arduino `optinoot_atmega328p.hex` on a typical Arduino Clone (using ATmega16U2 as USB to Serial chip) and I can see the current default timeout value of 80ms is not good for Windows.

I need the following patch to get it working. This is the same for the Nano clone (ATmega328P, 16MHz).
```
PS C:\work\avr\avrdude_test\avrdude_main> git diff
diff --git a/src/urclock.c b/src/urclock.c
index 7bfb2d7..f58a2d7 100644
--- a/src/urclock.c
+++ b/src/urclock.c
@@ -2154,7 +2154,11 @@ static int urclock_open(PROGRAMMER *pgm, const char *port) {
     usleep((80+ur.delay)*1000); // Wait until board comes out of reset

   // Drain any extraneous input
+#ifndef WIN32
   serial_drain_timeout = 80;    // ms
+#else
+  serial_drain_timeout = 150;    // ms
+#endif
   serial_drain(&pgm->fd, 0);

   if(urclock_getsync(pgm) < 0)
```

Testing results:
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p m328p -U .\optiboot_atmega328.hex -qq
 && echo OK
OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM3 -p m328p -qq -t
avrdude warning: attempt 1 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 2 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 3 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 4 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 5 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 6 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 7 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 8 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 9 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 10 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 11 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 12 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 13 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 14 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 15 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 16 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 17 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 18 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 19 of 20: not in sync
avrdude warning: programmer is not responding; try, eg, -xdelay=200
avrdude warning: attempt 20 of 20: not in sync
avrdude error: unable to open programmer urclock on port COM3

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM3 -p m328p -qq -t -xdelay=100
avrdude> quit
>>> quit
avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM3 -p m328p -qq -t
avrdude> quit
>>> quit
avrdude> avrdude error: programmer is not responding (Note: this is normal for `-c ardujno` because of time out)

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_150ms -c urclock -P COM3 -p m328p -xshowall

avrdude_150ms: AVR device initialized and ready to accept instructions
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 o4.4 --s-h-r-- vector 0 (RESET) ATmega328P

```

mcuee, 2022-11-23:
But I need to confirm under whether Linux or macOS is okay with 80ms or not.

stefanrueger, 2022-11-23:
I would really like to know what's behind this! Windows and Linux do not use the same draining function (well both are called `ser_drain()` but they work differently), and the variable `serial_drain_timeout` means different things in both.

I can set all to 150 ms, but it;'s better to understand *why* Windows needs these high values *and* why Windows cannot cope with a 500 ms WDT timeout. Any takers for an analysis what's needed for the handshake? We need to consider this both for fixed-baudrate bootloaders and for autobaud.

stefanrueger, 2022-11-23:
>  My initial testing results under Linux shows that both 80ms and 150ms timeout do not work. 

My tests under Linux are that almost any timeout works as long as it is, say, some 100 ms less than the WDT timeout. This variable is not critical at all, but because it delays the sync, I like to set it small. No difference fixed-baudrate/autobaud.

stefanrueger, 2022-11-23:
> Any takers for an analysis what's needed for the handshake?

With that I mean; provide a timing profile for the handshake process using, eg, my [timing branch](https://github.com/stefanrueger/avrdude/tree/timing) for this.

Under Linux there is the same timing profile for urboot v7.7 autobaud, urboot v7.6 fixed-baudrate and optiboot v8.3:
```
$ avrdude -p m328p -curclock -xshowall 
   0 ms: open
   1 ms: reset board
  21 ms: before 80+delay ms
 101 ms: after 80+delay ms
 182 ms: drained (1)
 182 ms: enter urclock_getsync()
 182 ms: sent initial sync
 262 ms: drained
 267 ms: all good, ready to rock

avrdude: AVR device initialized and ready to accept instructions
6c0065d30a63 0000-00-00 00.00  application 0 store 0 meta 0 boot 384 u7.7 weu-jPrac vector 25 (SPM_Ready) ATmega328P

$ avrdude -p m328p -curclock -xshowall 
   0 ms: open
   1 ms: reset board
  21 ms: before 80+delay ms
 102 ms: after 80+delay ms
 182 ms: drained (1)
 182 ms: enter urclock_getsync()
 182 ms: sent initial sync
 262 ms: drained
 267 ms: all good, ready to rock

avrdude: AVR device initialized and ready to accept instructions
00000bd39209 2022-11-09 18.24 blink.hex 354 store 32009 meta 21 boot 384 u7.6 weu-jpr-. vector 25 (SPM_Ready) ATmega328P

$ avrdude -p m328p -curclock -xshowall 
   0 ms: open
   1 ms: reset board
  21 ms: before 80+delay ms
 101 ms: after 80+delay ms
 182 ms: drained (1)
 182 ms: enter urclock_getsync()
 182 ms: sent initial sync
 262 ms: drained
 267 ms: all good, ready to rock

avrdude: AVR device initialized and ready to accept instructions
0 2022-11-09 18.24 blink.hex 354 store 31881 meta 21 boot 512 o8.3 --s-h-r-- vector 0 (RESET) ATmega328P
```

mcuee, 2022-11-24:
```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1188 -c usbasp -P COM5 -p m328p  -U .\atmega328p_16mhz_115200bps_ee_led+b1_fr_ce_ur_vbl.hex -qq && echo OK
OK
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1188 -c urclock -P COM5 -p m328p
  -U .\Blink_Uno.ino.hex -qq && echo OK
OK

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timing -c urclock -P COM5 -p m328p  -xshowall
   0 ms: open
   3 ms: reset board
  38 ms: before 80+delay ms
 130 ms: after 80+delay ms
 236 ms: drained (1)
 238 ms: enter urclock_getsync()
 239 ms: sent initial sync
 342 ms: drained
 350 ms: all good, ready to rock

avrdude_timing: AVR device initialized and ready to accept instructions
0000ffffffff 2022-10-17 09.23 Blink_Uno.ino.hex 924 store 31431 meta 29 boot 384 u7.7 
weu-jPr-c vector 25 (SPM_Ready) ATmega328P
```

mcuee, 2022-11-24:
Further timing test for Optiboot.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timing -c urclock -P COM5 -p m328p  -U .\Blink_Uno.ino.hex -xdelay=200
   0 ms: open
   3 ms: reset board
  27 ms: before 80+delay ms
 323 ms: after 80+delay ms
 491 ms: drained (1)
 494 ms: enter urclock_getsync()
 496 ms: sent initial sync
 658 ms: drained
avrdude_timing warning: attempt 1 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 2 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 3 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 4 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 5 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 6 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 7 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 8 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 9 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 10 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 11 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 12 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 13 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 14 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 15 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 16 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 17 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 18 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 19 of 20: not in sync
avrdude_timing warning: programmer is not responding; try, eg, -xdelay=200
avrdude_timing warning: attempt 20 of 20: not in sync
avrdude_timing error: unable to open programmer urclock on port COM5

avrdude_timing done.  Thank you.

```

mcuee, 2022-11-24:
Looks like ` urclock_getsync()` is the issue.
But the following patch does not work.
```
$ git diff
diff --git a/src/urclock.c b/src/urclock.c
index 90d27dc..f69b2f8 100644
--- a/src/urclock.c
+++ b/src/urclock.c
@@ -1923,6 +1923,8 @@ static int urclock_getsync(const PROGRAMMER *pgm) {
   serial_drain(&pgm->fd, 0);
   msg_info("%4d ms: drained\n", mstimestamp());

+  // Restrore 250ms timeout for establishing comms for Windows
+  serial_recv_timeout = 250;     // ms

   for(attempt = 0; attempt < MAX_SYNC_ATTEMPTS; attempt++) {
     iob[0] = Cmnd_STK_GET_SYNC;

```

mcuee, 2022-11-24:
Another board (Nano Clone with CH340) under Windows.
```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p m328p  -U .\optiboot_atmega328.hex
 -qq && echo OK
OK

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timing -c urclock -P COM4 -p m328p  -xshowall
   0 ms: open
   2 ms: reset board
  33 ms: before 80+delay ms
 127 ms: after 80+delay ms
 299 ms: drained (1)
 303 ms: enter urclock_getsync()
 305 ms: sent initial sync
 471 ms: drained
 477 ms: all good, ready to rock

avrdude_timing: AVR device initialized and ready to accept instructions
avrdude_timing error: protocol expects OK byte 0x10 but got 0x95 in urclock_read_sig_bytes()
avrdude_timing error: unable to read signature data for part ATmega328P, rc=-1
avrdude_timing error: unable to read signature data, rc=-1
avrdude_timing error: protocol expects sync byte 0x14 but got 0x0f

avrdude_timing done.  Thank you.

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timing -c urclock -P COM4 -p m328p  -xshowall -xdelay=100
   0 ms: open
   2 ms: reset board
  33 ms: before 80+delay ms
 221 ms: after 80+delay ms
 392 ms: drained (1)
 396 ms: enter urclock_getsync()
 397 ms: sent initial sync
 641 ms: drained
 651 ms: all good, ready to rock

avrdude_timing: AVR device initialized and ready to accept instructions
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 o4.4 --s-h-r-- vector 0 (RESET) ATmega328P

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timing -c urclock -P COM4 -p m328p  -U .\Blink_Uno.ino.hex -xdelay=100
   0 ms: open
   3 ms: reset board
  27 ms: before 80+delay ms
 214 ms: after 80+delay ms
 385 ms: drained (1)
 390 ms: enter urclock_getsync()
 392 ms: sent initial sync
 636 ms: drained
 647 ms: all good, ready to rock

avrdude_timing: AVR device initialized and ready to accept instructions
avrdude_timing: device signature = 0x1e950f (probably m328p)
avrdude_timing: Note: flash memory has been specified, an erase cycle will be performed.
                To disable this feature, specify the -D option.
avrdude_timing: erasing chip
                delaying chip erase until first -U upload to flash
avrdude_timing: reading input file .\Blink_Uno.ino.hex for flash
                with 924 bytes in 1 section within [0, 0x39b]
                using 8 pages and 100 pad bytes
avrdude_timing: preparing flash input for device bootloader
avrdude_timing: writing 924 bytes flash ...

Writing | ################################################## | 100% 6.36 s

avrdude_timing: 924 bytes of flash written
avrdude_timing: verifying flash memory against .\Blink_Uno.ino.hex

Reading | ################################################## | 100% 5.11 s

avrdude_timing: 924 bytes of flash verified

avrdude_timing done.  Thank you.

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timing -c urclock -P COM4 -p m328p  -U .\Blink_Uno.ino.hex -xdelay=200
   0 ms: open
   3 ms: reset board
  26 ms: before 80+delay ms
 321 ms: after 80+delay ms
 492 ms: drained (1)
 495 ms: enter urclock_getsync()
 497 ms: sent initial sync
 664 ms: drained
 674 ms: all good, ready to rock

avrdude_timing: AVR device initialized and ready to accept instructions
avrdude_timing: device signature = 0x1e950f (probably m328p)
avrdude_timing: Note: flash memory has been specified, an erase cycle will be performed.
                To disable this feature, specify the -D option.
avrdude_timing: erasing chip
                delaying chip erase until first -U upload to flash
avrdude_timing: reading input file .\Blink_Uno.ino.hex for flash
                with 924 bytes in 1 section within [0, 0x39b]
                using 8 pages and 100 pad bytes
avrdude_timing: preparing flash input for device bootloader
avrdude_timing: writing 924 bytes flash ...

Writing | ################################################## | 100% 6.27 s

avrdude_timing: 924 bytes of flash written
avrdude_timing: verifying flash memory against .\Blink_Uno.ino.hex

Reading | ################################################## | 100% 5.05 s

avrdude_timing: 924 bytes of flash verified

avrdude_timing done.  Thank you.

```

mcuee, 2022-11-24:
For the Nano Clone, no issue with STK500v1 urboot autobaud FW.
```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p m328p
  -U .\atmega328p_autobaud_ee_led+b1_fr_ce.hex -qq && echo OK
OK
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timing -c urclock -P COM4 -p m328p  -xshowall
   0 ms: open
   2 ms: reset board
  34 ms: before 80+delay ms
 127 ms: after 80+delay ms
 297 ms: drained (1)
 301 ms: enter urclock_getsync()
 302 ms: sent initial sync
 468 ms: drained
 477 ms: all good, ready to rock

avrdude_timing: AVR device initialized and ready to accept instructions
0000ffffffff 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 u7.7 wes-hprac vector 0 (RESET) ATmega328P
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timing -c urclock -P COM4 -p m328p  -U .\Blink_Uno.ino.hex
   0 ms: open
   3 ms: reset board
  30 ms: before 80+delay ms
 125 ms: after 80+delay ms
 296 ms: drained (1)
 298 ms: enter urclock_getsync()
 299 ms: sent initial sync
 470 ms: drained
 481 ms: all good, ready to rock

avrdude_timing: AVR device initialized and ready to accept instructions
avrdude_timing: device signature = 0x1e950f (probably m328p)
avrdude_timing: Note: flash memory has been specified, an erase cycle will be performed.
                To disable this feature, specify the -D option.
avrdude_timing: erasing chip
avrdude_timing: reading input file .\Blink_Uno.ino.hex for flash
                with 924 bytes in 1 section within [0, 0x39b]
                using 8 pages and 100 pad bytes
avrdude_timing: preparing flash input for device bootloader
avrdude_timing: writing 924 bytes flash ...

Writing | ################################################## | 100% 0.25 s

avrdude_timing: 924 bytes of flash written
avrdude_timing: verifying flash memory against .\Blink_Uno.ino.hex

Reading | ################################################## | 100% 0.18 s

avrdude_timing: 924 bytes of flash verified

avrdude_timing done.  Thank you.

```

mcuee, 2022-11-24:
New tests using the Arduino Uno clone (ATmega16U2 USB to Serial chip). As mentioned in the first post, the following patch seems to work fine. But it is kind of strange why it works.
```
$ git diff
diff --git a/src/urclock.c b/src/urclock.c
index 7bfb2d7..f58a2d7 100644
--- a/src/urclock.c
+++ b/src/urclock.c
@@ -2154,7 +2154,11 @@ static int urclock_open(PROGRAMMER *pgm, const char *port) {
     usleep((80+ur.delay)*1000); // Wait until board comes out of reset

   // Drain any extraneous input
+#ifndef WIN32
   serial_drain_timeout = 80;    // ms
+#else
+  serial_drain_timeout = 150;    // ms
+#endif
   serial_drain(&pgm->fd, 0);

   if(urclock_getsync(pgm) < 0)

S C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_150ms -c urclock -P COM3 -p m328p
 -U .\hex\Blink.ino.standard.hex -qq && echo OK
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_150ms -c urclock -P COM3 -p m328p -xshowall

avrdude_150ms: AVR device initialized and ready to accept instructions
0 2022-06-10 21.21 Blink.ino.standard.hex 924 store 31298 meta 34 boot 512 o4.4 --s-h-r-- 
vector 0 (RESET) ATmega328P

```

mcuee, 2022-11-24:
> But I need to confirm under whether Linux or macOS is okay with 80ms or not. My initial testing results under Linux shows that both 80ms and 150ms timeout do not work. Strange. I need to double check.

No issues with Linux, with either 80ms or 150ms. I may have a wrong fuse setting yesterday.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c urclock -P /dev/ttyUSB0 -p m328p
 -U ./Blink.ino.standard.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file ./Blink.ino.standard.hex for flash
         with 924 bytes in 1 section within [0, 0x39b]
         using 8 pages and 100 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 924 bytes flash ...

Writing | ################################################## | 100% 0.21 s 

avrdude: 924 bytes of flash written
avrdude: verifying flash memory against ./Blink.ino.standard.hex

Reading | ################################################## | 100% 0.15 s 

avrdude: 924 bytes of flash verified

avrdude done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_150ms -c urclock -P /dev/ttyUSB0
 -p m328p -U ./Blink.ino.standard.hex

avrdude_150ms: AVR device initialized and ready to accept instructions
avrdude_150ms: device signature = 0x1e950f (probably m328p)
avrdude_150ms: Note: flash memory has been specified, an erase cycle will be performed.
               To disable this feature, specify the -D option.
avrdude_150ms: erasing chip
avrdude_150ms: reading input file ./Blink.ino.standard.hex for flash
               with 924 bytes in 1 section within [0, 0x39b]
               using 8 pages and 100 pad bytes
avrdude_150ms: preparing flash input for device bootloader
avrdude_150ms: writing 924 bytes flash ...

Writing | ################################################## | 100% 0.21 s 

avrdude_150ms: 924 bytes of flash written
avrdude_150ms: verifying flash memory against ./Blink.ino.standard.hex

Reading | ################################################## | 100% 0.15 s 

avrdude_150ms: 924 bytes of flash verified

avrdude_150ms done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_timing -c urclock -P /dev/ttyUSB0 -p m328p
 -U ./Blink.ino.standard.hex
   0 ms: open
   1 ms: reset board
  22 ms: before 80+delay ms
 102 ms: after 80+delay ms
 182 ms: drained (1)
 182 ms: enter urclock_getsync()
 182 ms: sent initial sync
 263 ms: drained
 270 ms: all good, ready to rock

avrdude_timing: AVR device initialized and ready to accept instructions
avrdude_timing: device signature = 0x1e950f (probably m328p)
avrdude_timing: Note: flash memory has been specified, an erase cycle will be performed.
                To disable this feature, specify the -D option.
avrdude_timing: erasing chip
avrdude_timing: reading input file ./Blink.ino.standard.hex for flash
                with 924 bytes in 1 section within [0, 0x39b]
                using 8 pages and 100 pad bytes
avrdude_timing: preparing flash input for device bootloader
avrdude_timing: writing 924 bytes flash ...

Writing | ################################################## | 100% 0.21 s 

avrdude_timing: 924 bytes of flash written
avrdude_timing: verifying flash memory against ./Blink.ino.standard.hex

Reading | ################################################## | 100% 0.15 s 

avrdude_timing: 924 bytes of flash verified

avrdude_timing done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_timing -c urclock -P /dev/ttyUSB0 -p m328p -xshowall
   0 ms: open
   0 ms: reset board
  21 ms: before 80+delay ms
 101 ms: after 80+delay ms
 181 ms: drained (1)
 181 ms: enter urclock_getsync()
 181 ms: sent initial sync
 262 ms: drained
 269 ms: all good, ready to rock

avrdude_timing: AVR device initialized and ready to accept instructions
ffffffffffff 2022-06-10 21.21 Blink.ino.standard.hex 924 store 31298 meta 34 boot 512 u7.7 wes-hpr-c 
vector 0 (RESET) ATmega328P

```

stefanrueger, 2022-11-24:
@mcuee Thanks for careful and systematic timing profiles. Looks like the problems you are facing are almost exclusively with optiboot.

Turns out I can reproduce this under Linux. My initial tests were with optiboot v8.3, but I overlooked the culprit here is actually optiboot **v4.4**. Luckily, they shipped the source with Arduino, so I could read  what optiboot v4.4 does for handshake: turns out it
 - Entertains the user with flashing LEDs for ~ 300 ms
 - *After(!)* initialising the USART

These two together mean that any two-byte sync sequence sent in the first 300 ms get their first byte thundered over in the USART. Can be solved with a looooooooong initial delay. But who wants to wait for 250 ms drain + 250 ms initial delay (as in `-c arduino`) before anything happens? So, I changed  `urclock_getsync()` to cope with that kind of thing.

Updated the [timing branch](https://github.com/stefanrueger/avrdude/tree/timing) for you, so you can examine and test with different boards, OS and bootloaders.

My tests were:

```
for xdel in -20 0 100 200 300 400; do
  echo --- testing with $xdel+80 ms delay
  avrdude -qqcurclock -pm328p -xshowversion -xshowboot -xdelay=$xdel || echo failed
done
```

I am currently using decidedly short timeouts (rcv=40 ms, drain=10 ms). I hope they work with Windows!

Results for Optiboot v4.4 (shipped with Arduino)
```
--- testing with -20+80 ms delay
  91 ms: enter urclock_getsync()
 131 ms zzz(16 ms)
 188 ms zzz(32 ms)
 260 ms zzz(64 ms)
 364 ms zzz(128 ms)
 505 ms: all good, ready to rock
boot 512 o4.4 --s-h-r--
--- testing with 0+80 ms delay
 111 ms: enter urclock_getsync()
 152 ms zzz(16 ms)
 208 ms zzz(32 ms)
 280 ms zzz(64 ms)
 385 ms zzz(128 ms)
 526 ms: all good, ready to rock
boot 512 o4.4 --s-h-r--
--- testing with 100+80 ms delay
 212 ms: enter urclock_getsync()
 252 ms zzz(16 ms)
 309 ms zzz(32 ms)
 381 ms zzz(64 ms)
 479 ms: all good, ready to rock
boot 512 o4.4 --s-h-r--
--- testing with 200+80 ms delay
 311 ms: enter urclock_getsync()
 351 ms zzz(16 ms)
 408 ms zzz(32 ms)
 478 ms: all good, ready to rock
boot 512 o4.4 --s-h-r--
--- testing with 300+80 ms delay
 411 ms: enter urclock_getsync()
 451 ms zzz(16 ms)
 481 ms: all good, ready to rock
avrdude error: programmer is not responding; try and vary -xdelay=100
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 200kHz
        - use -F to override this check

avrdude error: programmer is not responding; try and vary -xdelay=100
failed
--- testing with 400+80 ms delay
 512 ms: enter urclock_getsync()
 538 ms: all good, ready to rock
boot 512 o4.4 --s-h-r--
```
Results for optiboot v8.3
```
--- testing with -20+80 ms delay
  91 ms: enter urclock_getsync()
 132 ms zzz(16 ms)
 166 ms: all good, ready to rock
boot 512 o8.3 --s-h-r--
--- testing with 0+80 ms delay
 111 ms: enter urclock_getsync()
 174 ms: all good, ready to rock
boot 512 o8.3 --s-h-r--
--- testing with 100+80 ms delay
 211 ms: enter urclock_getsync()
 251 ms zzz(16 ms)
 290 ms: all good, ready to rock
boot 512 o8.3 --s-h-r--
--- testing with 200+80 ms delay
 311 ms: enter urclock_getsync()
 361 ms: all good, ready to rock
boot 512 o8.3 --s-h-r--
--- testing with 300+80 ms delay
 412 ms: enter urclock_getsync()
 452 ms zzz(16 ms)
 481 ms: all good, ready to rock
avrdude error: programmer is not responding; try and vary -xdelay=100
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 200kHz
        - use -F to override this check

failed
--- testing with 400+80 ms delay
 512 ms: enter urclock_getsync()
 537 ms: all good, ready to rock
boot 512 o8.3 --s-h-r--
```

Results for urboot v7.7 autobaud

```
--- testing with -20+80 ms delay
  91 ms: enter urclock_getsync()
 131 ms zzz(16 ms)
 173 ms: all good, ready to rock
boot 384 u7.7 weu-jPrac
--- testing with 0+80 ms delay
 111 ms: enter urclock_getsync()
 151 ms zzz(16 ms)
 193 ms: all good, ready to rock
boot 384 u7.7 weu-jPrac
--- testing with 100+80 ms delay
 211 ms: enter urclock_getsync()
 252 ms zzz(16 ms)
 293 ms: all good, ready to rock
boot 384 u7.7 weu-jPrac
--- testing with 200+80 ms delay
 311 ms: enter urclock_getsync()
 352 ms zzz(16 ms)
 393 ms: all good, ready to rock
boot 384 u7.7 weu-jPrac
--- testing with 300+80 ms delay
 411 ms: enter urclock_getsync()
 452 ms zzz(16 ms)
 493 ms: all good, ready to rock
boot 384 u7.7 weu-jPrac
--- testing with 400+80 ms delay
 511 ms: enter urclock_getsync()
 551 ms zzz(16 ms)
 593 ms: all good, ready to rock
boot 384 u7.7 weu-jPrac
```

Results for urboot v7.7 fixed baudrate

```
--- testing with -20+80 ms delay
  91 ms: enter urclock_getsync()
 116 ms: all good, ready to rock
boot 256 u7.6 w-u-jpr-.
--- testing with 0+80 ms delay
 111 ms: enter urclock_getsync()
 137 ms: all good, ready to rock
boot 256 u7.6 w-u-jpr-.
--- testing with 100+80 ms delay
 212 ms: enter urclock_getsync()
 237 ms: all good, ready to rock
boot 256 u7.6 w-u-jpr-.
--- testing with 200+80 ms delay
 312 ms: enter urclock_getsync()
 338 ms: all good, ready to rock
boot 256 u7.6 w-u-jpr-.
--- testing with 300+80 ms delay
 411 ms: enter urclock_getsync()
 437 ms: all good, ready to rock
boot 256 u7.6 w-u-jpr-.
--- testing with 400+80 ms delay
 511 ms: enter urclock_getsync()
 537 ms: all good, ready to rock
boot 256 u7.6 w-u-jpr-.
```

Only rarely did an -xdelay value not work out. Almost all work, and certainly those around 0, so no need to specify them at all. I might remove that option.

mcuee, 2022-11-24:
@stefanrueger 

Great. Now it works without `-xdelay`. 

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timingv1 -c urclock -P COM3 -p m328p
 -U .\hex\Blink.ino.standard.hex
 134 ms: enter urclock_getsync()
 264 ms zzz(16 ms)
 415 ms zzz(32 ms)
 494 ms: all good, ready to rock

avrdude_timingv1: AVR device initialized and ready to accept instructions
avrdude_timingv1: device signature = 0x1e950f (probably m328p)
avrdude_timingv1: Note: flash memory has been specified, an erase cycle will be performed.
                  To disable this feature, specify the -D option.
avrdude_timingv1: erasing chip
                  delaying chip erase until first -U upload to flash
avrdude_timingv1: reading input file .\hex\Blink.ino.standard.hex for flash
                  with 924 bytes in 1 section within [0, 0x39b]
                  using 8 pages and 100 pad bytes
avrdude_timingv1: preparing flash input for device bootloader
avrdude_timingv1: writing 924 bytes flash ...

Writing | ################################################## | 100% 6.39 s

avrdude_timingv1: 924 bytes of flash written
avrdude_timingv1: verifying flash memory against .\hex\Blink.ino.standard.hex

Reading | ################################################## | 100% 4.94 s

avrdude_timingv1: 924 bytes of flash verified

avrdude_timingv1 done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timingv1 -c urclock -P COM3 -p m328p -xshowall
 139 ms: enter urclock_getsync()
 269 ms zzz(16 ms)
 419 ms zzz(32 ms)
 509 ms: all good, ready to rock

avrdude_timingv1: AVR device initialized and ready to accept instructions
0 2022-06-10 21.21 Blink.ino.standard.hex 924 store 31298 meta 34 boot 512 o4.4 --s-h-r-- 
vector 0 (RESET) ATmega328P

```

mcuee, 2022-11-24:
Second test with Arduino Nano clone (CH340)

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timingv1 -c urclock -P COM10 -p m328p
 -U .\hex\Blink.ino.standard.hex

 145 ms: enter urclock_getsync()
 276 ms zzz(16 ms)
 427 ms zzz(32 ms)
 497 ms: all good, ready to rock

avrdude_timingv1: AVR device initialized and ready to accept instructions
avrdude_timingv1: device signature = 0x1e950f (probably m328p)
avrdude_timingv1: Note: flash memory has been specified, an erase cycle will be performed.
                  To disable this feature, specify the -D option.
avrdude_timingv1: erasing chip
                  delaying chip erase until first -U upload to flash
avrdude_timingv1: reading input file .\hex\Blink.ino.standard.hex for flash
                  with 924 bytes in 1 section within [0, 0x39b]
                  using 8 pages and 100 pad bytes
avrdude_timingv1: preparing flash input for device bootloader
avrdude_timingv1: writing 924 bytes flash ...

Writing | ################################################## | 100% 6.63 s

avrdude_timingv1: 924 bytes of flash written
avrdude_timingv1: verifying flash memory against .\hex\Blink.ino.standard.hex

Reading | ################################################## | 100% 5.42 s

avrdude_timingv1: 924 bytes of flash verified

avrdude_timingv1 done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timingv1 -c urclock -P COM10 -p m328p -xshowall
 151 ms: enter urclock_getsync()
 281 ms zzz(16 ms)
 442 ms zzz(32 ms)
 512 ms: all good, ready to rock

avrdude_timingv1: AVR device initialized and ready to accept instructions
0 2022-06-10 21.21 Blink.ino.standard.hex 924 store 31298 meta 34 boot 512 o4.4 --s-h-r-- 
vector 0 (RESET) ATmega328P

$ for xdel in -20 0 100 200 300 400; do   echo --- testing with $xdel+80 ms delay;   ./avrdude_timingv1 -qq -c urclock -p m328p -P COM10 -xshowversion -xshowboo
t -xdelay=$xdel || echo failed; done
--- testing with -20+80 ms delay
 125 ms: enter urclock_getsync()
 255 ms zzz(16 ms)
 406 ms zzz(32 ms)
 497 ms: all good, ready to rock
boot 512 o4.4 --s-h-r--
--- testing with 0+80 ms delay
 145 ms: enter urclock_getsync()
 275 ms zzz(16 ms)
 435 ms zzz(32 ms)
 505 ms: all good, ready to rock
boot 512 o4.4 --s-h-r--
--- testing with 100+80 ms delay
 237 ms: enter urclock_getsync()
 367 ms zzz(16 ms)
 497 ms: all good, ready to rock
boot 512 o4.4 --s-h-r--
--- testing with 200+80 ms delay
 350 ms: enter urclock_getsync()
 532 ms: all good, ready to rock
boot 512 o4.4 --s-h-r--
--- testing with 300+80 ms delay
 435 ms: enter urclock_getsync()
 525 ms: all good, ready to rock
boot 512 o4.4 --s-h-r--
--- testing with 400+80 ms delay
 543 ms: enter urclock_getsync()
 613 ms: all good, ready to rock
boot 512 o4.4 --s-h-r--

```

mcuee, 2022-11-24:
Same Arduini Nano clone, using @MCUdude's mini core, Optiboot version 8.0.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timingv1 -c urclock -P COM10 -p m328p
 -U .\hex\Blink.ino.standard.hex -xbootsize=512
 137 ms: enter urclock_getsync()
 217 ms: all good, ready to rock

avrdude_timingv1: AVR device initialized and ready to accept instructions
avrdude_timingv1: device signature = 0x1e950f (probably m328p)
avrdude_timingv1: Note: flash memory has been specified, an erase cycle will be performed.
                  To disable this feature, specify the -D option.
avrdude_timingv1: erasing chip
                  delaying chip erase until first -U upload to flash
avrdude_timingv1: reading input file .\hex\Blink.ino.standard.hex for flash
                  with 924 bytes in 1 section within [0, 0x39b]
                  using 8 pages and 100 pad bytes
avrdude_timingv1: preparing flash input for device bootloader
avrdude_timingv1: writing 924 bytes flash ...

Writing | ################################################## | 100% 7.53 s

avrdude_timingv1: 924 bytes of flash written
avrdude_timingv1: verifying flash memory against .\hex\Blink.ino.standard.hex

Reading | ################################################## | 100% 5.39 s

avrdude_timingv1: 924 bytes of flash verified

avrdude_timingv1 done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timingv1 -c urclock -P COM10 -p m328p
 -xshowall -xbootsize=512
 137 ms: enter urclock_getsync()
 217 ms: all good, ready to rock

avrdude_timingv1: AVR device initialized and ready to accept instructions
0 2022-06-10 21.21 Blink.ino.standard.hex 924 store 31298 meta 34 boot 512 o8.0 -?s-?-r-- 
vector 0 (RESET) ATmega328P

$ for xdel in -20 0 100 200 300 400; do   echo --- testing with $xdel+80 ms delay;   ./avrdude_timingv1 -qq -c urclock -p m328p -P COM10 -xshowversion -xshowboot -xdelay=$xdel -xbootsize=512 || echo failed; done
--- testing with -20+80 ms delay
 127 ms: enter urclock_getsync()
 217 ms: all good, ready to rock
boot 512 o8.0 -?s-?-r--
--- testing with 0+80 ms delay
 151 ms: enter urclock_getsync()
 281 ms: all good, ready to rock
boot 512 o8.0 -?s-?-r--
--- testing with 100+80 ms delay
 251 ms: enter urclock_getsync()
 341 ms: all good, ready to rock
boot 512 o8.0 -?s-?-r--
--- testing with 200+80 ms delay
 352 ms: enter urclock_getsync()
 422 ms: all good, ready to rock
boot 512 o8.0 -?s-?-r--
--- testing with 300+80 ms delay
 441 ms: enter urclock_getsync()
 511 ms: all good, ready to rock
boot 512 o8.0 -?s-?-r--
--- testing with 400+80 ms delay
 544 ms: enter urclock_getsync()
 614 ms: all good, ready to rock
boot 512 o8.0 -?s-?-r--
```

mcuee, 2022-11-24:
@stefanrueger 

But there is a regression with picoboot.
https://github.com/nerdralph/picoboot/blob/master/arduino/picobootArduino328v3b2.hex

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timing -c urclock -P COM10 -p m328p
 -xshowall -xbootsize=512
   0 ms: open
   3 ms: reset board
  26 ms: before 80+delay ms
 117 ms: after 80+delay ms
 288 ms: drained (1)
 295 ms: enter urclock_getsync()
 296 ms: sent initial sync
 459 ms: drained
 725 ms: all good, ready to rock

avrdude_timing: AVR device initialized and ready to accept instructions
0 2022-06-10 21.21 Blink.ino.standard.hex 924 store 31298 meta 34 boot 512 x0.0 ......... 
vector 0 (RESET) ATmega328P

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timingv1 -c urclock -P COM10 -p m328p -xshowall -xbootsize=512
 150 ms: enter urclock_getsync()
 280 ms zzz(16 ms)
 441 ms zzz(32 ms)
 611 ms zzz(64 ms)
 811 ms zzz(128 ms)
1084 ms zzz(128 ms)
1354 ms zzz(128 ms)
1625 ms zzz(128 ms)
avrdude_timingv1 warning: attempt 1 of 10: not in sync
avrdude_timingv1 error: programmer is not responding; try and vary -xdelay=100
1894 ms zzz(128 ms)
avrdude_timingv1 warning: attempt 2 of 10: not in sync
avrdude_timingv1 error: programmer is not responding; try and vary -xdelay=100
2166 ms zzz(128 ms)
avrdude_timingv1 warning: attempt 3 of 10: not in sync
avrdude_timingv1 error: programmer is not responding; try and vary -xdelay=100
2439 ms zzz(128 ms)
avrdude_timingv1 warning: attempt 4 of 10: not in sync
avrdude_timingv1 error: programmer is not responding; try and vary -xdelay=100
2707 ms zzz(128 ms)
avrdude_timingv1 warning: attempt 5 of 10: not in sync
avrdude_timingv1 error: programmer is not responding; try and vary -xdelay=100
2976 ms zzz(128 ms)
avrdude_timingv1 warning: attempt 6 of 10: not in sync
avrdude_timingv1 error: programmer is not responding; try and vary -xdelay=100
3251 ms zzz(128 ms)
avrdude_timingv1 warning: attempt 7 of 10: not in sync
avrdude_timingv1 error: programmer is not responding; try and vary -xdelay=100
3534 ms zzz(128 ms)
avrdude_timingv1 warning: attempt 8 of 10: not in sync
avrdude_timingv1 error: programmer is not responding; try and vary -xdelay=100
3806 ms zzz(128 ms)
avrdude_timingv1 warning: attempt 9 of 10: not in sync
avrdude_timingv1 error: programmer is not responding; try and vary -xdelay=100
4070 ms zzz(128 ms)
avrdude_timingv1 warning: attempt 10 of 10: not in sync
avrdude_timingv1 error: programmer is not responding; try and vary -xdelay=100
avrdude_timingv1 error: unable to open programmer urclock on port COM10

avrdude_timingv1 done.  Thank you.

```

mcuee, 2022-11-25:
Tested the LGT8F328P bootloader as well (57600bps).

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timingv1 -c urclock -P COM6 -p lgt8f328p
 -b 57600  -xbootsize=4096 -U .\hex\Blink_Uno.ino.hex
 155 ms: enter urclock_getsync()
 296 ms zzz(16 ms)
 436 ms: all good, ready to rock

avrdude_timingv1: AVR device initialized and ready to accept instructions
avrdude_timingv1: device signature = 0x1e950f (probably lgt8f328p)
avrdude_timingv1: Note: flash memory has been specified, an erase cycle will be performed.
                  To disable this feature, specify the -D option.
avrdude_timingv1: erasing chip
avrdude_timingv1: reading input file .\hex\Blink_Uno.ino.hex for flash
                  with 924 bytes in 1 section within [0, 0x39b]
                  using 8 pages and 100 pad bytes
avrdude_timingv1: preparing flash input for device bootloader
avrdude_timingv1: writing 924 bytes flash ...

Writing | ################################################## | 100% 0.30 s

avrdude_timingv1: 924 bytes of flash written
avrdude_timingv1: verifying flash memory against .\hex\Blink_Uno.ino.hex

Reading | ################################################## | 100% 0.24 s

avrdude_timingv1: 924 bytes of flash verified

avrdude_timingv1 done.  Thank you.

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timing -c urclock -P COM6 -p lgt8f328p
 -b 57600  -xbootsize=4096 -xshowall
   0 ms: open
   3 ms: reset board
  31 ms: before 80+delay ms
 125 ms: after 80+delay ms
 299 ms: drained (1)
 303 ms: enter urclock_getsync()
 305 ms: sent initial sync
 581 ms: drained
 588 ms: all good, ready to rock

avrdude_timing: AVR device initialized and ready to accept instructions
0 2022-10-17 09.23 Blink_Uno.ino.hex 924 store 27719 meta 29 boot 4096 x0.0 ......... vector 0 (RESET) LGT8F328P
```

MCUdude, 2022-11-25:
@mcuee I have an LGT8F328P board I bought a few years ago just for curiosity. However, I was never able to use anything else than the pre-flashed `-carduino` compatible bootloader.

Which "ISP programmer" (Or is it ISP?) are you using for the LGT8?

mcuee, 2022-11-25:
@MCUdude 

You can make an LGT ISP programmer using the Arduino Nano.
https://github.com/dbuezas/lgt8fx
https://github.com/brother-yan/LGTISP

Before today I have only tried the factory bootloader and not the ISP programmer. But I just tried the LGT ISP programmer using the ATmega328P Nano Clone based LGTISP programmer and it works fine (using the hex file LGTISP.hex).

It is using SWD protocol.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v1 -P COM6 -p lgt8f328p

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably lgt8f328p)

avrdude done.  Thank you.

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v1 -P COM6 -p lgt8f328p -U flash:r:lgtisp_readback.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably lgt8f328p)
avrdude: reading flash memory ...

Reading | ################################################## | 100% 9.39 s

avrdude: writing output file lgtisp_readback.hex

avrdude done.  Thank you.

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v1 -P COM6 -p lgt8f328p -U .\optiboot_lgt8f328p.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably lgt8f328p)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\optiboot_lgt8f328p.hex for flash
         with 1002 bytes in 3 sections within [0, 0x77ff]
         using 9 pages and 150 pad bytes
avrdude: writing 1002 bytes flash ...

Writing | ################################################## | 100% 0.30 s

avrdude: 1002 bytes of flash written
avrdude: verifying flash memory against .\optiboot_lgt8f328p.hex

Reading | ################################################## | 100% 0.06 s

avrdude: 1002 bytes of flash verified

avrdude done.  Thank you.

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timingv1 -c urclock -P COM4 -b 57600 -p m328p
 -xbootsize=4096 -xshowall
 154 ms: enter urclock_getsync()
 277 ms zzz(16 ms)
 433 ms: all good, ready to rock

avrdude_timingv1: AVR device initialized and ready to accept instructions
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 4096 x0.0 ......... vector 0 (RESET) ATmega328P

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timingv1 -c urclock -P COM4 -b 57600 -p m328p -xbootsize=4096 -U .\lgt8f328p_blink.hex
 152 ms: enter urclock_getsync()
 276 ms zzz(16 ms)
 433 ms: all good, ready to rock

avrdude_timingv1: AVR device initialized and ready to accept instructions
avrdude_timingv1: device signature = 0x1e950f (probably m328p)
avrdude_timingv1: Note: flash memory has been specified, an erase cycle will be performed.
                  To disable this feature, specify the -D option.
avrdude_timingv1: erasing chip
avrdude_timingv1: reading input file .\lgt8f328p_blink.hex for flash
                  with 928 bytes in 1 section within [0, 0x39f]
                  using 8 pages and 96 pad bytes
avrdude_timingv1: preparing flash input for device bootloader
avrdude_timingv1: writing 928 bytes flash ...

Writing | ################################################## | 100% 0.29 s

avrdude_timingv1: 928 bytes of flash written
avrdude_timingv1: verifying flash memory against .\lgt8f328p_blink.hex

Reading | ################################################## | 100% 0.24 s

avrdude_timingv1: 928 bytes of flash verified

avrdude_timingv1 done.  Thank you.
```

mcuee, 2022-11-25:
@MCUdude 

You can also use the following version. 
https://github.com/LGTMCU/LarduinoISP

mcuee, 2022-11-25:
Arduino lgt8fx seems to work as well.
https://github.com/dbuezas/lgt8fx

```
C:\Users\xfchen\AppData\Local\Arduino15\packages\MegaCoreX\tools\avrdude\7.0.0-arduino2/bin/avrdude -CC:\Users\xfchen\AppData\Local\Arduino15\packages\MegaCoreX\tools\avrdude\7.0.0-arduino2/etc/avrdude.conf -v -patmega328p -cstk500v1 -PCOM6 -e -Ulock:w:0x3f:m -Uefuse:w:0x07:m -Uhfuse:w:0xff:m -Ulfuse:w:0xff:m 

avrdude: Version 7.0
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "C:\Users\xfchen\AppData\Local\Arduino15\packages\MegaCoreX\tools\avrdude\7.0.0-arduino2/etc/avrdude.conf"

         Using Port                    : COM6
         Using Programmer              : stk500v1
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration             0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 3
         Firmware Version: 6.4
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: erasing chip
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Writing |  ***failed;  
################################################## | 100% 0.05s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:

Reading | ################################################## | 100% 0.00s

avrdude: WARNING: ignoring mismatch in unused bits of "lock"
[... 103 lines cut ...]
avrdude: 30720 bytes of flash written
avrdude: verifying flash memory against C:\Users\xfchen\AppData\Local\Arduino15\packages\LGT8fx Boards\hardware\avr\1.0.6/bootloaders/lgt8fx8p/optiboot_lgt8f328p.hex:

Reading | ################################################## | 100% 0.04s

avrdude: 30720 bytes of flash verified
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Writing |  ***failed;  
################################################## | 100% 0.06s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:

Reading | ################################################## | 100% 0.00s

avrdude: WARNING: ignoring mismatch in unused bits of "lock"
         (0xff != 0x3f). To prevent this warning set unused bits
         to 1 when writing (double check with your datasheet first).
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.

```

stefanrueger, 2022-11-25:
> But there is a regression with picoboot.

Thanks for finding this! I think I know why. Will check, but it may be too difficult to cater for all b/l so they start up quickly.

stefanrueger, 2022-11-25:
> LGT8F328P bootloader

Could be added to hash table. 4096 bytes is a large bootloader. You had another one at 2048 bytes?

stefanrueger, 2022-11-25:
BTW, @MCUdude the `urboot` bootloader with autobaud won't work for LGT8F328P b/c some opcodes are faster, and the loop that measures the host baud rate drops from 5 cycles to 3 cycles. This can (in theory) be repaired by the AVRDUDE sending a different GET_SYNC byte in `urclock_getsync()`: 0x3c instead of 0x30. The same trick might be valid for XMEGA parts for which the loop will have 6 cycles: send 0x20 instead of 0x30. Maybe `avrdude.conf` could have an `sync_byte` entry for its parts?

stefanrueger, 2022-11-25:
Needs careful checking whether the `sync_byte` is a STK500 valid command, though. Would (not) be funny if that is chip erase for those bootloaders w/out autobaud detection. ;)

stefanrueger, 2022-11-25:
I have pushed new code to the timing test branch. Now picoboot should work, but *must* add `-xdelay=...` For optiboot this should work with no `-xdelay` or with `-xdelay=350` and higher.

Also added experimental code that should work for `urboot` autobaud bootloader on a LGT8F*P part

mcuee, 2022-11-25:
> I have pushed new code to the timing test branch. Now picoboot should work, but _must_ add `-xdelay=...` For optiboot this should work with no `-xdelay` or with `-xdelay=350` and higher.

Yes picoboot now works with `-xdelay=10` onwards under macOS (will try Windows later). Without `-xdelay` it does not work. BTW, git main does not need that.
```
cuee@mcuees-Mac-mini avrdude_bin % ./avrdude -c urclock -P /dev/tty.usbmodem2213701
 -p m328p -xshowall -xbootsize=512

avrdude: AVR device initialized and ready to accept instructions
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 x0.0 ......... vector 0 
(RESET) ATmega328P

mcuee@mcuees-Mac-mini avrdude_bin % ./avrdude_timingv2 -c urclock -P /dev/tty.usbmodem2213701
 -p m328p -xshowall -xbootsize=512 -xdelay=10
 135 ms: enter urclock_getsync()
 148 ms zzz(16 ms)
 183 ms zzz(32 ms)
 245 ms: all good, ready to rock

avrdude_timingv2: AVR device initialized and ready to accept instructions
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 x0.0 ......... vector 0 (RESET) ATmega328P
```

> Also added experimental code that should work for `urboot` autobaud bootloader on a LGT8F*P part
Hmm, do you have the urboot autobaud bootloader hex file for the LGT8F328P? The Nano LGT9F328P uses internal RC oscillator (default 32MHz).

mcuee, 2022-11-25:
> It would be interesting to see if Urboot EEPROM support works. IIRC, the LGT8F328P uses an emulated EEPROM, where it actually uses the flash memory as EEPROM.

The factory bootloader does not seem to work with EEPROM. So EEPROM support will be a nice new feature.

mcuee, 2022-11-25:
> The CH340* chip is for instance significantly slower than a CP2102N when uploading at higher speeds (115200 baud+).

@MCUdude 

Do you have some data for comparison of different USB to TTL chipset? 

For example, CH340, FT232RL, CP2102, PL2303 and the Arduino ATMega16U2 (for official Arduino Uno and Mega2560) are popular.

mcuee, 2022-11-25:
> Yes picoboot now works with -xdelay=10 onwards under macOS (will try Windows later). Without -xdelay it does not work. BTW, git main does not need that.

@stefanrueger 

Same results under Windows. This time I tried to push to the limit. `-xdelay=1` to `-xdelay=2300` are working. Without `-xdelay` it does not work.

I am okay with the `-xdelay` for picoboot as it is not popular.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timingv2 -c urclock -P COM3 -p m328p
 -U .\hex\Blink.ino.standard.hex -xbootsize=512 -xdelay=1
 142 ms: enter urclock_getsync()
 182 ms zzz(16 ms)
 272 ms: all good, ready to rock

avrdude_timingv2: AVR device initialized and ready to accept instructions
avrdude_timingv2: device signature = 0x1e950f (probably m328p)
avrdude_timingv2: Note: flash memory has been specified, an erase cycle will be performed.
                  To disable this feature, specify the -D option.
avrdude_timingv2: erasing chip
avrdude_timingv2: reading input file .\hex\Blink.ino.standard.hex for flash
                  with 924 bytes in 1 section within [0, 0x39b]
                  using 8 pages and 100 pad bytes
avrdude_timingv2: preparing flash input for device bootloader
avrdude_timingv2: writing 924 bytes flash ...

Writing | ################################################## | 100% 0.24 s

avrdude_timingv2: 924 bytes of flash written
avrdude_timingv2: verifying flash memory against .\hex\Blink.ino.standard.hex

Reading | ################################################## | 100% 0.21 s

avrdude_timingv2: 924 bytes of flash verified

avrdude_timingv2 done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timingv2 -c urclock -P COM3 -p m328p
 -xshowall -xbootsize=512 -xdelay=1
 141 ms: enter urclock_getsync()
 172 ms zzz(16 ms)
 262 ms: all good, ready to rock

avrdude_timingv2: AVR device initialized and ready to accept instructions
0 2022-06-10 21.21 Blink.ino.standard.hex 924 store 31298 meta 34 boot 512 x0.0 ......... 
vector 0 (RESET) ATmega328P
```

stefanrueger, 2022-11-25:
> emulated EEPROM, where it actually uses the flash memory as EEPROM

Well, urboot bootloaders certainly have *no* idea about parts with emulated EEPROM. I suspect, *but do not know*
 - These parts have no EEPROM. Full stop
 - Microchip's AVR EEPROM API does not work (using certain writes to certain registers to write to/read from EEPROM)
 - LGT provide a programming macro to write to high flash (possibly using their bootloader?)

I would just try to burn the [ATmega328p autobaud bootloader for LED at B5](https://raw.githubusercontent.com/stefanrueger/urboot/main/bootloaders/atmega328p/autobaud/urboot_atmega328p_autobaud_led+b5_ur_vbl.hex), or, if the board does not have an LED [this one](https://raw.githubusercontent.com/stefanrueger/urboot/main/bootloaders/atmega328p/autobaud/urboot_atmega328p_autobaud_lednop_ur_vbl.hex). These should, in theory, work for all F_CPU. No EEPROM support.

The `-c urclock` programmer creates a metadata section just below the bootloader. This contains info where the sketch ends. Together with the exported function `pgm_write_page(sram, progmem)` that each urboot bootloader exports at `FLASHEND-4+1` it is possible to use *all* unused flash space like EEPROM (I call this unused space *store*). I have written library functions to that effect that (probably) won't work in Arduino and that I have yet to publish. Here is a glimpse of the idea for the hardended developers like yourselves @mcuee and @MCUdude: [libraries.zip](https://github.com/avrdudes/avrdude/files/10094287/libraries.zip) [chaucer.zip](https://github.com/avrdudes/avrdude/files/10094288/chaucer.zip)

stefanrueger, 2022-11-25:
@mcuee Thanks for all the testing on this. The problem with backward compatibility is that all the different bootloaders do different things under the STK500v1 protocol.  It was *impossible* to get everything working without extra info by the user (and this is what `-xdelay=...` is all about). 

 Do you want to see a PR for the final changes or is it OK if I just push this timing test branch onto git main after cleaning the timing profile printouts? The PR makes sense if you want to (and can) test again.

stefanrueger, 2022-11-25:
> The [LGT8F328P] bootloader is actually not big at all, but it somehow starts at 0x7400.

Which makes it 3076 bytes long. It ends at 0x7800. I suspect the last 2k are reserved for, eg, "EEPROM"? I have just added the hashes for this bootloader.

mcuee, 2022-11-26:
> @mcuee Thanks for all the testing on this. The problem with backward compatibility is that all the different bootloaders do different things under the STK500v1 protocol. It was _impossible_ to get everything working without extra info by the user (and this is what `-xdelay=...` is all about).

Yes I agree. I have no issues with requiring `-xdelay=` for picoboot hex files.

mcuee, 2022-11-26:
> Do you want to see a PR for the final changes or is it OK if I just push this timing test branch onto git main after cleaning the timing profile printouts? The PR makes sense if you want to (and can) test again.

Yes I think it is better to have a PR as it is easier to pinpoint issues by PR number in case there are issues later. I can test again.

mcuee, 2022-11-26:
@stefanrueger 
The hash for lgt8f328p does not seem to work in git main. I still need to specify the bootloadersize.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c avrisp -P COM10 -p lgt8f328p -U .\optiboot_lgt8f328p.hex -qq && echo Okay
avrdude error: Yikes!  Invalid device signature.
        Double check connections and try again, or use -F to override
        this check.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c avrisp -P COM10 -p lgt8f328p
 -U .\optiboot_lgt8f328p.hex -qq && echo Okay
Okay

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM13 -b 57600 -p lgt8f328p
 -xshowall -xbootsize=4096

avrdude: AVR device initialized and ready to accept instructions
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 4096 x0.0 ......... vector 0 (RESET) LGT8F328P
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM13 -b 57600 -p lgt8f328p -xshowall

avrdude: AVR device initialized and ready to accept instructions
avrdude error: unknown bootloader ... please specify -xbootsize=<num>

avrdude error: unable to read signature data for part LGT8F328P, rc=-1
avrdude error: unable to read signature data, rc=-1

avrdude done.  Thank you.

```

mcuee, 2022-11-26:
> I would just try to burn the [ATmega328p autobaud bootloader for LED at B5](https://raw.githubusercontent.com/stefanrueger/urboot/main/bootloaders/atmega328p/autobaud/urboot_atmega328p_autobaud_led+b5_ur_vbl.hex), or, if the board does not have an LED [this one](https://raw.githubusercontent.com/stefanrueger/urboot/main/bootloaders/atmega328p/autobaud/urboot_atmega328p_autobaud_lednop_ur_vbl.hex). These should, in theory, work for all F_CPU [edit: but need the [timing test branch](https://github.com/stefanrueger/avrdude/tree/timing) for -c urclock`] No EEPROM support.

@stefanrueger 
This does not work. 

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c avrisp -P COM10 -p lgt8f328p
 -U .\urboot_atmega328p_autobaud_lednop_ur_vbl.hex -qq && echo Ok
Ok

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_timingv2 -C .\avrdude_timingv2.conf 
-c urclock -P COM13 -b 115200 -p lgt8f328p -xshowall -xbootsize=3072
 143 ms: enter urclock_getsync()
 183 ms zzz(16 ms)
 254 ms zzz(32 ms)
 333 ms zzz(64 ms)
 454 ms zzz(128 ms)
 635 ms zzz(128 ms)
 815 ms zzz(128 ms)
 996 ms zzz(128 ms)
avrdude_timingv2 warning: attempt 1 of 10: not in sync
avrdude_timingv2 error: programmer is not responding; try and vary -xdelay=100
1183 ms zzz(128 ms)
avrdude_timingv2 warning: attempt 2 of 10: not in sync
avrdude_timingv2 error: programmer is not responding; try and vary -xdelay=100
1364 ms zzz(128 ms)
avrdude_timingv2 warning: attempt 3 of 10: not in sync
avrdude_timingv2 error: programmer is not responding; try and vary -xdelay=100
1540 ms zzz(128 ms)
avrdude_timingv2 warning: attempt 4 of 10: not in sync
avrdude_timingv2 error: programmer is not responding; try and vary -xdelay=100
1726 ms zzz(128 ms)
avrdude_timingv2 warning: attempt 5 of 10: not in sync
avrdude_timingv2 error: programmer is not responding; try and vary -xdelay=100
1915 ms zzz(128 ms)
avrdude_timingv2 warning: attempt 6 of 10: not in sync
avrdude_timingv2 error: programmer is not responding; try and vary -xdelay=100
2103 ms zzz(128 ms)
avrdude_timingv2 warning: attempt 7 of 10: not in sync
avrdude_timingv2 error: programmer is not responding; try and vary -xdelay=100
2287 ms zzz(128 ms)
avrdude_timingv2 warning: attempt 8 of 10: not in sync
avrdude_timingv2 error: programmer is not responding; try and vary -xdelay=100
2469 ms zzz(128 ms)
avrdude_timingv2 warning: attempt 9 of 10: not in sync
avrdude_timingv2 error: programmer is not responding; try and vary -xdelay=100
2648 ms zzz(128 ms)
avrdude_timingv2 warning: attempt 10 of 10: not in sync
avrdude_timingv2 error: programmer is not responding; try and vary -xdelay=100
avrdude_timingv2 error: unable to open programmer urclock on port COM13
```

stefanrueger, 2022-11-26:
> BTW, shifting the bootloader hex to 0x7400 does not help either.

You cannot just shift the bootloader! The jump to the application is an **r**jmp forward over the end of the flash wrapping around to 0 and then further to the right vector to get to the application. Shifting the hex needs the `rjmp` adapting, too

mcuee, 2022-11-27:
> You cannot just shift the bootloader! The jump to the application is an **r**jmp forward over the end of the flash wrapping around to 0 and then further to the right vector to get to the [...]

I understand that. I was just trying my luck...

Further discussions of LGT8F328P urboot support is here.
* https://github.com/stefanrueger/urboot/issues/10