avrdudes/avrdude #1200: urclock support for optiboot_x and optiboot_dx

mcuee, 2022-11-27:
As or now urboot does not support AVR Mega0, Tiny01/2 and AVR DA/DB/DD yet.

What about the support of optiboot_x and optiboot_dx? One of the issue is that the bootloader is on top of the flash memory.

@stefanrueger 
Can I assumen it works out of the box?

```
$  ./avrdude -c urclock -P COM13 -p m4808 -xbootsize=512 -xshowall

avrdude: AVR device initialized and ready to accept instructions
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 x0.0 ......... vector 0 (RESET) ATmega4808

```

mcuee, 2022-11-27:
And it sems to work with ATtiny817 [XPlainedPro optiboot_x](https://github.com/avrdudes/avrdude/issues/1166#issuecomment-1303272023) as well.

mcuee, 2022-11-27:
@stefanrueger 
Found one issue with optiboot_dx. Hopefully there is an esay fix. Thanks.

```
$  ./avrdude -c urclock -P COM6 -p avr128db48 -xbootsize=512 -qqt
avrdude error: connected AVR128DB48's number 61 of interrupts differs from -p AVR128DB48's (0); use correct -p ... or override with -F
avrdude error: unable to read signature data for part AVR128DB48, rc=-1
avrdude error: unable to read signature data, rc=-1

$  ./avrdude -c arduino -P COM6 -p avr128db48  -qqt
avrdude>
```

stefanrueger, 2022-11-27:
Right now, `-curclock` works for urboot (and therefore parts for which urboot bootloaders exist). Currently these are 183 classic parts. 

>  AVR Mega0, Tiny01/2 and AVR DA/DB/DD

I have yet to study the data sheets of these parts. How would AVRDUDE know the difference? Is there something in avrdude.conf that would tell it that a part is one or the other (other than parsing the name). They all have `PM_UPDI` set in `prog_modes`.

`-c urclock` makes the assumption that the bootloader sits in top flash. This is wrong for Dx parts as @WestfW points out:
https://github.com/stefanrueger/urboot/issues/2#issuecomment-1312906322 So, I would assume `-curclock` not to work for these parts.

> Metadata location

`-c urclock` assumes 
  - The vector table sits at bottom flash (address 0) for vector bootloaders (urboot never uses own vector tables)
  - The bootloader sits in top flash
  - Applications grow from bottom to top in flash
  - Flash between the application end and the bootloader is "free"

Metadata (unless switched off by `-xnometadata`) are put just below the bootloader and grow from top to bottom. All the space between application end and metadata start is called *store* and can be utilised by the `pgm_write_page(sram, flash)` function that urboot bootloaders export. The metadata also say where the application ends. So in both your examples, the metadata sit where they should sit.

mcuee, 2022-11-27:
@stefanrueger 

Thanks for the explanations. I will label this as enhancement now.

Since `-c arduino` works with optiboot_x and optiboot_dx now, I was assuming that `-c urclock` would work out of the box other than the metadata thingy, or I need to add `-xstrict` and/or `-xdelay=`. Am I I wrong?

I think the AVR DA/DB/DD problem may be with `avrdude.conf` and not `-c urclock`.

stefanrueger, 2022-11-27:
To cut a long short: `-c urclock` currently supports classic parts, and is likely to work for other parts, but unlikely to work for Dx (as for these the bootloader does not sit in top flash).

I would suggest the community try `-c urboot` and/or urboot bootloaders first and see how well they work and how well they are adopted. All being well, if this idea catches on, there might be more AVR-literate people than me (such as @SpenceKonde or @MCUdude) who realise "Hang on a second, if we adopt this urprotocol idea we have all of a sudden freed up 100 bytes in the bootloader that I now can play with, I'll give it a shot".

So my personal strategy for solving this avrdude issue (as well as [Issue 1](https://github.com/stefanrueger/urboot/issues/1), [Issue 2](https://github.com/stefanrueger/urboot/issues/2) and [Issue 4](https://github.com/stefanrueger/urboot/issues/4) of the urboot project) is to do *nothing* about this right now, and let more talented people in the community play with this idea first.

mcuee, 2022-11-27:
@stefanrueger 

Fair enough. I totally agree with your thoughts. 

Our priority now is to get avrdude 7.1 release done by end of this year. I really hope that happens. I have long holiday in December so I can help on testing side.

I will carry out more tests but do not feel pressured to do anything on new parts.

stefanrueger, 2022-11-27:
> problem may be with avrdude.conf and not -c urclock.

Yes, I have pushed an update to avrdude.conf.in in PR #1198 (but `-c urclock` still assumes bootloaders are in top flash, so Dx parts are not currently working)

mcuee, 2022-11-27:
> > problem may be with avrdude.conf and not -c urclock.
> 
> Yes, I have pushed an update to avrdude.conf.in in PR #1198 (but `-c urclock` still assumes bootloaders are in top flash, so Dx parts are not currently working)

I will test it out later.

Actually it is the same situation for Mega0 and Tiny0/1/2 (optiboot_x). 

Please refer to the following run log posted before, the chip is ATtiny817 using optiboot_x from megaTinyCore. You can see that `-c urclock` works just fine.

0x00 - 0x200: Bootloader
0x200 onwards: application

Not exactly sure why the metadata goes to 0x1dc0 though, still since there is nothing below 0x00 so it has to go to the top of the flash. So it kind of makes sense as well. 

```
$  ./avrdude -c urclock -P COM23 -p t817 -xbootsize=512 -xshowall

avrdude: AVR device initialized and ready to accept instructions
0 2022-11-27 21.05 ./Blink_Pro817.ino.t817opti.20c0.mD0.v264.hex 980 store 6643 meta 57 boot 512
 x0.0 ......... vector 0 (RESET) ATtiny817

$  ./avrdude -c urclock -P COM23 -p t817 -xbootsize=512 -qqt
avrdude> dump flash 0 0x40
>>> dump flash 0 0x40
0000  01 c0 d8 c0 11 24 80 91  40 00 88 23 69 f0 28 2f  |.....$..@..#i.(/|
0010  30 e0 83 fd 03 c0 2d 7f  23 2b 31 f4 80 93 40 00  |0.....-.#+1...@.|
0020  8c bb 80 e0 af d0 ec c0  a8 95 22 9a 2a 9a 10 92  |..........".*...|
0030  01 02 80 91 82 12 83 70  81 30 61 f5 8c e5 90 e0  |.......p.0a.....|

avrdude> dump flash 0x200 0x40
>>> dump flash 0x200 0x40
0200  19 c0 34 c0 33 c0 32 c0  31 c0 30 c0 2f c0 2e c0  |..4.3.2.1.0./...|
0210  2d c0 2c c0 2b c0 2a c0  29 c0 28 c0 63 c0 26 c0  |-.,.+.*.).(.c.&.|
0220  25 c0 24 c0 23 c0 22 c0  21 c0 20 c0 1f c0 1e c0  |%.$.#.".!. .....|
0230  1d c0 1c c0 11 24 1f be  cf ef cd bf df e3 de bf  |.....$..........|

avrdude> dump flash 0x1dc0 0x40
>>> dump flash 0x1dc0 0x40
1dc0  ff ff ff ff ff ff ff 2e  2f 42 6c 69 6e 6b 5f 50  |......../Blink_P|
1dd0  72 6f 38 31 37 2e 69 6e  6f 2e 74 38 31 37 6f 70  |ro817.ino.t817op|
1de0  74 69 2e 32 30 63 30 2e  6d 44 30 2e 76 32 36 34  |ti.20c0.mD0.v264|
1df0  2e 68 65 78 00 e6 07 0b  1b 15 05 d4 03 f3 19 2e  |.hex... ........|

avrdude> quit
>>> quit
avrdude>

```

mcuee, 2022-11-27:
> > problem may be with avrdude.conf and not -c urclock.
> 
> Yes, I have pushed an update to avrdude.conf.in in PR #1198 (but `-c urclock` still assumes bootloaders are in top flash, so Dx parts are not currently working)

@stefanrueger 

Guess what, it works now. 👍 

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM6 -p avr128db48
 -xshowall -xbootsize=512

avrdude: AVR device initialized and ready to accept instructions
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 x0.0 ......... vector 0 (RESET) AVR128DB48

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM6 -p avr128db48 -xbootsize=512 -U .\Blink.ino.avr128db48opti.16c1.mB2.wO.v1410.hex
avrdude warning: attempt 1 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 2 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 3 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 4 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM6 -p avr128db48 -xbootsize=512
 -U .\Blink.ino.avr128db48opti.16c1.mB2.wO.v1410.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e970c (probably avr128db48)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink.ino.avr128db48opti.16c1.mB2.wO.v1410.hex for flash
         with 1040 bytes in 1 section within [0x200, 0x60f]
         using 3 pages and 496 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 1040 bytes flash ...

Writing | ################################################## | 100% 0.25 s

avrdude: 1040 bytes of flash written
avrdude: verifying flash memory against .\Blink.ino.avr128db48opti.16c1.mB2.wO.v1410.hex

Reading | ################################################## | 100% 0.19 s

avrdude: 1040 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM6 -p avr128db48
 -xshowall -xbootsize=512

avrdude: AVR device initialized and ready to accept instructions
0 2022-10-11 21.15 Blink.ino.avr128db48opti.16c1.mB2.wO.v1410.hex 1552 store 128946 meta 62 boot 512 
x0.0 ......... vector 0 (RESET) AVR128DB48

```

stefanrueger, 2022-11-27:
OK, understood. `-c urclock` will currently not work with any part that has the bootloader at the bottom of flash. It assumes the bootloader is in top flash. And it assumes the vector table is at the bottom of flash at address 0. So, only classic and xmega parts.

Use `-c urclock -xnometadata` or use `-c arduino` (which is ignorant about the bootloader and does not protect it from overwriting).

>  You can see that -c urclock works just fine.

Well it does in a sense, but **not correctly so:** `-c urclock` is being told the bootloader is 512 bytes long. So `-c urclock` *fiercly*  protects the top 512 bytes (where it thinks the bootloader is) and puts metadata just below it (if the size of the application allows it). Clearly that is wrong. It also assumes the reset vector is at address 0 and the IRS vectors thereafter. And it writes to the reset vector and writes to the dedicated jump to the application vector.

I realise there need changes to be made to `-c urclock` that are not too big *but* I need to read the data sheets first:
  - Is the bootloader in bottom flash in all `PM_UPDI` parts?
  - Does the vector table follow directly after the bootloader?

mcuee, 2022-11-27:
@MCUdude 

I think you should be able to answer the question from @stefanrueger 

- > Is the bootloader in bottom flash in all PM_UPDI parts?

I believe the answer is yes. Correct me if I am wrong.

- > Does the vector table follow directly after the bootloader

I believe the answer is yes. Correct me if I am wrong.

stefanrueger, 2022-11-27:
And
 - How does the compiler/linker know how big the bootloader is? [It needs to place the application thereafter and needs to resolve the symbols to absolute addresses]
 - Does avr-gcc support AVR Dx parts? I thought not.

mcuee, 2022-11-28:
> And
> 
> * How does the compiler/linker know how big the bootloader is? [It needs to place the application thereafter and needs to resolve the symbols to absolute addresses]

Not sure if @dl8dtl can help to answer your question or not.

> * Does avr-gcc support AVR Dx parts? I thought not.

At least DxCore works fine to build the blink sketch for my testing.

```
PS C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4b\avr\include\avr> ls ioavr*

    Directory: C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4b\avr\include\avr

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---            6/4/2021  7:10 pm         308268 ioavr128da28.h
-a---            6/4/2021  7:10 pm         310900 ioavr128da32.h
-a---            6/4/2021  7:10 pm         354404 ioavr128da48.h
-a---            6/4/2021  7:10 pm         366896 ioavr128da64.h
-a---            6/4/2021  7:10 pm         339634 ioavr128db28.h
-a---            6/4/2021  7:10 pm         341708 ioavr128db32.h
-a---            6/4/2021  7:10 pm         392395 ioavr128db48.h
-a---            6/4/2021  7:10 pm         401810 ioavr128db64.h
-a---            6/4/2021  7:10 pm         307794 ioavr32da28.h
-a---            6/4/2021  7:10 pm         310426 ioavr32da32.h
-a---            6/4/2021  7:10 pm         353930 ioavr32da48.h
-a---            6/4/2021  7:10 pm         339428 ioavr32db28.h
-a---            6/4/2021  7:10 pm         341502 ioavr32db32.h
-a---            6/4/2021  7:10 pm         392189 ioavr32db48.h
-a---            6/4/2021  7:10 pm         308062 ioavr64da28.h
-a---            6/4/2021  7:10 pm         310694 ioavr64da32.h
-a---            6/4/2021  7:10 pm         354198 ioavr64da48.h
-a---            6/4/2021  7:10 pm         366690 ioavr64da64.h
-a---            6/4/2021  7:10 pm         339428 ioavr64db28.h
-a---            6/4/2021  7:10 pm         341502 ioavr64db32.h
-a---            6/4/2021  7:10 pm         392189 ioavr64db48.h
-a---            6/4/2021  7:10 pm         401604 ioavr64db64.h
-a---            6/4/2021  7:10 pm         304067 ioavr64dd14.h
-a---            6/4/2021  7:10 pm         304859 ioavr64dd20.h
-a---            6/4/2021  7:10 pm         305960 ioavr64dd28.h
-a---            6/4/2021  7:10 pm         306488 ioavr64dd32.h

```

SpenceKonde, 2022-11-28:
Yeah avr-gcc works with modern AVRs - you just need to build the toolchain with the correct ATpacks (there is nothing magic about the AVR Dx-series as far as the compiler is concerned - the compiler doesn't provide any handling of flash mapping or anything - you have to implement anything like that that yourself. But thats a design decision microchip made, because they want to promote their XC8 compiler, which nobody else wants to come within 10 meters of because it's a closed source (hence inherently untrustworthy) toolchain. They recently added some features to support using the flash in mapped mode automatically there)
I build the toolchains (but not the executable binaries, as I get output that doesn't produce working .hex files - I stole the binaries from Arduino7 version of the toolchain, the last one they released over 2 years ago. I need to update the toolchain typically twice for each new product family they introduce; this is the latest version, same directory structure that the Arduino IDE uses, this is what gets installed when DxCore is installed through board manager.

https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino6-aarch64-pc-linux-gnu.tar.bz2
https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino6-arm-linux-gnueabihf.tar.bz2
https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino6-i686-pc-linux-gnu.tar.bz2
https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-apple-darwin14.tar.bz2
https://github.com/SpenceKonde/DxCore/raw/gh-pages/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2 <--- hosted on github so that the traffic involved in the CI testing remains internal)
https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino6-i686-w64-mingw32.tar.bz2

stefanrueger, 2022-11-28:
> The *compiler* needs to know about the bootloader's size when it compiles the application on modern AVRs

That's what I thought. And *how* does the compiler know? Which option does the user need when compiling blink.c? Linker options like `-Wl,-section-start=.text=0x200` that kind of stuff?

> modern AVRs

*How* can avrdude tell from `AVRPART *part` that it's a modern AVR? Is it those with UPDI i/face, ie,  `!!(part->prog_modes & PM_UPDI)`  or is it a subset of these? I suspect reset always goes to 0x0000 (ie, the bootloader if one is there). 

> This was not imo the best decision that went into the design of the AVR Dx-series.

So no one in the design team has asked the question *And how would the compiler know the start address of the application?*. I see poor design like this so often that I am convinced that a higher proportion of people in the IT industry would have failed the [Sally-Anne test](https://en.wikipedia.org/wiki/Sally%E2%80%93Anne_test) as children than the average population.

> actual test for entry conditions and jumping to app then is simply

Simply? I don't know. Optiboot's default conditions easily lead to WDT timeout reset loops, alternating between application/bootloader. It's the reason that arduino.c needs to sleep 250 ms before it resets the board b/c otherwise two successive avrdude calls don't work; it leads to a more complicated getsync routine than necessary etc. But hey, ho each to its own.

WestfW, 2022-11-28:
> The compiler needs to know about the bootloader's size when it compiles the application on modern AVRs

I don't like it either; I believe "the bootloader should be transparent and the application shouldn't even have to know  (at either compile or run time) whether there is a bootloder in use."  The old AVR scheme that allowed the bootloader to be in high memory was ideal.
Unfortunately, it seems to be a lost battle.  Most ARM chips also have their bootloaders at 0x0

I did come up with a scheme that lets the start address be specified in SOURCE CODE rather than linker switches. ( https://github.com/Optiboot/optiboot/blob/master/optiboot/bootloaders/optiboot/optiboot.c#L605 )
But it requires new linker scripts that use a symbol for __TEXT_REGION_START instead of hardwiring it to 0x0 (as is done in all of the current AVR linker scripts.  Sigh.)

stefanrueger, 2022-11-28:
> a lost battle

That's OK. Who knows what optimisation this buys the hardware developers; they *might* actually have a reason for this which we fail to see - I simply don't know.

> I did come up with a scheme that lets the start address be specified in SOURCE CODE rather than linker switches.

... of the bootloader (not the application), I assume. Mmh, yes, this is a perennial problem for bootloader writers to which the [urboot](https://github.com/stefanrueger/urboot) project found a different solution. That project uses a wrapper [`urboot-gcc`](https://github.com/stefanrueger/urboot/blob/main/src/urboot-gcc) that figures out the size of the compiled bootloader and *explicitly* tells the source with `-DSTART=...` where it is going to sit. More optimisation potential than resolving a linker symbol. This method can and will cause the size of the bootloader to change as the compiler may use the then compile-time constant for optimisation (eg, when protecting itself from being overwritten, different compile-time constant start & end address lend themselves to different compile-time optimisation), so there might be a couple of rounds of compilation before this converges.

mcuee, 2022-11-28:
> Unfortunately, it seems to be a lost battle. Most ARM chips also have their bootloaders at 0x0

It seems to me many ARM Cortex MCUs will have factory installed bootloader (some USB AVRs have factory installed bootloader as well). 

STM32: factory installed bootloader, activated by boot pins
https://www.st.com/resource/en/application_note/cd00167594-stm32-microcontroller-system-memory-boot-mode-stmicroelectronics.pdf

NXP: many have factory installed bootloader
https://www.nxp.com/design/software/development-software/mcuxpresso-software-and-tools-/mcu-bootloader-for-nxp-microcontrollers:MCUBOOT

Microchip/Atmel SAM-BA, usually reside in ROM monitor, but some need to be in flash (eg: SAM D21)
https://www.microchip.com/en-us/development-tool/SAM-BA-In-system-Programmer

mcuee, 2022-12-09:
> > The _compiler_ needs to know about the bootloader's size when it compiles the application on modern AVRs
> 
> That's what I thought. And _how_ does the compiler know? Which option does the user need when compiling blink.c? Linker options like `-Wl,-section-start=.text=0x200` that kind of stuff?

 Yes, I believe that is the way for MegaCoreX, megaTinyCore and DxCore.

Example from megaTinyCore
```
Linking everything together...
"C:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-gcc" 
-Wall -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -Wl,--section-start=.text=0x200 -mmcu=attiny817 
-o "C:\\Users\\xiaof\\AppData\\Local\\Temp\\arduino_build_803015/Blink_Pro817.ino.elf" 
```

mcuee, 2022-12-09:
>> modern AVRs
>>
> How can avrdude tell from AVRPART *part that it's a modern AVR? Is it those with UPDI i/face, ie, !!(part->prog_modes & PM_UPDI) or is it a subset of these? I suspect reset always goes to 0x0000 (ie, the bootloader if one is there).

I believe that should be fine, basically Mega0/1 (optiboot_x), Tiny0/1/2 (optiboot_x) and AVR DA/DB/DD (optiboot_dx) as of now.

closed 2022-12-14 by stefanrueger