avrdudes/avrdude #1201: Terminal does not continue read/dump correctly

stefanrueger, 2022-11-28:
Try
```
$ avrdude -t

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude> d fl 0x7f00 32

Reading | ################################################## | 100% 0.03 s 

7f00  80 bd fa 9a f9 9a 31 96  da 13 f4 cf 23 c0 a0 e0  |......1.....#...|
7f10  b1 e0 4d d0 1f c0 c3 30  41 f4 33 d0 c8 2f 3d d0  |..M....0A.3../=.|

avrdude> d fl

Reading | ################################################## | 100% 0.01 s 

0000  3f cf ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |?...............|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> d 
Usage: d <memory> <addr> <len>
       d <memory> <addr> ...
       d <memory> <addr>
       d <memory> ...
       d <memory>

```

I would expect the second command to continue where the first command left off. This is a minor bug that I wanted to fix before 7.1 but I now won't have the time to do.

Here an idea for an extension: when using the `read`/`dump` without arguments it would be great if the terminal could dump the next page of the last memory read with the size the user had last specified for this memory: if the user specifies `d ee` it should continue reading the last eeprom read, again with the eeprom size specified. An ideal implementation would be able to alternate between two memories and continue where the last memory left off with the last used page size for that memory:
```
avrdude> read application 0x3000 32
avrdude> read boot 0x0000 32
avrdude> r app   <- shows the next 32 bytes where the last read application left off
avrdude> r boo   <- shows the next 32 bytes where the last read boot left off
avrdude> r ap
avrdude> r bo
avrdude> r       <- continues with the last read
```

mcuee, 2022-11-29:
@stefanrueger 
I agree that this is a good extension. I also agree that this is just a minor issue and we can delay the enhancement post 7.1 release if you do not have time.

stefanrueger, 2022-12-01:
Maybe @MCUdude can have a look?

MCUdude, 2022-12-01:
I took a quick look, and it appears that it's a very simple fix. I've barely tested it.

https://github.com/avrdudes/avrdude/blob/ea2d7eec95b24d2b9d8f969c0e0f60c88b1b491d/src/term.c#L283-L292

It (almost) works as @stefanrueger describes by removing:

```c
  addr = 0; 
  len  = read_size;
```

However, Avrdude doesn't keep separate addresses, and no. bytes to read from each memory. But it's a start!

```
$ ./avrdude -cxplainedmini_tpi -pattiny40 -t

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e920e (probably t40)
avrdude> read flash
>>> read flash 

Reading | ################################################## | 100% 0.79 s 

0000  48 65 6c 6c 6f 20 57 6f  72 6c 64 00 ff ff ff ff  |Hello World.....|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> read flash 0x10 0x40
>>> read flash 0x10 0x40 

Reading | ################################################## | 100% 0.00 s 

0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> read flash 
>>> read flash 

Reading | ################################################## | 100% 0.00 s 

0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> read flash
>>> read flash 

Reading | ################################################## | 100% 0.00 s 

0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> read flash
>>> read flash 

Reading | ################################################## | 100% 0.20 s 

00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0100  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> quit
>>> quit 
avrdude> 
avrdude done.  Thank you.
```

MCUdude, 2022-12-02:
I got something that works now. Would be great if you could give it a try @stefanrueger and @mcuee.

https://github.com/MCUdude/avrdude/tree/terminal-read

There's one thing I'm not so thrilled about is that I couldn't quickly figure out how to solve. At the moment there is a hard-coded limit of 32 memories you can read from. It would be awesome if this number matches the actual number of memories a target has, but I can't malloc and free every time `cmd_dump` is run, since data needs to be preserved between each reading.

``` c
  // Preserve memory start address and length info for every readable memory
  static struct mem_addr_len {
    int addr;
    int len;
    AVRMEM *mem;
  } read_mem[32]; // Can the number of memories be determined at runtime?

  // Iterate though the structs to find relevant address and length info
  int i;
  for(i = 0; i < 32; i++) {
```

```
$ ./avrdude -pavr128db48 -cpkobn_updi -t

         Vtarget                      : 3.31 V
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e970c (probably avr128db48)
avrdude> read flash
>>> read flash 

Reading | ################################################## | 100% 0.09 s 

0000  47 c0 00 00 55 c0 00 00  53 c0 00 00 51 c0 00 00  |G...U...S...Q...|
0010  4f c0 00 00 4d c0 00 00  4b c0 00 00 49 c0 00 00  |O...M...K...I...|
0020  47 c0 00 00 45 c0 00 00  43 c0 00 00 41 c0 00 00  |G...E...C...A...|
0030  3f c0 00 00 3d c0 00 00  3b c0 00 00 39 c0 00 00  |?...=...;...9...|
0040  37 c0 00 00 35 c0 00 00  33 c0 00 00 31 c0 00 00  |7...5...3...1...|
0050  2f c0 00 00 2d c0 00 00  2b c0 00 00 29 c0 00 00  |/...-...+...)...|
0060  27 c0 00 00 97 c0 00 00  23 c0 00 00 21 c0 00 00  |'.......#...!...|
0070  1f c0 00 00 1d c0 00 00  1b c0 00 00 19 c0 00 00  |................|
0080  17 c0 00 00 15 c0 00 00  13 c0 00 00 11 c0 00 00  |................|
0090  11 24 1f be cf ef cd bf  df e3 de bf 28 e2 a0 e0  |.$..........(...|
00a0  b8 e2 01 c0 1d 92 a4 30  b2 07 e1 f7 9b d0 26 c1  |.......0......&.|
00b0  a7 cf 90 91 01 0a 9f 7e  90 93 01 0a 90 91 60 04  |..... .~... ..`.|
00c0  93 ff 0a c0 81 11 04 c0  88 e0 80 93 66 04 08 95  |.. .........f...|
00d0  88 e0 80 93 65 04 08 95  9f b7 f8 94 81 11 07 c0  |....e...........|
00e0  80 91 73 04 87 7f 80 93  73 04 9f bf 08 95 80 91  |..s.....s.......|
00f0  73 04 88 60 f8 cf 8f b7  f8 94 20 91 00 28 30 91  |s..`...... ..(0.|

avrdude> read flash 0 0x20
>>> read flash 0 0x20 

Reading | ################################################## | 100% 0.00 s 

0000  47 c0 00 00 55 c0 00 00  53 c0 00 00 51 c0 00 00  |G...U...S...Q...|
0010  4f c0 00 00 4d c0 00 00  4b c0 00 00 49 c0 00 00  |O...M...K...I...|

avrdude> read flash 
>>> read flash 

Reading | ################################################## | 100% 0.00 s 

0020  47 c0 00 00 45 c0 00 00  43 c0 00 00 41 c0 00 00  |G...E...C...A...|
0030  3f c0 00 00 3d c0 00 00  3b c0 00 00 39 c0 00 00  |?...=...;...9...|

avrdude> read eeprom 0 0x40
>>> read eeprom 0 0x40 

Reading | ################################################## | 100% 0.26 s 

0000  48 65 6c 6c 6f 20 57 6f  72 6c 64 20 45 45 50 52  |Hello World EEPR|
0010  4f 4d 20 77 72 69 74 65  20 74 65 73 74 00 ff ff  |OM write test...|
0020  41 56 52 44 55 44 45 20  2d 20 41 56 52 20 44 6f  |AVRDUDE - AVR Do|
0030  77 6e 6c 6f 61 64 65 72  20 55 70 6c 6f 61 64 65  |wnloader Uploade|

avrdude> read flash
>>> read flash 

Reading | ################################################## | 100% 0.00 s 

0040  37 c0 00 00 35 c0 00 00  33 c0 00 00 31 c0 00 00  |7...5...3...1...|
0050  2f c0 00 00 2d c0 00 00  2b c0 00 00 29 c0 00 00  |/...-...+...)...|

avrdude> read eeprom
>>> read eeprom 

Reading | ################################################## | 100% 0.26 s 

0040  72 20 2d 20 69 73 20 61  20 70 72 6f 67 72 61 6d  |r - is a program|
0050  20 66 6f 72 20 64 6f 77  6e 6c 6f 61 64 69 6e 67  | for downloading|
0060  20 61 6e 64 20 75 70 6c  6f 61 64 69 6e 67 20 74  | and uploading t|
0070  68 65 20 6f 6e 2d 63 68  69 70 20 6d 65 6d 6f 72  |he on-chip memor|

avrdude> read flash
>>> read flash 

Reading | ################################################## | 100% 0.00 s 

0060  27 c0 00 00 97 c0 00 00  23 c0 00 00 21 c0 00 00  |'.......#...!...|
0070  1f c0 00 00 1d c0 00 00  1b c0 00 00 19 c0 00 00  |................|

avrdude> read eeprom
>>> read eeprom 

Reading | ################################################## | 100% 0.26 s 

0080  69 65 73 20 6f 66 20 4d  69 63 72 6f 63 68 69 70  |ies of Microchip|
0090  e2 80 99 73 20 41 56 52  20 6d 69 63 72 6f 63 6f  |...s AVR microco|
00a0  6e 74 72 6f 6c 6c 65 72  73 2e 20 49 74 20 63 61  |ntrollers. It ca|
00b0  6e 20 70 72 6f 67 72 61  6d 20 74 68 65 20 46 6c  |n program the Fl|

avrdude> read flash
>>> read flash 

Reading | ################################################## | 100% 0.00 s 

0080  17 c0 00 00 15 c0 00 00  13 c0 00 00 11 c0 00 00  |................|
0090  11 24 1f be cf ef cd bf  df e3 de bf 28 e2 a0 e0  |.$..........(...|

avrdude> read eeprom
>>> read eeprom 

Reading | ################################################## | 100% 0.26 s 

00c0  61 73 68 20 61 6e 64 20  45 45 50 52 4f 4d 2c 20  |ash and EEPROM, |
00d0  61 6e 64 20 77 68 65 72  65 20 73 75 70 70 6f 72  |and where suppor|
00e0  74 65 64 20 62 79 20 74  68 65 20 70 72 6f 67 72  |ted by the progr|
00f0  61 6d 6d 69 6e 67 20 70  72 6f 74 6f 63 6f 6c 2c  |amming protocol,|

avrdude> quit
>>> quit 
avrdude> 
avrdude done.  Thank you.
```

MCUdude, 2022-12-02:
Alternatively, I can do this, but then there's no place to call `free()`. What to do?

```c
  // Preserve memory start address and length info for every readable memory
  struct mem_addr_len {
    int addr;
    int len;
    AVRMEM *mem;
  };
  static struct mem_addr_len* read_mem;
  if (!read_mem) {
    int mems = 0;
    for (LNODEID ln=lfirst(p->mem); ln; ln=lnext(ln))
      mems++;
    read_mem = calloc(mems, sizeof(struct mem_addr_len));
  }
```

mcuee, 2022-12-03:
> 
As far as testing and edge cases go, I'd try to see what happens 
 - In small memories, eg, `signature`
 - Near the end of a memory, eg, you look at flash in chunks of 0x80 and come to the end of the memory in the middle of a chunk (r fl 0x7f40 0x80; r fl; r fl) for the m328p.  Is that intuitive?
 - What happens with `read` alone; does it carry on with the last memory used?

@stefanrueger 
Nice test cases. It seems to work fine.

I think it is okay that `read` along only prints out the usage.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1201 -c urclock -P COM3 -p m328p -qt

avrdude_issue1201: AVR device initialized and ready to accept instructions
avrdude_issue1201: device signature = 0x1e950f (probably m328p)
avrdude> read flash
>>> read flash
0000  0c 94 5c 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .\. .n. .n. .n.|
0010  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|
0020  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|
0030  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|
0040  0c 94 13 01 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | ... .n. .n. .n.|
0050  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|
0060  0c 94 6e 00 0c 94 6e 00  00 00 00 00 24 00 27 00  | .n. .n.....$.'.|
0070  2a 00 00 00 00 00 25 00  28 00 2b 00 04 04 04 04  |*.....%.(.+.....|
0080  04 04 04 04 02 02 02 02  02 02 03 03 03 03 03 03  |................|
0090  01 02 04 08 10 20 40 80  01 02 04 08 10 20 01 02  |..... @...... ..|
00a0  04 08 10 20 00 00 00 08  00 02 01 00 00 03 04 07  |... ............|
00b0  00 00 00 00 00 00 00 00  11 24 1f be cf ef d8 e0  |.........$......|
00c0  de bf cd bf 21 e0 a0 e0  b1 e0 01 c0 1d 92 a9 30  |....!..........0|
00d0  b2 07 e1 f7 0e 94 5d 01  0c 94 cc 01 0c 94 00 00  |......]. ... ...|
00e0  e1 eb f0 e0 24 91 ed e9  f0 e0 94 91 e9 e8 f0 e0  |....$...........|
00f0  e4 91 ee 23 c9 f0 22 23  39 f0 23 30 01 f1 a8 f4  |...#.."#9.#0....|

avrdude> read flash 0 0x20
>>> read flash 0 0x20
0000  0c 94 5c 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .\. .n. .n. .n.|
0010  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|

avrdude> read flash
>>> read flash
0020  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|
0030  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|

avrdude> read eeprom 0 0x40
>>> read eeprom 0 0x40
avrdude_issue1201 error: bootloader does not seem to have paged EEPROM read capability
avrdude_issue1201 error: bootloader does not seem to have EEPROM access capability
avrdude_issue1201 error: unable to read eeprom page at addr 0x0000
avrdude_issue1201 error: (dump) error reading eeprom address 0x00000 of part ATmega328P
                         read operation not supported on memory type eeprom
avrdude> read flash
>>> read flash
0040  0c 94 13 01 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | ... .n. .n. .n.|
0050  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|

avrdude> sig
>>> sig
Device signature = 0x1e950f

avrdude> read signature 0
>>> read signature 0
0000  1e 95 0f                                          |...             |

avrdude> read signature 0 0x01
>>> read signature 0 0x01
0000  1e                                                |.               |

avrdude> read signature
>>> read signature
0001  95                                                |.               |

avrdude> read signature
>>> read signature
0002  0f                                                |.               |

avrdude> read flash 0x7f40 0x80
>>> read flash 0x7f40 0x80
7f40  f1 10 f9 cf 0e c0 85 37  39 f4 27 d0 8e e1 0c d0  |.......79.'... .|
7f50  85 e9 0a d0 8f e0 9c cf  81 35 11 f4 88 e0 17 d0  |.. ......5......|
7f60  1c d0 80 e1 01 d0 89 cf  90 91 c0 00 95 ff fc cf  |................|
7f70  80 93 c6 00 08 95 80 91  c0 00 87 ff fc cf 80 91  |................|
7f80  c0 00 84 fd 01 c0 a8 95  80 91 c6 00 08 95 e0 e6  |................|
7f90  f0 e0 98 e1 90 83 80 83  08 95 ed df 80 32 19 f0  |.............2..|
7fa0  88 e0 f5 df ff cf 84 e1  df cf cf 93 c8 2f e3 df  |............./..|
7fb0  c1 50 e9 f7 cf 91 f1 cf  fc 01 0a 01 67 bf e8 95  |.P........ .g...|

avrdude> read flash
>>> read flash
7fc0  11 24 07 b6 00 fc fd cf  66 70 29 f0 45 2b 19 f4  |.$......fp).E+..|
7fd0  81 e1 87 bf e8 95 08 95  ff ff ff ff ff ff ff ff  |................|
7fe0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7ff0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff 03 08  |................|

avrdude> read flash
>>> read flash
0000  0c 94 5c 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .\. .n. .n. .n.|
0010  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|
0020  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|
0030  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|

avrdude> read
>>> read
Usage: read <memory> <addr> <len>
       read <memory> <addr> ...
       read <memory> <addr>
       read <memory> ...
       read <memory>
avrdude> quit
>>> quit
avrdude>
avrdude_issue1201 done.  Thank you.

```

stefanrueger, 2022-12-03:
> I think it is okay that read alone only prints out the usage.

That's a matter of taste. If I want to step through memory regions, I'd probably prefer the shortcut `r` for "carry on"; either way is fine with me, though (seeing it would be an enhancement). The first `read` alone before any memory was addressed should show usage, though.

Wrapping around memory end works nicely, that's neat. I personally would have expected that the second `read flash` uses the last user-specified chunk size of 0x80 rather than how many bytes were left at the end of memory previously.

MCUdude, 2022-12-03:
> If you don't want to limit the code to 32, you might consider a growing array using something to the effect of
  `read_mem = cfg_realloc(__func__, read_mem, ++n_mems*sizeof*read_mem);`
keeping a tally in n_mems of how many memories the code currently caters for. You can then free(read_mem) when the terminal ends, but it doesn't matter if you don't: it's small fry and not recurring, so not a memory leak.

Why would I do this when I can use `calloc` directly? And when looking at the `cfg_realloc` function, it seems as `realloc` would always be used since `ret = p` is always true.

https://github.com/avrdudes/avrdude/blob/ea2d7eec95b24d2b9d8f969c0e0f60c88b1b491d/src/config.c#L123-L132

MCUdude, 2022-12-03:
> If I want to step through memory regions, I'd probably prefer the shortcut r for "carry on"; either way is fine with me, though (seeing it would be an enhancement). The first read alone before any memory was addressed should show usage, though.

I've just pushed a commit that lets you use read/dump alone

```
$ ./avrdude -pavr128db48 -cpkobn_updi -t

         Vtarget                      : 3.31 V
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e970c (probably avr128db48)
avrdude> read
>>> read 
Usage: read <memory> <addr> <len>
       read <memory> <addr> ...
       read <memory> <addr>
       read <memory> ...
       read <memory>
avrdude> read flash 0 0x10
>>> read flash 0 0x10 

Reading | ################################################## | 100% 0.09 s 

0000  47 c0 00 00 55 c0 00 00  53 c0 00 00 51 c0 00 00  |G...U...S...Q...|

avrdude> read eeprom 0 0x20
>>> read eeprom 0 0x20 

Reading | ################################################## | 100% 0.13 s 

0000  48 65 6c 6c 6f 20 57 6f  72 6c 64 20 45 45 50 52  |Hello World EEPR|
0010  4f 4d 20 77 72 69 74 65  20 74 65 73 74 00 ff ff  |OM write test...|

avrdude> r
>>> r 

Reading | ################################################## | 100% 0.13 s 

0020  41 56 52 44 55 44 45 20  2d 20 41 56 52 20 44 6f  |AVRDUDE - AVR Do|
0030  77 6e 6c 6f 61 64 65 72  20 55 70 6c 6f 61 64 65  |wnloader Uploade|

avrdude> r
>>> r 

Reading | ################################################## | 100% 0.13 s 

0040  72 20 2d 20 69 73 20 61  20 70 72 6f 67 72 61 6d  |r - is a program|
0050  20 66 6f 72 20 64 6f 77  6e 6c 6f 61 64 69 6e 67  | for downloading|

avrdude> read flash
>>> read flash 

Reading | ################################################## | 100% 0.00 s 

0010  4f c0 00 00 4d c0 00 00  4b c0 00 00 49 c0 00 00  |O...M...K...I...|

avrdude> r
>>> r 

Reading | ################################################## | 100% 0.00 s 

0020  47 c0 00 00 45 c0 00 00  43 c0 00 00 41 c0 00 00  |G...E...C...A...|

avrdude> r
>>> r 

Reading | ################################################## | 100% 0.00 s 

0030  3f c0 00 00 3d c0 00 00  3b c0 00 00 39 c0 00 00  |?...=...;...9...|

[... 73 lines cut ...]
01d0  74 75 72 65 20 6f 66 20  61 20 70 61 72 74 69 63  |ture of a partic|
01e0  75 6c 61 72 20 63 68 69  70 2e 00 ff ff ff ff ff  |ular chip.......|
01f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0200  00 65 6c 6c 6f 20 57 6f  72 6c 64 20 45 45 50 52  |.ello World EEPR|
0210  4f 4d 20 77 72 69 74 65  20 74 65 73 74 00 ff ff  |OM write test...|
0220  41 56 52 44 55 44 45 20  2d 20 41 56 52 20 44 6f  |AVRDUDE - AVR Do|
0230  77 6e 6c 6f 61 64 65 72  20 55 70 6c 6f 61 64 65  |wnloader Uploade|
0240  72 20 2d 20 69 73 20 61  20 70 72 6f 67 72 61 6d  |r - is a program|
0250  20 66 6f 72 20 64 6f 77  6e 6c 6f 61 64 69 6e 67  | for downloading|
0260  20 61 6e 64 20 75 70 6c  6f 61 64 69 6e 67 20 74  | and uploading t|
0270  68 65 20 6f 6e 2d 63 68  69 70 20 6d 65 6d 6f 72  |he on-chip memor|
0280  69 65 73 20 6f 66 20 4d  69 63 72 6f 63 68 69 70  |ies of Microchip|
0290  e2 80 99 73 20 41 56 52  20 6d 69 63 72 6f 63 6f  |...s AVR microco|
02a0  6e 74 72 6f 6c 6c 65 72  73 2e 20 49 74 20 63 61  |ntrollers. It ca|
02b0  6e 20 70 72 6f 67 72 61  6d 20 74 68 65 20 46 6c  |n program the Fl|
02c0  61 73 68 20 61 6e 64 20  45 45 50 52 4f 4d 2c 20  |ash and EEPROM, |
02d0  61 6e 64 20 77 68 65 72  65 20 73 75 70 70 6f 72  |and where suppor|
02e0  74 65 64 20 62 79 20 74  68 65 20 70 72 6f 67 72  |ted by the progr|
02f0  61 6d 6d 69 6e 67 20 70  72 6f 74 6f 63 6f 6c 2c  |amming protocol,|

avrdude> quit
>>> quit 
avrdude> 
avrdude done.  Thank you.
```

MCUdude, 2022-12-03:
> Although it is possible to wrap around and continue reading from address 0, the problem is that hexdump_buf doesn't handle this. Look what happens when I read past EEPROM address 0x200 on the AVR128DB48. I'm not sure how to solve this in an elegant way.

I think I came up with an OK solution to this problem, and I've just pushed this fix. Please test and review!

One issue I'd like to discuss is what to di with the misalignment when the memory rolls over:

```
$ ./avrdude -pavr128db48 -cpkobn_updi -t -B0.5

         Vtarget                      : 3.31 V
         PDI/UPDI clock Xmega/megaAVR : 2000 kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e970c (probably avr128db48)
avrdude> read flash 0x1ffde 0x100
>>> read flash 0x1ffde 0x100 

Reading | ################################################## | 100% 0.06 s 

1ffde  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ffee  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1fffe  ff ff                                             |..              |
0000  47 c0 00 00 55 c0 00 00  53 c0 00 00 51 c0 00 00  |G...U...S...Q...|
0010  4f c0 00 00 4d c0 00 00  4b c0 00 00 49 c0 00 00  |O...M...K...I...|
0020  47 c0 00 00 45 c0 00 00  43 c0 00 00 41 c0 00 00  |G...E...C...A...|
0030  3f c0 00 00 3d c0 00 00  3b c0 00 00 39 c0 00 00  |?...=...;...9...|
0040  37 c0 00 00 35 c0 00 00  33 c0 00 00 31 c0 00 00  |7...5...3...1...|
0050  2f c0 00 00 2d c0 00 00  2b c0 00 00 29 c0 00 00  |/...-...+...)...|
0060  27 c0 00 00 97 c0 00 00  23 c0 00 00 21 c0 00 00  |'.......#...!...|
0070  1f c0 00 00 1d c0 00 00  1b c0 00 00 19 c0 00 00  |................|
0080  17 c0 00 00 15 c0 00 00  13 c0 00 00 11 c0 00 00  |................|
0090  11 24 1f be cf ef cd bf  df e3 de bf 28 e2 a0 e0  |.$..........(...|
00a0  b8 e2 01 c0 1d 92 a4 30  b2 07 e1 f7 9b d0 26 c1  |.......0......&.|
00b0  a7 cf 90 91 01 0a 9f 7e  90 93 01 0a 90 91 60 04  |..... .~... ..`.|
00c0  93 ff 0a c0 81 11 04 c0  88 e0 80 93 66 04 08 95  |.. .........f...|
00d0  88 e0 80 93 65 04 08 95  9f b7 f8 94 81 11        |....e.........  |

avrdude> quit
>>> quit 
avrdude> 
avrdude done.  Thank you.
```

One alternative is "dynamic padding", where a space is added when the memory address crosses 0xffff:

```
$ ./avrdude -pavr128db48 -cpkobn_updi -t -B0.5

         Vtarget                      : 3.31 V
         PDI/UPDI clock Xmega/megaAVR : 2000 kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e970c (probably avr128db48)
avrdude> read flash 0 0x80
>>> read flash 0 0x80 

Reading | ################################################## | 100% 0.03 s 

0000  47 c0 00 00 55 c0 00 00  53 c0 00 00 51 c0 00 00  |G...U...S...Q...|
0010  4f c0 00 00 4d c0 00 00  4b c0 00 00 49 c0 00 00  |O...M...K...I...|
0020  47 c0 00 00 45 c0 00 00  43 c0 00 00 41 c0 00 00  |G...E...C...A...|
0030  3f c0 00 00 3d c0 00 00  3b c0 00 00 39 c0 00 00  |?...=...;...9...|
0040  37 c0 00 00 35 c0 00 00  33 c0 00 00 31 c0 00 00  |7...5...3...1...|
0050  2f c0 00 00 2d c0 00 00  2b c0 00 00 29 c0 00 00  |/...-...+...)...|
0060  27 c0 00 00 97 c0 00 00  23 c0 00 00 21 c0 00 00  |'.......#...!...|
0070  1f c0 00 00 1d c0 00 00  1b c0 00 00 19 c0 00 00  |................|

avrdude> read flash 0x1ff00 0x200 
>>> read flash 0x1ff00 0x200 

Reading | ################################################## | 100% 0.03 s 

1ff00  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ff10  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ff20  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ff30  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ff40  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ff50  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ff60  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ff70  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ff80  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ff90  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ffa0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ffb0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ffc0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ffd0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ffe0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1fff0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
 0000  47 c0 00 00 55 c0 00 00  53 c0 00 00 51 c0 00 00  |G...U...S...Q...|
 0010  4f c0 00 00 4d c0 00 00  4b c0 00 00 49 c0 00 00  |O...M...K...I...|
 0020  47 c0 00 00 45 c0 00 00  43 c0 00 00 41 c0 00 00  |G...E...C...A...|
 0030  3f c0 00 00 3d c0 00 00  3b c0 00 00 39 c0 00 00  |?...=...;...9...|
 0040  37 c0 00 00 35 c0 00 00  33 c0 00 00 31 c0 00 00  |7...5...3...1...|
 0050  2f c0 00 00 2d c0 00 00  2b c0 00 00 29 c0 00 00  |/...-...+...)...|
 0060  27 c0 00 00 97 c0 00 00  23 c0 00 00 21 c0 00 00  |'.......#...!...|
 0070  1f c0 00 00 1d c0 00 00  1b c0 00 00 19 c0 00 00  |................|
 0080  17 c0 00 00 15 c0 00 00  13 c0 00 00 11 c0 00 00  |................|
 0090  11 24 1f be cf ef cd bf  df e3 de bf 28 e2 a0 e0  |.$..........(...|
 00a0  b8 e2 01 c0 1d 92 a4 30  b2 07 e1 f7 9b d0 26 c1  |.......0......&.|
 00b0  a7 cf 90 91 01 0a 9f 7e  90 93 01 0a 90 91 60 04  |..... .~... ..`.|
 00c0  93 ff 0a c0 81 11 04 c0  88 e0 80 93 66 04 08 95  |.. .........f...|
 00d0  88 e0 80 93 65 04 08 95  9f b7 f8 94 81 11 07 c0  |....e...........|
 00e0  80 91 73 04 87 7f 80 93  73 04 9f bf 08 95 80 91  |..s.....s.......|
 00f0  73 04 88 60 f8 cf 8f b7  f8 94 20 91 00 28 30 91  |s..`...... ..(0.|

avrdude> quit
>>> quit 
avrdude> 
avrdude done.  Thank you.
```

stefanrueger, 2022-12-03:
> pushed a commit that lets you use read/dump alone

neat

> realloc

Three ways to cater for the memories list:
  1. Fixed array (always uses 32*16 global bytes)
  2. Count all existing n memories and allocate n*16 bytes at runtime - don't free
  3. Grow the array of memories every time a new unseen memory is *used*  (12 bytes+ k*16 bytes with k uses)
      - Declare static outside functions
        ```
        static struct ... *read_mems = NULL;
        static int n_mems = 0;
        ```
      - Find and/or enter memory mem with associated data
         ```
         int findEnter(AVRMEM *mem) {
           for(int i=0; i<n_mems; i++)
              if(read_mems[i].mem == mem)
                 return i;
           // Not found, expand array (address read_mem may change), increment n_mems
           read_mem = cfg_realloc(__func__, read_mem, ++n_mems*sizeof*read_mem);
           read_mem[n_mems-1].mem = mem;
           // Initialise data associated to mem ...
          return n_mems-1;
         }
         ```
      - Use in dump_cmd()
        ```
        int k = findEnter(mem); 
        last_addr = read_mems[k].addr; // ...
        ```

> ret = p is always true.

No:
```
ret = p? realloc(p, n): calloc(1, n)
```
If p is NULL, eg, on first call, then calloc is used otherwise realloc (could have just used realloc, but that uses malloc on NULL).

>  possible to wrap around and continue reading from address 0

Not necessary, you spotted the mis-alignment that might happen. I only meant to say if a user gets towards the end of the memory and the last printed page was less than the user had originally selected then when starting from 0 it should be the old user-selected size not whatever was left at end of memory. Example:
```
avrdude> read flash 0x7f40 0x80
>>> read flash 0x7f40 0x80
7f40  f1 10 f9 cf 0e c0 85 37  39 f4 27 d0 8e e1 0c d0  |.......79.'... .|
7f50  85 e9 0a d0 8f e0 9c cf  81 35 11 f4 88 e0 17 d0  |.. ......5......|
7f60  1c d0 80 e1 01 d0 89 cf  90 91 c0 00 95 ff fc cf  |................|
7f70  80 93 c6 00 08 95 80 91  c0 00 87 ff fc cf 80 91  |................|
7f80  c0 00 84 fd 01 c0 a8 95  80 91 c6 00 08 95 e0 e6  |................|
7f90  f0 e0 98 e1 90 83 80 83  08 95 ed df 80 32 19 f0  |.............2..|
7fa0  88 e0 f5 df ff cf 84 e1  df cf cf 93 c8 2f e3 df  |............./..|
7fb0  c1 50 e9 f7 cf 91 f1 cf  fc 01 0a 01 67 bf e8 95  |.P........ .g...|

avrdude> read flash
>>> read flash
7fc0  11 24 07 b6 00 fc fd cf  66 70 29 f0 45 2b 19 f4  |.$......fp).E+..|
7fd0  81 e1 87 bf e8 95 08 95  ff ff ff ff ff ff ff ff  |................|
7fe0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7ff0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff 03 08  |................|

avrdude> read flash
>>> read flash
0000  0c 94 5c 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .\. .n. .n. .n.|
0010  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|
0020  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|
0030  0c 94 6e 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .n. .n. .n. .n.|

avrdude> read
```

User chose a page size of 0x80; memory comes to an end; last memory page is only 0x40 data (good); next read will continue at 0 (good); but it should show 0x80 bytes from 0 (the user selected size) not 0x40 (what happened to be left over at end of memory

stefanrueger, 2022-12-03:
> dynamic padding

Probably best to assume addresses are either 4 or 5 hex digits (4 when memory size id 64 kB or less, 5 if more).

Crossing posts; ahh - I see you have turned the memory into a circle, probably displaying data[addr % size]; that's very  neat.

MCUdude, 2022-12-03:
> Probably best to assume addresses are either 4 or 5 hex digits (4 when memory size is 64 kB or less, 5 if more). There is
printf("%0*x", size > 65536? 5: 4, addr % size)

Thanks for the input. I've added a zero if the memory that's being read is greater than 64kiB:
`term_out("%0*x  %s  |%s|\n", m->size > 0xffff ? 5: 4, addr, dst1, dst2);`

> realloc

I think I'll end up using statically allocated memory instead, and limit the number of memories to 32. This makes the code much easier to read, there are fewer lines of code, and there's no memory that should be freed.

stefanrueger, 2022-12-03:
>  using statically allocated memory 

Great decision. What I sometimes do is to output an error at runtime in the vvv unlikely case the array size turned out to be insufficient, eg, https://github.com/avrdudes/avrdude/blob/ea2d7eec95b24d2b9d8f969c0e0f60c88b1b491d/src/avr.c#L1341-L1352 Not the prettiest, but there you go.

MCUdude, 2022-12-03:
Should we allow users to read the same memory address more than once in a single command, or should the number os bytes be capped by the size of the memory?

```
avrdude> read sig 0 0x10
>>> read sig 0 0x10 

Reading | ################################################## | 100% 0.02 s 

0000  1e 97 0c                                          |..              |
0000  1e 97 0c 1e 97 0c 1e 97  0c 1e 97 0c 1e           |.. .. .. .. .   |
```

```
>>> read sig 0 0x10 

Reading | ################################################## | 100% 0.00 s 

0000  1e 97 0c                                          |..              |
```

stefanrueger, 2022-12-03:
I like the capped version more (also gives the user a sense of an end of memory that they wouldn't get otherwise)

MCUdude, 2022-12-03:
> I like the capped version more (also gives the user a sense of an end of memory that they wouldn't get otherwise)

Thanks, I agree. I've pushed a fix for this as well.

I think most of your wishes have been granted. If @mcuee would stress test my branch a little, it may even be ready to be submitted as a PR.

Please let me know if there are other things I should look into regarding the terminal read command. Do the docs needs to be updated if by branch ends up as a PR?

Off-topic terminal-related question to @stefanrueger: I realize that #1180 is probably a lot of work and will absolutely not make it into Avrdude 7.1. But is a feature like this even realistic/possible to implement in terminal interactive mode? Wouldn't this require a terminal command similar to `ls`, so files and folders can be browsed in terminal mode? Even though #1180 would be an incredible feature, it sounds to me like a lot of work and a potential can of worms.

stefanrueger, 2022-12-04:
Thanks for all the hard work, @MCUdude! I may not be able to review until Weds, though, possibly not until next w/end. 

Yes, now that one can write to flash from the terminal #1180 is a natural extension. Absolutely makes sense. And it may be less work than it looks. After all, there are a lot of useful functions in fileio.c and update.c. Would be a cool thing for v8.0. As to `ls`, these days it's easy to switch windows on the desktop and do the `ls` elsewhere. The 90s way would have been to implement a command `!` in term.c that calls `system()`:
```
avrdude> !ls *hex
blink-mega2560+lext-test.hex  rsketch-ending-in-ff.hex	sketch-ending-in-ff.hex  zeros16kib.hex 
```
That's probably only a few lines code and gives you access to all that your shell can muster (but depending how you implement it might time out the bootloader)

Maybe it is a good exercise before attempting something like `write flash file.hex` or `load file.hex` (where load is an alias to write flash) in terminal is to look closely at `-U file.elf` that we discussed in another PR and learn from our experience how to implement *that*.

Re updating docs; I created this issue b/c the docs promised to continue where the previous command left off. Maybe review and update if the implementation is much better than the current promise in the docs.

stefanrueger, 2022-12-04:
Just because I see it right now: m->size > 0xffff? 5: 4 should be m->size > 0x10000? 5: 4

mcuee, 2022-12-04:
It seems to work pretty well now.

The suggestion from @stefanrueger here has not been implemented yet (eg: wrapping around at the end of flash).
https://github.com/avrdudes/avrdude/issues/1201#issuecomment-1336263408

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1201v1 -c xplainedpro_updi -p t817 -qqt
avrdude> dump fuses
>>> dump fuses
0000  00 00 02 ff 00 c4 04 00  02                       |.........       |

avrdude> dump fuses 0 0x01
>>> dump fuses 0 0x01
0000  00                                                |.               |

avrdude> dump fuses
>>> dump fuses
0001  00                                                |.               |

avrdude> dump fuses
>>> dump fuses
0002  02                                                |.               |

avrdude> dump
>>> dump
0003  ff                                                |.               |

avrdude> read signature 0x10
>>> read signature 0x10
avrdude_issue1201v1 error: (dump) signature address 0x00010 is out of range [0, 0x00002]
avrdude> read signature 0 0x10
>>> read signature 0 0x10
0000  1e 93 20

avrdude> dump flash 0x1f40 0x80
>>> dump flash 0x1f40 0x80
1f40  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1f50  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1f60  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1f70  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1f80  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1f90  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1fa0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1fb0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> dump flash
>>> dump flash                                               (supposed to stop at 0x1fff)
1fc0  ff ff ff ff ff ff ff ff  ff 42 6c 69 6e 6b 5f 50  |.........Blink_P|
1fd0  72 6f 38 31 37 2e 69 6e  6f 2e 74 38 31 37 6f 70  |ro817.ino.t817op|
1fe0  74 69 2e 32 30 63 30 2e  6d 44 30 2e 76 32 36 34  |ti.20c0.mD0.v264|
1ff0  2e 68 65 78 00 e6 07 0b  1b 15 05 d4 03 f5 1b 2c  |.hex... .......,|
0000  01 c0 d8 c0 11 24 80 91  40 00 88 23 69 f0 28 2f  |.....$..@..#i.(/|
0010  30 e0 83 fd 03 c0 2d 7f  23 2b 31 f4 80 93 40 00  |0.....-.#+1...@.|
0020  8c bb 80 e0 af d0 ec c0  a8 95 22 9a 2a 9a 10 92  |..........".*...|
0030  01 02 80 91 82 12 83 70  81 30 61 f5 8c e5 90 e0  |.......p.0a.....|

```

mcuee, 2022-12-04:
@MCUdude and @stefanrueger 

I am thinking it would be good to explicitly echo back "dump mem_type" even when the user only uses `dump`.

MCUdude, 2022-12-04:
Stefan wrote:
> Near the end of a memory, eg, you look at flash in chunks of 0x80 and come to the end of the memory in the middle of a chunk (r fl 0x7f40 0x80; r fl; r fl) for the m328p. Is that intuitive?

> The suggestion from @stefanrueger here has not been fully implemented yet or there is a bug (eg: wrapping around at the end of flash).

I thought this was what he wanted? You're asking Avrdude to read past the end of the memory, and then it rolls over. If the address isn't a "round number", it's very obvious what's happening. Maybe I should add a line break to make it _very_ clear that we just rolled over?

```
avrdude> read flash 0x1ffc3 0x100
>>> read flash 0x1ffc3 0x100 

Reading | ################################################## | 100% 0.03 s 

1ffc3  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ffd3  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ffe3  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1fff3  ff ff ff ff ff ff ff ff  ff ff ff ff ff           |.............   |
00000  47 c0 00 00 55 c0 00 00  53 c0 00 00 51 c0 00 00  |G...U...S...Q...|
00010  4f c0 00 00 4d c0 00 00  4b c0 00 00 49 c0 00 00  |O...M...K...I...|
00020  47 c0 00 00 45 c0 00 00  43 c0 00 00 41 c0 00 00  |G...E...C...A...|
00030  3f c0 00 00 3d c0 00 00  3b c0 00 00 39 c0 00 00  |?...=...;...9...|
00040  37 c0 00 00 35 c0 00 00  33 c0 00 00 31 c0 00 00  |7...5...3...1...|
00050  2f c0 00 00 2d c0 00 00  2b c0 00 00 29 c0 00 00  |/...-...+...)...|
00060  27 c0 00 00 97 c0 00 00  23 c0 00 00 21 c0 00 00  |'.......#...!...|
00070  1f c0 00 00 1d c0 00 00  1b c0 00 00 19 c0 00 00  |................|
00080  17 c0 00 00 15 c0 00 00  13 c0 00 00 11 c0 00 00  |................|
00090  11 24 1f be cf ef cd bf  df e3 de bf 28 e2 a0 e0  |.$..........(...|
000a0  b8 e2 01 c0 1d 92 a4 30  b2 07 e1 f7 9b d0 26 c1  |.......0......&.|
000b0  a7 cf 90 91 01 0a 9f 7e  90 93 01 0a 90 91 60 04  |..... .~... ..`.|
000c0  93 ff 0a                                          |..              |
```

This is also an alternative that makes it more obvious that we rolled over:
```
avrdude> read flash 0x1ffc3 0x100
>>> read flash 0x1ffc3 0x100 

Reading | ################################################## | 100% 0.03 s 

1ffc3  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ffd3  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1ffe3  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1fff3  ff ff ff ff ff ff ff ff  ff ff ff ff ff           |.............   |

00000  47 c0 00 00 55 c0 00 00  53 c0 00 00 51 c0 00 00  |G...U...S...Q...|
00010  4f c0 00 00 4d c0 00 00  4b c0 00 00 49 c0 00 00  |O...M...K...I...|
00020  47 c0 00 00 45 c0 00 00  43 c0 00 00 41 c0 00 00  |G...E...C...A...|
00030  3f c0 00 00 3d c0 00 00  3b c0 00 00 39 c0 00 00  |?...=...;...9...|
00040  37 c0 00 00 35 c0 00 00  33 c0 00 00 31 c0 00 00  |7...5...3...1...|
00050  2f c0 00 00 2d c0 00 00  2b c0 00 00 29 c0 00 00  |/...-...+...)...|
00060  27 c0 00 00 97 c0 00 00  23 c0 00 00 21 c0 00 00  |'.......#...!...|
00070  1f c0 00 00 1d c0 00 00  1b c0 00 00 19 c0 00 00  |................|
00080  17 c0 00 00 15 c0 00 00  13 c0 00 00 11 c0 00 00  |................|
00090  11 24 1f be cf ef cd bf  df e3 de bf 28 e2 a0 e0  |.$..........(...|
000a0  b8 e2 01 c0 1d 92 a4 30  b2 07 e1 f7 9b d0 26 c1  |.......0......&.|
000b0  a7 cf 90 91 01 0a 9f 7e  90 93 01 0a 90 91 60 04  |..... .~... ..`.|
000c0  93 ff 0a                                          |..              |
```

mcuee, 2022-12-04:
@MCUdude 

We can ask @stefanrueger to clarify what he would like to see.

My understand is that he wants the following. 
```
avrdude> dump flash 0x1f40 0x80
>>> dump flash 0x1f40 0x80
1f40  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1f50  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1f60  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1f70  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1f80  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1f90  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1fa0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
1fb0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> dump flash                                                                     (stopped at the end, not to roll over)
>>> dump flash             
1fc0  ff ff ff ff ff ff ff ff  ff 42 6c 69 6e 6b 5f 50  |.........Blink_P|
1fd0  72 6f 38 31 37 2e 69 6e  6f 2e 74 38 31 37 6f 70  |ro817.ino.t817op|
1fe0  74 69 2e 32 30 63 30 2e  6d 44 30 2e 76 32 36 34  |ti.20c0.mD0.v264|
1ff0  2e 68 65 78 00 e6 07 0b  1b 15 05 d4 03 f5 1b 2c  |.hex... .......,|

avrdude> dump flash                                                                   (now restart at zero, dump 0x80 bytes again).                                           
>>> dump flash
0000  01 c0 d8 c0 11 24 80 91  40 00 88 23 69 f0 28 2f  |.....$..@..#i.(/|
0010  30 e0 83 fd 03 c0 2d 7f  23 2b 31 f4 80 93 40 00  |0.....-.#+1...@.|
0020  8c bb 80 e0 af d0 ec c0  a8 95 22 9a 2a 9a 10 92  |..........".*...|
0030  01 02 80 91 82 12 83 70  81 30 61 f5 8c e5 90 e0  |.......p.0a.....|
0040  80 93 08 08 90 93 09 08  81 e0 80 93 0b 08 83 e0  |...... ..... ...|
0050  80 93 07 08 10 92 05 08  80 ec 80 93 06 08 8b e0  |................|
0060  91 d0 07 9a 87 e0 81 50  c1 f4 a8 95 0d e9 83 e0  |.......P.... ...|
0070  d8 2e 7c d0 81 34 e9 f4  79 d0 18 2f 8c d0 81 e0  |..|..4..y../....|

```

stefanrueger, 2022-12-04:
> `>>> dump (Is it possible to say "dump flash" here?)`

Basically, no. The ">>>" is *just* an echo of what the user typed, and at the point in code where this is written the code has no idea of what it means. `<rant attribute="opinionated">` The ">>>" line is probably to cater for when `fgets()` is used that does not echo (I suspect Windows has a line editor built into `fgets()` when it reads from a terminal and that editor does echo).  I find this echo *terrible* b/c it is just redundant and steals one line of my screen estate. I have always wondered why no one complains about this line, but seeing that it does not appear in Linux I did nothing about it. Anyone finds a good use of this line, let me know. The only use I can see is when the terminal does not echo what the user types; IMO there should be *no* echo when input is from a pipe or a file. OK `</rant>` 

How *does* the user know which memory is shown? Well, they asked for a memory to start with. And when they type `read` the terminal continues with that memory. That's fair enough. I could imagine a luxurious extension in verbose mode when the address is prepended by an abbreviated memory string:
```
avrdude> verbose 1
New verbosity level: 1
avrdude> read flash
fl 1fc0  ff ff ff ff ff ff ff ff  ff 42 6c 69 6e 6b 5f 50  |.........Blink_P|
fl 1fd0  72 6f 38 31 37 2e 69 6e  6f 2e 74 38 31 37 6f 70  |ro817.ino.t817op|
fl 1fe0  74 69 2e 32 30 63 30 2e  6d 44 30 2e 76 32 36 34  |ti.20c0.mD0.v264|
fl 1ff0  2e 68 65 78 00 e6 07 0b  1b 15 05 d4 03 f5 1b 2c  |.hex... .......,|
```
The abbreviation would have as few characters as are needed to disambiguate all memories of the part, right justified for extra neatness.

MCUdude, 2022-12-04:
> Basically, no. The ">>>" is just an echo of what the user typed, and at the point in code where this is written the code has no idea of what it means

I'd like to remove the `>>>`echo as well, since it serves no purpose to me, for the same reasons as @stefanrueger mentioned.
Instead, we can print the "actual" command if the user runs `read` or `read mem` if the verbosity level is greater than 0.

Here's an example of what this could look like:

```
$ ./avrdude -pavr128db48 -cpkobn_updi -t -B0.5

         Vtarget                      : 3.31 V
         PDI/UPDI clock Xmega/megaAVR : 2000 kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e970c (probably avr128db48)
avrdude> read flash 0 0x40

Reading | ################################################## | 100% 0.03 s 

00000  47 c0 00 00 55 c0 00 00  53 c0 00 00 51 c0 00 00  |G...U...S...Q...|
00010  4f c0 00 00 4d c0 00 00  4b c0 00 00 49 c0 00 00  |O...M...K...I...|
00020  47 c0 00 00 45 c0 00 00  43 c0 00 00 41 c0 00 00  |G...E...C...A...|
00030  3f c0 00 00 3d c0 00 00  3b c0 00 00 39 c0 00 00  |?...=...;...9...|

avrdude> read

Reading | ################################################## | 100% 0.00 s 

00040  37 c0 00 00 35 c0 00 00  33 c0 00 00 31 c0 00 00  |7...5...3...1...|
00050  2f c0 00 00 2d c0 00 00  2b c0 00 00 29 c0 00 00  |/...-...+...)...|
00060  27 c0 00 00 97 c0 00 00  23 c0 00 00 21 c0 00 00  |'.......#...!...|
00070  1f c0 00 00 1d c0 00 00  1b c0 00 00 19 c0 00 00  |................|

avrdude> verbose 1
New verbosity level: 1

avrdude> read
>>> read flash 0x80 0x40

Reading | ################################################## | 100% 0.00 s 

00080  17 c0 00 00 15 c0 00 00  13 c0 00 00 11 c0 00 00  |................|
00090  11 24 1f be cf ef cd bf  df e3 de bf 28 e2 a0 e0  |.$..........(...|
000a0  b8 e2 01 c0 1d 92 a4 30  b2 07 e1 f7 9b d0 26 c1  |.......0......&.|
000b0  a7 cf 90 91 01 0a 9f 7e  90 93 01 0a 90 91 60 04  |..... .~... ..`.|

avrdude> quit

avrdude done.  Thank you.
```

closed 2022-12-14 by stefanrueger