avrdudes/avrdude #1202: Read all memory sections to single elf file

mcuee, 2022-11-29:
https://github.com/avrdudes/avrdude/pull/1195#issuecomment-1329967636

@MCUdude @stefanrueger @McMCCRU @dl8dtl 

PR #1195 adds a convenient way to write all the memory sections using a single elf file. 

What about memory reading?

I am thinking of a new memory name called `all` and I think it is nice to have the following command. Any comments? This can be in a new enhancement.

```
avrdude -c xxx -p xxx -U:all:r:readhackall.elf
```

mcuee, 2022-12-01:
From here:
https://github.com/avrdudes/avrdude/issues/1204#issuecomment-1332866351

We may even be able to read the Flash and EEPROM into single hex file (currently not supported), this should be easier than ELF file.

```
0x00000000 up to 0x0007FFFF for Flash memory,
0x00800000 up to 0x0080FFFF for RAM, and
0x00810000 up to 0x0081FFFF for EEPROM.
```

stefanrueger, 2022-12-01:
Even better, looking at [`elf_mem_limits()`](https://github.com/avrdudes/avrdude/blob/ea2d7eec95b24d2b9d8f969c0e0f60c88b1b491d/src/fileio.c#L763-L796)  in `fileio.c`  we can map the writeable memories to 

|Low|High|Memories|
| --: | --: | :-- |
| 0x00000000 | 0x0007FFFF | Flash |
| 0x00800000 | 0x0080FFFF |RAM |
| 0x00810000 | 0x0081FFFF | EEPROM |
| 0x00820000 | 0x0082FFFF | Fuses(*) |
| 0x00830000 | 0x0083FFFF | Lock byte(s) |

(*) The mapping of the respective fuse bytes to offset locations follows historic name convention as in `elf_mem_limits()`.

This means we can, indeed, transform our .hex files to act as *container* files for how a part has been prepared. Above is only valid for classic parts. More modern parts have their own, *different* mapping into a linear address space, so clearly that mapping needs to be per part family (or is worse comes to worst) per part. 

Useful applications would be to make a backup of all (writeable) memories of a part in a single .hex file. And, extending the `-U file.hex` syntax into "Write all memories that are mentioned in the .hex file" a la PR #1195 would have an obvious benefit, too.

Comments welcome from those who are more AVR-literate than me: @MCUdude, @SpenceKonde, @WestfW, @dl8dtl, ...

mcuee, 2023-06-20:
I think this feature request does not gain much traction and I will close this for now as not planned.

We can always revisit this in the future if someone is interested and prepare a PR for this.

closed 2023-06-20 by mcuee