avrdudes/avrdude #1203: Error under Windows in terminal mode: avrdude OS error: No such file or directory

mcuee, 2022-11-29:
```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p m168p -qqt
avrdude> d
>>> d
Usage: d <memory> <addr> <len>
       d <memory> <addr> ...
       d <memory> <addr>
       d <memory> ...
       d <memory>
avrdude> dump flash 0 0x10
>>> dump flash 0 0x10
0000  0c 94 61 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | .a. .s. .s. .s.|

avrdude> dump flash
>>> dump flash
0000  0c 94 61 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | .a. .s. .s. .s.|
0010  0c 94 73 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | .s. .s. .s. .s.|
0020  0c 94 73 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | .s. .s. .s. .s.|
0030  0c 94 73 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | .s. .s. .s. .s.|
0040  0c 94 8d 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | ... .s. .s. .s.|
0050  0c 94 73 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | .s. .s. .s. .s.|
0060  0c 94 73 00 0c 94 73 00  00 00 00 09 00 03 02 00  | .s. .s.... ....|
0070  00 04 05 08 00 00 00 00  00 00 00 00 00 00 00 01  |................|
0080  02 04 08 10 20 40 80 01  02 04 08 10 20 01 02 04  |.... @...... ...|
0090  08 10 20 40 80 40 04 04  04 04 04 04 04 04 02 02  |.. @.@..........|
00a0  02 02 02 02 03 03 03 03  03 03 02 02 03 00 00 00  |................|
00b0  00 25 00 28 00 2b 00 00  00 00 00 24 00 27 00 2a  |.%.(.+.....$.'.*|
00c0  00 00 11 24 1f be cf ef  d4 e0 de bf cd bf 21 e0  |...$..........!.|
00d0  a0 e0 b1 e0 01 c0 1d 92  a9 30 b2 07 e1 f7 0e 94  |.........0......|
00e0  f1 01 0c 94 01 02 0c 94  00 00 61 e0 8d e0 0c 94  |.. ... ...a... .|
00f0  86 01 61 e0 8d e0 0e 94  c2 01 68 ee 73 e0 80 e0  |..a.......h.s...|

avrdude> dump flash
>>> dump flash
0100  90 e0 0e 94 fc 00 60 e0  8d e0 0e 94 c2 01 68 ee  |......`.......h.|
0110  73 e0 80 e0 90 e0 0c 94  fc 00 1f 92 0f 92 0f b6  |s..... .........|
0120  0f 92 11 24 2f 93 3f 93  8f 93 9f 93 af 93 bf 93  |...$/.?.........|
0130  80 91 05 01 90 91 06 01  a0 91 07 01 b0 91 08 01  |................|
0140  30 91 04 01 23 e0 23 0f  2d 37 68 f1 26 e8 23 0f  |0...#.#.-7h.&.#.|
0150  02 96 a1 1d b1 1d 20 93  04 01 80 93 05 01 90 93  |...... .........|
0160  06 01 a0 93 07 01 b0 93  08 01 80 91 00 01 90 91  |................|
0170  01 01 a0 91 02 01 b0 91  03 01 01 96 a1 1d b1 1d  |................|
0180  80 93 00 01 90 93 01 01  a0 93 02 01 b0 93 03 01  |................|
0190  bf 91 af 91 9f 91 8f 91  3f 91 2f 91 0f 90 0f be  |........?./.....|
01a0  0f 90 1f 90 18 95 01 96  a1 1d b1 1d d4 cf 3f b7  |..............?.|
01b0  f8 94 80 91 00 01 90 91  01 01 a0 91 02 01 b0 91  |................|
01c0  03 01 26 b5 a8 9b 05 c0  2f 3f 19 f0 01 96 a1 1d  |..&...../?......|
01d0  b1 1d 3f bf ba 2f a9 2f  98 2f 88 27 bc 01 cd 01  |..?../././.'....|
01e0  62 0f 71 1d 81 1d 91 1d  42 e0 66 0f 77 1f 88 1f  |b.q.....B.f.w...|
01f0  99 1f 4a 95 d1 f7 08 95  8f 92 9f 92 af 92 bf 92  |..J.............|

avrdude> quit
>>> quit
avrdude> avrdude OS error: No such file or directory
``

mcuee, 2022-11-29:
Looks like this is specific to usbasp.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p m168p -t -v

avrdude: Version 7.0-20221127 (2f2a6c0)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega168P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no        512    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     16384  128    128  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e940b (probably m168p)
avrdude> dump flash
>>> dump flash

Reading | ################################################## | 100% 0.33 s

0000  0c 94 61 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | .a. .s. .s. .s.|
0010  0c 94 73 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | .s. .s. .s. .s.|
0020  0c 94 73 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | .s. .s. .s. .s.|
0030  0c 94 73 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | .s. .s. .s. .s.|
0040  0c 94 8d 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | ... .s. .s. .s.|
0050  0c 94 73 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | .s. .s. .s. .s.|
0060  0c 94 73 00 0c 94 73 00  00 00 00 09 00 03 02 00  | .s. .s.... ....|
0070  00 04 05 08 00 00 00 00  00 00 00 00 00 00 00 01  |................|
0080  02 04 08 10 20 40 80 01  02 04 08 10 20 01 02 04  |.... @...... ...|
0090  08 10 20 40 80 40 04 04  04 04 04 04 04 04 02 02  |.. @.@..........|
00a0  02 02 02 02 03 03 03 03  03 03 02 02 03 00 00 00  |................|
00b0  00 25 00 28 00 2b 00 00  00 00 00 24 00 27 00 2a  |.%.(.+.....$.'.*|
00c0  00 00 11 24 1f be cf ef  d4 e0 de bf cd bf 21 e0  |...$..........!.|
00d0  a0 e0 b1 e0 01 c0 1d 92  a9 30 b2 07 e1 f7 0e 94  |.........0......|
00e0  f1 01 0c 94 01 02 0c 94  00 00 61 e0 8d e0 0c 94  |.. ... ...a... .|
00f0  86 01 61 e0 8d e0 0e 94  c2 01 68 ee 73 e0 80 e0  |..a.......h.s...|

avrdude> dump flash
>>> dump flash

Reading | ################################################## | 100% 0.31 s

0100  90 e0 0e 94 fc 00 60 e0  8d e0 0e 94 c2 01 68 ee  |......`.......h.|
0110  73 e0 80 e0 90 e0 0c 94  fc 00 1f 92 0f 92 0f b6  |s..... .........|
0120  0f 92 11 24 2f 93 3f 93  8f 93 9f 93 af 93 bf 93  |...$/.?.........|
0130  80 91 05 01 90 91 06 01  a0 91 07 01 b0 91 08 01  |................|
0140  30 91 04 01 23 e0 23 0f  2d 37 68 f1 26 e8 23 0f  |0...#.#.-7h.&.#.|
0150  02 96 a1 1d b1 1d 20 93  04 01 80 93 05 01 90 93  |...... .........|
0160  06 01 a0 93 07 01 b0 93  08 01 80 91 00 01 90 91  |................|
0170  01 01 a0 91 02 01 b0 91  03 01 01 96 a1 1d b1 1d  |................|
0180  80 93 00 01 90 93 01 01  a0 93 02 01 b0 93 03 01  |................|
0190  bf 91 af 91 9f 91 8f 91  3f 91 2f 91 0f 90 0f be  |........?./.....|
01a0  0f 90 1f 90 18 95 01 96  a1 1d b1 1d d4 cf 3f b7  |..............?.|
01b0  f8 94 80 91 00 01 90 91  01 01 a0 91 02 01 b0 91  |................|
01c0  03 01 26 b5 a8 9b 05 c0  2f 3f 19 f0 01 96 a1 1d  |..&...../?......|
01d0  b1 1d 3f bf ba 2f a9 2f  98 2f 88 27 bc 01 cd 01  |..?../././.'....|
01e0  62 0f 71 1d 81 1d 91 1d  42 e0 66 0f 77 1f 88 1f  |b.q.....B.f.w...|
01f0  99 1f 4a 95 d1 f7 08 95  8f 92 9f 92 af 92 bf 92  |..J.............|

avrdude> quit
>>> quit
avrdude> avrdude usbasp_transmit() OS error: No such file or directory

avrdude done.  Thank you.

```

MCUdude, 2022-11-29:
I remember getting a warning/error when using USBasp with Avrdude on MacOS. It's in the very beginning, and it doesn't affect the programming. I'm also not getting the warning/error every time I think.

dl8dtl, 2022-11-29:
Can you run that through a debugger, and see where it is complaining?

dl8dtl, 2022-11-29:
Ah, I have an idea. I guess upon saying "goodbye" to USBasp, the firmware causes the USB to be too slow to respond (keep in mind this is just firmware USB), so the operating systems kicks the USB device out while AVRDUDE attempts to perform some final cleanup action.

mcuee, 2022-11-29:
> Can you run that through a debugger, and see where it is complaining?

It is difficult as this seldom happens. 

> Ah, I have an idea. I guess upon saying "goodbye" to USBasp, the firmware causes the USB to be too slow to respond (keep in mind this is just firmware USB), so the operating systems kicks the USB device out while AVRDUDE attempts to perform some final cleanup action.

I think this is a very possible cause. Thanks for the insight.

mcuee, 2023-01-02:
I have not encountered this recently. Strange.

mariusgreuel, 2023-01-09:
I just tried it with v7.1, and I can not reproduce it on my PC. I agree with Jörg, it is probably a V-USB issue.

mcuee, 2023-01-09:
I agree. I will cloe this one.

closed 2023-01-09 by mcuee