avrdudes/avrdude #1210: Flash usb_out error:Connection timed out (expected 128, got -110)

paulcrawford, 2022-12-05:
From a 64bit Ubuntu 20.04 Odroid n2+ I am programming an ATmega328p with a USBTiny programmer and an Intel hex file. Until recently everything worked fine. But when I edited the source c file, which actually made the hex file shrink from 8646 bytes to 8478 bytes I started to get the error noted in the title. Going back the original larger hex file still works perfectly so it seems that the issue is the hex file itself. I have tried both version 6.3 and 7.0 and can reproduce the same error on both. I am attaching three text files of output, one from the error in v. 6.3, one from the error in v. 7.0 and one success with v. 7.0.  Also included are the large hex file (no error) and the small hex file (error).

I am not sure if this is really a bug or not, but to continue with my project I need to find out how to prevent the error.

The command line text for the compile is:

avr-gcc -mmcu=atmega328p -Wall -Os -o sg.elf sg.c -w -Wl,-u,vfprintf -lprintf_flt -lm

(it runs with no errors)

and for the hex file creation is:

avr-objcopy -j .text -j .data -O ihex sg.elf sg.hex

The programming command is:

avrdude -P usb -c usbtiny -p m328p -F -v -v -U flash:w:"sg.hex":a

[Large hex file.txt](https://github.com/avrdudes/avrdude/files/10150290/Large.hex.file.txt)
[small hex file.txt](https://github.com/avrdudes/avrdude/files/10150291/small.hex.file.txt)
[v 6.3 error.txt](https://github.com/avrdudes/avrdude/files/10150300/v.6.3.error.txt)
[v 7.0 error.txt](https://github.com/avrdudes/avrdude/files/10150301/v.7.0.error.txt)
[v 7.0 success.txt](https://github.com/avrdudes/avrdude/files/10150302/v.7.0.success.txt)

mcuee, 2022-12-05:
Can you try latest git main?

paulcrawford, 2022-12-05:
I tried Version 7.0-20221121 and got early the same error. The command and output is in the attached test file using the same small hex file.
[Version 7.0-20221121 error.txt](https://github.com/avrdudes/avrdude/files/10154552/Version.7.0-20221121.error.txt)

paulcrawford, 2022-12-05:
I just realized that the v. 7.0 that I used when I found the error is the same version that I sent a few minutes ago. So I guess that I never used the latest release dated May 8, 2022. I did the build because the Ubuntu repository had only version 6.3. What I should have done is download the latest May 8 version rather than the Linux build. In any case, I have definitely been using the nightly build version already. Sorry for the confusion.

paulcrawford, 2022-12-05:
Having just read Issue #1204, which had some discussion of the avr toolchain, I want to add that the toolchain I used is:

$ avr-gcc -v
Using built-in specs.
Reading specs from /usr/lib/gcc/avr/5.4.0/device-specs/specs-avr2
COLLECT_GCC=avr-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/avr/5.4.0/lto-wrapper
Target: avr
Configured with: ../gcc/configure -v --enable-languages=c,c++ --prefix=/usr/lib --infodir=/usr/share/info --mandir=/usr/share/man --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --enable-shared --with-system-zlib --enable-long-long --enable-nls --without-included-gettext --disable-libssp --build=aarch64-linux-gnu --host=aarch64-linux-gnu --target=avr CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-KOUpdi/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong -Wformat ' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-KOUpdi/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong -Wformat ' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-KOUpdi/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-KOUpdi/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-KOUpdi/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-KOUpdi/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong -Wformat ' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-KOUpdi/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong -Wformat '
Thread model: single
gcc version 5.4.0 (GCC)  

and 

$ avr-objcopy -V
GNU objcopy (GNU Binutils) 2.26.20160125
Copyright (C) 2015 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.

mcuee, 2022-12-05:
@paulcrawford 
There are different types of USBtinyISP programmer. My one is based on ATtiny2313 (most common one). There are others which are based on ATtiny44.

Please also try a different USB cable or USB port to see if that makes any difference. The error seems to point to USB communication error.

mcuee, 2022-12-05:
In any case, I just tried another USBtinyISP based on ATtiny44 and it works fine as well.

paulcrawford, 2022-12-05:
@mcuee

Thank you for your efforts so far. I had, of course, hoped that you could reproduce the error that is so reproducible on my hardware. I checked my USBTinyISP programmer and it is the same as yours, an ATtiny2313:

I have tried two different cables and both give the error and success depending on the hex file used. Unfortunately there is only one USB port available on the Odroid n2+ (long story) so I cannot try another.

I would have thought that the fact that the flash command works perfectly with the one hex file but not the other would verify that there are no issues with the hardware, i.e. the USB port, USB cable and USB programmer. But there is one more thing to try and that is the cable from the USB programmer to the ATmega328p. I will try to change that as presently the pins are controlled by a relay so that I can do the programming without changing anything on the live MCU. I can put a fixed cable in place to check if that is the issue.

The other strange thing is that the error always occurs at the same point in the writing to flash. One would think that a bad connection would be more random, but a software issue would be at a specific point and repeatable, just as it is.

paulcrawford, 2022-12-05:
Just tested again with a direct connection from the USBtiny to the 328p. Writing to flash succeeded with the large hex file and failed with the small hex file. I can only conclude that the issue is software related and in the content of the small hex file. The question still remains, how to correct the behaviour?

Definitely the error occurs as a usb_out error according to avrdude messages. Looking at the source code it seems that avrdude is "expected 128, got -110" where the 128 is buflen, which is the "CHUNKSIZE" or amount of data to be sent in one USB packet, and -110 is nbytes. See /src/usbtiny.c line 170. There must be a reason why nbytes is not as expected, which again appears to be related to software in the creation of the hex file.

The variable nbytes comes from the function usb_control_msg and if it is equal to buflen there would be no error. I cannot find the definition of this function in the source code. If someone knows about this function and can determine how it would get a value of -110, we might be a little closer to the cause of the issue.

MCUdude, 2022-12-05:
Can you try to upload the small hex file to a different ATmega328P board if you have one? The programmer _should_ be able to write any value from 0 to 0xff to any byte in flash, even if it isn't a valid program. I'm note sure, but there _may_ be an issue with the 328P.

Fro

MCUdude, 2022-12-05:
I only have an ATtiny44 based USBtiny programmer, but I can't reproduce the issue on macOS:

```
$ ./avrdude -cusbtiny -patmega328p -Uflash:w:/Users/hans/Downloads/Large.hex.file.txt:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
erasing chip
avrdude: reading input file /Users/hans/Downloads/Large.hex.file.txt for flash
         with 8646 bytes in 1 section within [0, 0x21c5]
         using 68 pages and 58 pad bytes
avrdude: writing 8646 bytes flash ...

Writing | ################################################## | 100% 13.88 s 

avrdude: 8646 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/Large.hex.file.txt

Reading | ################################################## | 100% 8.08 s 

avrdude: 8646 bytes of flash verified

avrdude done.  Thank you.
```

paulcrawford, 2022-12-05:
@Hans,

Thank you for testing my hex file on your hardware. Did you try the small hex file as well? If not, could you please try it since it is the one with the error. The large hex file works fine on my hardware just as it did on yours and mcuee’s. The small hex file should give 8478 bytes to be written to flash memory.

Thanks again.

paulcrawford, 2022-12-05:
Sorry, should have been @MCUdude above.

paulcrawford, 2022-12-05:
@MCUdude,

So I tried two other 328p MCUs in my possession, and neither one was able to flash either the large or small hex file. Going back to the original mcu, the large hex file worked, as expected, and the small hex file failed, as expected. So everything is reproducible at least. 

I think it is time order new ATmega328p MCUs.

Thanks again for your suggestion.

MCUdude, 2022-12-05:
@paulcrawford sorry, I have no idea why I thought Large.hex was the file you couldn't upload to your board.

But even if I upload small.hex, I still can't reproduce the issue:

```
$ ./avrdude -cusbtiny -patmega328p -Uflash:w:/Users/hans/Downloads/small.hex.file.txt:i -vv

avrdude: Version 7.0-20220508
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/hans/Downloads/avrdude-terminal/src/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): found USBtinyISP, bus:device: 020:016
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, https://learn.adafruit.com/usbtinyisp
avrdude pgm_default() error: programmer operation not supported

avrdude: using SCK period of 10 usec
CMD: [ac 53 00 00] [ff fe 53 00]
avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00 s CMD: [30 00 00 00] [00 30 00 1e]
CMD: [30 00 01 00] [00 30 00 95]
Reading | #################                                  | 33% 0.00 s CMD: [30 00 02 00] [00 30 00 0f]
Reading | ################################################## | 100% 0.01 s 

avrdude: device signature = 0x1e950f (probably m328p)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
erasing chip
CMD: [ac 80 00 00] [00 ac 80 00]
avrdude: using SCK period of 10 usec
CMD: [ac 53 00 00] [00 ac 53 00]
avrdude: reading input file /Users/hans/Downloads/small.hex.file.txt for flash
         with 8478 bytes in 1 section within [0, 0x211d]
         using 67 pages and 98 pad bytes
avrdude: writing 8478 bytes flash ...

[... 55 lines cut ...]
Writing | #########################################          | 82% 10.86 s CMD: [4c 0d c0 00] [95 4c 0d c0]
Writing | ##########################################         | 83% 11.06 s CMD: [4c 0e 00 00] [0b 4c 0e 00]
Writing | ###########################################        | 85% 11.27 s CMD: [4c 0e 40 00] [05 4c 0e 40]
Writing | ###########################################        | 86% 11.48 s CMD: [4c 0e 80 00] [1e 4c 0e 80]
Writing | ############################################       | 88% 11.67 s CMD: [4c 0e c0 00] [0b 4c 0e c0]
Writing | #############################################      | 89% 11.87 s CMD: [4c 0f 00 00] [e3 4c 0f 00]
Writing | ##############################################     | 91% 12.08 s CMD: [4c 0f 40 00] [81 4c 0f 40]
Writing | ##############################################     | 92% 12.27 s CMD: [4c 0f 80 00] [1d 4c 0f 80]
Writing | ###############################################    | 94% 12.47 s CMD: [4c 0f c0 00] [35 4c 0f c0]
Writing | ################################################   | 95% 12.68 s CMD: [4c 10 00 00] [d9 4c 10 00]
Writing | #################################################  | 97% 12.90 s CMD: [4c 10 40 00] [a3 4c 10 40]
Writing | #################################################  | 98% 13.09 s CMD: [4c 10 80 00] [ff 4c 10 80]
Writing | ################################################## | 100% 13.30 s 

avrdude: 8478 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/small.hex.file.txt
avrdude: reading on-chip flash data ...

Reading | ################################################## | 100% 8.04 s 

avrdude: verifying ...
avrdude: 8478 bytes of flash verified

avrdude done.  Thank you.
```

mcuee, 2022-12-05:
@paulcrawford 

I tend to think this is a USB related problem and there are probably two reasons.

1) You are using an embedded Linux system -- ODROID-N2+ 
2) USBtinyISP is using FW based USB which does not comply to USB standards

If possible, please test with an Intel/AMD CPU based system. Thanks.

paulcrawford, 2022-12-06:
@mcuee 

While I agree with your three issues as possibilities, they do not explain why the large hex file is consistently successfully flashed.

I will try to do the flashing with an Intel based Linux machine to see what happens.

Your are correct about the four USB ports on the Odroid, but my board was one of the many that had hardware failures of all four of them. That leaves me with only the OTG port. It has worked fine with two USB to RS485 converters that I run as well as the flashing of the large hex file.

As discussed with @MCUdude I am also going to replace my ATmega328p to confirm that the existing one is perhaps the culprit.

mcuee, 2022-12-07:
I will close this issue. Feel free to open if you have new info to show that avrdude may be the problem.

closed 2022-12-07 by mcuee