avrdudes/avrdude #1211: How to install libusb-1.0 in Windows

am-ar, 2022-12-05:
I'm using vcpkg to install the libraries on windows, but cmake doesn't find libusb-1.0: `[cmake] -- DON'T HAVE libusb_1_0`

What is the correct way to install?

<details>

`vcpkg.exe list`
```console
libconfuse:x64-windows                            2019-07-14#2        Small configuration file parser library for C
libftdi1:x64-windows                              1.5#3               FTDI USB driver with bitbang mode
libusb-win32:x64-windows                          1.2.6.0#8           Allows user space applications to access many US...
libusb-win32:x86-windows                          1.2.6.0#8           Allows user space applications to access many US...
libusb:x64-windows                                1.0.26#2            a cross-platform library to access USB devices
libusb:x86-windows                                1.0.26#2            a cross-platform library to access USB devices
readline-win32:x64-windows                        5.0#5               Implementation of readline for Windows Desktop
readline-win32:x86-windows                        5.0#5               Implementation of readline for Windows Desktop
readline:x64-windows                              0#3                 GNU readline and history libraries
readline:x86-windows                              0#3                 GNU readline and history libraries
vcpkg-cmake-config:x64-windows                    2022-02-06#1
vcpkg-cmake:x64-windows                           2022-10-30
```

`cmake --no-warn-unused-cli -DUSE_LIBUSBWIN32=TRUE -DUSE_EXTERNAL=TRUE -DCMAKE_TOOLCHAIN_FILE=vcpkg/scripts/buildsystems/vcpkg.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=TRUE -S. -Bbuild -G "Visual Studio 17 2022" -T host=x64 -A x64`
```console
[main] Configuring folder: avrdude 
[driver] Removing c:/Users/Dados/Environment/eclipse-cpp/avrdude/build/CMakeCache.txt
[driver] Removing c:\Users\Dados\Environment\eclipse-cpp\avrdude\build\CMakeFiles
[proc] Executing command: "C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" --no-warn-unused-cli -DUSE_LIBUSBWIN32:BOOL=TRUE -DUSE_EXTERNAL:BOOL=TRUE -DCMAKE_TOOLCHAIN_FILE:STRING=C:/Users/Dados/Environment/eclipse-cpp/avrdude/vcpkg/scripts/buildsystems/vcpkg.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -Sc:/Users/Dados/Environment/eclipse-cpp/avrdude -Bc:/Users/Dados/Environment/eclipse-cpp/avrdude/build -G "Visual Studio 17 2022" -T host=x64 -A x64
[cmake] Not searching for unused variables given on the command line.
[cmake] -- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.22000.
[cmake] -- The C compiler identification is MSVC 19.34.31935.0
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.34.31933/bin/Hostx64/x64/cl.exe - skipped
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.38.1.windows.1") 
[cmake] -- Found FLEX: C:/Users/Dados/Environment/msys32/usr/bin/flex.exe (found version "2.6.4") 
[cmake] -- Found BISON: C:/Users/Dados/Environment/msys32/usr/bin/bison.exe (found version "3.4.1")
[cmake] -- Looking for ftdi_tcioflush
[cmake] -- Looking for ftdi_tcioflush - found
[cmake] -- Fetching external libraries, please wait...
[cmake] -- The CXX compiler identification is MSVC 19.34.31935.0
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.34.31933/bin/Hostx64/x64/cl.exe - skipped
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] -- Using external library 'libelf'
[cmake] -- Using external library 'libusb'
[cmake] -- Using external library 'libhidapi'
[cmake] -- Using external library 'libftdi'
[cmake] -- Configuration summary:
[cmake] -- ----------------------
[cmake] -- DO HAVE    libelf
[cmake] -- DO HAVE    libusb
[cmake] -- DON'T HAVE libusb_1_0
[cmake] -- DO HAVE    libhidapi
[cmake] -- DO HAVE    libftdi (but prefer to use libftdi1)
[cmake] -- DO HAVE    libftdi1
[cmake] -- DON'T HAVE libreadline
[cmake] -- DISABLED   doc
[cmake] -- DISABLED   parport
[cmake] -- DISABLED   linuxgpio
[cmake] -- DISABLED   linuxspi
[cmake] -- ----------------------
[cmake] -- Configuring done
[cmake] -- Generating done
[cmake] -- Build files have been written to: C:/Users/Dados/Environment/eclipse-cpp/avrdude/build
[visual-studio] Patch Windows SDK path from C:\Program Files (x86)\Windows Kits\10\bin\x64 to C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\x64 for C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvarsall.bat
```
</details>

mcuee, 2022-12-05:
If you use the option `Using external library`, you will not be able to use libusb-1.0, rather you will use the special version of libusb (which works with libusb0.sys and WinUSB driver) from avrdude project.

If you want to build with libusb-1.0, my recommendation will be using MSYS2 mingw32/mingw64.

mcuee, 2022-12-06:
Test using vcpkg, it does not seem to work well.

```
PS C:\src\vcpkg> .\vcpkg\vcpkg list
hidapi:x64-windows                                 0.11.2#1         A Simple library for communicating with USB and ...
libconfuse:x64-windows                             2019-07-14#2     Small configuration file parser library for C
libftdi1:x64-windows                               1.5#3            FTDI USB driver with bitbang mode
libusb-win32:x64-windows                           1.2.6.0#8        Allows user space applications to access many US...
libusb:x64-windows                                 1.0.26           a cross-platform library to access USB devices
readline-win32:x64-windows                         5.0#5            Implementation of readline for Windows Desktop
readline:x64-windows                               0#3              GNU readline and history libraries
vcpkg-cmake-config:x64-windows                     2022-02-06
vcpkg-cmake:x64-windows                            2022-05-10

PS C:\work\avr\avrdude_test\avrdude_main> & 'C:\Program Files\CMake\bin\cmake.exe' --no-warn-unused-cli -DCMAKE_TOOLCHAIN_FILE=c:\src\vcpkg\vcpkg\scripts\buildsystems\vcpkg.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=TRUE -S. -Bbuild -G "Visual Studio 17 2022" -T host=x64 -A x64
Not searching for unused variables given on the command line.
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22621.
-- The C compiler identification is MSVC 19.31.31105.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.33.0.windows.2")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "3.7.4")
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - not found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - not found
-- The CXX compiler identification is MSVC 19.31.31105.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuration summary:
-- ----------------------
-- DON'T HAVE libelf
-- DON'T HAVE libusb
-- DON'T HAVE libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DON'T HAVE libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude_main/build
```

mcuee, 2022-12-06:
If you follow the official instruction, it should work our of the box.

```
PS C:\work\avr\avrdude_test\avrdude_main> & 'C:\Program Files\CMake\bin\cmake.exe' -DUSE_EXTERNAL=TRUE -B build -G "Visual Studio 17 2022" -T host=x64 -A x6
4
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22621.
-- The C compiler identification is MSVC 19.31.31105.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.33.0.windows.2")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "3.7.4")
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.31.31105.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using external library 'libelf'
-- Using external library 'libusb'
-- Using external library 'libhidapi'
-- Using external library 'libftdi'
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DON'T HAVE libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi
-- DON'T HAVE libftdi1
-- DON'T HAVE libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude_main/build
```

VS2022 build log (x64 release).
```
Build started...
1>------ Build started: Project: ZERO_CHECK, Configuration: Release x64 ------
1>Checking Build System
2>------ Build started: Project: libusb, Configuration: Release x64 ------
3>------ Build started: Project: libhidapi, Configuration: Release x64 ------
4>------ Build started: Project: libftdi, Configuration: Release x64 ------
5>------ Build started: Project: libelf, Configuration: Release x64 ------
6>------ Build started: Project: conf, Configuration: Release x64 ------
2>Building Custom Rule C:/work/avr/avrdude_test/avrdude_main/build/_deps/libusb-src/CMakeLists.txt
3>Building Custom Rule C:/work/avr/avrdude_test/avrdude_main/build/_deps/libhidapi-src/CMakeLists.txt
4>Building Custom Rule C:/work/avr/avrdude_test/avrdude_main/build/_deps/libftdi-src/CMakeLists.txt
2>descriptors.c
3>hid.c
4>ftdi.cpp
5>Building Custom Rule C:/work/avr/avrdude_test/avrdude_main/build/_deps/libelf-src/CMakeLists.txt
6>Generating avrdude.conf
2>error.c
5>32.fsize.c
5>32.getehdr.c
6>Building Custom Rule C:/work/avr/avrdude_test/avrdude_main/src/CMakeLists.txt
5>32.getphdr.c
5>32.getshdr.c
5>32.newehdr.c
[... 123 lines cut ...]
7>wiring.c
7>xbee.c
7>lexer.c
7>config_gram.c
7>getopt.c
7>gettimeofday.c
7>Generating Code...
7>usleep.cpp
7>libavrdude.vcxproj -> C:\work\avr\avrdude_test\avrdude_main\build\src\Release\libavrdude.lib
8>------ Build started: Project: avrdude, Configuration: Release x64 ------
8>Building Custom Rule C:/work/avr/avrdude_test/avrdude_main/src/CMakeLists.txt
8>main.c
8>term.c
8>avrintel.c
8>developer_opts.c
8>whereami.c
8>Generating Code...
8>   Creating library C:/work/avr/avrdude_test/avrdude_main/build/src/Release/avrdude.lib and object C:/work/avr/avrdude_test/avrdude_main/build/src/Release/avrdude.exp
8>avrdude.vcxproj -> C:\work\avr\avrdude_test\avrdude_main\build\src\Release\avrdude.exe
9>------ Build started: Project: ALL_BUILD, Configuration: Release x64 ------
9>Building Custom Rule C:/work/avr/avrdude_test/avrdude_main/CMakeLists.txt
10>------ Skipped Build: Project: INSTALL, Configuration: Release x64 ------
10>Project not selected to build for this solution configuration 
========== Build: 9 succeeded, 0 failed, 0 up-to-date, 1 skipped ==========
```

mcuee, 2022-12-06:
Result:
```
PS C:\work\avr\avrdude_test\avrdude_main\build\src\Release> ls

    Directory: C:\work\avr\avrdude_test\avrdude_main\build\src\Release

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---           6/12/2022  7:42 pm         941568 avrdude.exe
-a---           6/12/2022  7:42 pm           3214 avrdude.exp
-a---           6/12/2022  7:42 pm           5822 avrdude.lib
-a---           6/12/2022  7:42 pm        2460470 libavrdude.lib

PS C:\work\avr\avrdude_test\avrdude_main\build\src\Release> .\avrdude.exe
Usage: avrdude [options]
Options:
  -p <partno>                Specify AVR device
  -b <baudrate>              Override RS-232 baud rate
  -B <bitclock>              Specify bit clock period (us)
  -C <config-file>           Specify location of configuration file
  -c <programmer>            Specify programmer type
  -A                         Disable trailing-0xff removal from file and AVR read
  -D                         Disable auto erase for flash memory; implies -A
  -i <delay>                 ISP Clock Delay [in microseconds]
  -P <port>                  Specify connection port
  -F                         Override invalid signature or initialisation check
  -e                         Perform a chip erase
  -O                         Perform RC oscillator calibration (see AVR053)
  -U <memtype>:r|w|v:<filename>[:format]
                             Memory operation specification
                             Multiple -U options are allowed, each request
                             is performed in the order specified
  -n                         Do not write anything to the device
  -V                         Do not verify
  -t                         Enter terminal mode
  -E <exitspec>[,<exitspec>] List programmer exit specifications
  -x <extended_param>        Pass <extended_param> to programmer
  -v                         Verbose output; -v -v for more
  -q                         Quell progress output; -q -q for less
  -l logfile                 Use logfile rather than stderr for diagnostics
  -?                         Display this usage

avrdude version 7.0-20221129 (ea2d7ee), URL: <https://github.com/avrdudes/avrdude>

```

am-ar, 2022-12-06:
@mcuee, I'm not having trouble compiling. My problem is enabling usblib-1.0 support.

But right now I'm using MinGW as you suggested yesterday and apparently it's working for me.

mcuee, 2022-12-07:
vcpkg running example:

```
PS C:\work\avr\avrdude_test\avrdude_main> & 'C:\Program Files\CMake\bin\cmake.exe' -D HAVE_LIBREADLINE:FILEPATH=C:\work\avr\avrdude_test\avrdude_main\src\msvc\readline\readline.lib -D HAVE_LIBUSB_1_0:FILEPATH=C:\src\vcpkg\vcpkg\packages\libusb_x64-windows\lib\libusb-1.0.lib -D USE_EXTERNAL=TRUE -B build
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22621.
-- The C compiler identification is MSVC 19.31.31105.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.33.0.windows.2")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "3.7.4")
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - not found
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.31.31105.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using external library 'libelf'
-- Using external library 'libusb'
-- Using external library 'libhidapi'
-- Using external library 'libftdi'
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi
-- DON'T HAVE libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude_main/build

```

mcuee, 2022-12-07:
A simple workaround is to copy `libusb.h` to `src\msvc`.
```
PS C:\work\avr\avrdude_test\avrdude_main> ls .\src\msvc\libusb*

    Directory: C:\work\avr\avrdude_test\avrdude_main\src\msvc

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---           10/4/2022  4:57 pm          76208 libusb.h

PS C:\work\avr\avrdude_test\avrdude_main> ls .\src\msvc\readline\

    Directory: C:\work\avr\avrdude_test\avrdude_main\src\msvc\readline

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---           30/3/2017  1:17 am           1500 ansi_stdlib.h
-a---           30/3/2017  1:17 am           4853 chardefs.h
-a---           14/5/2022  8:26 pm           1496 config.h
-a---           30/3/2017  1:17 am           2481 histlib.h
-a---           30/3/2017  1:17 am          10805 history.h
-a---           30/3/2017  1:17 am           3620 keymaps.h
-a---           30/3/2017  1:17 am           2278 posixdir.h
-a---           30/3/2017  1:17 am           1298 posixjmp.h
-a---           30/3/2017  1:17 am           4381 posixstat.h
-a---           17/7/2022 11:18 am         176128 readline.dll
-a---           30/3/2017  1:17 am          38631 readline.h
-a---           17/7/2022 11:18 am         105056 readline.lib
-a---           17/7/2022 11:18 am        1290240 readline.pdb
-a---           30/3/2017  1:17 am           2182 rlconf.h
-a---           30/3/2017  1:17 am           5637 rldefs.h
-a---           30/3/2017  1:17 am           4455 rlmbutil.h
-a---           30/3/2017  1:17 am          11530 rlprivate.h
-a---           30/3/2017  1:17 am           1450 rlshell.h
-a---           30/3/2017  1:17 am           2246 rlstdc.h
-a---           30/3/2017  1:17 am           2668 rltty.h
-a---           30/3/2017  1:17 am           2977 rltypedefs.h
-a---           30/3/2017  1:17 am           2195 rlwinsize.h
-a---           30/3/2017  1:17 am           1703 tcap.h
-a---           30/3/2017  1:17 am           3194 tilde.h
-a---           30/3/2017  1:17 am           1439 xmalloc.h

However, there are still errors when building the solution.
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	C2011	'timeval': 'struct' type redefinition	libavrdude	C:\Program Files (x86)\Windows Kits\10\Include\10.0.22000.0\um\winsock.h	108	
Message		see declaration of 'timeval'	libavrdude	C:\work\avr\avrdude_test\avrdude_main\src\msvc\sys\time.h	27	
Error	LNK1181	cannot open input file 'Release\libavrdude.lib'	avrdude	C:\work\avr\avrdude_test\avrdude_main\build\src\LINK	1	
```

So I gave up in the end.

mcuee, 2022-12-07:
All in all, vcpkg is not supported as of now. If someone figures out the solution, please update here.

mcuee, 2022-12-07:
BTW, if you do not use libusb-1.0 and just Readline, then it works.

```
PS C:\work\avr\avrdude_test\avrdude_main> & 'C:\Program Files\CMake\bin\cmake.exe' -D HAVE_LIBREADLINE:FILEPATH=C:\src\vcpkg\vcpkg\packages\readline-win32_x64-windows\lib\readline.lib -D USE_EXTERNAL=TRUE -B build
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22621.
-- The C compiler identification is MSVC 19.31.31105.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.33.0.windows.2")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "3.7.4")
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.31.31105.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using external library 'libelf'
-- Using external library 'libusb'
-- Using external library 'libhidapi'
-- Using external library 'libftdi'
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DON'T HAVE libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi
-- DON'T HAVE libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude_main/build

```

mcuee, 2022-12-07:
> However, there are still errors when building the solution.
>
> ```
> Severity	Code	Description	Project	File	Line	Suppression State
> Error	C2011	'timeval': 'struct' type redefinition	libavrdude	C:\Program [...]

The issue was the same as last time.
* https://github.com/avrdudes/avrdude/issues/961#issue-1235316964

mcuee, 2022-12-07:
A few reasons why this is a low priority.
1) [avrdude-libusb](https://github.com/avrdudes/libusb) can be used for MSVC build, it supports devices using libusb0.sys or WinUSB driver.
2) MSYS2 MinGW build supports libusb-1.0

mcuee, 2022-12-07:
The reason is explained here: `This problem is caused when including <windows.h> before <winsock2.h>. `
https://stackoverflow.com/questions/1372480/c-redefinition-header-files-winsock2-h

Why is this not caught by github action? That is because the default MSVC build does not use libusb-1.0, rather use avrdude-libusb.

The following simple fix is good to sort out the issue.
```
PS C:\work\avr\avrdude_test\avrdude_pr1207> git diff
diff --git a/src/usbasp.c b/src/usbasp.c
index 35d3a2e..dff3470 100644
--- a/src/usbasp.c
+++ b/src/usbasp.c
@@ -31,7 +31,7 @@
 #include <string.h>
 #include <stdint.h>
 #include <errno.h>
-#include <sys/time.h>
+
 #include <unistd.h>

 #include "avrdude.h"
@@ -62,6 +62,8 @@
 # endif
 #endif

+#include <sys/time.h>
+
 #ifdef USE_LIBUSB_1_0

 static libusb_context *ctx = NULL;
```

@mariusgreuel 
Any objection to the above simple fix? I can create a PR for this one if you think this is okay.

mcuee, 2022-12-07:
But the header file detection for vcpkg files are still a problem (libub-1.0 and readline). What I did (to copy files to `src\msvc`) is just a hack.

mcuee, 2022-12-07:
@am-ar 

Please try the above simple patch to `usbasp.c` and see if you can build avrdude with libusb-1.0 support using MSVC. Thanks.

mcuee, 2022-12-08:
If I use the similar command as @am-ar, I am able to reproduce the same issue for git main. Looks like current CMake scripts can not find libusb-1.0 and readline.

```
PS C:\work\avr\avrdude_test\avrdude_main> & 'C:\Program Files\CMake\bin\cmake.exe' --no-warn-unused-cli 
-DUSE_EXTERNAL=TRUE 
-DCMAKE_TOOLCHAIN_FILE=C:\src\vcpkg\vcpkg\scripts\buildsystems\vcpkg.cmake 
-DCMAKE_EXPORT_COMPILE_COMMANDS=TRUE -S. -G "Visual Studio 17 2022" 
-T host=x64 -A x64 -D USE_EXTERNAL=TRUE -B build

Not searching for unused variables given on the command line.
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22621.
-- The C compiler identification is MSVC 19.31.31105.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.33.0.windows.2")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "3.7.4")
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - not found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - not found
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.31.31105.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using external library 'libelf'
-- Using external library 'libusb'
-- Using external library 'libhidapi'
-- Using external library 'libftdi'
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DON'T HAVE libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- DON'T HAVE libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude_main/build
```

mcuee, 2022-12-08:
Workaround is to explicitly point CMake to GNU Readline and libusb-1.0 vcpkg lib files.
```
PS C:\work\avr\avrdude_test\avrdude_main> & 'C:\Program Files\CMake\bin\cmake.exe' --no-warn-unused-cli 
-DUSE_EXTERNAL=TRUE -DCMAKE_TOOLCHAIN_FILE=C:\src\vcpkg\vcpkg\scripts\buildsystems\vcpkg.cmake 
-D HAVE_LIBUSB_1_0:FILEPATH=C:\src\vcpkg\vcpkg\packages\libusb_x64-windows\lib\libusb-1.0.lib 
-D HAVE_LIBREADLINE:FILEPATH=C:\src\vcpkg\vcpkg\packages\readline-win32_x64-windows\lib\readline.lib 
-DCMAKE_EXPORT_COMPILE_COMMANDS=TRUE -S. -G "Visual Studio 17 2022" 
-T host=x64 -A x64 -D USE_EXTERNAL=TRUE -B build_external

Not searching for unused variables given on the command line.
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22621.
-- The C compiler identification is MSVC 19.31.31105.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.33.0.windows.2")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "3.7.4")
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - not found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - not found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - not found
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.31.31105.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using external library 'libelf'
-- Using external library 'libusb'
-- Using external library 'libhidapi'
-- Using external library 'libftdi'
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude_main/build_external

```

mcuee, 2022-12-08:
Full steps.
1) Install necessary vcpkg packages, do not install libftdi1 if we want to use avrdude-libftdi.
```
PS C:\src> .\vcpkg\vcpkg\vcpkg.exe list
hidapi:x64-windows                                 0.11.2#1         A Simple library for communicating with USB and ...
libusb:x64-windows                                 1.0.26           a cross-platform library to access USB devices
readline-win32:x64-windows                         5.0#5            Implementation of readline for Windows Desktop
readline:x64-windows                               0#3              GNU readline and history libraries
vcpkg-cmake-config:x64-windows                     2022-02-06
vcpkg-cmake:x64-windows                            2022-05-10
```

2) apply the work-around above (copy the necessary header files to `src\msvc`) since we also want to use libusb-1.0.

3) apply the following patch to `usbasp.c`.
```
PS C:\work\avr\avrdude_test\avrdude_pr1207> git diff
diff --git a/src/usbasp.c b/src/usbasp.c
index 35d3a2e..dff3470 100644
--- a/src/usbasp.c
+++ b/src/usbasp.c
@@ -31,7 +31,7 @@
 #include <string.h>
 #include <stdint.h>
 #include <errno.h>
-#include <sys/time.h>
+
 #include <unistd.h>

 #include "avrdude.h"
@@ -62,6 +62,8 @@
 # endif
 #endif

+#include <sys/time.h>
+
 #ifdef USE_LIBUSB_1_0

 static libusb_context *ctx = NULL;
```

3) use the following command or similar.
```
PS C:\work\avr\avrdude_test\avrdude_main> & 'C:\Program Files\CMake\bin\cmake.exe' --no-warn-unused-cli -DUSE_EXTERNAL=TRUE -DCMAKE_TOOLCHAIN_FILE=C:\src\vcpkg\vcpkg\scripts\buildsystems\vcpkg.cmake -D HAVE_LIBUSB_1_0:FILEPATH=C:\src\vcpkg\vcpkg\packages\libusb_x64-windows\lib\libusb-1.0.lib -D HAVE_LIBREADLINE:FILEPATH=C:\src\vcpkg\vcpkg\packages\readline-win32_x64-windows\lib\readline.lib -DCMAKE_EXPORT_COMPILE_COMMANDS=TRUE -S. -G "Visual Studio 17 2022" -T host=x64 -A x64
 -D USE_EXTERNAL=TRUE -B build_external
Not searching for unused variables given on the command line.
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22621.
-- The C compiler identification is MSVC 19.31.31105.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.33.0.windows.2")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "3.7.4")
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - not found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - not found
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.31.31105.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using external library 'libelf'
-- Using external library 'libusb'
-- Using external library 'libhidapi'
-- Using external library 'libftdi'
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi
-- DON'T HAVE libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude_main/build_external
```
4) Build the solution using VS2022. You can also use msbuild if you like.

mcuee, 2022-12-08:
The result binary works fine. Take note you need to copy the file `avrdude.conf` into the directory.

```
PS C:\work\avr\avrdude_test\avrdude_bin\avrdude_vcpkg_libusb1_readline> ls

    Directory: C:\work\avr\avrdude_test\avrdude_bin\avrdude_vcpkg_libusb1_readline

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---           8/12/2022  8:51 am         547542 avrdude.conf
-a---           8/12/2022  8:56 am         941056 avrdude.exe
-a---           8/12/2022  8:56 am           3232 avrdude.exp
-a---           8/12/2022  8:56 am           5822 avrdude.lib
-a---           8/12/2022  8:56 am        2461462 libavrdude.lib
-a---           13/5/2022  9:40 pm         174080 libusb-1.0.dll
-a---           17/7/2022 11:18 am         176128 readline.dll

PS C:\work\avr\avrdude_test\avrdude_bin\avrdude_vcpkg_libusb1_readline> .\avrdude -c urclock -P COM1 -p m328p -xbootsize=512 -qqt
avrdude> dump flash 0 0x20
>>> dump flash 0 0x20
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> dump flash 0x7e00 0x20
>>> dump flash 0x7e00 0x20
7e00  11 24 84 b7 14 be 81 ff  f0 d0 85 e0 80 93 81 00  |.$..............|
7e10  82 e0 80 93 c0 00 88 e1  80 93 c1 00 86 e0 80 93  |................|

avrdude> quit
>>> quit
avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin\avrdude_vcpkg_libusb1_readline> .\avrdude
 -c urclock -P COM1 -p m328p -xbootsize=512 -xshowall

avrdude: AVR device initialized and ready to accept instructions
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 o4.4 -?s-?-r-- vector 0 (RESET) ATmega328P

```
[avrdude_vcpkg_libusb1_readline.zip](https://github.com/avrdudes/avrdude/files/10181260/avrdude_vcpkg_libusb1_readline.zip)

mcuee, 2023-01-06:
@mariusgreuel 

Please look at this one as well when you got the time. Or do you think we can close this as not planned?

mariusgreuel, 2023-01-06:
> Or do you think we can close this as not planned?

Yes. I'd say this request is outside the scope. Our MSVC build is only supported using the USE_EXTERNAL_LIBS switch and libraries.

@am-ar When setting up AVRDUDE for Windows I looked at vcpkg and other packages, too. But they were pretty much all unusable. You are welcome to submit a PR, though if you think this would add some extra value over our Windows build.

am-ar, 2023-01-06:
@mariusgreuel when I opened the issue I was trying to compile using Visual Studio, however after understanding that there was a complication due to the libraries I chose to use it via MinGW.

For me the result was satisfactory and I don't feel the need to work more on it. So much so that the result was PR #1215.

mariusgreuel, 2023-01-06:
@am-ar Thanks for the feedback. I am closing the issue then.

closed 2023-01-06 by mariusgreuel

mcuee, 2023-01-16:
Just a note to build using command line.
1) Open "Developer Powershell for VS2022"
2) Run log.
```
PS C:\work\avr\avrdude_test\avrdude_main> cmake.exe  -G "Ninja"  -DCMAKE_BUILD_TYPE:STRING="RelWithDebInfo" -DCMAKE_INSTALL_PREFIX:PATH="C:\work\avr\avrdude_test\avrdude_main\out\install\x64-Release" -DUSE_EXTERNAL_LIBS:BOOL="true" -DCMAKE_C_COMPILER:FILEPATH="C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe"  -DCMAKE_MAKE_PROGRAM="C:\PROGRAM FILES\MICROSOFT VISUAL STUDIO\2022\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\Ninja\ninja.exe" -B build_vs2022
-- The C compiler identification is MSVC 19.31.31105.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.33.0.windows.2")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "3.7.4")
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.31.31105.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using external library 'libelf'
-- Using external library 'libusb'
-- Using external library 'libhidapi'
-- Using external library 'libftdi'
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DON'T HAVE libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi
-- DON'T HAVE libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude_main/build_vs2022
PS C:\work\avr\avrdude_test\avrdude_main> cmake --build build_vs2022
[137/137] Linking CXX executable src\avrdude.exe
PS C:\work\avr\avrdude_test\avrdude_main> cmake.exe  -G "Ninja"  -DCMAKE_BUILD_TYPE:STRING="RelWithDebInfo" -DCMAKE_INSTALL_PREFIX:PATH="C:\work\avr\avrdude_test\avrdude_main\out\install\x64-Release" -DUSE_EXTERNAL_LIBS:BOOL="true" -B build_vs2022_1
-- The C compiler identification is MSVC 19.31.31105.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.33.0.windows.2")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "3.7.4")
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.31.31105.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using external library 'libelf'
-- Using external library 'libusb'
-- Using external library 'libhidapi'
-- Using external library 'libftdi'
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DON'T HAVE libusb_1_0
[... 82 lines cut ...]
  -c <programmer>            Specify programmer type
  -c <wildcard>/<flags>      Run developer options for matched programmers
  -A                         Disable trailing-0xff removal from file and AVR read
  -D                         Disable auto erase for flash memory; implies -A
  -i <delay>                 ISP Clock Delay [in microseconds]
  -P <port>                  Specify connection port
  -F                         Override invalid signature or initialisation check
  -e                         Perform a chip erase
  -O                         Perform RC oscillator calibration (see AVR053)
  -U <memtype>:r|w|v:<filename>[:format]
                             Memory operation specification
                             Multiple -U options are allowed, each request
                             is performed in the order specified
  -n                         Do not write anything to the device
  -V                         Do not verify
  -t                         Enter terminal mode
  -E <exitspec>[,<exitspec>] List programmer exit specifications
  -x <extended_param>        Pass <extended_param> to programmer
  -v                         Verbose output; -v -v for more
  -q                         Quell progress output; -q -q for less
  -l logfile                 Use logfile rather than stderr for diagnostics
  -?                         Display this usage

avrdude version 7.1-20230114 (5ec72e2), URL: <https://github.com/avrdudes/avrdude>
```

mcuee, 2023-01-17:
VS2019 build

```
c:\work\avr\avrdude_test\avrdude>cmake -D USE_EXTERNAL_LIBS=ON -B build_msvc
-- Building for: Visual Studio 16 2019
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
-- The C compiler identification is MSVC 19.29.30147.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.35.3.windows.1")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "3.7.4")
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.29.30147.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using external library 'libelf'
-- Using external library 'libusb'
-- Using external library 'libhidapi'
-- Using external library 'libftdi'
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DON'T HAVE libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi
-- DON'T HAVE libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude/build_msvc

c:\work\avr\avrdude_test\avrdude>cmake --build build_msvc
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Generating avrdude.conf
  Building Custom Rule C:/work/avr/avrdude_test/avrdude/src/CMakeLists.txt
  Building Custom Rule C:/work/avr/avrdude_test/avrdude/build_msvc/_deps/libelf-src/CMakeLists.txt
  32.fsize.c
  32.getehdr.c
  32.getphdr.c
  32.getshdr.c
  32.newehdr.c
  32.newphdr.c
  32.xlatetof.c
  64.xlatetof.c
  assert.c
  begin.c
  checksum.c
  cntl.c
  cook.c
  data.c
  end.c
  errmsg.c
  errno.c
  fill.c
[... 133 lines cut ...]
  gettimeofday.c
C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\um\winbase.h(9531,5): warning C5105: macro expansion produc
ing 'defined' has undefined behavior [C:\work\avr\avrdude_test\avrdude\build_msvc\src\libavrdude.vcxproj]
  Generating Code...
  usleep.cpp
  readline.cpp
  Generating Code...
  libavrdude.vcxproj -> C:\work\avr\avrdude_test\avrdude\build_msvc\src\Debug\libavrdude.lib
  Building Custom Rule C:/work/avr/avrdude_test/avrdude/src/CMakeLists.txt
  main.c
  term.c
C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\um\winbase.h(9531,5): warning C5105: macro expansion produc
ing 'defined' has undefined behavior [C:\work\avr\avrdude_test\avrdude\build_msvc\src\avrdude.vcxproj]
  avrintel.c
  developer_opts.c
  whereami.c
C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\um\winbase.h(9531,5): warning C5105: macro expansion produc
ing 'defined' has undefined behavior [C:\work\avr\avrdude_test\avrdude\build_msvc\src\avrdude.vcxproj]
  Generating Code...
     Creating library C:/work/avr/avrdude_test/avrdude/build_msvc/src/Debug/avrdude.lib and object C:/work/avr/avrdude_
  test/avrdude/build_msvc/src/Debug/avrdude.exp
  avrdude.vcxproj -> C:\work\avr\avrdude_test\avrdude\build_msvc\src\Debug\avrdude.exe
  Building Custom Rule C:/work/avr/avrdude_test/avrdude/CMakeLists.txt

```

mcuee, 2023-01-17:
VS2019 build using Ninja.

```
c:\work\avr\avrdude_test\avrdude>cmake.exe  -G "Ninja"  -DCMAKE_BUILD_TYPE:STRING="RelWithDebInfo" -DUSE_EXTERNAL_LIBS:BOOL="true" -B build_vs2019
-- The C compiler identification is MSVC 19.29.30147.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.35.3.windows.1")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "3.7.4")
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.29.30147.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using external library 'libelf'
-- Using external library 'libusb'
-- Using external library 'libhidapi'
-- Using external library 'libftdi'
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DON'T HAVE libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi
-- DON'T HAVE libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude/build_vs2019

c:\work\avr\avrdude_test\avrdude>cmake --build build_vs2019
[100/137] Building C object src\CMakeFiles\libavrdude.dir\confwin.c.obj
C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um\winbase.h(9531): warning C5105: macro expansion producing 'defined' has undefined behavior
[102/137] Building C object src\CMakeFiles\libavrdude.dir\bitbang.c.obj
C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um\winbase.h(9531): warning C5105: macro expansion producing 'defined' has undefined behavior
[120/137] Building C object src\CMakeFiles\libavrdude.dir\serbb_win32.c.obj
C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um\winbase.h(9531): warning C5105: macro expansion producing 'defined' has undefined behavior
[122/137] Building C object src\CMakeFiles\libavrdude.dir\pickit2.c.obj
C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um\winbase.h(9531): warning C5105: macro expansion producing 'defined' has undefined behavior
[123/137] Building C object src\CMakeFiles\libavrdude.dir\ser_win32.c.obj
C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um\winbase.h(9531): warning C5105: macro expansion producing 'defined' has undefined behavior
[127/137] Building RC object src\CMakeFiles\avrdude.dir\windows.rc.res
Microsoft (R) Windows (R) Resource Compiler Version 10.0.10011.16384
Copyright (C) Microsoft Corporation.  All rights reserved.

[131/137] Building C object src\CMakeFiles\libavrdude.dir\msvc\gettimeofday.c.obj
C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um\winbase.h(9531): warning C5105: macro expansion producing 'defined' has undefined behavior
[132/137] Building C object src\CMakeFiles\libavrdude.dir\msvc\getopt.c.obj
C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um\winbase.h(9531): warning C5105: macro expansion producing 'defined' has undefined behavior
[134/137] Building C object src\CMakeFiles\avrdude.dir\whereami.c.obj
C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um\winbase.h(9531): warning C5105: macro expansion producing 'defined' has undefined behavior
[136/137] Building C object src\CMakeFiles\avrdude.dir\term.c.obj
C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um\winbase.h(9531): warning C5105: macro expansion producing 'defined' has undefined behavior
[137/137] Linking CXX executable src\avrdude.exe

```