avrdudes/avrdude #1212: JTAG3 programmer info not available in ISP mode

MCUdude, 2022-12-08:
When a device is programmed with a JTAG3-compatible programmer, Avrdude prints out some programmer-related information in verbose mode, like serial number and firmware version. However, this does not happen in ISP mode.

The fix is quite simple, and I plan to submit a PR that addresses this once #1205 has been merged.

This is the information that's lacking in ISP mode.
```
         ICE HW version  : 4
         ICE FW version  : 1.14 (rel. 5)
         Serial number   : 
         Vtarget         : 5.05 V
```

JTAG3 in ISP mode:
```
$ ./avrdude -cpickit4_isp -patmega16 -v

avrdude: Version 7.0-20220508
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/hans/Downloads/avrdude-terminal/src/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : usb
         Using Programmer              : pickit4_isp
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega16
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                  4    10   128    0 no        512    4      0  9000  9000 0xff 0xff
           flash                  33     6   128    0 yes     16384  128    128  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

         Programmer Type : JTAG3_ISP
         Description     : MPLAB(R) PICkit 4 in ISP mode
         Vtarget         : 5.07 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
         SCK period                   : 8.00 us

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9403 (probably m16)

avrdude done.  Thank you.
```

JTAG3 in UPDI mode:
```
$ ./avrdude -cpickit4_updi -patmega4808 -v

avrdude: Version 7.0-20220508
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/hans/Downloads/avrdude-terminal/src/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : usb
         Using Programmer              : pickit4_updi
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4808
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         64   64      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   64      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      49152  128      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : MPLAB(R) PICkit 4 in UPDI mode
         ICE HW version  : 4
         ICE FW version  : 1.14 (rel. 5)
         Serial number   : 
         Vtarget         : 5.05 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: partial Family_ID returned: "mega"
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)

avrdude done.  Thank you.
```

mcuee, 2022-12-14:
Indeed this is an issue.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c pickit4_isp -p m8a -v

avrdude_git: Version 7.0-20221214 (159d126)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : usb
             Using Programmer              : pickit4_isp
avrdude_git: found CMSIS-DAP compliant device, using EDBG protocol
             AVR Part                      : ATmega8A
             Chip Erase delay              : 10000 us
......
             Programmer Type : JTAG3_ISP
             Description     : MPLAB(R) PICkit 4 in ISP mode
             Vtarget         : 4.90 V
             JTAG clock megaAVR/program   : 1000 kHz
             JTAG clock megaAVR/debug     : 100 kHz
             PDI/UPDI clock Xmega/megaAVR : 100 kHz
             SCK period                   : 8.00 us

avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e9307 (probably m8a)

avrdude_git done.  Thank you.

```

mcuee, 2022-12-16:
@MCUdude 

Now that #1205 has been merged, you can now create a PR to fix this issue.

mcuee, 2022-12-17:
This is now addressed in PR #1223 as well.

closed 2022-12-21 by stefanrueger