avrdudes/avrdude #1213: Still confused where to collect my AVRDUDE libraries

kristofmulier, 2022-12-10:
Hi @mcuee ,
I discovered your new GitHub repo `avrdude-packing`. It's great you did this, I applaud the effort. However, right now, I see the following sources where I can collect AVRDUDE binaries:
 - https://github.com/arduino/avrdude-packing/releases
 - https://github.com/avrdudes/avrdude/releases
 - https://github.com/mcuee/avrdude-packing/releases

Please help me - where should I collect the binaries? How do I make the decision?

PS: I need binaries that *just work*, without installation. It's for our [Embeetle IDE](https://embeetle.com) project, and IDE that works on both Windows and Linux. Embeetle should just drop a zipped folder on the user's computer (in any location the user wishes), then unzip the folder and invoke the `avrdude` binary - no installation required.

kristofmulier, 2022-12-10:
Thank you @mcuee for your clarifications.

I currently need Linux and Windows. Not yet MacOS (but perhaps later, I do as well). Unfortunately, the official release on https://github.com/avrdudes/avrdude/releases only offers Windows binaries. Any idea if that would ever change?

For the moment, I'll probably stick with the Arduino release at: https://github.com/arduino/avrdude-packing/releases

Kind regards,
Kristof

mcuee, 2022-12-11:
> I currently need Linux and Windows. Not yet MacOS (but perhaps later, I do as well). Unfortunately, the official release on https://github.com/avrdudes/avrdude/releases only offers Windows binaries. Any idea if that would ever change?

I do not think that it will happen anytime soon. Take note the two required patches are not in a good shape to be adopted by avrdude project yet.

Patch 06 may need some work. Patch 07 is just a hack.We need CMake experts like @mariusgreuel to come out with the real solution.

https://github.com/arduino/avrdude-packing/tree/main/patches
Patch 06: https://github.com/arduino/avrdude-packing/blob/main/patches/0006-Cmake-add-libtinfo-and-libncurses-as-dependencies.patch
Patch 07: https://github.com/arduino/avrdude-packing/blob/main/patches/0007-Cmake-fix-build-not-working-with-mingw-toolchain.patch

(Patch 08 is just Arduino branding)

> For the moment, I'll probably stick with the Arduino release at: https://github.com/arduino/avrdude-packing/releases

Yes that is the way to go.

mcuee, 2022-12-14:
I will close this issue. For users who are interested in this topic, please go to #987.
* #987

closed 2022-12-14 by mcuee