avrdudes/avrdude #1221: PICKit 4  only works with hidapi and not libusb

mcuee, 2022-12-16:
From here:
* https://github.com/avrdudes/avrdude/issues/1038#issuecomment-1354195120

For this test, I build avrdude without hidapi support. And I make sure the hid drver is already detached and then libusb is used.

hidapi version works fine (with hidapi-libusb backend). So there is a bug here in git main.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_main$ cat build_nohidapi.sh 
#!/bin/sh
cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -D HAVE_LIBHIDAPI=OFF  -B build_linux
cmake --build build_linux

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ldd ./avrdude_git_nohidapi 
	linux-vdso.so.1 (0x00007ffc52b2f000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fa6169f6000)
	libelf.so.1 => /lib/x86_64-linux-gnu/libelf.so.1 (0x00007fa6169da000)
	libusb-0.1.so.4 => /usr/local/lib/libusb-0.1.so.4 (0x00007fa6169d2000)
	libusb-1.0.so.0 => /usr/local/lib/libusb-1.0.so.0 (0x00007fa6169b2000)
	libftdi.so.1 => /lib/x86_64-linux-gnu/libftdi.so.1 (0x00007fa6169a8000)
	libftdi1.so.2 => /usr/local/lib/libftdi1.so.2 (0x00007fa616995000)
	libreadline.so.8 => /lib/x86_64-linux-gnu/libreadline.so.8 (0x00007fa616945000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fa616753000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fa616c5f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fa616737000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fa616714000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fa6166e7000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007fa6166b5000)

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_git_nohidapi -c pickit4_updi -p m4808 -v

avrdude_git_nohidapi: Version 7.0-20221214 (159d126)
                      Copyright the AVRDUDE authors;
                      see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                      System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude.conf
                      User configuration file is /home/mcuee/.avrduderc
                      User configuration file does not exist or is not a regular file, skipping

                      Using Port                    : usb
                      Using Programmer              : pickit4_updi
avrdude_git_nohidapi: usbdev_open(): found MPLAB PICkit 4 CMSIS-DAP, serno: BUR200973052
avrdude_git_nohidapi usbdev_open() warning: unable to set configuration 1: Device or resource busy
avrdude_git_nohidapi: found CMSIS-DAP compliant device, using EDBG protocol
avrdude_git_nohidapi usbdev_send() error: wrote -5 out of 912 bytes, err = Input/output error
avrdude_git_nohidapi jtag3_edbg_prepare() error: unable to send command to serial port
avrdude_git_nohidapi main() error: unable to open programmer pickit4_updi on port usb

avrdude_git_nohidapi done.  Thank you.
```

mcuee, 2022-12-16:
Same result under Windows.

I built avrdude without hidapi and readline support in this test.

I also tried with the default Windows HID driver (libusb-1.0 under Windows supports HID driver), or with WinUSB driver. The results are the same.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git_nohidapi -c pickit4_updi -p m4808 -v

avrdude_git_nohidapi: Version 7.0-20221214 (159d126)
                      Copyright the AVRDUDE authors;
                      see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                      System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

                      Using Port                    : usb
                      Using Programmer              : pickit4_updi
avrdude_git_nohidapi: usbdev_open(): found MPLAB PICkit 4 CMSIS-DAP, serno: BUR200973052
avrdude_git_nohidapi usbdev_open() error: SET_IDLE failed
avrdude_git_nohidapi: found CMSIS-DAP compliant device, using EDBG protocol
avrdude_git_nohidapi usbdev_send() error: wrote -2 out of 912 bytes, err = No such file or directory
avrdude_git_nohidapi jtag3_edbg_prepare() error: unable to send command to serial port
avrdude_git_nohidapi main() error: unable to open programmer pickit4_updi on port usb

avrdude_git_nohidapi done.  Thank you.
```

MCUdude, 2022-12-16:
I'll assume the SNAP is also affected.

But how about other, "older" JTAG3 compatible programmers such as the Xplained Mini ATtiny817?

mcuee, 2022-12-16:
> I'll assume the SNAP is also affected.
> 
> But how about other, "older" JTAG3 compatible programmers such as the Xplained Mini ATtiny817?

It seems to work with WinUSB driver.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git_nohidapi -c xplainedpro_updi -p t817

                      Vtarget                      : 3.34 V
                      JTAG clock megaAVR/program   : 1000 kHz
                      JTAG clock megaAVR/debug     : 1000 kHz
                      JTAG clock Xmega             : 1000 kHz
                      PDI/UPDI clock Xmega/megaAVR : 1000 kHz
avrdude_git_nohidapi: AVR device initialized and ready to accept instructions
avrdude_git_nohidapi: device signature = 0x1e9320 (probably t817)

avrdude_git_nohidapi done.  Thank you.
```

mcuee, 2022-12-16:
> > I'll assume the SNAP is also affected.
> > But how about other, "older" JTAG3 compatible programmers such as the Xplained Mini ATtiny817?
> 
> It seems to work with WinUSB driver. I am using the Xplained Pro Attiny817 for this test.

It also works under Linux even though there is a warning.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_git -c xplainedpro_updi -p t817 -v

avrdude_git: Version 7.0-20221214 (159d126)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude.conf
             User configuration file is /home/mcuee/.avrduderc
             User configuration file does not exist or is not a regular file, skipping

             Using Port                    : usb
             Using Programmer              : xplainedpro_updi
avrdude_git: found CMSIS-DAP compliant device, using EDBG protocol
             AVR Part                      : ATtiny817
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             ....
             Programmer Type : JTAGICE3_UPDI
             Description     : Atmel AVR XplainedPro in UPDI mode
             ICE HW version  : 240
             ICE FW version  : 3.34 (rel. 435)
             Serial number   : ATMLxxxxxxxxxxxxxxxxxxx
             Vtarget         : 3.34 V
             JTAG clock megaAVR/program   : 1000 kHz
             JTAG clock megaAVR/debug     : 1000 kHz
             JTAG clock Xmega             : 1000 kHz
             PDI/UPDI clock Xmega/megaAVR : 1000 kHz
avrdude_git: partial Family_ID returned: "tiny"
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e9320 (probably t817)

avrdude_git done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_git_nohidapi -c xplainedpro_updi -p t817 -qqt
avrdude_git_nohidapi warning: unable to set configuration 1: Device or resource busy
avrdude> dump fuses
0000  00 00 02 ff 00 c4 04 00  02                       |.........       |

avrdude> quit
avrdude> mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ 

```

MCUdude, 2022-12-16:
> It also works under Linux even though there is a warning.

Cool! I think we're on our own though. IIRC, no JTAG3 programmer will work with Microchip Studio if not using the hidapi driver.

mcuee, 2022-12-16:
> Cool! I think we're on our own though. IIRC, no JTAG3 programmer will work with Microchip Studio if not using the hidapi driver.

Yes we are on our own as Microchip Studio and MPLAB X use native Windows HID API under Windows and not hidapi/libusb. But I think we should be able to figure out the issue as hidapi-libusb works under Linux -- therefore libusb should also work.

And this is not an important issue either as HIDAPI works, so Windows/macOS/Linux all work. FreeBSD should also work.

mcuee, 2022-12-16:
BTW, this is not only applicable to UPDI, but also ISP.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git_nohidapi -c pickit4_isp -p m8a -v

avrdude_git_nohidapi: Version 7.0-20221214 (159d126)
                      Copyright the AVRDUDE authors;
                      see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                      System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

                      Using Port                    : usb
                      Using Programmer              : pickit4_isp
avrdude_git_nohidapi: usbdev_open(): found MPLAB PICkit 4 CMSIS-DAP, serno: BUR2009xxxxx
avrdude_git_nohidapi usbdev_open() error: SET_IDLE failed
avrdude_git_nohidapi: found CMSIS-DAP compliant device, using EDBG protocol
avrdude_git_nohidapi usbdev_send() error: wrote -2 out of 912 bytes, err = No such file or directory
avrdude_git_nohidapi jtag3_edbg_prepare() error: unable to send command to serial port
avrdude_git_nohidapi stk500v2_jtag3_open() error: unable to sync with the JTAGICE3 in ISP mode
avrdude_git_nohidapi main() error: unable to open programmer pickit4_isp on port usb

avrdude_git_nohidapi done.  Thank you.
```

mcuee, 2022-12-16:
> I'll assume the SNAP is also affected.

Yes it is also affected.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git_nohidapi -c snap_isp -p m8a -v

avrdude_git_nohidapi: Version 7.0-20221214 (159d126)
                      Copyright the AVRDUDE authors;
                      see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                      System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

                      Using Port                    : usb
                      Using Programmer              : snap_isp
avrdude_git_nohidapi: usbdev_open(): found MPLAB Snap ICD CMSIS-DAP, serno: BURxxxxxxxxx
avrdude_git_nohidapi: found CMSIS-DAP compliant device, using EDBG protocol
avrdude_git_nohidapi usbdev_send() error: wrote -2 out of 912 bytes, err = No such file or directory
avrdude_git_nohidapi jtag3_edbg_prepare() error: unable to send command to serial port
avrdude_git_nohidapi stk500v2_jtag3_open() error: unable to sync with the JTAGICE3 in ISP mode
avrdude_git_nohidapi main() error: unable to open programmer snap_isp on port usb

avrdude_git_nohidapi done.  Thank you.
```

mcuee, 2022-12-16:
More detailed debug log under Linux.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ sudo ./avrdude_git_nohidapi -c pickit4_isp -p m8a -vvvv
executable_abspath = /home/mcuee/build/avr/avrdude_bin/avrdude_git_nohidapi
executable_abspath_len = 54
executable_dirpath = /home/mcuee/build/avr/avrdude_bin
executable_dirpath_len = 33
sys_config = /home/mcuee/build/avr/avrdude_bin/avrdude.conf
sys_config_found = true

avrdude_git_nohidapi: Version 7.0-20221214 (159d126)
                      Copyright the AVRDUDE authors;
                      see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                      System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude.conf
                      User configuration file is /root/.avrduderc
                      User configuration file does not exist or is not a regular file, skipping

                      Using Port                    : usb
                      Using Programmer              : pickit4_isp
avrdude_git_nohidapi: stk500v2_jtag3_open()
avrdude_git_nohidapi: usbdev_open(): found MPLAB PICkit 4 CMSIS-DAP, serno: BUR2009xxxxx
avrdude_git_nohidapi usbdev_open() [usb_libusb.c:197] warning: unable to set configuration 1: Device or resource busy
avrdude_git_nohidapi: found CMSIS-DAP compliant device, using EDBG protocol
avrdude_git_nohidapi: jtag3_getsync()

avrdude_git_nohidapi: jtag3_edbg_prepare()
avrdude_git_nohidapi usbdev_send() [usb_libusb.c:344] error: wrote -5 out of 912 bytes, err = Input/output error
avrdude_git_nohidapi jtag3_edbg_prepare() [jtag3.c:542] error: unable to send command to serial port
avrdude_git_nohidapi stk500v2_jtag3_open() [stk500v2.c:3510] error: unable to sync with the JTAGICE3 in ISP mode
avrdude_git_nohidapi main() [main.c:1123] error: unable to open programmer pickit4_isp on port usb

avrdude_git_nohidapi done.  Thank you.

```

mcuee, 2022-12-16:
Simple troubleshooting patch (to bypass the first warning and error) does not help too much.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude_main$ git diff
diff --git a/src/usb_libusb.c b/src/usb_libusb.c
index 858db21..e12046a 100644
--- a/src/usb_libusb.c
+++ b/src/usb_libusb.c
@@ -192,13 +192,13 @@ static int usbdev_open(const char *port, union pinfo pinfo, union filedescriptor
                      goto trynext;
                    }
 
-                 if (usb_set_configuration(udev, dev->config[0].bConfigurationValue))
-                   {
-                     pmsg_warning("unable to set configuration %d: %s\n",
-                        dev->config[0].bConfigurationValue, usb_strerror());
+                 //if (usb_set_configuration(udev, dev->config[0].bConfigurationValue))
+                 //  {
+                 //    pmsg_warning("unable to set configuration %d: %s\n",
+                   //     dev->config[0].bConfigurationValue, usb_strerror());
                      /* let's hope it has already been configured */
                      // goto trynext;
-                   }
+                 //  }
 
                  for (iface = 0; iface < dev->config[0].bNumInterfaces; iface++)
                    {
@@ -341,7 +341,7 @@ static int usbdev_send(const union filedescriptor *fd, const unsigned char *bp,
       rv = usb_bulk_write(udev, fd->usb.wep, (char *)bp, tx_size, 10000);
     if (rv != tx_size)
     {
-        pmsg_error("wrote %d out of %d bytes, err = %s\n", rv, tx_size, usb_strerror());
+        pmsg_warning("wrote %d out of %d bytes, err = %s\n", rv, tx_size, usb_strerror());
         return -1;
     }
     bp += tx_size;

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ sudo ./avrdude_issue1221 -c pickit4_isp -p m8a -vvvv
executable_abspath = /home/mcuee/build/avr/avrdude_bin/avrdude_issue1221
executable_abspath_len = 51
executable_dirpath = /home/mcuee/build/avr/avrdude_bin
executable_dirpath_len = 33
sys_config = /home/mcuee/build/avr/avrdude_bin/avrdude.conf
sys_config_found = true

avrdude_issue1221: Version 7.0-20221214 (159d126)
                   Copyright the AVRDUDE authors;
                   see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                   System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude.conf
                   User configuration file is /root/.avrduderc
                   User configuration file does not exist or is not a regular file, skipping

                   Using Port                    : usb
                   Using Programmer              : pickit4_isp
avrdude_issue1221: stk500v2_jtag3_open()
avrdude_issue1221: usbdev_open(): found MPLAB PICkit 4 CMSIS-DAP, serno: BUR200973052
avrdude_issue1221: found CMSIS-DAP compliant device, using EDBG protocol
avrdude_issue1221: jtag3_getsync()

avrdude_issue1221: jtag3_edbg_prepare()
avrdude_issue1221 usbdev_send() [usb_libusb.c:344] warning: wrote -5 out of 912 bytes, err = Input/output error
avrdude_issue1221 jtag3_edbg_prepare() [jtag3.c:542] error: unable to send command to serial port
avrdude_issue1221 stk500v2_jtag3_open() [stk500v2.c:3510] error: unable to sync with the JTAGICE3 in ISP mode
avrdude_issue1221 main() [main.c:1123] error: unable to open programmer pickit4_isp on port usb

avrdude_issue1221 done.  Thank you.

```

mcuee, 2023-05-12:
Somehow for PICKit 4, the warning becomes an error when using WinUSB driver/libusb. Need to dig further.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c pickit4_isp -p m8a -v

avrdude: Version 7.1-20230504 (e6a2f3c0)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

         Using Port                    : usb
         Using Programmer              : pickit4_isp
avrdude usbhid_open() warning: USB device with VID: 0x03eb and PID: 0x2177 not found
avrdude usbhid_open() warning: USB device with VID: 0x03eb and PID: 0x2178 not found
avrdude usbhid_open() warning: USB device with VID: 0x03eb and PID: 0x2179 not found
avrdude: usbdev_open(): found MPLAB PICkit 4 CMSIS-DAP, serno: BUR200973052
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude usbdev_send() error: wrote -2 out of 912 bytes, err = No such file or directory
avrdude jtag3_edbg_prepare() error: unable to send command to serial port
avrdude stk500v2_jtag3_open() error: unable to sync with the JTAGICE3 in ISP mode
avrdude main() error: unable to open programmer pickit4_isp on port usb

avrdude done.  Thank you.
```

mcuee, 2023-05-12:
In any case, this is not of high priority since the interface is an HID interface and we should use HIDAPI anyway and not libusb.

mcuee, 2023-05-12:
From hidapi hidtest utility (under Windows, which only generate simulated HID report, I will test under Linux later which will dump the real HID report.

Input report size should only be 64 bytes. No idea why `912 bytes` was there in the libusb output.

```
Device Found
  type: 03eb 2177
  path: \\?\HID#VID_03EB&PID_2177&MI_00#8&36594c76&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: BUR200973052
  Manufacturer: Microchip Technology Incorporated
  Product:      MPLAB PICkit 4 CMSIS-DAP
  Release:      100
  Interface:    0
  Usage (page): 0x1 (0xff00)
  Bus type: 1 (USB)

  Report Descriptor: (47 bytes)
0x06, 0x00, 0xff, 0x09, 0x01, 0xa1, 0x01, 0x09, 0x01, 0x15,
0x00, 0x26, 0xff, 0x00, 0x75, 0x08, 0x95, 0x40, 0x81, 0x02,
0x09, 0x01, 0x15, 0x00, 0x26, 0xff, 0x00, 0x75, 0x08, 0x95,
0x40, 0x91, 0x02, 0x09, 0x01, 0x15, 0x00, 0x26, 0xff, 0x00,
0x75, 0x08, 0x95, 0x04, 0xb1, 0x02, 0xc0,
```

Feed the above through an online HID report parser
https://eleccelerator.com/usbdescreqparser/

```
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x09, 0x01,        //   Usage (0x01)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x40,        //   Report Count (64)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x01,        //   Usage (0x01)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x40,        //   Report Count (64)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x09, 0x01,        //   Usage (0x01)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x04,        //   Report Count (4)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection

// 47 bytes
```

mcuee, 2023-05-12:
Full debug log under Linux using hidapi.

<details>
<summary> Correct run log with hidapi </summary>

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c pickit4_isp -p m8a -vvvv

avrdude: Version 7.1-20230504 (e6a2f3c0)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude.conf
         User configuration file is /home/mcuee/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : pickit4_isp
avrdude: stk500v2_jtag3_open()
avrdude: usbhid_open(): probing for max packet size
avrdude: usbhid_open(): setting max_xfer from DAP_Info response to 64
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_getsync()

avrdude: jtag3_edbg_prepare()
avrdude: jtag3_edbg_prepare(): connection status 0x01
avrdude: sending sign-on command: 
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 0 (command_sequence == 0)

Raw message:
01  80  00  
[general] OK
         AVR Part                      : ATmega8A
         Chip Erase delay              : 10000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                  4    20   128    0 no        512    4      0  9000  9000 0xff 0xff
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                  33    10    64    0 yes      8192   64    128  4500  4500 0xff 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse                   0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse                   0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock                    0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
[... 120 lines cut ...]
avrdude: stk500v2_jtag3_close()
avrdude: jtag3_close()
avrdude: sending AVR sign-off command: 
avrdude: jtag3_edbg_send(): sending 3 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 15 (command_sequence == 15)

Raw message:
12  80  00  
[AVR] OK
avrdude: sending sign-off command: 
avrdude: jtag3_edbg_send(): sending 4 bytes
avrdude: jtag3_edbg_recv():
avrdude: jtag3_recv(): Got message seqno 16 (command_sequence == 16)

Raw message:
01  80  00  
[general] OK

avrdude: jtag3_edbg_signoff()

avrdude done.  Thank you.
```

</details>

mcuee, 2023-05-12:
I will need to debug the issue under Linux a bit further. The error output is actually as expected since avrdude did not detach the kernel HID driver, that is why we have the warning like this.
```
usbdev_open() warning: unable to set configuration 1: Device or resource busy
```

I will try later to manually detach the kernel HID driver.

mcuee, 2023-05-13:
The following is the step to manually unbind kernel usbhid driver.

Ref: Manual driver binding and unbinding
https://lwn.net/Articles/143397/

```
mcuee@UbuntuSwift3:/sys/bus/usb/drivers/usbhid$ ls
3-2.1:1.0  3-3:1.0  3-3:1.1  3-3:1.2  bind  module  new_id  remove_id  uevent  unbind
mcuee@UbuntuSwift3:/sys/bus/usb/drivers/usbhid$ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 10, If 0, Class=Human Interface Device, Driver=usbhid, 480M
        |__ Port 1: Dev 10, If 1, Class=Communications, Driver=cdc_acm, 480M
        |__ Port 1: Dev 10, If 2, Class=CDC Data, Driver=cdc_acm, 480M
    |__ Port 3: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 3: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 3: Dev 3, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 5: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 6, If 0, Class=Vendor Specific Class, Driver=, 480M
    |__ Port 10: Dev 7, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 10: Dev 7, If 1, Class=Wireless, Driver=btusb, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
mcuee@UbuntuSwift3:/sys/bus/usb/drivers/usbhid$ cd 3-2.1\:1.0
mcuee@UbuntuSwift3:/sys/bus/usb/drivers/usbhid/3-2.1:1.0$ ls
0003:03EB:2177.0016  bAlternateSetting  bInterfaceNumber    bInterfaceSubClass  driver  ep_81     power      supports_autosuspend  usbmisc
authorized           bInterfaceClass    bInterfaceProtocol  bNumEndpoints       ep_02   modalias  subsystem  uevent

mcuee@UbuntuSwift3:/sys/bus/usb/drivers/usbhid$ tree /sys/bus/usb/drivers/usbhid/
/sys/bus/usb/drivers/usbhid/
├── 3-2.1:1.0 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.1/3-2.1:1.0
├── 3-3:1.0 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0
├── 3-3:1.1 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.1
├── 3-3:1.2 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2
├── bind
├── module -> ../../../../module/usbhid
├── new_id
├── remove_id
├── uevent
└── unbind

5 directories, 5 files
mcuee@UbuntuSwift3:/sys/bus/usb/drivers/usbhid$ sudo su
root@UbuntuSwift3:/sys/bus/usb/drivers/usbhid# echo -n "3-2.1:1.0" >/sys/bus/usb/drivers/usbhid/unbind 
root@UbuntuSwift3:/sys/bus/usb/drivers/usbhid# lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 10, If 0, Class=Human Interface Device, Driver=, 480M
        |__ Port 1: Dev 10, If 1, Class=Communications, Driver=cdc_acm, 480M
        |__ Port 1: Dev 10, If 2, Class=CDC Data, Driver=cdc_acm, 480M
    |__ Port 3: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 3: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 3: Dev 3, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 5: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 6, If 0, Class=Vendor Specific Class, Driver=, 480M
    |__ Port 10: Dev 7, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 10: Dev 7, If 1, Class=Wireless, Driver=btusb, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M

```

mcuee, 2023-05-13:
After detaching the kernel usbhid driver, default avrdude build will still work as it is using hidapi-libusb (because of CMake library search order.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c pickit4_isp -p m8a -v

avrdude: Version 7.1-20230504 (e6a2f3c0)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude.conf
         User configuration file is /home/mcuee/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : pickit4_isp
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega8A
         Chip Erase delay              : 10000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                  4    20   128    0 no        512    4      0  9000  9000 0xff 0xff
           flash                  33    10    64    0 yes      8192   64    128  4500  4500 0xff 0x00
           lfuse                   0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

         Programmer Type : JTAG3_ISP
         Description     : MPLAB(R) PICkit 4 in ISP mode
         ICE HW version  : 4
         ICE FW version  : 1.14 (rel. 268)
         Serial number   : BUR200973052
         SCK period      : 8.00 us
         Vtarget         : 4.93 V

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9307 (probably m8a)

avrdude done.  Thank you.
```

Buf if I change the sequence and use hidapi-hidraw, then it will not work with hidapi as the hidraw driver has been detached. Then it 

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_main$ git diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index a50207fd..d3f05e18 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -130,7 +130,7 @@ else()
     set(PREFERRED_LIBELF elf)
     set(PREFERRED_LIBUSB usb)
     set(PREFERRED_LIBUSB_1_0 usb-1.0)
-    set(PREFERRED_LIBHIDAPI hidapi hidapi-libusb hidapi-hidraw)
+    set(PREFERRED_LIBHIDAPI hidapi hidapi-hidraw hidapi-libusb)
     set(PREFERRED_LIBFTDI ftdi)
     set(PREFERRED_LIBFTDI1 ftdi1)
     set(PREFERRED_LIBREADLINE readline)

```

mcuee, 2023-05-13:
It is very strange that the code path for the fall-back libusb is different.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_mod2_hidraw -c pickit4_isp -p m8a -vvvv

avrdude_mod2_hidraw: Version 7.1-20230504 (e6a2f3c0)
                     Copyright the AVRDUDE authors;
                     see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                     System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude.conf
                     User configuration file is /home/mcuee/.avrduderc
                     User configuration file does not exist or is not a regular file, skipping

                     Using Port                    : usb
                     Using Programmer              : pickit4_isp
avrdude_mod2_hidraw: stk500v2_jtag3_open()
avrdude_mod2_hidraw usbhid_open() [usb_hidapi.c:126] warning: USB device with VID: 0x03eb and PID: 0x2177 not found
avrdude_mod2_hidraw usbhid_open() [usb_hidapi.c:126] warning: USB device with VID: 0x03eb and PID: 0x2178 not found
avrdude_mod2_hidraw usbhid_open() [usb_hidapi.c:126] warning: USB device with VID: 0x03eb and PID: 0x2179 not found
avrdude_mod2_hidraw: usbdev_open(): found MPLAB PICkit 4 CMSIS-DAP, serno: BUR200973052
avrdude_mod2_hidraw: found CMSIS-DAP compliant device, using EDBG protocol
avrdude_mod2_hidraw: jtag3_getsync()

avrdude_mod2_hidraw: jtag3_edbg_prepare()
libusb: error [submit_bulk_transfer] submiturb failed, errno=2
avrdude_mod2_hidraw usbdev_send() [usb_libusb.c:345] error: wrote -5 out of 912 bytes, err = Input/output error
avrdude_mod2_hidraw: sent: . [02] . [01] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00] . [00 [... cut ...]  
libusb: error [submit_bulk_transfer] submiturb failed, errno=2
avrdude_mod2_hidraw: usbdev_recv_frame(): usb_interrupt_read(): Input/output error
avrdude_mod2_hidraw jtag3_edbg_prepare() [jtag3.c:567] error: unable to read from serial port (-1)
avrdude_mod2_hidraw stk500v2_jtag3_open() [stk500v2.c:4051] error: unable to sync with the JTAGICE3 in ISP mode
avrdude_mod2_hidraw main() [main.c:1174] error: unable to open programmer pickit4_isp on port usb

avrdude_mod2_hidraw done.  Thank you.

```

mcuee, 2023-05-13:
@MCUdude and @stefanrueger 
It seems to me that there is a bug in the fall back code path which some initialization steps were missing. If you got the time, please take a look. Thanks.

mcuee, 2023-06-20:
I am not so sure why we get the following error under libusb. 
```
usbdev_send() error: wrote -5 out of 912 bytes, err = Input/output error
```

There is a code when using hidapi which seems to be missing when using libusb.
```
      /* No timely response, assume 512 byte size */
      hid_write(dev, usbbuf, (512 - 64) + 1);
      fd->usb.max_xfer = 512;
      fd->usb.max_xfer = usbbuf[2] + (usbbuf[3] << 8);
      pmsg_debug("usbhid_open(): setting max_xfer from DAP_Info response to %d\n",
        fd->usb.max_xfer);
```

mcuee, 2023-11-02:
> I think we need to set PICKIt4 and SNAP max packet size to 64 to see if that helps.

@MCUdude and @stefanrueger,

I believe I have found the root cause of the issue, however, I do not know how to fix the issue. Just wondering if you can look at my comments above to see if there is a simple fix or not. Thanks.

mcuee, 2024-01-19:
* #1617 may be the same issue.

mcuee, 2024-03-03:
Close this one as not planned until someone is actibely working on this issue.

This issue is also linked here.
https://github.com/avrdudes/avrdude/discussions/1710

mcuee, 2024-08-02:
@askn37 

Please take a look at this issue and my comments to see if you can come out with a patch. Thanks.
* https://github.com/avrdudes/avrdude/issues/1221#issuecomment-1546496650

askn37, 2024-08-02:
Descriptor when ATMELICE3 is recognized at High-Speed ​​on macos/M1/sonoma14.5

```
High Speed device @ 1 (0x02300000): .............................................   Miscellaneous/Common Class device: "Atmel-ICE CMSIS-DAP"
    Port Information:   0x001a
           Not Captive
           Attached to Root Hub
           External Device
           Connected
           Enabled
    Number Of Endpoints (includes EP0):   
        Total Endpoints for Configuration 1 (current):   5
    Device Descriptor   
        Descriptor Version Number:   0x0200
        Device Class:   239   (Miscellaneous)
        Device Subclass:   2   (Common Class)
        Device Protocol:   1   (Interface Association)
        Device MaxPacketSize:   64
        Device VendorID/ProductID:   0x03EB/0x2141   (unknown vendor)
        Device Version Number:   0x0101
        Number of Configurations:   1
        Manufacturer String:   1 "Atmel Corp."
        Product String:   2 "Atmel-ICE CMSIS-DAP"
        Serial Number String:   3 "J42700015230"
    Configuration Descriptor (current config)   
        Length (and contents):   64
            Raw Descriptor (hex)    0000: 09 02 40 00 02 01 00 80  32 09 04 00 00 02 03 00  
            Raw Descriptor (hex)    0010: 00 04 09 21 11 01 00 01  22 23 00 07 05 01 03 00  
            Raw Descriptor (hex)    0020: 02 01 07 05 82 03 00 02  01 09 04 01 00 02 FF FF  
            Raw Descriptor (hex)    0030: FF 05 07 05 84 02 00 02  FF 07 05 03 02 00 02 FF  
            Unknown Descriptor   0040: 
        Number of Interfaces:   2
        Configuration Value:   1
        Attributes:   0x80 (bus-powered)
        MaxPower:   100 mA
        Interface #0 - HID ..............................................   "Atmel-ICE CMSIS-DAP"
            Alternate Setting   0
            Number of Endpoints   2
            Interface Class:   3   (HID)
            Interface Subclass;   0
            Interface Protocol:   0
            HID Descriptor   
                Descriptor Version Number:   0x0111
                Country Code:   0
                Descriptor Count:   1
                Descriptor 1   
                    Type:   0x22  (Report Descriptor)
                    Length (and contents):   35
                        Raw Descriptor (hex)    0000: 06 00 FF 09 01 A1 01 15  00 26 FF 00 75 08 96 00  
                        Raw Descriptor (hex)    0010: 02 09 01 81 02 96 00 02  09 01 91 02 95 04 09 01  
                        Raw Descriptor (hex)    0020: B1 02 C0 
                    Parsed Report Descriptor:   
                          Usage Page    (Vendor defined 0) 
                          Usage 1 (0x1)    
                              Collection (Application)    
                                Logical Minimum.........    (0)  
                                Logical Maximum.........    (255)  
                                Report Size.............    (8)  
                                Report Count............    (512)  
                                Usage 1 (0x1)    
                                Input...................   (Data, Variable, Absolute, No Wrap, Linear, Preferred State, No Null Position, Bitfield) 
                                Report Count............    (512)  
                                Usage 1 (0x1)    
                                Output..................   (Data, Variable, Absolute, No Wrap, Linear, Preferred State, No Null Position, Nonvolatile, Bitfield) 
                                Report Count............    (4)  
                                Usage 1 (0x1)    
                                Feature.................   (Data, Variable, Absolute, No Wrap, Linear, Preferred State, No Null Position, Nonvolatile, Bitfield) 
                              End Collection     
            Endpoint 0x01 - Interrupt Output   
                Address:   0x01  (OUT)
[... 76 lines cut ...]
                Polling Interval:   255 ms
            Endpoint 0x03 - Bulk Output   
                Address:   0x03  (OUT)
                Attributes:   0x02  (Bulk)
                Max Packet Size:   64
                Polling Interval:   255 ms
```

We can see that the "HID Report Descriptor" allows a fixed payload of 512 bytes.

`Report Count............    (512) `

And endpoint 0x01/0x82 allows 512 bytes per communication.

`Max Packet Size:   0x0200  (512 x 1  transactions opportunities per microframe)`

On the other hand, the descriptor for Full-Speed ​​exists as "Other Speed ​​Configuration Descriptor".

`Max Packet Size:   64`

In this case, will HID communication be performed using 8 sets of 64 bytes? The "HID Report Descriptor" for Full-Speed ​​is not listed here.

You should be able to select/switch which one is active in the "Device Qualifier Descriptor". That means there should be an API somewhere to downgrade/upgrade between High-Speed ​​and Full-Speed, and you should be able to distinguish between them. The Windows API should be able to do it. But I don't know about HIDAPI or libusb.

This display is displayed by the usbdiagnose command on macos, but it can be used without privileges, so it's clear that there is an API that can be used with user privileges at the OS level. It's probably not built into HIDAPI for policy reasons. And libusb's descriptor-related API requires privileges, so we can see that there is a problem somewhere in the implementation.

mcuee, 2024-08-02:
> You should be able to select/switch which one is active in the "Device Qualifier Descriptor". That means there should be an API somewhere to downgrade/upgrade between High-Speed ​​and Full-Speed, [...]

I do not think you can chage the speed using libusb or hidapi. And for Windows, I do not think it is possible to change the negotiated speed by the OS either.

@Youw 
Please share your thoughts as well. Thanks.

Youw, 2024-08-02:
> Please share your thoughts as well. Thanks.

I'm not aware if such method.
Last time I needed to do something like that - I've sent a special command to my USB device and it rebooted/reinitialized at different speed. Although, I have no idea how - that was done by a different team, who made the FW for that device.

stefanrueger, 2024-08-02:
@mcuee So best to close this issue again?

mcuee, 2024-08-03:
> @mcuee So best to close this issue again?

No, there is a bug here in avrdude.

mcuee, 2024-08-04:
This is still good to be fixed as hdiapi does not work under NetBSD/OpenBSD.

AndreiCherniaev, 2024-12-19:
On Ubuntu 24 x86_64 with [PICkit 4](https://www.microchip.com/en-us/development-tool/pg164140) doesn't works. Why?

How to build
```
sudo apt purge libusb-dev
sudo apt install flex bison libhidapi-dev
git clone https://github.com/avrdudes/avrdude && cd avrdude
./build.sh -j8
sudo cmake --build build_linux --target install
```
How to test pickit4. Why "USB device with VID: 0x03eb and PID: 0x2177 not found"?
```
q@PC:~$ /usr/local/bin/avrdude -c pickit4 -vvvv
Avrdude version 8.0-20241213 (f1555aef)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/q/.avrduderc does not exist

Using port            : usb
Using programmer      : pickit4
Jtag3_open()
USB device with VID: 0x03eb and PID: 0x2177 not found
USB device with VID: 0x03eb and PID: 0x2178 not found
USB device with VID: 0x03eb and PID: 0x2179 not found
USB device with VID: 0x04d8 and PID: 0x9012 not found
USB device with VID: 0x04d8 and PID: 0x9017 not found
USB device with VID: 0x04d8 and PID: 0x9012 not found
USB device with VID: 0x04d8 and PID: 0x9017 not found
Error jtag3_open_common() jtag3.c 1735: no device found matching VID 0x04d8 and PID list: 0x2177, 0x2178, 0x2179
Error main() main.c 1450: unable to open port usb for programmer pickit4

Avrdude done.  Thank you.
```

But Linux can detect pickit4 as device with VID: 0x03eb and PID: 0x2177
```
q@PC:~$ lsusb -vvv -d 03eb:2177

Bus 003 Device 019: ID 03eb:2177 Atmel Corp. MPLAB PICkit 4 CMSIS-DAP
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x03eb Atmel Corp.
  idProduct          0x2177 
  bcdDevice            1.00
  iManufacturer           1 Microchip Technology Incorporated
  iProduct                2 MPLAB PICkit 4 CMSIS-DAP
  iSerial                 3 BUR225174055
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
[... 84 lines cut ...]
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
```

mcuee, 2025-04-28:
Issue reported here too.
* https://github.com/avrdudes/avrdude/issues/1938

askn37, 2026-11-04:
I discovered that the `hid_get_report_descriptor` function is available in `hidapi` v0.14.0 and later. I decided to retrieve and parse the HID report to see if I could obtain a value of 64 or 512. In this sample code, the parsing process is halted as soon as an "Input" report tag is encountered; this is because continuing to the end would result in retrieving the size of a "Feature" report tag instead.

```diff
diff --git a/src/usb_hidapi.c b/src/usb_hidapi.c
index cae30979..4d4615b2 100644
--- a/src/usb_hidapi.c
+++ b/src/usb_hidapi.c
@@ -158,6 +158,96 @@ static int usbhid_open(const char *port, union pinfo pinfo, union filedescriptor
     }
   }
 
+#if defined(HID_API_VERSION_MAJOR) && defined(HID_API_VERSION_MINOR)
+
+  pmsg_debug("%s(): HID_API_VERSION: %s\n", __func__, HID_API_VERSION_STR);
+
+  #if (HID_API_VERSION_MAJOR > 0) || (HID_API_VERSION_MAJOR == 0 && HID_API_VERSION_MINOR >= 14)
+
+  /*
+   * Retrieve the HID report descriptor.
+   * "hid_get_report_descriptor" function is Available in hidapi v0.14.0 and later.
+   */
+  unsigned long _report_count = 0, _report_size = 0;
+  unsigned char _hid_report[HID_API_MAX_REPORT_DESCRIPTOR_SIZE];
+  int _desc_size = hid_get_report_descriptor(dev, _hid_report, sizeof(_hid_report));
+  pmsg_trace("%s(): get HID Report size %d\n", __func__, _desc_size);
+  if (_desc_size > 0) {   // In the case of EDBG, it is typically 35 bytes.
+    for (int _index = 0; _index < _desc_size; _index++) {
+      // The lower 2 bits of the Prefix-Byte indicate the argument type.
+      // (0=none, 1=char, 2=short, 3=long)
+      unsigned char _prefix = _hid_report[_index + 0];
+      unsigned char _tag_type = _prefix & 0xFC;
+      unsigned char _tag_size = _prefix & 0x03;
+      unsigned long _tag_value = 0;
+
+      // Perform a signed type conversion on the value, and skip the next index.
+      if (_tag_size == 0) {
+        pmsg_trace("%s(): tag: %02X\n", __func__, _hid_report[_index + 0]);
+        // A value with a length of 0 is interpreted as 0.
+      }
+      else if (_tag_size == 1) {
+        pmsg_trace("%s(): tag: %02X %02X\n", __func__, _hid_report[_index + 0],
+          _hid_report[_index + 1]);
+
+        _tag_value = ((unsigned long)_hid_report[_index + 1]);
+        _index += 1;
+      }
+      else if (_tag_size == 2) {
+        pmsg_trace("%s(): tag: %02X %02X %02X\n", __func__, _hid_report[_index + 0],
+          _hid_report[_index + 1], _hid_report[_index + 2]);
+
+        _tag_value = ((unsigned long)_hid_report[_index + 1])
+                   | ((unsigned long)_hid_report[_index + 2] << 8);
+        _index += 2;
+      }
+      else {
+        pmsg_trace("%s(): tag: %02X %02X %02X %02X %02X\n", __func__, _hid_report[_index],
+          _hid_report[_index + 1], _hid_report[_index + 2], _hid_report[_index + 3], _hid_report[_index + 4]);
+
+        _tag_value = ((unsigned long)_hid_report[_index + 1])
+                   | ((unsigned long)_hid_report[_index + 2] << 8)
+                   | ((unsigned long)_hid_report[_index + 3] << 16)
+                   | ((unsigned long)_hid_report[_index + 4] << 24);
+        _index += 4;
+      }
+
+      // If an "Output" tag is encountered, stop the search.
+      if (_tag_type == 0x80) {
+        pmsg_trace("%s(): found Input(%lX)\n", __func__, _tag_value);
+        break;
+      }
[... 64 lines cut ...]
Usbhid_open(): max_xfer is (presumably) 64 bytes.
Found CMSIS-DAP compliant device, using EDBG protocol
Jtag3_getsync()

Jtag3_edbg_prepare()
Usbhid_send: 02 01 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_recv: 02 01 00 06 0e 38 02 01  80 00 00 00 00 00 00 00  |.....8..........|
Usbhid_recv: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_recv: 00 00 00 00 00 00 00 00  00 00 00 ff ff ff ff ff  |................|
Usbhid_recv: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
Jtag3_edbg_prepare(): connection status 0x01
Usbhid_send: 01 00 01 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_recv: 01 00 00 06 0e 38 02 01  80 00 00 00 00 00 00 00  |.....8..........|
Usbhid_recv: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_recv: 00 00 00 00 00 00 00 00  00 00 00 ff ff ff ff ff  |................|
Usbhid_recv: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
Jtag3_command(sign-on): 
Jtag3_edbg_send(): sending 3 bytes
```

stefanrueger, 2026-11-07:
> I have a feeling this will yield quite good results.

@askn37 Great discovery and neat implementation! Would you like to issue a PR? The only changes I would advise
 - Not use variable names that start with `_` (that is not wrong, but doesn't fit the coding style in AVRDUDE)
 - Move up the `else` behind the closing `}` bracket (AVRDUDE has a compact style)
 
Thank you!

askn37, 2026-11-13:
#### Background leading to the creation of the previous patch

As you may know, I have been working on the design—covering both hardware and software—of EDBG-based devices using the AVR-DU. While these devices function as intended on most macOS and Windows systems, the same cannot be said for Linux-aarch64. My investigation into the issue points to a compatibility problem between `avrdude` and `libusb`.
I am currently testing this on Debian 13, as it is one of the few distributions that can be installed on Apple Silicon via VMware Fusion. However, the version of `libusb` that comes pre-installed with it is 0.14, and its behavior is highly unreliable.

The issue is that running `avrdude` often yields completely different results each time. When the device is configured for 64-byte packets and I specify `-c pickit4_updi`—forcing a fallback from 512-byte to 64-byte packets—the operation alternates between success and failure. From the device's perspective, a 512-byte packet appears as eight 64-byte packets, which are then interpreted as eight separate `DAP_CMD` commands; naturally, this prevents proper synchronization. I have been using tools like https://github.com/ataradov/usb-sniffer to observe the USB packets, but I am still very much in the dark as to the root cause.

<details>

```
askn@debian:~/Develop/avrdude/build_linux/src$ avrdude -Pusb:03eb:2177 -cpickit4_updi -pavr64du28 -vqvvvv
$ avrdude -Pusb:03eb:2177 -cpickit4_updi -pavr64du28 -vqvvvv
Avrdude version 8.3-20261109 (d5b309b5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

dirpath = /usr/local/bin
sysconfig = /usr/local/etc/avrdude.conf
User configuration file is not provided/readable

Using port            : usb:03eb:2177
Using programmer      : pickit4_updi
Jtag3_open_updi(usb:03eb:2177)
HV UPDI support: 0 1 2
Usbhid_open(): product string: UPDI4AVR:AVR-DU:mEDBG:CMSIS-DAP
Usbhid_open(): probing for max packet size
Error usbhid_open() usb_hidapi.c 211: unexpected reply to DAP_Info: 0x02 0x01
Found CMSIS-DAP compliant device, using EDBG protocol
Jtag3_getsync()

Jtag3_edbg_prepare()
Usbhid_send: 02 01 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Warning usbhid_recv() usb_hidapi.c 269: hid_read_timeout() experienced a timeout after 10 s
Error jtag3_edbg_prepare() jtag3.c 559: unable to read from serial port (0)
Error main() main.c 1287: unable to open port usb:03eb:2177 for programmer pickit4_updi

Avrdude done.  Thank you.
```

</details>

<details>

```
askn@debian:~/Develop/avrdude/build_linux/src$ avrdude -Pusb:03eb:2177 -cpickit4_updi -pavr64du28 -vqvvvv
$ avrdude -Pusb:03eb:2177 -cpickit4_updi -pavr64du28 -vqvvvv
Avrdude version 8.3-20261109 (d5b309b5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

dirpath = /usr/local/bin
sysconfig = /usr/local/etc/avrdude.conf
User configuration file is not provided/readable

Using port            : usb:03eb:2177
Using programmer      : pickit4_updi
Jtag3_open_updi(usb:03eb:2177)
HV UPDI support: 0 1 2
Usbhid_open(): product string: UPDI4AVR:AVR-DU:mEDBG:CMSIS-DAP
Usbhid_open(): probing for max packet size
Error usbhid_open() usb_hidapi.c 211: unexpected reply to DAP_Info: 0x00 0x00
Found CMSIS-DAP compliant device, using EDBG protocol
Jtag3_getsync()

Jtag3_edbg_prepare()
Usbhid_send: 02 01 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
[... 14 lines cut ...]
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Usbhid_send: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
Warning usbhid_recv() usb_hidapi.c 269: hid_read_timeout() experienced a timeout after 10 s
Error jtag3_edbg_prepare() jtag3.c 559: unable to read from serial port (0)
Error main() main.c 1287: unable to open port usb:03eb:2177 for programmer pickit4_updi

Avrdude done.  Thank you.
```

</details>

- The intention behind the patch I mentioned earlier is to reduce this unwanted phenomenon. (Though it cannot yet be considered perfect.)
- Regarding the reported issue where `hid_read_timeout()` times out after 10 seconds: in reality, the function returns in less than one millisecond. This is likely a bug in libusb 0.14. That said, we would prefer to resolve the issue while continuing to use this version.

askn37, 2026-11-13:
> Regarding the reported issue where hid_read_timeout() times out after 10 seconds: in reality, the function returns in less than one millisecond. This is likely a bug in libusb 0.14. That said, we would prefer to resolve the issue while continuing to use this version.

```diff
--- a/src/usb_hidapi.c
+++ b/src/usb_hidapi.c
@@ -358,7 +358,8 @@ static int usbhid_recv(const union filedescriptor *fd, unsigned char *buf, size_
   if(udev == NULL)
     return -1;
 
-  int rv = hid_read_timeout(udev, buf, nbytes, 10000);
+  int rv = hid_read_timeout(udev, buf, nbytes, 100);
+  if(rv == 0) rv = hid_read_timeout(udev, buf, nbytes, 10000);
   if(rv <= 0 || (size_t) rv > nbytes) {
     if(rv < 0)
       pmsg_error("hid_read_timeout(usb, %lu, 10000) failed\n", (unsigned long) nbytes);
```

This will eliminate this instability. TRUE?

askn37, 2026-11-15:
> This will eliminate this instability. TRUE?

My assessment is as follows:

`usb_detach_kernel_driver_np` unloads the driver, and `usb_claim_interface` reloads it. Unless the device is simultaneously being used by another driver, it will effectively disappear from the USB bus for a brief moment.

#### 1. When only one interface exists, and it is for HID.

Without root privileges, `usb_detach_kernel_driver_np` will fail—meaning the unloading of the HID driver is denied. From the OS's perspective, the device is treated as a desktop peripheral, similar to a mouse or keyboard.

The most appropriate solution for handling devices like EDBG or USBasp is to register a `udev` rule.

#### 2. When there are three or more interfaces (e.g., HID + VCP + Debug + Mass Storage):

All current official EDBG products should fall into this category. In this case, `usb_detach_kernel_driver_np` and `usb_claim_interface` will succeed even without a `udev` rule.

However, this might just be working by coincidence; under certain conditions—such as when using a virtual machine or running the PC at a lower clock speed—the situation can mirror the case described below.

#### 3. When there are exactly two interfaces (e.g., HID + VCP):

While `usb_detach_kernel_driver_np` and `usb_claim_interface` succeed even without a `udev` rule, the first subsequent call to `hid_read_timeout` will occasionally fail. It appears that the driver reload process has not yet fully taken effect.

For now, a simple workaround is to retry any `hid_read_timeout` call that fails (i.e., returns zero).

stefanrueger, 2026-11-15:
Very cool analysis, @askn37. Thanks for the detailed and thorough tests. 

I am hoping that the AVRDUDE's Linux support for adding udev rules helps
```
$ avrdude -c snap/u
```
or
```
$ avrdude -c "*/u"
```