avrdudes/avrdude #1227: Behavior of Optiboot firmware without EEPROM capability when reading eeprom

mcuee, 2022-12-18:
I think this is not a real issue but kind of strange behavior of the optiboot bootloader firmware.

I jusgt bought a few Uno Clones with CH340. Apparently due to chip shortages the boards come with ATmega328PB and not ATmega328P. The 4.4 version of Optiboot Arduino FW is still used wihout any changes. It does not support EEPROM and it is meant for ATmega328P.
https://github.com/arduino/ArduinoCore-avr/blob/master/bootloaders/optiboot/optiboot_atmega328.hex

A few interesting behaviors here.

1) `-c arduino` and `-c urclock` will insist that the chip connected is ATmega328P and not ATmega328PB.
2) `-c arduino` will read Flash contents as EEPROM contents. `-c urclock` will do that as well if I specify `-xeepromrw`.

@stefanrueger 
Do you think this is a problem for avrdude? If yes what can we do here?

```
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump flash 0 0x10" | .\avrdude -c avrisp2 -p m328pb -qqt
avrdude> dump flash 0 0x10
0000  0c 94 61 00 0c 94 7e 00  0c 94 7e 00 0c 94 7e 00  | .a. .~. .~. .~.|

avrdude>
avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude -c avrisp2 -p m328pb -qqt
avrdude> dump eeprom 0 0x10
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude -c urclock -P COM17 -p m328p -qqt -xeepromrw
avrdude> dump eeprom 0 0x10
0000  0c 94 61 00 0c 94 7e 00  0c 94 7e 00 0c 94 7e 00  | .a. .~. .~. .~.|

avrdude>
avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude -c arduino -P COM17 -p m328p -qqt
avrdude> dump eeprom 0 0x10
0000  0c 94 61 00 0c 94 7e 00  0c 94 7e 00 0c 94 7e 00  | .a. .~. .~. .~.|

avrdude>
avrdude>
```

mcuee, 2022-12-18:
Even if I change the bootloader FW to @MCUdude's MiniCore and specified the right chip (ATmega328PB), then the first behavior is gone but the second behavior is still there.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM5 -p m328pb -qq -xbootsize=512 -xshowall
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 o8.0 -?s-?-r-- vector 0 (RESET) ATmega328PB

PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude -c urclock -P COM5 -p m328pb -qqt -xbootsize=512
avrdude> dump eeprom 0 0x10
avrdude error: bootloader does not seem to have paged EEPROM read capability
avrdude error: bootloader does not seem to have EEPROM access capability
avrdude error: unable to read eeprom page at addr 0x0000
avrdude error: (dump) error reading eeprom address 0x00000 of part ATmega328PB
               read operation not supported on memory type eeprom
avrdude>
avrdude>

PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude -c urclock -P COM5 -p m328pb -qqt -xbootsize=512 -xeepromrw
avrdude> dump eeprom 0 0x10
0000  0c 94 8f 00 0c 94 a1 00  0c 94 a1 00 0c 94 a1 00  | ... ... ... ...|

avrdude>
avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude -c arduino -P COM5 -p m328pb -qqt
avrdude> dump eeprom 0 0x10
0000  0c 94 8f 00 0c 94 a1 00  0c 94 a1 00 0c 94 a1 00  | ... ... ... ...|

avrdude>
avrdude>

PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude -c usbasp -p m328pb -qqt
avrdude> dump eeprom 0 0x10
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump flash 0 0x10" | .\avrdude -c usbasp -p m328pb -qqt
avrdude> dump flash 0 0x10
0000  0c 94 8f 00 0c 94 a1 00  0c 94 a1 00 0c 94 a1 00  | ... ... ... ...|

avrdude>
avrdude>

```

mcuee, 2022-12-18:
Same for an Nano Clone with ATmega328P with the original 4.4 Optiboot bootloader FW.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM10 -p m328p -qq -xbootsize=512 -xshowall
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 o4.4 -?s-?-r-- vector 0 (RESET) ATmega328P
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude -c urclock -P COM10 -p m328p -qqt -xbootsize=512 -xeepromrw
avrdude> dump eeprom 0 0x10
0000  0c 94 5c 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .\. .n. .n. .n.|

avrdude>
avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude -c arduino -P COM10 -p m328p -qqt
avrdude> dump eeprom 0 0x10
0000  0c 94 5c 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .\. .n. .n. .n.|

avrdude>
avrdude>
```

mcuee, 2022-12-18:
@stefanrueger 

Now I actually think the second issue is a bug for `-c arduino` and it is arguable for `-c urclock`. What is your opinion?

stefanrueger, 2022-12-18:
> 1) `-c arduino` and `-c urclock` will insist that the chip connected is ATmega328P and not ATmega328PB. I think this is because of the bootloader firmware and we can not do anything about it.

Correct. The bootloader firmware actually does not read the signature memory of the part. The bootloader returns hard-coded signature bytes of the part *for which it was compiled*. This is normal bootloader practice to save space. If the user burns an ATmega328P bootloader onto an ATmega328PB chip then AVRDUDE will see an m328p signature.

> 2) `-c arduino` will read Flash contents as EEPROM contents. `-c urclock` will do that as well if I specify `-xeepromrw`.
> @stefanrueger 
> Do you think if No. 2 is a problem for avrdude? If yes what can we do here? Is it also the limitation of the paticular old optiboot firmware shipped by Arduino?

This is a well-known **optiboot bug**. It returns flash for EEPROM if optiboot has been compiled to *not* handle EEPROM. This bug has been around for a long time (and possibly still is today but defo was 2016 when I last looked closely at optiboot).

It is not actually AVRDUDE's job to invent workarounds for all the bugs in bootloaders or programmer FW (though I have provided a fair deal of workarounds in the past). In this specific case, knowing about this bug, `-c urclock` uses the *justified* and *sane* default assumption that optiboot cannot deal EEPROM *unless* there is evidence to the contrary either by
 - A *known* bootloader in the hash table specifying that indeed it can deal with EEPROM
 - The user specifically promising AVRDUDE with `-xeeprom` that the bootloader can handle EEPROM

Now, @mcuee, your example is one where the user tells AVRDUDE `-xeepromrw` that indeed the bootloader can handle EEPROM only to be given flash by the *buggy* optiboot that actually cannot handle EEPROM.

Soooooo, can you guess what my opinion is?

Ditch optiboot. Never look back. Use urboot. Be happy ever after.

As there is a good alternative in this case I believe we should not spend our energy dancing around optiboot bugs.

mcuee, 2022-12-18:
@stefanrueger 

Haha, I support urboot. We need to get avrdude 7.1 released to enable `-c urclock`.

Hopefully @MCUdude can start to distribute urboot early next year to make urboot more popular.

MCUdude, 2022-12-18:
> Hopefully @MCUdude can start to distribute urboot early next year to make urboot more popular.

Yes, this is the plan. However, I need to make sure that my cores will continue to support legacy optiboot bootloaders while using -c urclock, so users aren't _forced_ to replace their bootloaders, even though it's highly recommended. It's also important that potential bugs found by mass-testing urboot gets fixed quickly.

stefanrueger, 2022-12-18:
> legacy optiboot bootloaders while using -c urclock

You are welcome to add to the [hash table](https://github.com/avrdudes/avrdude/blob/159d12693e8787fab5bf68d87f331bedf85164a1/src/urclock.c#L1146-L1173) those legacy optiboot bootloaders that were shipped with your boards and/or frequently downloaded from your github core; I am happy to help you [compute the hashes](https://github.com/avrdudes/avrdude/blob/main/tools/bootloader-hash) if you struggle with that.

And our legacy bootloaders will continue to work with `-c arduino` though with fewer benefits than `-c urclock`

> urboot gets fixed quickly

That is my intention...

MCUdude, 2022-12-18:
> You are welcome to add to the [hash table](https://github.com/avrdudes/avrdude/blob/159d12693e8787fab5bf68d87f331bedf85164a1/src/urclock.c#L1146-L1173) those legacy optiboot bootloaders that were [...]

Is this something you could help me automate? The [optiboot flash](https://github.com/MCUdude/optiboot_flash) repo is pretty easy for a script to run through.

From what I know, these are the most popular bootloaders:

20 MHz, 115200 baud
16 MHz, 115200 baud
8 MHz, 57600 baud
8 MHz 38400 baud
1 MHz 9600 baud

For UART 0, 1, 2, and 3 where present.

The only parts that should _not_ be added are the AT90USB* parts, as I provide no Arduino core for them.

stefanrueger, 2022-12-18:
The [bootloader-hashes script](https://github.com/avrdudes/avrdude/blob/main/tools/bootloader-hash) finds all bootloaders recursively in the current directory and outputs the hash lines that can be cut and paste into the table. Sounds like the only think you'd need to do is grep out the ones you are interested in. BTW, I think the bootloaders for 16 MHz, 115200 baud are probably identical to the ones 8 MHz, 57600 baud (so will have the same hash)

MCUdude, 2022-12-18:
@stefanrueger this is the list of bootloader combinations I think are the most common.
However, some of the files have the same hash, even though they are different parts.
I've not removed any duplicates, but you can easily spot the duplicated by using `sort -k 5` on the list below:

<details>
<summary>Optiboot flash hash</summary>

```
    {  512, 0, 0x2afe013b, 0x5775955f }, // optiboot_flash_atmega169_UART0_38400_8000000L_B5.hex
    {  512, 0, 0x2afe013b, 0xd5de2de6 }, // optiboot_flash_atmega169_UART0_57600_8000000L_B5.hex
    {  512, 0, 0x7de5e119, 0xecba44da }, // optiboot_flash_atmega169_UART0_9600_1000000L_B5.hex
    {  512, 0, 0x2afe013b, 0x0ebf1682 }, // optiboot_flash_atmega169_UART0_115200_20000000L_B5.hex
    {  512, 0, 0x2afe013b, 0xee9f5b33 }, // optiboot_flash_atmega169_UART0_115200_16000000L_B5.hex
    { 1024, 1, 0x5790d602, 0x728c740e }, // optiboot_flash_at90can64_UART1_57600_8000000L_B5_BIGBOOT.hex
    { 1024, 1, 0x5790d602, 0x5419d99f }, // optiboot_flash_at90can64_UART1_38400_8000000L_B5_BIGBOOT.hex
    { 1024, 1, 0x5790d602, 0x9ff3235b }, // optiboot_flash_at90can64_UART0_38400_8000000L_B5_BIGBOOT.hex
    { 1024, 1, 0x5790d602, 0xd1e8d0b9 }, // optiboot_flash_at90can64_UART0_57600_8000000L_B5_BIGBOOT.hex
    { 1024, 1, 0x74b9c238, 0xc76f76e5 }, // optiboot_flash_at90can64_UART0_9600_1000000L_B5_BIGBOOT.hex
    { 1024, 1, 0x74b9c238, 0xeafeb59b }, // optiboot_flash_at90can64_UART1_9600_1000000L_B5_BIGBOOT.hex
    { 1024, 1, 0x5790d602, 0x4ad22f5b }, // optiboot_flash_at90can64_UART1_115200_20000000L_B5_BIGBOOT.hex
    { 1024, 1, 0x5790d602, 0xb495f1ea }, // optiboot_flash_at90can64_UART0_115200_20000000L_B5_BIGBOOT.hex
    { 1024, 1, 0x5790d602, 0x246dbc35 }, // optiboot_flash_at90can64_UART1_115200_16000000L_B5_BIGBOOT.hex
    { 1024, 1, 0x5790d602, 0xd92f11b5 }, // optiboot_flash_at90can64_UART0_115200_16000000L_B5_BIGBOOT.hex
    {  512, 0, 0xb83f0d83, 0x3d3da465 }, // optiboot_flash_atmega32_UART0_57600_8000000L_B7.hex
    {  512, 0, 0x8790003e, 0x9f82a66b }, // optiboot_flash_atmega32_UART0_57600_8000000L_B0.hex
    {  512, 0, 0x8790003e, 0x65d86300 }, // optiboot_flash_atmega32_UART0_38400_8000000L_B0.hex
    {  512, 0, 0xb83f0d83, 0x87ac0178 }, // optiboot_flash_atmega32_UART0_38400_8000000L_B7.hex
    {  512, 0, 0xb83f0d83, 0xfae2fc26 }, // optiboot_flash_atmega32_UART0_9600_1000000L_B7.hex
    {  512, 0, 0x8790003e, 0x333e86ec }, // optiboot_flash_atmega32_UART0_9600_1000000L_B0.hex
    {  512, 0, 0x8790003e, 0xc286cdc8 }, // optiboot_flash_atmega32_UART0_115200_20000000L_B0.hex
    {  512, 0, 0xb83f0d83, 0x3330e675 }, // optiboot_flash_atmega32_UART0_115200_20000000L_B7.hex
    {  512, 0, 0x8790003e, 0x55b98295 }, // optiboot_flash_atmega32_UART0_115200_16000000L_B0.hex
    {  512, 0, 0xb83f0d83, 0xde792fb2 }, // optiboot_flash_atmega32_UART0_115200_16000000L_B7.hex
    {  512, 0, 0x9e762fa8, 0xe2724ea2 }, // optiboot_flash_atmega3250p_UART0_57600_8000000L_B7.hex
    {  512, 0, 0x9e762fa8, 0xb21ddc89 }, // optiboot_flash_atmega3250p_UART0_38400_8000000L_B7.hex
    {  512, 0, 0xdaa198fa, 0x70fcc8e9 }, // optiboot_flash_atmega3250p_UART0_9600_1000000L_B7.hex
    {  512, 0, 0x9e762fa8, 0xca6c9014 }, // optiboot_flash_atmega3250p_UART0_115200_20000000L_B7.hex
    {  512, 0, 0x9e762fa8, 0x680e4729 }, // optiboot_flash_atmega3250p_UART0_115200_16000000L_B7.hex
    { 1024, 1, 0xae42ebb8, 0xe11ad957 }, // optiboot_flash_atmega644a_UART1_38400_8000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0xdc558cfc }, // optiboot_flash_atmega644a_UART1_57600_8000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0xc4b9120d }, // optiboot_flash_atmega644a_UART0_57600_8000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0x1327cc64 }, // optiboot_flash_atmega644a_UART0_38400_8000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0x93195537 }, // optiboot_flash_atmega644a_UART0_57600_8000000L_B0_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0x1224da19 }, // optiboot_flash_atmega644a_UART0_38400_8000000L_B0_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0x1ec7bc40 }, // optiboot_flash_atmega644a_UART1_38400_8000000L_B0_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0xe9d90a4f }, // optiboot_flash_atmega644a_UART1_57600_8000000L_B0_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0x9d361c4e }, // optiboot_flash_atmega644a_UART0_9600_1000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0x1e9a5d21 }, // optiboot_flash_atmega644a_UART1_9600_1000000L_B0_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0x3b651e15 }, // optiboot_flash_atmega644a_UART1_9600_1000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0xea8fe774 }, // optiboot_flash_atmega644a_UART0_9600_1000000L_B0_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0xb49eaa1f }, // optiboot_flash_atmega644a_UART0_115200_20000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0xbdbcaa5b }, // optiboot_flash_atmega644a_UART1_115200_20000000L_B0_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0x8298e287 }, // optiboot_flash_atmega644a_UART1_115200_20000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0x50418be4 }, // optiboot_flash_atmega644a_UART0_115200_20000000L_B0_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0xf9f61eb2 }, // optiboot_flash_atmega644a_UART0_115200_16000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0x7b6e9bb9 }, // optiboot_flash_atmega644a_UART1_115200_16000000L_B0_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0xe27428c7 }, // optiboot_flash_atmega644a_UART1_115200_16000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0xdab405bd }, // optiboot_flash_atmega644a_UART0_115200_16000000L_B0_BIGBOOT.hex
[... cut ...]
    { 1024, 1, 0xae42ebb8, 0x82288400 }, // optiboot_flash_atmega6450_UART0_38400_8000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0xe8704cb1 }, // optiboot_flash_atmega6450_UART0_57600_8000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0xb80318bc }, // optiboot_flash_atmega6450_UART0_9600_1000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0x31550abd }, // optiboot_flash_atmega6450_UART0_115200_20000000L_B7_BIGBOOT.hex
    { 1024, 1, 0xae42ebb8, 0x28381b17 }, // optiboot_flash_atmega6450_UART0_115200_16000000L_B7_BIGBOOT.hex
```

</details>

stefanrueger, 2022-12-18:
> same hash, even though they are different parts

Hash function clashes are vvv unlikely (1/4bn). The different parts will have the same signature, and therefore the bootloader will be identical. Here one example:
```
$ grep 0xee9f5b33 /tmp/x
    {  512, 0, 0x2afe013b, 0xee9f5b33 }, // optiboot_flash_atmega169_UART0_115200_16000000L_B5.hex
    {  512, 0, 0x2afe013b, 0xee9f5b33 }, // optiboot_flash_atmega169p_UART0_115200_16000000L_B5.hex
    {  512, 0, 0x2afe013b, 0xee9f5b33 }, // optiboot_flash_atmega165_UART0_115200_16000000L_B5.hex
```
My bet is that the bootloaders for these three parts are identical, and in particular that the signatures are the same. Your list has 480 different hashes. (And I very much predict your 542-strong bootloader list has only 480 unique ones.) You get the 57 hashes that occur multiple times by 
```
$ cut -f4 -d, /tmp/hash-list  | sort | uniq -c | grep -v " 1 " | sort -n
```

It makes sense to keep only one representative of bootoaders that are identical (though there is no harm in having identical hashes in the list, the algorithm copes).

stefanrueger, 2022-12-18:
The more important question is: which of these bootloaders have EEPROM support and which don't. The script uses the heuristic(!) that all bootloaders that have bigboot or BIGBOOT in its names cope with EEPROM (ie, get a 1 in the second column).

MCUdude, 2022-12-18:
> My bet is that the bootloaders for these three parts are identical, and in particular that the signatures are the same. Your list has 480 different hashes. (And I very much predict your 542-strong bootloader list has only 480 unique ones.) You get the 57 hashes that occur multiple times by

I realize now that I should have built these bootloaders for ATmega165A and ATmega169A instead of the non-A models. There seems to be a bug in the avr-libc version I used where the non-A parts incorrectly has the same signature values in their io.h file. I should fix this, re-build the bootloader and update the hash table

> The more important question is: which of these bootloaders have EEPROM support and which don't. The script uses the heuristic(!) that all bootloaders that have bigboot or BIGBOOT in its names cope with EEPROM (ie, get a 1 in the second column).

All bootloaders that have BIGBOOT in their name support EEPROM read/write, so the script is correct.

stefanrueger, 2022-12-18:
|  I should fix this, re-build the bootloader and update the hash table

We had the discussion before: [signatures are no good indicator for the part](https://github.com/avrdudes/avrdude/pull/1091#issuecomment-1231678867). So, this cannot be fixed!

Unless you switch to urboot that is. :grinning:

And for good measure, signatures can be different between avr-libc  and avrdude.

stefanrueger, 2022-12-19:
@MCUdude I've been wondering why your (8 MHz, 57600 baud) bootloaders are different from (16 MHz, 115200 baud) bootloaders, but I now suspect that optiboot flashes the LED for a while with a certain frequency - and that therefore the code will be different for these two bootloader versions. For the comms alone the code would be identical.

mcuee, 2022-12-19:
I agree with change the lable to `wontfix`. We can keep this issue open for a while and then close it. 

We can even create a FAQ page to include information like this one.

mcuee, 2022-12-19:
> This is a well-known optiboot bug. It returns flash for EEPROM if optiboot has been compiled to not handle EEPROM. This bug has been around for a long time (and possibly still is today but defo was 2016 when I last looked closely at optiboot).

I can confirm the issue is still there with latest optiboot git.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM4 -p m328p -xbootsize=512 -xshowall

avrdude: AVR device initialized and ready to accept instructions
0 2022-06-10 21.21 Blink.ino.standard.hex 924 store 31298 meta 34 boot 512 o8.3 -?s-?-r-- vector 0 (RESET) ATmega328P

PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude -c arduino -P COM4 -p m328p -qqt
avrdude> dump eeprom 0 0x10
0000  0c 94 5c 00 0c 94 6e 00  0c 94 6e 00 0c 94 6e 00  | .\. .n. .n. .n.|

avrdude>
avrdude>
PS C:\work\avr\avrdude_test\avrdude_bin> echo "dump eeprom 0 0x10" | .\avrdude -c usbasp -p m328p -qqt
avrdude> dump eeprom 0 0x10
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude>

```

mcuee, 2022-12-19:
@WestfW
Just wondering if you have any intentions to fix this paticular optiboot bug. Thanks.

@MCUdude and @SpenceKonde
Just wondering if you know this bug or not. I consider it as a serious bug.

MCUdude, 2022-12-19:
> We had the discussion before: https://github.com/avrdudes/avrdude/pull/1091#issuecomment-1231678867. So, this cannot be fixed!

Signatures aren't the best, I agree. Atmel is to blame here for their inconsistency. However, in my case, uploading to an ATmega169A _will fail_, because the signature hardcoded into the device differs from the one in the bootloader. I'll ditch the non-A parts for now. I'm doing this because I want a last "known-good" [MegaCore](https://github.com/MCUdude/MegaCore) release before migrating everything to Urboot.

> Just wondering if you know this bug or not. I consider it as a serious bug.

I wasn't aware of it, but since Arduino IDE doesn't support uploading directly to EEPROM, it has never really been an issue. If I'm going to switch to Urboot later, I don't think I'll bother fixing it and compiling all new hex files.

mcuee, 2022-12-19:
> I wasn't aware of it, but since Arduino IDE doesn't support uploading directly to EEPROM, it has never really been an issue. 

Ah, that explains why this issue was not noticed for so long.

mcuee, 2022-12-20:
https://github.com/Optiboot/optiboot/issues/357 raised against Optiboot.

mcuee, 2022-12-20:
Re-open this issue so that we can continue which hash to be included in avrdude.

SpenceKonde, 2022-12-20:
@MCUdude:  What is this "Urboot" thing"?

mcuee, 2022-12-20:
@SpenceKonde 
It is here.
https://github.com/stefanrueger/urboot

I am thinking it may be useful for your ATtinyCore.

MCUdude, 2022-12-20:
@SpenceKonde It's a new bootloader written _together_ with the Avrdude implementation of the protocol. It has _all_ the bells and whistles you might dream of (Only for classic AVRs for now), and it occupies very little flash space. The virtual bootloaders are very interesting and are impossible to brick as long as you don't wipe or modify the flash contents with an ISP programmer.

It lets you:
* Read and write to flash and EEPROM using -U flash or -U eeprom
* Read and write single bytes to flash and EEPROM in Avrdude terminal mode -t
* Stores metadata of your program, like flashing date, program name, and size.
* For virtual bootloaders, Avrdude uploads a patched version of the user program to ensure there always is a reset vector present
* Supports auto baud, with no upper limit other than the AVR clock speed.
* Option to make it backward compatible with -c arduino
* plus more

More about the Urboot bootloader here: https://github.com/stefanrueger/urboot/blob/b068ad0a16ac6e11690a72172c43ef689c8b8425/src/urboot.c#L1-L709

More about the Avrdude protocol implementation (Urclock) and what it supports here:
https://github.com/avrdudes/avrdude/pull/1171

The Avrdude docs (avrdude.texi and avrdude.1) also contain lots of information.

Bottom line: It's an incredibly feature-rich and packed bootloader designed around a custom protocol _specifically_ for bootloader usage. It's an impressive piece of software, and I'm planning to migrate all my Arduino cores to use Urboot and ditch Optiboot completely. And for ATtinyCore where lots of parts use virtual bootloaders, It's a no-brainer.

mcuee, 2022-12-21:
> Re-open this issue so that we can continue which hash to be included in avrdude.

That has been addressed in PR #1226.
* #1226

closed 2022-12-21 by mcuee