avrdudes/avrdude #1228: linuxgpio cannot use pin 0

wijnen, 2022-12-18:
Most devices with gpio pins start counting from 0. However, when defining a linuxgpio programmer to use e.g. reset=0, it fails with the message "avrdude: error: no pin has been assigned for AVR RESET". Apparently a value of 0 counts as "undefined".

Because of this, it is impossible to use pin 0, even though it is a perfectly valid pin to use.

I'm not sure if there needs to be a way to specify "undefined" for a pin, but if there is, it should be something that cannot be a real pin, such as the word "undefined", or a negative value.

For reference, here's a sample avrdude.conf file:

```
programmer
  id	= "opi";
  desc	= "Program using OrangePi Zero GPIO pins";
  type	= "linuxgpio";
  reset	= 0;
  sck	= 11;
  mosi	= 12;
  miso	= 6;
;
```

I use it with "sudo avrdude -C +avrdude.conf -c opi -p attiny24 -U hfuse:r:-:h"

MCUdude, 2022-12-18:
Hi @wijnen, and thanks for reporting!

This check is what's causing the error:

https://github.com/avrdudes/avrdude/blob/159d12693e8787fab5bf68d87f331bedf85164a1/src/bitbang.c#L611-L617

Can you try to modify the source code (for instance, replace the code with the code below) just to check if using GPIO 0 will actually work?

```c
static int verify_pin_assigned(const PROGRAMMER *pgm, int pin, char *desc) {
  return 0;
}
```

wijnen, 2022-12-19:
To do it right, the value 0 should be valid and not mean "unused". I've written a patch (for the Debian stable version, so it doesn't exactly apply to the code here,  but that should be easy to fix).

If I have time, I'll turn it into a pull request. But if it gets merged before I get around to it, that's even better. :-)

Here is the [patch file](https://github.com/avrdudes/avrdude/files/10255074/avrdude.patch.txt).

The system I use it on is too weak to compile avrdude on, so I couldn't test it there. I did however test it on my regular PC and it works as I would expect. That means:
  - it does not complain about undefined optional pins, such as vcc.
  - it does complain about missing required pins (such as mosi).
  - it attempts to create gpio0 when asked to.
  - it also still attempts to create other gpio pins when asked to.

So I'm pretty confident that this will work. If you have no way of testing it on hardware, I can do it, but it's a lot of work which I'd rather avoid.

mcuee, 2022-12-19:
@wijnen 
PR will be most welcome. You can carry out the test on your target later. 
I have a few OrangePi boards and I will see if I can get Linux GPIO work on one of them.

wijnen, 2022-12-19:
I made the PR. I haven't had time to test it at all yet and I'm not sure when I will, so I thought I'd just send it now. Please check and test it before merging.

wijnen, 2022-12-21:
I saw a comment that asked for a #define NO_PIN -1. That seemed like a good idea, so I did that. Now it says there is a review that requests a change, but I can't see the review, or the requested change. I did a commit --amend and a push --force, so perhaps that caused some things to get lost.

If there is still a request for a change, please let me know what that is (and where I could have found it).

Anyway, what seems weird to me is the pin mask. The code expects pins to be able to be masked out of the pin number with a binary and operation, it seems. I'm guessing this would work for other systems, but it will certainly not work for linuxgpio. So while the change (from (pinno & mask) == 0 to pinno == NO_PIN) makes no difference for linuxgpio, I'm not sure if I'm breaking anything for other targets.

mcuee, 2022-12-24:
@wijnen 

@stefanrueger has updated your PR. Please review as well. Thanks.

mcuee, 2022-12-26:
I need to find out how to use GPIO_0 under Raspberry Pi, it seems to be reserved (no GPIO_0/GPIO_1).
https://projects.raspberrypi.org/en/projects/physical-computing/1

mcuee, 2022-12-26:
Using GPIO_25 as RESET Pin:

```
pi@raspberrypi:~/Desktop/build/avr/avrdude_bin $ ./avrdude -c linuxgpio -P GPIO -p m328p -v -U flash:r:m328p_read.hex:i

avrdude: Version 7.0-20221223 (8573442)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /home/pi/Desktop/build/avr/avrdude_bin/avrdude.conf
         User configuration file is /home/pi/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : GPIO
         Using Programmer              : linuxgpio
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : linuxgpio
         Description     : Use the Linux sysfs interface to bitbang GPIO lines
         Pin assignment  : /sys/class/gpio/gpio{n}
           RESET   =  25
           SCK     =  11
           SDO     =  10
           SDI     =  9

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: reading flash memory ...

Reading | ################################################## | 100% 6.48 s 

avrdude: writing output file m328p_read.hex

avrdude done.  Thank you.

```

mcuee, 2022-12-26:
PR #1231 is good.

But I think it is too verbose.

```
pi@raspberrypi:~/Desktop/build/avr/avrdude_bin $ ./avrdude_pr1231 -c linuxgpio -p m328p -v

avrdude_pr1231: Version 7.0-20221223 (184c62c)
                Copyright the AVRDUDE authors;
                see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                System wide configuration file is /home/pi/Desktop/build/avr/avrdude_bin/avrdude.conf
                User configuration file is /home/pi/.avrduderc
                User configuration file does not exist or is not a regular file, skipping

                Using Port                    : /dev/parport0
                Using Programmer              : linuxgpio
                AVR Part                      : ATmega328P
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PC2
                RESET disposition             : possible i/o
                RETRY pulse                   : SCK
                Serial program mode           : yes
                Parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                                  Block Poll               Page                       Polled
                  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
                  flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
                  lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                  hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                  efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                  lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                  signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                  calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

                Programmer Type : linuxgpio
                Description     : Use the Linux sysfs interface to bitbang GPIO lines
                Pin assignment  : /sys/class/gpio/gpio{n}
                  RESET   =  0
                  SCK     =  11
                  SDO     =  10
                  SDI     =  9

11110avrdude_pr1231: AVR device initialized and ready to accept instructions
011011011avrdude_pr1231: device signature = 0x1e950f (probably m328p)
1
avrdude_pr1231 done.  Thank you.

```

mcuee, 2022-12-26:
Hmm, I think this is related to GPIO_0 being a special pin for Raspberry Pi (I am testing with Raspberry Pi 3B+). And it has nothing to do with the PR.

I have tried without `-v` and even `-qq`, it still prints out those binary streams.

stefanrueger, 2022-12-26:
> I have tried without `-v` and even `-qq`, it still prints out those binary streams.

The PR certainly does not interfere with messaging. Have you ensured you rebuilt avrdude from scratch? Not doing so is a common cause for tests that go wrong.

mcuee, 2022-12-26:
> > I have tried without `-v` and even `-qq`, it still prints out those binary streams.
> 
> The PR certainly does not interfere with messaging. Have you ensured you rebuilt avrdude from scratch? Not doing so is a common cause for tests that go wrong.

Yes I have done that. I believe it has something to do with the change to `config.txt` even though I have reverted it and reboot.

mcuee, 2022-12-26:
I have also tested `-c linuxspi` and confirmed it is not affected by this issue. PR #1231 is still good so there is no regression.
* #1231
For others who want to use `linuxspi` or `linuxgpio`, please take note that there is a bug in git main.
* #1107 

Work-around for #1107, please do not use `ON` but rather `1` to enable `linuxspi` and `linuxgpio`.
```
pi@raspberrypi:~/Desktop/build/avr/avrdude_git $ cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo
 -D HAVE_LINUXGPIO=1 -D HAVE_LINUXSPI=1 -B build_linux
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.20.1") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "3.3.2")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- ENABLED    linuxgpio
-- ENABLED    linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Desktop/build/avr/avrdude_git/build_linux

pi@raspberrypi:~/Desktop/build/avr/avrdude_git $ cmake --build build_linux
[  1%] [BISON][Parser] Building parser with bison 3.3.2
[  3%] [FLEX][Parser] Building scanner with flex 2.6.4
Scanning dependencies of target libavrdude
[  4%] Building C object src/CMakeFiles/libavrdude.dir/arduino.c.o
[  6%] Building C object src/CMakeFiles/libavrdude.dir/avr.c.o
[  7%] Building C object src/CMakeFiles/libavrdude.dir/avr910.c.o
[  9%] Building C object src/CMakeFiles/libavrdude.dir/avrcache.c.o
[ 10%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi.c.o
[ 12%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.o
[ 13%] Building C object src/CMakeFiles/libavrdude.dir/avrpart.c.o
[ 15%] Building C object src/CMakeFiles/libavrdude.dir/bitbang.c.o
[ 16%] Building C object src/CMakeFiles/libavrdude.dir/buspirate.c.o
[ 18%] Building C object src/CMakeFiles/libavrdude.dir/butterfly.c.o
[ 19%] Building C object src/CMakeFiles/libavrdude.dir/config.c.o
[ 21%] Building C object src/CMakeFiles/libavrdude.dir/confwin.c.o
[ 22%] Building C object src/CMakeFiles/libavrdude.dir/crc16.c.o
[... 32 lines cut ...]
[ 72%] Building C object src/CMakeFiles/libavrdude.dir/updi_state.c.o
[ 74%] Building C object src/CMakeFiles/libavrdude.dir/urclock.c.o
[ 75%] Building C object src/CMakeFiles/libavrdude.dir/usbasp.c.o
[ 77%] Building C object src/CMakeFiles/libavrdude.dir/usb_hidapi.c.o
[ 78%] Building C object src/CMakeFiles/libavrdude.dir/usb_libusb.c.o
[ 80%] Building C object src/CMakeFiles/libavrdude.dir/usbtiny.c.o
[ 81%] Building C object src/CMakeFiles/libavrdude.dir/update.c.o
[ 83%] Building C object src/CMakeFiles/libavrdude.dir/wiring.c.o
[ 84%] Building C object src/CMakeFiles/libavrdude.dir/xbee.c.o
[ 86%] Building C object src/CMakeFiles/libavrdude.dir/__/lexer.c.o
[ 87%] Building C object src/CMakeFiles/libavrdude.dir/__/config_gram.c.o
[ 89%] Linking C static library libavrdude.a
[ 89%] Built target libavrdude
Scanning dependencies of target avrdude
[ 90%] Building C object src/CMakeFiles/avrdude.dir/main.c.o
[ 92%] Building C object src/CMakeFiles/avrdude.dir/term.c.o
[ 93%] Building C object src/CMakeFiles/avrdude.dir/avrintel.c.o
[ 95%] Building C object src/CMakeFiles/avrdude.dir/developer_opts.c.o
[ 96%] Building C object src/CMakeFiles/avrdude.dir/whereami.c.o
[ 98%] Linking C executable avrdude
[ 98%] Built target avrdude
Scanning dependencies of target conf
[100%] Generating avrdude.conf
[100%] Built target conf
```

mcuee, 2022-12-26:
> > The PR certainly does not interfere with messaging. Have you ensured you rebuilt avrdude from scratch? Not doing so is a common cause for tests that go wrong.
> 
> Yes I have done that. I believe it has something to do with the change to `config.txt` even though I have reverted it and reboot.
> 
> In any case, as you mentioned, it is not related to PR #1231 and it is good to merge PR #1231 now.

Further tests with a new setup with Raspberry Pi 400 confirms this PR is the issue. But I am not so sure how to fix it.

MCUdude, 2023-01-02:
Fixed in #1231

closed 2023-01-02 by MCUdude