avrdudes/avrdude #1229: `-c arduino` keep-alive feature similar to `-c urclock`.

mcuee, 2022-12-19:
As of now `-c urclock can keep bootloaders in terminal mode `-t` without the bootloader resetting itself through the watchdog timer.

This is a nice feature which would be nice to bring to `-c arduino` as well.

stefanrueger, 2022-12-19:
Why? There is `-c urclock` that can be used instead. Duplicating code for `-c arduino` costs effort and increases the maintenance surface. Our energy is better spent on bug fixes, or implementing keep-alive for `-c wiring`, `-c xbee` etc. And before you rush raising issues for that just a note that I for one don't know what NOPs look like for these bootloaders, so this needs people using these bootloaders to chip in with their knowledge.

mcuee, 2022-12-19:
I think this request came from @MCUdude. Let's see if he is still interested in this improvement on `-c arduino` or not. If not we can close this issue as not planned.

mcuee, 2022-12-25:
@MCUdude 

Do you still think this feature is useful? If not I will close this issue. Thanks,

MCUdude, 2022-12-27:
Even though Urboot is vastly superior, there are still millions(?) of microcontrollers out there that run Optiboot. I'd argue that this _would_ be a nice feature for those who are stuck with Optiboot or can't/don't want to switch to a new and improved bootloader.

mcuee, 2022-12-28:
From here.
* https://github.com/avrdudes/avrdude/pull/1171#issuecomment-1308724541

Being able to use the pgm->term_keep_alive() function relies on the following:

1. AVRDUDE is compiled with libreadline
2. The programmer provides a function term_keep_alive() to keep the bootloader happy
3. The timing of reading characters and draining serial input does not get in the way
4. AVRDUDE is able to figure out whether or not the next read from stdin would block

stefanrueger, 2022-12-28:
> there are still millions(?) of microcontrollers out there that run Optiboot. 

And they can use `-c urclock` if they want to use the keep-alive feature, can't they? So, I'd rather spend time on other things.

MCUdude, 2022-12-29:
> And they can use -c urclock if they want to use the keep-alive feature, can't they?

Ah, of course. Totally forgot this, sorry

closed 2022-12-29 by MCUdude