avrdudes/avrdude #1230: `-c urclock` does not write the meta data

mcuee, 2022-12-19:
This is a strange issue I first encountered on a new Nano clone (ATmega328P, 5V, 16MHz crystal, CH340). It does not write the meta data.

@stefanrueger 
Do you think this is a specific issue for this board?

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM12 -p m328p  -xshowall

avrdude: AVR device initialized and ready to accept instructions
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 o4.4 --s-h-r-- vector 0 (RESET) ATmega328P
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM12 -p m328p  -U .\hex\Blink.ino.standard.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
         delaying chip erase until first -U upload to flash
avrdude: reading input file .\hex\Blink.ino.standard.hex for flash
         with 924 bytes in 1 section within [0, 0x39b]
         using 8 pages and 100 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 924 bytes flash ...

Writing | ################################################## | 100% 5.76 s

avrdude: 924 bytes of flash written
avrdude: verifying flash memory against .\hex\Blink.ino.standard.hex

Reading | ################################################## | 100% 4.57 s

avrdude: 924 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM12 -p m328p  -xshowall

avrdude: AVR device initialized and ready to accept instructions
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 o4.4 --s-h-r-- vector 0 (RESET) ATmega328P
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM12 -p m328p  -qqt
avrdude> dump flash 0x7d00
7d00  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7d10  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7d20  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7d30  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7d40  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7d50  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7d60  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7d70  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7d80  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7d90  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7da0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7db0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7dc0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7dd0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7de0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7df0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> dump flash 0x7e00
7e00  11 24 84 b7 14 be 81 ff  f0 d0 85 e0 80 93 81 00  |.$..............|
7e10  82 e0 80 93 c0 00 88 e1  80 93 c1 00 86 e0 80 93  |................|
7e20  c2 00 80 e1 80 93 c4 00  8e e0 c9 d0 25 9a 86 e0  |............%...|
7e30  20 e3 3c ef 91 e0 30 93  85 00 20 93 84 00 96 bb  | .<...0... .....|
7e40  b0 9b fe cf 1d 9a a8 95  81 50 a9 f7 cc 24 dd 24  |.........P...$.$|
7e50  88 24 83 94 b5 e0 ab 2e  a1 e1 9a 2e f3 e0 bf 2e  |.$..............|
7e60  a2 d0 81 34 61 f4 9f d0  08 2f af d0 02 38 11 f0  |...4a..../...8..|
7e70  01 38 11 f4 84 e0 01 c0  83 e0 8d d0 89 c0 82 34  |.8.............4|
7e80  11 f4 84 e1 03 c0 85 34  19 f4 85 e0 a6 d0 80 c0  |.......4........|
7e90  85 35 79 f4 88 d0 e8 2e  ff 24 85 d0 08 2f 10 e0  |.5y......$.../..|
7ea0  10 2f 00 27 0e 29 1f 29  00 0f 11 1f 8e d0 68 01  |./.'.).)......h.|
7eb0  6f c0 86 35 21 f4 84 e0  90 d0 80 e0 de cf 84 36  |o..5!..........6|
7ec0  09 f0 40 c0 70 d0 6f d0  08 2f 6d d0 80 e0 c8 16  | .@.p.o../m.....|
7ed0  80 e7 d8 06 18 f4 f6 01  b7 be e8 95 c0 e0 d1 e0  |................|
7ee0  62 d0 89 93 0c 17 e1 f7  f0 e0 cf 16 f0 e7 df 06  |b... ...........|
7ef0  18 f0 f6 01 b7 be e8 95  68 d0 07 b6 00 fc fd cf  |........h.......|

avrdude> dump flash 0x7f00
7f00  a6 01 a0 e0 b1 e0 2c 91  30 e0 11 96 8c 91 11 97  |......,.0.......|
7f10  90 e0 98 2f 88 27 82 2b  93 2b 12 96 fa 01 0c 01  |.../.'.+.+.... .|
7f20  87 be e8 95 11 24 4e 5f  5f 4f f1 e0 a0 38 bf 07  |.....$N__O...8..|
7f30  51 f7 f6 01 a7 be e8 95  07 b6 00 fc fd cf 97 be  |Q...............|
7f40  e8 95 26 c0 84 37 b1 f4  2e d0 2d d0 f8 2e 2b d0  |..&..7....-...+.|
7f50  3c d0 f6 01 ef 2c 8f 01  0f 5f 1f 4f 84 91 1b d0  |<....,..._.O....|
7f60  ea 94 f8 01 c1 f7 08 94  c1 1c d1 1c fa 94 cf 0c  |............... |
7f70  d1 1c 0e c0 85 37 39 f4  28 d0 8e e1 0c d0 85 e9  |.....79.(... ...|
7f80  0a d0 8f e0 7a cf 81 35  11 f4 88 e0 18 d0 1d d0  | ...z..5........|
7f90  80 e1 01 d0 65 cf 98 2f  80 91 c0 00 85 ff fc cf  |....e../........|
7fa0  90 93 c6 00 08 95 80 91  c0 00 87 ff fc cf 80 91  |................|
7fb0  c0 00 84 fd 01 c0 a8 95  80 91 c6 00 08 95 e0 e6  |................|
7fc0  f0 e0 98 e1 90 83 80 83  08 95 ed df 80 32 19 f0  |.............2..|
7fd0  88 e0 f5 df ff cf 84 e1  de cf 1f 93 18 2f e3 df  |............./..|
7fe0  11 50 e9 f7 f2 df 1f 91  08 95 80 e0 e8 df ee 27  |.P.............'|
7ff0  ff 27 09 94 ff ff ff ff  ff ff ff ff ff ff 04 04  |.' .............|

avrdude> quit
avrdude>

```

mcuee, 2022-12-19:
So I get the Arduino Uno clone (more proper clone using ATmega16U2 USB to Serial chip, similar design as the official Uno). And it has no issues.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM1 -p m328p  -qqt -xshowall
0 0000-00-00 00.00  application 0 store 0 meta 0 boot 512 o4.4 --s-h-r-- vector 0 (RESET) ATmega328P
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM1 -p m328p  -U .\hex\Blink.ino.standard.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
         delaying chip erase until first -U upload to flash
avrdude: reading input file .\hex\Blink.ino.standard.hex for flash
         with 924 bytes in 1 section within [0, 0x39b]
         using 8 pages and 100 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 924 bytes flash ...

Writing | ################################################## | 100% 6.41 s

avrdude: 924 bytes of flash written
avrdude: verifying flash memory against .\hex\Blink.ino.standard.hex

Reading | ################################################## | 100% 4.89 s

avrdude: 924 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM1 -p m328p  -qqt -xshowall
0 2022-06-10 21.21 Blink.ino.standard.hex 924 store 31298 meta 34 boot 512 o4.4 --s-h-r-- vector 0 (RESET) ATmega328P

```

mcuee, 2022-12-19:
I have not got the chance to solder the 6-pin ISP connector on the Nano clone. I will need to check the fuse settings to see if that plays a part. But this is kind of strange.

stefanrueger, 2022-12-19:
If the only change you made was changing the board then it is likely the fault of the ATmega328P (counterfeit?) part. I have had that before: some parts you get on Aliexpress won't write from the bootloader section to the NRWW section [0x7000, 0x7fff] neither for love nor for money. You can use any bootloader you like, any AVRDUDE bootloader programmer (`-c urclock`, `-c arduino`, `-c wiring`, ...), stand on your head, try during new moon. The part just won't write anything to the NRWW area from the bootloader. 

Try creating a bespoke blink.hex that has a large 0x55 data section in the top flash up to, say, 0x7fc80, and try writing that .hex with any bootloader you like. _If nothing is written above 0x7000_ then it's a dud part: Raise an issue with the counterfeit manufacturer, raise a stink with your dealer and change the subject of this issue (as it's not an urclock issue).

mcuee, 2022-12-20:
@stefanrueger 

Interesting you have encountered counterfeit chip problem. I have not encountered that myself. I think there is no counterfeit chip but rather there maybe old defective chip problems out there.

Anyway, I found the issue to be fuse setting issues by attaching 2x3 pin headers. Somehow the fuse setting use 1024 Words bootloader (2048 Bytes).
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM12 -p m328p  -qqt
avrdude> dump flash 0x7000 0x10
7000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> write flash 0x7000 0xaa 0xff
avrdude> flush
avrdude> dump flash 0x7000 0x10
7000  aa ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> dump flash 0x7800 0x10
7800  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> write flash 0x7800 0xaa 0xff
avrdude> flush
avrdude error: verification mismatch at flash page addr 0x7800
avrdude> quit
avrdude> avrdude error: verification mismatch at flash page addr 0x7800

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p m328p  -qqt
avrdude> dump lfuse
0000  ff                                                |.               |

avrdude> dump hfuse
0000  da                                                |.               |

avrdude> dump efuse
0000  fd                                                |.               |

avrdude> dump lock
0000  cf                                                |.               |

avrdude> quit
avrdude>
```

stefanrueger, 2022-12-20:
OK, so the fuses/lock bits were set to protect the top 2048 bytes from writing, so the issue description is wrong and the issue invalid?

closed 2022-12-20 by mcuee

mcuee, 2022-12-20:
Interesting that I encountered another Nano clone (ATmega328P, 5V, 16MHz, USB Type C).

mcuee, 2022-12-20:
The more common issue seems to be non-working bootloaders. That was why I usually reflash the bootloaders from Chinese Nano Clones.

This is one example (Nano 328P, 5V, 16MHz crystal, CH340, USB Type C clone). I can see the LEDs

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p m328p -qqt
avrdude> dump lfuse
0000  ff                                                |.               |

avrdude> dump hfuse
0000  da                                                |.               |

avrdude> dump lock
0000  cf                                                |.               |

avrdude> dump efuse
0000  fd                                                |.               |

avrdude> dump flash 0x7800 2048
7800  0c 94 34 3c 0c 94 51 3c  0c 94 51 3c 0c 94 51 3c  | .4< .Q< .Q< .Q<|
7810  0c 94 51 3c 0c 94 51 3c  0c 94 51 3c 0c 94 51 3c  | .Q< .Q< .Q< .Q<|
7820  0c 94 51 3c 0c 94 51 3c  0c 94 51 3c 0c 94 51 3c  | .Q< .Q< .Q< .Q<|
7830  0c 94 51 3c 0c 94 51 3c  0c 94 51 3c 0c 94 51 3c  | .Q< .Q< .Q< .Q<|
7840  0c 94 51 3c 0c 94 51 3c  0c 94 51 3c 0c 94 51 3c  | .Q< .Q< .Q< .Q<|
7850  0c 94 51 3c 0c 94 51 3c  0c 94 51 3c 0c 94 51 3c  | .Q< .Q< .Q< .Q<|
7860  0c 94 51 3c 0c 94 51 3c  11 24 1f be cf ef d8 e0  | .Q< .Q<.$......|
7870  de bf cd bf 11 e0 a0 e0  b1 e0 ec e9 ff e7 02 c0  |................|
7880  05 90 0d 92 a2 30 b1 07  d9 f7 12 e0 a2 e0 b1 e0  |.. ..0..........|
7890  01 c0 1d 92 ad 30 b1 07  e1 f7 0e 94 2d 3d 0c 94  |.....0......-= .|
78a0  cc 3f 0c 94 00 3c 98 2f  95 95 95 95 95 95 95 95  |.? ..<./........|
78b0  90 5d 8f 70 8a 30 7c f0  28 2f 29 5a 80 91 c0 00  |.].p.0|.(/)Z....|
78c0  85 ff fc cf 90 93 c6 00  80 91 c0 00 85 ff fc cf  |................|
78d0  20 93 c6 00 08 95 28 2f  20 5d f0 cf 98 2f 80 91  | .....(/ ].../..|
78e0  c0 00 85 ff fc cf 90 93  c6 00 08 95 ef 92 ff 92  |................|
78f0  0f 93 1f 93 ee 24 ff 24  87 01 80 91 c0 00 87 fd  |.....$.$........|
7900  17 c0 08 94 e1 1c f1 1c  01 1d 11 1d 81 e4 e8 16  |................|
7910  82 e4 f8 06 8f e0 08 07  80 e0 18 07 70 f3 e0 91  |............p...|
7920  04 01 f0 91 05 01 09 95  80 91 c0 00 87 ff e9 cf  |...... .........|
7930  80 91 c6 00 1f 91 0f 91  ff 90 ef 90 08 95 0e 94  |................|
7940  76 3c 98 2f 80 91 c0 00  85 ff fc cf 90 93 c6 00  |v<./............|
7950  91 36 2c f4 90 33 0c f0  90 53 89 2f 08 95 97 55  |.6,..3 ..S./...U|
7960  89 2f 08 95 1f 93 0e 94  9f 3c 18 2f 0e 94 9f 3c  |./.......<./...<|
7970  12 95 10 7f 81 0f 1f 91  08 95 1f 93 18 2f 88 23  |............./.#|
7980  21 f0 0e 94 76 3c 11 50  e1 f7 1f 91 08 95 1f 93  |!...v<.P........|
7990  18 2f 0e 94 76 3c 80 32  49 f0 80 91 03 01 8f 5f  |./..v<.2I......_|
79a0  80 93 03 01 85 30 c1 f0  1f 91 08 95 80 91 c0 00  |.....0..........|
79b0  85 ff fc cf 84 e1 80 93  c6 00 80 91 c0 00 85 ff  |................|
79c0  fc cf 10 93 c6 00 80 91  c0 00 85 ff fc cf 80 e1  |................|
79d0  80 93 c6 00 1f 91 08 95  e0 91 04 01 f0 91 05 01  |................|
79e0  09 95 1f 91 08 95 0e 94  76 3c 80 32 41 f0 80 91  | .......v<.2A...|
79f0  03 01 8f 5f 80 93 03 01  85 30 81 f0 08 95 80 91  |..._.....0......|
7a00  c0 00 85 ff fc cf 84 e1  80 93 c6 00 80 91 c0 00  |................|
7a10  85 ff fc cf 80 e1 80 93  c6 00 08 95 e0 91 04 01  |................|
7a20  f0 91 05 01 09 95 08 95  40 e9 51 e0 88 23 a1 f0  |.... ...@.Q..#..|
7a30  2d 9a 28 ee 33 e0 fa 01  31 97 f1 f7 21 50 30 40  |-.(.3...1...!P0@|
7a40  d1 f7 2d 98 28 ee 33 e0  fa 01 31 97 f1 f7 21 50  |..-.(.3...1...!P|
7a50  30 40 d1 f7 81 50 61 f7  08 95 3f 92 4f 92 5f 92  |0@...Pa...?.O._.|
7a60  6f 92 7f 92 8f 92 9f 92  af 92 bf 92 cf 92 df 92  |o...............|
7a70  ef 92 ff 92 0f 93 1f 93  cf 93 df 93 00 00 80 e1  |................|
7a80  80 93 c4 00 10 92 c5 00  88 e1 80 93 c1 00 86 e0  |................|
7a90  80 93 c2 00 50 98 58 9a  25 9a 81 e0 0e 94 14 3d  |....P.X.%......=|
7aa0  24 e1 f2 2e 9e e1 e9 2e  85 e9 d8 2e 0f e0 c0 2e  |$...............|
7ab0  10 e1 b1 2e aa 24 a3 94  b1 e4 9b 2e a6 e5 8a 2e  |.....$..........|
7ac0  f2 e5 7f 2e e0 e2 6e 2e  79 e4 57 2e 63 e5 46 2e  |......n.y.W.c.F.|
7ad0  50 e5 35 2e 0e 94 76 3c  80 33 b1 f1 81 33 b9 f1  |P.5...v<.3...3..|
7ae0  80 34 09 f4 6f c0 81 34  09 f4 76 c0 82 34 09 f4  |.4 .o..4 .v..4 .|
7af0  85 c0 85 34 09 f4 88 c0  80 35 31 f1 82 35 21 f1  |...4 ....51..5!.|
7b00  81 35 11 f1 85 35 09 f4  85 c0 86 35 09 f4 8d c0  |.5...5 ....5 ...|
7b10  84 36 09 f4 96 c0 84 37  09 f4 03 c1 85 37 09 f4  |.6 ....7 ....7 .|
7b20  72 c1 86 37 09 f4 66 c0  80 91 03 01 8f 5f 80 93  |r..7 .f......_..|
[... 56 lines cut ...]
7eb0  88 0b 8a 21 80 93 0b 02  80 91 06 01 90 91 07 01  |. .!.. .........|
7ec0  88 0f 99 1f 90 93 07 01  80 93 06 01 80 91 08 02  |................|
7ed0  80 ff 09 c0 80 91 08 02  90 91 09 02 01 96 90 93  |.. ....... .....|
7ee0  09 02 80 93 08 02 f8 94  f9 99 fe cf 11 27 e0 91  | ............'..|
7ef0  06 01 f0 91 07 01 c8 e0  d1 e0 80 91 08 02 90 91  |................|
7f00  09 02 10 30 91 f4 00 91  57 00 01 70 01 30 d9 f3  | ..0....W..p.0..|
7f10  03 e0 00 93 57 00 e8 95  00 91 57 00 01 70 01 30  |....W.....W..p.0|
7f20  d9 f3 01 e1 00 93 57 00  e8 95 09 90 19 90 00 91  |......W... .....|
7f30  57 00 01 70 01 30 d9 f3  01 e0 00 93 57 00 e8 95  |W..p.0......W...|
7f40  13 95 10 34 98 f0 11 27  00 91 57 00 01 70 01 30  |...4...'..W..p.0|
7f50  d9 f3 05 e0 00 93 57 00  e8 95 00 91 57 00 01 70  |......W.....W..p|
7f60  01 30 d9 f3 01 e1 00 93  57 00 e8 95 32 96 02 97  |.0......W...2...|
[... cut ...]
7f90  85 e9 0e 94 c7 3c 9e cd  f8 94 ff cf 80 00 ff ff  |.....<..........|
7fa0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7fb0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7fc0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7fd0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7fe0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
7ff0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> quit
avrdude>
```