avrdudes/avrdude #1235: Write fuses and toggle target power on the Xplained Mini

MCUdude, 2022-12-22:
It turns out that Xplained Mini's (ATmega32U4 based) can't change the target fuse settings by default.
In order to do this, the SUFFER register needs to be modified. What an awesome name by the way!

The SUFFER register can either be modified using the atprogram.exe that's bundled with Microchip Studio 7, or by using the open-source [pyedbgutils](https://github.com/mraardvark/pyedbgutils) script. It would be really neat if Avrdude could support writing to the SUFFER register, perhaps through the -x flag: `-xsuffer=0xff`.

The Xplained Mini does also have a software-controlled target power switch that can be controlled using Atmel Studio. It would also be very neat if this could be controlled using Avrdude: `-xtargetpower=1`.

It _should_ be possible to implement this. I'll see what I can do.

From the [ATmega328PB Xplained Mini user guide](https://ww1.microchip.com/downloads/en/DeviceDoc/ATmega328PB-Xplained-Mini-UG-DS50002660B.pdf):
[image]

MCUdude, 2022-12-23:
I got SUFFER read/write support working on Xplained Mini's with UPDI. I'll have to look more on how to do it in stk500v2.c.

https://github.com/MCUdude/avrdude/tree/xplainedmini-suffer

MCUdude, 2022-12-26:
I now have SUFFER support for Xplained Mini's with ISP as well, so now I only need to figure out how to control the target power switch. Should absolutely be doable, but it takes some trial and error because the protocol isn't all that well documented

MCUdude, 2022-12-26:
Controlling the target power was quite easy once I read the [EDBG documentation](https://onlinedocs.microchip.com/pr/GUID-33422CDF-8B41-417C-9C31-E4521ADAE9B4-en-US-2/index.html). However, there are several different boards that can control the target power. All Xplained Pros, Xplained Minis, and Curiosity Nanos support this.

```c
  unsigned char buf[1];
  buf[0] = 0x01; // 0x00 -> power off, 0x01 -> power on
  jtag3_setparm(pgm, SCOPE_EDBG, EDBG_CTXT_CONTROL, EDBG_CONTROL_TARGET_POWER, buf, 1);
```

@stefanrueger It would be really great if the programmer that supported this had their own field in avrdude.conf, perhaps `target_power_ctrl = true` or similar instead of hard-coding it to the programmer name. I've tried to add this property to config_gram.y and lexer.l, but I couldn't figure it out how to add an "independent field" like this.

MCUdude, 2022-12-26:
Perhaps it would be even better to have a dedicated field for "special programmer features"? Then we could clean up existing code that depends on the programmer name, for instance here:

https://github.com/avrdudes/avrdude/blob/857344295c0fbb5259bc7c9f3293e9a5d2b218bd/src/jtag3.c#L3119-L3125

This could look something like this:

avrdude.conf
```
#------------------------------------------------------------
# pkobn_updi
#------------------------------------------------------------

programmer
    id                     = "pkobn_updi";
    desc                   = "Curiosity nano (nEDBG) in UPDI mode";
    type                   = "jtagice3_updi";
    prog_modes             = PM_UPDI;
    extra_features         = HAS_VTARG_ADJ | HAS_VTARG_SWITCH;
    connection_type        = usb;
    usbpid                 = 0x2175;
    hvupdi_support         = 1;
;

#------------------------------------------------------------
# xplainedmini
#------------------------------------------------------------

programmer
    id                     = "xplainedmini";
    desc                   = "Atmel AVR XplainedMini in ISP mode";
    type                   = "jtagice3_isp";
    prog_modes             = PM_ISP;
    extra_features         =  HAS_VTARG_SWITCH | HAS_SUFFER;
    connection_type        = usb;
    usbpid                 = 0x2145;
;
```

The obvious "extra features" I'm aware of are these, but there are probably more that other programmers might have.
```
#define HAS_VTARG_SWITCH 1
#define HAS_VTARG_ADJ    2
#define HAS_SUFFER       4
```

stefanrueger, 2022-12-26:
SUFFER! Was this added to the data sheet on 1 April? :smile:

The second commit of PR #1238 demonstrates how a true/false variable can be added to the programmer structure. The first commit adds a general Boolean type for `avrdude.conf` (the corresponding `libavrdude.h` field can be `bool` or `int`, I ever so slightly prefer `int`).

@MCUdude Feel free to model either as distinct Boolean components in avrdude.h (such as `target_power_ctrl`, `vtarget_switch` or `suffer_enable` or some such) or a single int with bits (that would be modelled after PM_xyz).

If these feature bits are related to each other (like the programming modes are) then an int with bits is more appropriate. If these are "random" features that some programmers have, then I think independent Boolean variables are more appropriate.

MCUdude, 2022-12-26:
> SUFFER! Was this added to the data sheet on 1 April? 

Super User Fantastic Feature Enable Register is just hilarious.

> If these feature bits are related to each other (like the programming modes are) then an int with bits is more appropriate. If these are "random" features that some programmers have, then I think independent Boolean variables are more appropriate.

SUFFER is a completely separate thing, so this can indeed be a boolean value. Target power switch control and target voltage adjustment are absolutely related, but since there are only these two that are related, bool is probably fine.

On the other side, having _one_ field that contains all special features is also convenient. Fewer lines to add to avrdude.conf, and fewer "independent flags" to check; all special features are tucked in one variable.

stefanrueger, 2022-12-27:
Have added `target_v_adjust` and `suffer` as programmer parameters. It's easier to document and carry out. See how you like PR #1238.  You would still need to put the parameters into `avrdude.conf.in`. If that doesn't fit the bill, you could consider using the technique that was used for `prog_modes` (in the programmer structure) and for dealing with the `PM_XYZ` bits. Either way is fine for me. If you implement the latter, at I'd still merge the single [commit c683a2ae30](https://github.com/avrdudes/avrdude/pull/1238/commits/c683a2ae301769f046cc906ff6f51f00e7d502ae) as that might come in handy later. If you are OK with the former, please feel free to merge PR #1238.

mcuee, 2022-12-27:
> Have added target_v_adjust and suffer as programmer parameters. It's easier to document and carry out. See how you like PR https://github.com/avrdudes/avrdude/pull/1238. You would still need to put the parameters into avrdude.conf.in.

@stefanrueger and @MCUdude 

I think this approach is more flexible.

MCUdude, 2022-12-27:
@stefanrueger thanks for the PR. However, I'd argue that a single `special_features` parameter might be more suitable in the long run. SUFFER is something that _only two_ programmers have, and it feels unnecessary to make this available as a dedicated parameter. The same is true for the target power switch as well. Older boards like the STK500 don't have this, as it's physically controlled by a jumper instead.

Other things we might use `special_features` / `extra_features` for:

* Programmer can provide (XTAL) clock signal to the target (STK500, STK600, probably others)
* Programmer can provide AREF voltage
* Programmer can _actually_ read the target voltage
* Programmer can detect if the target is too slow to initialize communication (high-end JTAG3 programmers only)

Adding boolean parameters for all this may be a bit too much as it's only relevant for a handful of programmers.

stefanrueger, 2022-12-27:
@MCUdude I am not invested in either. Feel free to solve this as you see fit. You now have a blueprint for Boolean variables in PR #1238 and you have a blueprint for bitfields in `prog_modes` and `PM_...` constants.

MCUdude, 2023-01-03:
>... and you have a blueprint for bitfields in prog_modes and PM_... constants.

I'm not planning to add this functionality before the 7.1 release, so there's no hurry. 
However, I'm having a hard time adding `extra_features` to avrdude.conf.

I've just pushed a commit where I attempt to add this, but I'm just getting a segmentation fault when trying to run Avrdude.
@stefanrueger could you take a quick look and see if you figure out where the issue is? Thanks!

https://github.com/MCUdude/avrdude/commit/dcb109d2e0d2a6070f79ccaddccbe8ba86156a53

stefanrueger, 2023-01-03:
I think you need to implement a lot more: 
 - Implement extra_features as prog_modes was done (lexer.l, config.c, libavrdude.h, dev_options.c, ...), except it only needs to be put into the table for the programmer components [here](https://github.com/avrdudes/avrdude/blob/2ca87126e5a7cb27a0160cb4822543f85b45a2bc/src/config.c#L66) and *not* the part components a few lines below
 - Implement the constants HAS_SUFFER as the PM_... constants, in particular, the function new_constant() needs to know about them! This will be the cause of the seg fault. Don't forget to implement them in dev_options.c, too (modelled after the programmer prog_modes), otherwise `avrude -c*/s` will be broken
 - config_gram.y does not need touching if you do above (the only mention of prog_modes in config_gram.y is to be backward compatible with the previous syntax)
 - Documentation: Follow what was done for prog_modes in avrdude.conf.in in the preamble, and in avrdude.texi

stefanrueger, 2023-01-03:
... and in lexer.l remove the paragraph about extra_features, just put extra_features next to prog_modes - this will take care of it.

stefanrueger, 2023-01-03:
Though it is correct to have two sections for constants in lexer.c I think it might work better to have only one:
```
(?x: PM_(SPM|TPI|ISP|PDI|UPDI|HVSP|HVPP|debugWIRE|JTAG|JTAGmkI|XMEGAJTAG|AVR32JTAG|aWire) |
     HAS_(SUFFER|VTARG_SWITCH|VTARG_ADJ) 
) {  /* Constants */
  yylval = new_constant(yytext);
  return TKN_NUMBER;
}
```

MCUdude, 2023-01-09:
> Implement extra_features as prog_modes was done (lexer.l, config.c, libavrdude.h, dev_options.c, ...), except it only needs to be put into the table for the programmer components [...]
Implement the constants HAS_SUFFER as the PM_... constants, in particular, the function new_constant() needs to know about them! This will be the cause of the seg fault. Don't forget to implement them in dev_options.c, too (modelled after the programmer prog_modes), otherwise avrude -c*/s will be broken
config_gram.y does not need touching if you do above (the only mention of prog_modes in config_gram.y is to be backward compatible with the previous syntax)
Documentation: Follow what was done for prog_modes in avrdude.conf.in in the preamble, and in avrdude.texi

As an attempt to get it working, I haven't touched dev_options.c or the documentation. However, I _think_ I've done what you told me to do, but I'm still getting a segmentation fault error. Could you look at my last commit and see if you can spot the error? Thanks!

MCUdude, 2023-01-09:
I think I figured out the issue. `extra_features` was missing in `lexer.l`.

stefanrueger, 2023-01-10:
Cool!

The new system of adding components to the grammar is meant to be a slick few one-liners in
 - `libavrdude.h` (obvs)
 - `config.c` in the components table listing the type
 - `lexer.l` (that's just dropping in the component name amongst other, not even one line)
 - `avrpart.c` or `pgm.c` if that component needs fancy initialisation in the `_new()` functions
 - `dev_options.c` to output the component on `avrdude -p*/s -c*/s` (normally just one line)
 - avrdude.conf.in in the preamble to document this component in situ
 - avrdude.texi (basically where it has the preamble of avrdude.conf.in)

As you wanted a generic bit-field with symbolic constants it's a bit more work to write that function that prints symbolic value strings such as `HAS_VTARG_ADJ | HAS_VTARG_SWITCH`. But once finished, you'll be amazed how neat all looks.

Note that `avrconfig_gram.y` does not need touching in the new system.

MCUdude, 2023-01-29:
@stefanrueger @mcuee another extra feature that could be useful is `PM_HAS_VTARG_READ`. Most official Atmel/Microchip programmers have this feature (can't supply power but can read the target voltage). We can use this feature to help the user understand what's going on by showing the user what the target voltage is if the target fails to respond

mcuee, 2023-01-29:
@MCUdude 

Yes I agree. This is a good feature to have. Right now quite some of the Microchip tools already display Vtarget but it is good to make it more generic.

mcuee, 2023-03-26:
@MCUdude 

Now that #1286 has been merged. I think this issue can be closed. Right?

mcuee, 2023-03-26:
I found my ATmega328PB-Xmini. I think PR #1286 has already addressed all the issues mentioned here (changing the SUFFER register, writing fuse and toggling target power).

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini_isp -p m328pb -x suffer=0xff

         SUFFER register value changed from 0xfe to 0xff
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini_isp -p m328pb -x suffer=0xfe

         SUFFER register value changed from 0xff to 0xfe
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini_isp -p m328pb -qqt
avrdude> dump lfuse
0000  e0                                                |.               |

avrdude> dump hfuse
0000  de                                                |.               |

avrdude> dump efuse
0000  fd                                                |.               |

avrdude> write hfuse 0 0xd6
avrdude> dump hfuse
0000  d6                                                |.               |

avrdude> quit
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini_isp -p m328pb -qqt
avrdude> dump hfuse
0000  d6                                                |.               |

avrdude> quit

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini_isp -p m328pb -xvtarg_switch=0

         Vtarg switch setting changed from 1 to 0
         Turn on the Vtarg switch to establish connection with the target

avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower the bit clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini_isp -p m328pb -xvtarg_switch=1

         Vtarg switch setting changed from 0 to 1
avrdude error: command failed
avrdude error: device is locked; chip erase required to unlock
avrdude warning: target prepared for ISP, signed off
        now retrying without power-cycling the target
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)

avrdude done.  Thank you.

```

closed 2023-03-26 by MCUdude