avrdudes/avrdude #1236: `-U` command path under Windows

mcuee, 2022-12-24:
It is interesting that the following two commands (supposed to be equivalent) produce different results under Windows (tested under both Windows Terminal Powershell and CMD).

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p m2560 -U C:\work\avr\urboot_pr18\bootloaders\atmega2560\autobaud\urboot_atmega2560_autobaud_ee_led+b7_fr_ce_ur_vbl.hex -qq && echo OK
avrdude error: invalid I/O mode '\' in update specification
  allowed values are:
    r = read device
    w = write device
    v = verify device
avrdude error: unable to parse update operation 'C:\work\avr\urboot_pr18\bootloaders\atmega2560\autobaud\urboot_atmega2560_autobaud_ee_led+b7_fr_ce_ur_vbl.hex'

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p m2560 -U flash:w:C:\work\avr\urboot_pr18\bootloaders\atmega2560\autobaud\urboot_atmega2560_autobaud_ee_led+b7_fr_ce_ur_vbl.hex:i -qq && echo OK
OK

```

dl8dtl, 2022-12-24:
It is documented (somewhere): if you want to use a "drive" letter under Windows, you cannot use the abbreviated form of the `-U` command; you have to use the full form that includes all the delimiting colons (`-U mem:mode:path:format`).

After a lot of bad experience with Powershell damaging pipeline and IO redirection data, I have completely given up on trusting it in any way. The only thing Powershell seems to be designed and good for is passing .NET objects around.

dl8dtl, 2022-12-24:
I turned it to "unconfirmed" – please review the existing documentation, and if something is not clear, we should improve it.
Quoting the man page:
```
                   As an abbreviation, the form -U filename is equivalent to
                   specifying -U flash:w:filename:a.  This will only work if
                   filename does not have a colon in it.
```

mcuee, 2022-12-24:
@dl8dtl 

I think the existing documentation is okay, I will close this issue as `invalid`.

closed 2022-12-24 by mcuee