avrdudes/avrdude #1244: Micronucleus programmer shows 'error: unsupported memory type flash'

mariusgreuel, 2022-12-27:
When using the Micronucleus programmer, the error message 'error: unsupported memory type flash' appears:

```console
D:\Temp>avrdude -c micronucleus -p t167 -V -x wait -U flash:w:digistump_digisparkpro_blink_fast.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9487 (probably t167)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file digistump_digisparkpro_blink_fast.hex for flash
         with 170 bytes in 1 section within [0, 0xa9]
         using 2 pages and 86 pad bytes
avrdude: writing 170 bytes flash ...

Writing | -------------------------------------------------- | 0% 0.00 s

avrdude error: unsupported memory type flash
avrdude: 170 bytes of flash written

avrdude done.  Thank you.
```

I just took a quick, and it appears that the line is printed here: <https://github.com/avrdudes/avrdude/blob/main/src/micronucleus.c#L827>

which is called from 
<https://github.com/avrdudes/avrdude/blob/main/src/avr.c#L988>

As the Micronucleus bootloader does not support reading from flash, the option `-V` must be used. It appears there is now some extra code that reads from flash in `avr_write_mem`.

with -vvv

```console
avrdude micronucleus_read_byte() [micronucleus.c:827] error: unsupported memory type flash
avrdude: cannot read flash [0x0080, 0x00ff] to pad page
```

Not sure how to fix this. It appears that the flash is written, however, the padding code is not doing what it is supposed to.

mcuee, 2022-12-27:
@stefanrueger 

Hopefully you have some good idea why this happens. This is a regression so we need to fix it before release.

mcuee, 2022-12-28:
Let me see if I can reproduce using ATtiny85 DigiSpark CLone.

Firstly I made sure the bootloader works under ATTinyCore (LED is blinking).

```
C:\Users\xiaof\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\xiaof\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/avrdude.conf -v -pattiny85 -cusbasp -e -Uefuse:w:0xFE:m -Uhfuse:w:0b11011111:m -Ulfuse:w:0xF1:m -Uflash:w:C:\Users\xiaof\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/bootloaders/micronucleus/t85_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex:i 

avrdude: Version 7.0-20221101 (4c92030)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is C:\Users\xiaof\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2\avrdude.conf

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATtiny85
         Chip Erase delay              : 4500 us
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     6     4    0 no        512    4      0  4000  4500 0xff 0xff
           flash                  65     6    32    0 yes      8192   64    128  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e930b (probably t85)
erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file 0xFE for efuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte efuse ...
avrdude: 1 byte of efuse written
avrdude: verifying efuse memory against 0xFE
avrdude: 1 byte of efuse verified
avrdude: reading input file 0b11011111 for hfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte hfuse ...
avrdude: 1 byte of hfuse written
avrdude: verifying hfuse memory against 0b11011111
avrdude: 1 byte of hfuse verified
avrdude: reading input file 0xF1 for lfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lfuse ...
avrdude: 1 byte of lfuse written
avrdude: verifying lfuse memory against 0xF1
avrdude: 1 byte of lfuse verified
avrdude: reading input file C:\Users\xiaof\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/bootloaders/micronucleus/t85_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex for flash
         with 1590 bytes in 1 section within [0x19c0, 0x1ff5]
         using 25 pages and 10 pad bytes
avrdude: writing 1590 bytes flash ...

Writing | ################################################## | 100% 0.00s

avrdude: 1590 bytes of flash written
avrdude: verifying flash memory against C:\Users\xiaof\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/bootloaders/micronucleus/t85_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex

Reading | ################################################## | 100% 0.00s

avrdude: 1590 bytes of flash verified

avrdude done.  Thank you.

C:\Users\xiaof\AppData\Local\Arduino15\packages\ATTinyCore\tools\micronucleus\2.5-azd1b/micronucleus --no-ansi --run --timeout 60 C:\Users\xiaof\AppData\Local\Temp\arduino_build_21212/Blink.ino.hex 
> Please plug in the device (will time out in 60 seconds) ... 
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 2.5
> Device signature: 0x1e930b 
> Available space for user applications: 6586 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 103  page size: 64
> Erase function sleep duration: 721ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
writing: 70% complete
writing: 75% complete
writing: 80% complete
> Starting the user app ...
running: 100% complete
>> Micronucleus done. Thank you!

```

mcuee, 2022-12-28:
@mariusgreuel and @stefanrueger 
```
avrdude micronucleus_read_byte() [micronucleus.c:827] error: unsupported memory type flash
avrdude: cannot read flash [0x0080, 0x00ff] to pad page
```

This may be caused by PR #1188. I will double check.
* https://github.com/avrdudes/avrdude/pull/1188

mcuee, 2022-12-28:
The next suspect is PR #1106.
* #1106

mcuee, 2022-12-28:
Same problem with ATtiny88 (MH Tiny88). ATtinyCore works fine as it does not avrdude for programming.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c micronucleus -p t88 -V -x wait -U .\Blink_tiny88.ino.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9311 (probably t88)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink_tiny88.ino.hex for flash
         with 468 bytes in 1 section within [0, 0x1d3]
         using 8 pages and 44 pad bytes
avrdude: writing 468 bytes flash ...

Writing | -------------------------------------------------- | 0% 0.00 s

avrdude error: unsupported memory type flash
avrdude error: unable to transfer page: Input/output error
 ***failed;
 ***failed;
...
avrdude error: Micronucleus V2.0 programmer uses avr_write_page() but does not
        provide a cmd() method
 *** page 7 (addresses 0x0194 - 0x01d3) failed to write

avrdude: 468 bytes of flash written
avrdude error: unable to transfer page: Input/output error
avrdude warning: start command failed: Input/output error

avrdude done.  Thank you.
```

stefanrueger, 2022-12-28:
| It appears that the flash is written, however, the padding code is not doing what it is supposed to

Well, it does. Read on.

`avr_write()` for paged memories pads pages with contents that it reads from the memory (ie, uses a read/modify/write cycle). Micronucleus cannot read flash, so it complains. `avr_write()` does not mind, it just does not pad flash with contents read from the device. However, micronucleus complains that someone tried reading from flash. This error message is a false alarm.

Padding fixes Issue [#1050 that AVRDUDE used to treat flash and EEPROM as NOR-memory](https://github.com/avrdudes/avrdude/issues/1050). Treating all flash as if it was NOR memory is wrong. Only some flash looks like NOR-memory where writing 0xff is a NOP. Whenever the programmer uses a page erase (newer devices) or when it's a bootloader that uses SPM page erase, writing 0xff, well, writes 0xff. This means if you ask these programmers to program 3.5 pages of flash, 4 pages are written and the remaining 0.5 page is overwritten with 0xff. In general, I'd argue that this is wrong. The user might have previously put an important table there. It is worse that this happens to EEPROM.

So, this behaviour (false alarm) is an unfortunate combination of 
 - Micronucleus being unable to read flash
 - AVRDUDE not having a fine-grained model of whether a programmer can read some memories but not others
 - Micronucleus complaining in a low-level routine that someone asked to read (rather than letting the caller complain)
 - AVRDUDE not having a good model which programmer/memories combos make writing 0xff a NOP

I'd suggest converting line https://github.com/avrdudes/avrdude/blob/main/src/micronucleus.c#L827 from `pmsg_error()` to `pmsg_notice()` or `pmsg_notice2()`. If you agree @mariusgreuel that can be done straight away without a PR.

mcuee, 2022-12-28:
@stefanrueger 

I think you can proceed with the change. It is good with an ATTiny85.

I still got a warning in the end though, `warning: start command failed: Input/output error`.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c micronucleus -p t85 -V -U .\Blink_t85.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e930b (probably t85)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink_t85.hex for flash
         with 542 bytes in 1 section within [0, 0x21d]
         using 9 pages and 34 pad bytes
avrdude: writing 542 bytes flash ...

Writing | -------------------------------------------------- | 0% 0.00 s

avrdude error: unsupported memory type flash
avrdude: 542 bytes of flash written
avrdude warning: start command failed: Input/output error

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1244 -c micronucleus -p t85 -V -U .\Blink_t85.hex
avrdude_issue1244 warning: unresponsive Micronucleus device detected, please reconnect ...
avrdude_issue1244 error: cannot find device with Micronucleus bootloader (16D0:0753)
avrdude_issue1244 error: unable to open programmer micronucleus on port usb

avrdude_issue1244 done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1244 -c micronucleus -p t85 -V -U .\Blink_t85.hex

avrdude_issue1244: AVR device initialized and ready to accept instructions
avrdude_issue1244: device signature = 0x1e930b (probably t85)
avrdude_issue1244: Note: flash memory has been specified, an erase cycle will be performed.
                   To disable this feature, specify the -D option.
avrdude_issue1244: erasing chip
avrdude_issue1244: reading input file .\Blink_t85.hex for flash
                   with 542 bytes in 1 section within [0, 0x21d]
                   using 9 pages and 34 pad bytes
avrdude_issue1244: writing 542 bytes flash ...

Writing | ################################################## | 100% 0.18 s

avrdude_issue1244: 542 bytes of flash written
avrdude_issue1244 warning: start command failed: Input/output error

avrdude_issue1244 done.  Thank you.

```

mcuee, 2022-12-28:
However, the current git main has more issues with the Micronucleus firmware bundled with ATTinyCore. Tested with both ATtiny85 and ATtiny167.

ATtinyCore, no problem with flash upload and the LED is blinking nicely.
```
C:\Users\xiaof\AppData\Local\Arduino15\packages\ATTinyCore\tools\micronucleus\2.5-azd1b/micronucleus --no-ansi --run --timeout 60 C:\Users\xiaof\AppData\Local\Temp\arduino_build_474993/Blink.ino.hex 
> Please plug in the device (will time out in 60 seconds) ... 
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 2.5
> Device signature: 0x1e930b 
> Available space for user applications: 6586 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 103  page size: 64
> Erase function sleep duration: 721ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
writing: 70% complete
writing: 75% complete
writing: 80% complete
> Starting the user app ...
running: 100% complete
>> Micronucleus done. Thank you!

```

avrdude git main with minor changes. The LED is blinking in the end, but there are many error messages.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1244 -c micronucleus -p t85 -V -U .\Blink_t85.hex

avrdude_issue1244: AVR device initialized and ready to accept instructions
avrdude_issue1244: device signature = 0x1e930b (probably t85)
avrdude_issue1244: Note: flash memory has been specified, an erase cycle will be performed.
                   To disable this feature, specify the -D option.
avrdude_issue1244: erasing chip
avrdude_issue1244: reading input file .\Blink_t85.hex for flash
                   with 542 bytes in 1 section within [0, 0x21d]
                   using 9 pages and 34 pad bytes
avrdude_issue1244: writing 542 bytes flash ...

Writing | -------------------------------------------------- | 0% 0.00 s

avrdude_issue1244 error: unable to transfer page: Input/output error
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
[... 539 lines cut ...]
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
avrdude_issue1244 error: Micronucleus V2.0 programmer uses avr_write_page() but does not
                  provide a cmd() method
 *** page 8 (addresses 0x01de - 0x021d) failed to write

avrdude_issue1244: 542 bytes of flash written
avrdude_issue1244 error: unable to transfer page: Input/output error
avrdude_issue1244 warning: start command failed: Input/output error

avrdude_issue1244 done.  Thank you.

```

mcuee, 2022-12-28:
@stefanrueger and @mariusgreuel 

In the end, it is due to the old libusb-win32 driver. If I switch to libusbk or WinUSB driver, there is no issue.

git main will have one warning and the proposed changes by @stefanrueger is good.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c micronucleus -V -p t167 -U .\Blink_t167.ino.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9487 (probably t167)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink_t167.ino.hex for flash
         with 738 bytes in 1 section within [0, 0x2e1]
         using 6 pages and 30 pad bytes
avrdude: writing 738 bytes flash ...

Writing | -------------------------------------------------- | 0% 0.00 s

avrdude error: unsupported memory type flash
avrdude: 738 bytes of flash written

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1244 -c micronucleus -V -p t167 -U .\Blink_t167.ino.hex

avrdude_issue1244: AVR device initialized and ready to accept instructions
avrdude_issue1244: device signature = 0x1e9487 (probably t167)
avrdude_issue1244: Note: flash memory has been specified, an erase cycle will be performed.
                   To disable this feature, specify the -D option.
avrdude_issue1244: erasing chip
avrdude_issue1244: reading input file .\Blink_t167.ino.hex for flash
                   with 738 bytes in 1 section within [0, 0x2e1]
                   using 6 pages and 30 pad bytes
avrdude_issue1244: writing 738 bytes flash ...

Writing | ################################################## | 100% 0.19 s

avrdude_issue1244: 738 bytes of flash written

avrdude_issue1244 done.  Thank you.

```

stefanrueger, 2022-12-28:
The other warning is unrelated and may be useful.

I suggest changing the message as well:
```
diff --git a/src/micronucleus.c b/src/micronucleus.c
index fc6b1c1..6f205e6 100644
--- a/src/micronucleus.c
+++ b/src/micronucleus.c
@@ -824,7 +824,7 @@ static int micronucleus_read_byte(const PROGRAMMER *pgm, const AVRPART *p, const
     }
     else
     {
-        pmsg_error("unsupported memory type %s\n", mem->desc);
+        pmsg_notice("reading not supported for %s memory\n", mem->desc);
         return -1;
     }
 }
```
to emphasise that it's the reading that is not supported.

mcuee, 2022-12-28:
@stefanrueger 

Yes I agree with you that the other messgae is probably due to older Micronucleus bootloader firmware. I have since upgraded to the latest FIrmware (either the version from Micronucleus or the version from @mariusgreuel), the second error message goes away.

And it is important to use WinUSB driver (preferred) and not libusb-win32 driver. That is already mentioned by @mariusgreuel in the wiki.

MCUdude, 2022-12-28:
> And it is important to use WinUSB driver (preferred) and not libusb-win32 driver. That is already mentioned by @mariusgreuel in the wiki.

Maybe Avrdude can warn the user that the USB driver used for Micronucleus isn't preferred?

stefanrueger, 2022-12-28:
Downgraded flash read unsupported err msg to notice.

| Maybe Avrdude can warn the user that the USB driver used for Micronucleus isn't preferred?

Good idea. Someone else to do who knows where to warn and what wording.

closed 2022-12-28 by stefanrueger

mcuee, 2022-12-28:
> Downgraded flash read unsupported err msg to notice.
> 
> | Maybe Avrdude can warn the user that the USB driver used for Micronucleus isn't preferred?
> 
> Good idea. Someone else to do who knows where to warn and what wording.

I will create a new issue to see if we can warn the user for this easily or not.

@mariusgreuel has the avrdude-libusb for MSVC and there is no problem with either WInUSB driver or libusb-win32 driver in this case.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git_msvc -c micronucleus -V -p t167 -vv -U .\Blink_t167.ino.hex

avrdude_git_msvc: Version 7.0-20221228 (92f980d)
                  Copyright the AVRDUDE authors;
                  see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                  System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

                  Using Port                    : usb
                  Using Programmer              : micronucleus
avrdude_git_msvc: found device with Micronucleus V2.5, bus:device: bus-0:\\?\usb#vid_16d0&pid_0753#6&8694159&0&1#{a5dcbf10-6530-11d2-901f-00c04fb951ed}--libusb0
avrdude_git_msvc: defaulting memtype in -U w:.\Blink_t167.ino.hex option to "flash"
avrdude_git_msvc: input file .\Blink_t167.ino.hex auto detected as Intel Hex
                  AVR Part                      : ATtiny167
                  Chip Erase delay              : 15000 us
                  RESET disposition             : possible i/o
                  RETRY pulse                   : SCK
                  Serial program mode           : yes
                  Parallel program mode         : yes
                  Timeout                       : 200
                  StabDelay                     : 100
                  CmdexeDelay                   : 25
                  SyncLoops                     : 32
                  PollIndex                     : 3
                  PollValue                     : 0x53
                  Memory Detail                 :

                                                    Block Poll               Page                       Polled
                    Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                    ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                    eeprom                 65    10     4    0 no        512    4      0  4000  4500 0xff 0xff
                    flash                  65    10    64    0 yes     16384  128    128  4500  4500 0xff 0xff
                    lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                    hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                    efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                    lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                    signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                    calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

                  Programmer Type : Micronucleus V2.0
                  Description     : Micronucleus for bootloader

avrdude_git_msvc: Bootloader version: 2.5
                  Available flash size: 14844
                  Page size: 128
                  Bootloader start: 0x3A00
                  Write sleep: 7ms
                  Erase sleep: 812ms
                  Signature1: 0x94
                  Signature2: 0x87
avrdude_git_msvc: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00 s

avrdude_git_msvc: device signature = 0x1e9487 (probably t167)
avrdude_git_msvc: Note: flash memory has been specified, an erase cycle will be performed.
                  To disable this feature, specify the -D option.
avrdude_git_msvc: erasing chip
avrdude_git_msvc: reading input file .\Blink_t167.ino.hex for flash
                  with 738 bytes in 1 section within [0, 0x2e1]
                  using 6 pages and 30 pad bytes
avrdude_git_msvc: writing 738 bytes flash ...

Writing |                                                    | 0% 0.00 s avrdude_git_msvc: reading not supported for flash memory
avrdude_git_msvc: cannot read flash [0x0280, 0x02ff] to pad page
Writing | ################################################## | 100% 0.15 s

avrdude_git_msvc: 738 bytes of flash written

avrdude_git_msvc done.  Thank you.
```

The issue is probably specific to libusb-1.0+libusb-compat-0.1 with MSYS2.

mcuee, 2022-12-28:
MSYS2 build with libusb-1.0 + libusb-compat-0.1.

You can see these is no way to differentiate the three drivers now.

1) No issues with WinUSB driver or libusbK driver.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c micronucleus -V -p t167 -vv -U .\Blink_t167.ino.hex

avrdude_git: Version 7.0-20221228 (92f980d)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : usb
             Using Programmer              : micronucleus
avrdude_git: found device with Micronucleus V2.5, bus:device: 001:046
avrdude_git: defaulting memtype in -U w:.\Blink_t167.ino.hex option to "flash"
avrdude_git: input file .\Blink_t167.ino.hex auto detected as Intel Hex
             AVR Part                      : ATtiny167
             Chip Erase delay              : 15000 us
             RESET disposition             : possible i/o
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     4    0 no        512    4      0  4000  4500 0xff 0xff
               flash                  65    10    64    0 yes     16384  128    128  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

             Programmer Type : Micronucleus V2.0
             Description     : Micronucleus for bootloader

avrdude_git: Bootloader version: 2.5
             Available flash size: 14844
             Page size: 128
             Bootloader start: 0x3A00
             Write sleep: 7ms
             Erase sleep: 812ms
             Signature1: 0x94
             Signature2: 0x87
avrdude_git: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00 s

avrdude_git: device signature = 0x1e9487 (probably t167)
avrdude_git: Note: flash memory has been specified, an erase cycle will be performed.
             To disable this feature, specify the -D option.
avrdude_git: erasing chip
avrdude_git: reading input file .\Blink_t167.ino.hex for flash
             with 738 bytes in 1 section within [0, 0x2e1]
             using 6 pages and 30 pad bytes
avrdude_git: writing 738 bytes flash ...

Writing |                                                    | 0% 0.00 s avrdude_git: reading not supported for flash memory
[... 71 lines cut ...]
             with 738 bytes in 1 section within [0, 0x2e1]
             using 6 pages and 30 pad bytes
avrdude_git: writing 738 bytes flash ...

Writing |                                                    | 0% 0.00 s avrdude_git: reading not supported for flash memory
avrdude_git: cannot read flash [0x0280, 0x02ff] to pad page
Writing | -------------------------------------------------- | 0% 0.01 s

avrdude_git micronucleus_write_page_v2() error: unable to transfer page: Input/output error
 ***failed;
 ***failed;
...
...
 ***failed;
 ***failed;
avrdude_git avr_write_page() error: Micronucleus V2.0 programmer uses avr_write_page() but does not
            provide a cmd() method
 *** page 5 (addresses 0x0262 - 0x02e1) failed to write

avrdude_git: 738 bytes of flash written
avrdude_git micronucleus_write_page_v2() error: unable to transfer page: Input/output error
avrdude_git micronucleus_start() warning: start command failed: Input/output error

avrdude_git done.  Thank you.
```

mcuee, 2022-12-31:
Two issues raised against libusb project

* https://github.com/libusb/libusb/issues/1225
* https://github.com/libusb/libusb/issues/1226

mcuee, 2022-12-31:
avrdude Wiki has multiple places to recommend using MSVC build.
https://github.com/avrdudes/avrdude/wiki/The-story-on-libusb-for-Windows
https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Windows-using-Visual-Studio
https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Windows-using-MSYS2
https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Windows-using-Cygwin
https://github.com/avrdudes/avrdude/wiki/Troubleshooting-USB-devices-in-Windows

mcuee, 2023-02-16:
Interestingly I can not reproduce the issue with ATtiny88.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbtiny -p t88 -U .\t88_default.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9311 (probably t88)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\t88_default.hex for flash
         with 1350 bytes in 1 section within [0x1a40, 0x1f85]
         using 22 pages and 58 pad bytes
avrdude: writing 1350 bytes flash ...

Writing | ################################################## | 100% 0.11 s

avrdude: 1350 bytes of flash written
avrdude: verifying flash memory against .\t88_default.hex

Reading | ################################################## | 100% 0.06 s

avrdude: 1350 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c micronucleus -p t88 -V -x wait -U .\Blink_tiny88.ino.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9311 (probably t88)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink_tiny88.ino.hex for flash
         with 468 bytes in 1 section within [0, 0x1d3]
         using 8 pages and 44 pad bytes
avrdude: writing 468 bytes flash ...

Writing | ################################################## | 100% 0.16 s

avrdude: 468 bytes of flash written

avrdude done.  Thank you.

```

mcuee, 2023-02-16:
I can reproduce the issue with libusb0.sys 1.2.6.0 version.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c micronucleus -p t88 -V -x wait -U .\Blink_tiny88.ino.hex
avrdude warning: unresponsive Micronucleus device detected, please reconnect ...
avrdude error: no device found, waiting for device to be plugged in ...
avrdude error: press CTRL-C to terminate

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9311 (probably t88)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink_tiny88.ino.hex for flash
         with 468 bytes in 1 section within [0, 0x1d3]
         using 8 pages and 44 pad bytes
avrdude: writing 468 bytes flash ...

Writing | -------------------------------------------------- | 0% 0.00 s

avrdude error: unable to transfer page: Input/output error
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
[... 433 lines cut ...]
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
avrdude error: Micronucleus V2.0 programmer uses avr_write_page() but does not
        provide a cmd() method
 *** page 7 (addresses 0x0194 - 0x01d3) failed to write

avrdude: 468 bytes of flash written
avrdude error: unable to transfer page: Input/output error
avrdude warning: start command failed: Input/output error

avrdude done.  Thank you.

```