avrdudes/avrdude #1245: urclock communication issues in Windows

mariusgreuel, 2022-12-27:
I have been tinkering with urboot on Windows, and there seem to be issues with the bootloader communication.

I set up an Digispark Pro with a ATtiny167 and flashed it with `urboot_digispark-pro_16mhz_115200bps_rxb6_txb3_led+b1_ur_vbl.hex`.

I am running the command

```console
avrdude -c urclock -p t167 -P COM3
```
and get the error 

```console
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100

avrdude done.  Thank you.
```

and sometimes

```console
avrdude warning: uP_table does not know mcuid 1306

avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100

avrdude done.  Thank you.
```

Depending on the USB serial converter I am using, the connection is very unreliable. My trusted FT232H-based serial converter seems to cause the most problems (>50% failure rate), a CP2102-based serial converter seems to work as expected.

The first thing I noticed is that the WIN32 implementation of `ser_recv` is broken. As `ser_recv` does not return the number of bytes read, it must return the exact same number of bytes requested or -1 on failure/timeout. When a timeout occurs, the current WIN32 `ser_recv` may return success, but only a partial amount of bytes requested.

`urclock_getsync` sets a very short timeout of 10ms, so quite often, `ser_recv` only returns one byte instead of two, and everything goes downhill from there.

Initially, I thought I can fix this by changing `ser_recv` to either receive the full amount of bytes requested or -1 on timeout, but that did not fix it. There seems to be a problem related to the small 10ms timeout, and the FTDI driver sometimes returning zero bytes.

Also tried `-xstrict`, did not help. I do not think its cabling, signals look good.

Using -vvvv and some WIN32 patches, here is a success (notice there is still a `ser_recv(): programmer is not responding`)

```console
         Using Port                    : COM3
         Using Programmer              : urclock
avrdude:  234 ms: enter urclock_getsync()
avrdude: send: 0 [30]   [20] 
avrdude: serial_recv_timeout: 10
avrdude: ser_recv(): programmer is not responding
avrdude:  283 ms: sleeping for 32 ms
avrdude: send:   [20]   [20] 
avrdude: serial_recv_timeout: 10
avrdude: recv(2 of 2): . [ea] . [fc] 
avrdude: send:   [20]   [20] 
avrdude: serial_recv_timeout: 10
avrdude: recv(2 of 2):   [20] . [1d] 
avrdude: send:   [20]   [20] 
avrdude: serial_recv_timeout: 10
avrdude: recv(2 of 2):   [20] . [1d] 
avrdude: send:   [20] 
avrdude: serial_recv_timeout: 10
avrdude: ser_recv(): programmer is not responding
avrdude: send:   [20] 
avrdude:  436 ms: all good, ready to rock
         AVR Part                      : ATtiny167
         Chip Erase delay              : 15000 us
<snip>
         Programmer Type : Urclock
         Description     : Urclock programmer for urboot bootloaders using urprotocol
         Urboot protocol for ATtiny167

avrdude: send: P [50]   [20] 
avrdude: serial_recv_timeout: 500
avrdude: recv(1 of 1):   [20] 
avrdude: serial_recv_timeout: 500
avrdude: recv(1 of 1): . [1d] 
avrdude: AVR device initialized and ready to accept instructions

Reading | avrdude: ur_readEF(urclock, ATtiny167, flash, 000000C4583AE240, 0x003ffa, 6, F)
avrdude: send: . [03] . [fa] ? [3f] . [06]   [20] 
avrdude: serial_recv_timeout: 500
avrdude: recv(1 of 1):   [20] 
avrdude: serial_recv_timeout: 500
avrdude: recv(6 of 6): . [02] . [10] . [e1] . [cf] $ [24] ? [3f] 
avrdude: serial_recv_timeout: 500
avrdude: recv(1 of 1): . [1d] 
################################################## | 100% 0.01s

avrdude: device signature = 0x1e9487 (probably t167)
avrdude: send: Q [51]   [20] 
avrdude: serial_recv_timeout: 500
avrdude: recv(1 of 1):   [20] 
avrdude: serial_recv_timeout: 500
avrdude: recv(1 of 1): . [1d] 

avrdude done.  Thank you.
```

and a failure

```console
avrdude:  234 ms: enter urclock_getsync()
avrdude: send: 0 [30]   [20] 
avrdude: serial_recv_timeout: 10
avrdude: ser_recv(): programmer is not responding
avrdude:  266 ms: sleeping for 32 ms
avrdude: send:   [20]   [20] 
avrdude: serial_recv_timeout: 10
avrdude: recv(2 of 2): . [00] . [ea] 
avrdude: send:   [20]   [20] 
avrdude: serial_recv_timeout: 10
avrdude: recv(2 of 2): . [fc]   [20] 
avrdude: send:   [20]   [20] 
avrdude: serial_recv_timeout: 10
avrdude: recv(2 of 2): . [1d]   [20] 
avrdude: send:   [20]   [20] 
avrdude: serial_recv_timeout: 10
avrdude: recv(2 of 2): . [1d]   [20] 
avrdude: send:   [20] 
avrdude: serial_recv_timeout: 10
avrdude: recv(1 of 1): . [1d] 
avrdude set_uP() [urclock.c:1054] warning: uP_table does not know mcuid 1306
avrdude:  387 ms: all good, ready to rock
         AVR Part                      : ATtiny167
         Chip Erase delay              : 15000 us
<snip>
         Programmer Type : Urclock
         Description     : Urclock programmer for urboot bootloaders using urprotocol
         Urboot protocol for ATtiny167

avrdude: send: P [50]   [20] 
avrdude: serial_recv_timeout: 500
avrdude: ser_recv(): programmer is not responding
avrdude urclock_recv() [urclock.c:1923] error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude main() [main.c:1246] error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude: send: Q [51]   [20] 
avrdude: serial_recv_timeout: 500
avrdude: ser_recv(): programmer is not responding
avrdude urclock_recv() [urclock.c:1923] error: programmer is not responding; try -xstrict and/or vary -xdelay=100

avrdude done.  Thank you.
```

Saving further analysis for some other time...

mcuee, 2022-12-28:
Ah, I do not have good testing facility to carry out test for urboot for ATtiny chips. My tests were more on ATmega328P and ATmega2560.

mcuee, 2022-12-28:
I will try ATTiny88 (MH Tiny) first.

1) ATTinyCore Optiboot seems to work (PD6 as TXD and PD7 as RXD).
https://github.com/SpenceKonde/ATTinyCore/blob/v2.0.0-devThis-is-the-head-submit-PRs-against-this/avr/extras/Pinout_x8.jpg
```
C:\Users\xiaof\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\xiaof\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/avrdude.conf -v -pattiny88 -cusbasp -e -Uefuse:w:0xfe:m -Uhfuse:w:0b11011111:m -Ulfuse:w:0xE2:m -Uflash:w:C:\Users\xiaof\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/bootloaders/optiboot/optiboot_attiny88_8000000L.hex:i 

avrdude: Version 7.0-20221101 (4c92030)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is C:\Users\xiaof\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2\avrdude.conf

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATtiny88
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no         64    4      0  3600  3600 0xff 0xff
           flash                  65     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9311 (probably t88)
erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file 0xfe for efuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte efuse ...
avrdude: 1 byte of efuse written
avrdude: verifying efuse memory against 0xfe
avrdude: 1 byte of efuse verified
avrdude: reading input file 0b11011111 for hfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte hfuse ...
avrdude: 1 byte of hfuse written
avrdude: verifying hfuse memory against 0b11011111
avrdude: 1 byte of hfuse verified
avrdude: reading input file 0xE2 for lfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lfuse ...
avrdude: 1 byte of lfuse written
avrdude: verifying lfuse memory against 0xE2
avrdude: 1 byte of lfuse verified
avrdude: reading input file C:\Users\xiaof\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/bootloaders/optiboot/optiboot_attiny88_8000000L.hex for flash
[... 53 lines cut ...]

         Programmer Type : Arduino
         Description     : Arduino for bootloader using STK500 v1 protocol
         Hardware Version: 3
         Firmware Version: 59.0

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9311 (probably t88)
avrdude: reading input file C:\Users\xiaof\AppData\Local\Temp\arduino_build_694615/Blink_tiny88.ino.hex for flash
         with 468 bytes in 1 section within [0, 0x1d3]
         using 8 pages and 44 pad bytes
avrdude: writing 468 bytes flash ...

Writing | ################################################## | 100% 0.41s

avrdude: 468 bytes of flash written
avrdude: verifying flash memory against C:\Users\xiaof\AppData\Local\Temp\arduino_build_694615/Blink_tiny88.ino.hex

Reading | ################################################## | 100% 0.36s

avrdude: 468 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-12-28:
However, I can not get urboot hex file to work.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t88 -U ..\..\urboot\bootloaders\attiny88\fcpu_8mhz\19200_bps\attiny88_8mhz_19200bps_rxb0_txb1_ee_lednop_fr_ce_ur_vbl.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9311 (probably t88)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file ..\..\urboot\bootloaders\attiny88\fcpu_8mhz\19200_bps\attiny88_8mhz_19200bps_rxb0_txb1_ee_lednop_fr_ce_ur_vbl.hex for flash
         with 376 bytes in 2 sections within [0x1e80, 0x1fff]
         using 6 pages and 8 pad bytes
avrdude: writing 376 bytes flash ...

Writing | ################################################## | 100% 0.03 s

avrdude: 376 bytes of flash written
avrdude: verifying flash memory against ..\..\urboot\bootloaders\attiny88\fcpu_8mhz\19200_bps\attiny88_8mhz_19200bps_rxb0_txb1_ee_lednop_fr_ce_ur_vbl.hex

Reading | ################################################## | 100% 0.02 s

avrdude: 376 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM6 -b 19200 -p t88 -xshowall -xstrict -xdelay=100
avrdude warning: attempt 1 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 2 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 3 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 4 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 5 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 6 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 7 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 8 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 9 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 10 of 10: not in sync
avrdude error: unable to open programmer urclock on port COM6

avrdude done.  Thank you.
```

mcuee, 2022-12-28:
@mariusgreuel and @stefanrueger 

On the other hand, my test can not confirm whether it is a urboot bootloader FW problem, or avrdude `-c clock` issue.

mcuee, 2022-12-28:
I build one hex file which seems to work under Linux.
```
mcuee@UbuntuSwift3:~/build/avr/urboot/src$ make MCU=attiny88 F_CPU=8000000L WDTO=8S AUTOBAUD=0 LED=AtmelPB0 LEDPOLARITY=1  DUAL=0 EEPROM=1 URPROTOCOL=1 FRILLS=7 SWIO=1 TX=AtmelPD6 RX=AtmelPD7 
./avr-toolchain/5.4.0/bin/avr-gcc -DSTART=0x1e80UL -DRJMPWP=0xcfd9 -Wl,--section-start=.text=0x1e80 -Wl,--section-start=.version=0x1ffa -D_urboot_AVAILABLE=20 -g -Wundef -Wall -Os -fno-split-wide-types -mrelax -mmcu=attiny88 -DF_CPU=8000000L -Wno-clobbered -DWDTO=8S -DAUTOBAUD=0 -DLED=AtmelPB0 -DLEDPOLARITY=1 -DDUAL=0 -DEEPROM=1 -DURPROTOCOL=1 -DFRILLS=7 -DVBL=0 -DSWIO=1 -DTX=AtmelPD6 -DRX=AtmelPD7 -Wl,--relax -nostartfiles -nostdlib -o attiny88_8mhz_115200bps_ef7_swio1_txPD6_rxPD7_ledPB0_wdto8s_ur.elf urboot.c
./avr-toolchain/5.4.0/bin/avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex attiny88_8mhz_115200bps_ef7_swio1_txPD6_rxPD7_ledPB0_wdto8s_ur.elf attiny88_8mhz_115200bps_ef7_swio1_txPD6_rxPD7_ledPB0_wdto8s_ur.hex
./avr-toolchain/5.4.0/bin/avr-objdump -h -S attiny88_8mhz_115200bps_ef7_swio1_txPD6_rxPD7_ledPB0_wdto8s_ur.elf > attiny88_8mhz_115200bps_ef7_swio1_txPD6_rxPD7_ledPB0_wdto8s_ur.lst
-- 364 384 u7.7 weu-hpr-c attiny88_8mhz_115200bps_ef7_swio1_txPD6_rxPD7_ledPB0_wdto8s_ur.hex

mcuee@UbuntuSwift3:~/build/avr/urboot/src$ avrdude -c usbasp -p t88 -U attiny88_8mhz_115200bps_ef7_swio1_txPD6_rxPD7_ledPB0_wdto8s_ur.hex 

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9311 (probably t88)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file attiny88_8mhz_115200bps_ef7_swio1_txPD6_rxPD7_ledPB0_wdto8s_ur.hex for flash
         with 364 bytes in 2 sections within [0x1e80, 0x1fff]
         using 6 pages and 20 pad bytes
avrdude: writing 364 bytes flash ...

Writing | ################################################## | 100% 0.01 s 

avrdude: 364 bytes of flash written
avrdude: verifying flash memory against attiny88_8mhz_115200bps_ef7_swio1_txPD6_rxPD7_ledPB0_wdto8s_ur.hex

Reading | ################################################## | 100% 0.00 s 

avrdude: 364 bytes of flash verified

avrdude done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/urboot/src$ avrdude -c urclock -P /dev/ttyUSB0 -p t88 -xshowall -xstrict

avrdude: AVR device initialized and ready to accept instructions
ffffffffffff 0000-00-00 00.00  application 0 store 0 meta 0 boot 384 u7.7 weu-hpr-c vector 0 (RESET) ATtiny88

```

[attiny88_8mhz_115200bps_ef7_swio1_txPD6_rxPD7_ledPB0_wdto8s_ur.zip](https://github.com/avrdudes/avrdude/files/10312439/attiny88_8mhz_115200bps_ef7_swio1_txPD6_rxPD7_ledPB0_wdto8s_ur.zip)

mcuee, 2022-12-28:
And then it seems to work under Windows as well. So I can not reproduce the isse on my side.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM6 -b 19200 -p t88 -xshowall

avrdude: AVR device initialized and ready to accept instructions
0000ffffffff 0000-00-00 00.00  application 0 store 0 meta 0 boot 384 u7.7 weu-hpr-c vector 0 (RESET) ATtiny88

```

stefanrueger, 2022-12-28:
@mariusgreuel Have you tried to set and vary `-xdelay=100`? If the board takes an uncharacteristically long time to get out of reset that can be helpful. Does your USB/serial chip reset the board when RTS/DTS is plucked? 

Does the same board board work under Linux/MacOS?

Yes, it's a pity that the windows timeout behaves in a different way to the Linux/Posix timeout. AVRDUDE use the same number of milliseconds for Windows timeout and Posix timeout but there is a different behaviour when reading bytes. I once tried to understand  `serial_w32SetTimeOut()` but the documentation that I looked at was not entirely clear to me. There is a complexity with timeouts of between-character timeouts etc that is unwanted/not needed here and makes `read()` behave differently in both worlds.

Unfortunately, I cannot test what that Windows timeout does to `read()`. Ideally, Windows read and Posix read would behave in the same way. The urclock/urboot synchronisation for the host/AVR communication was developed with Posix behaviour of read() and Posix timeout.

>  timeout of 10ms, so quite often, ser_recv only returns one byte instead of two

When the AVR sends two bytes as a reply to the GET_SYNC command they will be sent within microseconds of each other. The timeout should not matter. If one byte can be read, then the other should follow.

Catering for autobaud requires a bit of fiddling in the synchronisation process, which makes the synchronisation more complex. Urboot bootloaders can gobble up the first sync request from AVRDUDE in order to determine and set the baud rate. However, sometimes the first sync request is gobbled up by the board needing longer than expected to come out of reset (this is what `-xdelay=...` can correct). There is careful draining that happens to get both sides into synchronisation.

@mcuee In this case I find the way you present the tests confusing and hard to process. [This comment](https://github.com/avrdudes/avrdude/issues/1245#issuecomment-1366394115) says  "But it does not work very well" without giving an indication what does not work, followed by many lines of -v` output that seem to indicate all works well?

> However, I can not get urboot hex file to work.

I suspect you picked a pre-compiled bootloader with the wrong TX and RX pins of the board? You need to compile this one. 

Please edit comments when they give the wrong conclusions. As the comments stay for a long time, people who come here from a search engine may be mislead...

mcuee, 2022-12-28:
@stefanrueger 

Sorry for the confusion. This board is not very good for Optiboot/urboot -- it is difficult to go into the bootloader after flashing the blinking LED sketch. So the easies way is to flash the bootloader everytime for the testing. It is more meant to run Micronucleus bootloader.

I have deleted two messages and edited the remaining messgaes. Please check if there are still confusions or not. Thanks.

stefanrueger, 2022-12-28:
> the pre-compiled bootloader and they do not seem to work for my board

The Attiny88 has no "designated" communication pins as it has no USART; in that case the pre-compiled urboot bootloader assumes RX=PB0 and TX=PB1 and you'd either have to hook up comms with the host at PB0/PB1 or, if the board has an USB/serial converter, compile the bootloader to use the correct pins of the particular board.

stefanrueger, 2022-12-28:
Thanks for clarifying the tests and how you got there, @mcuee. Off topic: I am also pleased to learn that urboot has been tested to work well with ATtiny88. Also good to know the MH Tiny board has `LED=AtmelPB0 LEDPOLARITY=1 SWIO=1 TX=AtmelPD6 RX=AtmelPD7`. I can create pre-compiled bootloaders for that board...

mcuee, 2022-12-28:
> The Attiny88 has no "designated" communication pins as it has no USART; in that case the pre-compiled urboot bootloader assumes RX=PB0 and TX=PB1 and you'd either have to hook up comms with the host [...]

It does not have USB Serial Converter on-board. And I did use PB0/PB1 as well. I tend to think the issue is with 1s WDT timeout.

mcuee, 2022-12-28:
> Thanks for clarifying the tests and how you got there, @mcuee. 

Off topic: I am also pleased to learn that urboot has been tested to work well with ATtiny88. Also good to know the MH Tiny board has `LED=AtmelPB0 LEDPOLARITY=1 SWIO=1 TX=AtmelPD6 RX=AtmelPD7`. I can create pre-compiled bootloaders for that board...

That will be good. I will suggest you create two versions, one with WDT=8s (for people who do not have a proper USB-TTL converter with DTR pin and not able to hook up proper reset circuitry) and the other wil WDT=1s (for people who hook-up proper reset circuitry with the DTR pin and 0.1uF autoreset capacitor).

stefanrueger, 2022-12-28:
> And I did use PB0/PB1 as well

Ahh - OK. Understood. I looked at the differences of what is working (TX=AtmelPD6 RX=AtmelPD7) and what was not (TX=AtmelPB1 RX=AtmelPB0), so concluded that was the issue. So, the only board-specific thing is the position of the LED (P**D**0, not PB0 as you compiled the urboot bootloader for). What's the LED polarity? Is it low-active or high-active?

| two versions, one with WDT=8s [and] 1s

About time I thought about how to use the template versions of urboot bootloaders. Their binary are done in a way, so that the LED pin, LED polarity, WDTO, TX, RX can be recognised and exchanged in the binary. I have an urloader sketch for a ProMini that acts as bootloader programmer, which creates the right bootloader and correct fuses. Need to publish that.

Maybe I can transfer that idea to AVRDUDE that then can patch bootloaders on the fly for these things. One of the ideas for v8.0 ...

mcuee, 2022-12-28:
@stefanrueger 

Yes you are rgth that LED pin is PD0 and it is high active.

stefanrueger, 2022-12-28:
@mariusgreuel 

I now can reproduce the issue in Linux with the FTDI USB/serial converter. Turns out the read timeout of 10 ms is cutting it too fine for the FTDI. See PR #1246

mcuee, 2022-12-28:
@stefanrueger and @mariusgreuel 

Did you change the default latency timer from 16ms to 1ms?

https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md#serial-adapter-requirements

stefanrueger, 2022-12-28:
@mcuee Great shout! I had not known FTDI has a default latency of 16 ms. Actually, I had not imagined that such a parameter existed, let alone is set to 16 ms per default! One never ceases to learn....

This must be the reason why the FTDI didn't work: I can confirm that setting a receive timeout of 16 ms just about works (15 ms does not), so I set it to 25 ms in PR #1246 to have a margin.

I still think `urclock.c` should increase its receive timeout for establishing a sync as there will be users who do not know which chip is used to manage the USB/serial conversion, let alone about the *default* latency timer of the FTDI, let alone that it's 16 ms, let alone that it is not useful on this occasion, let alone how it can be changed.

stefanrueger, 2022-12-28:
OK, there might still be issues with Windows around the fact that timeouts behave differently so I leave the subject of this issue and await @mariusgreuel's response to whether PR #1246 significantly changes the experienced issues.

mcuee, 2022-12-28:
> I still think urclock.c should increase its receive timeout for establishing a sync as there will be users who do not know which chip is used to manage the USB/serial conversion, let alone about the [...]

@stefanrueger 
Totally agree with you here. I think PR #1246 should be safe to merge.

> OK, there might still be issues with Windows around the fact that timeouts behave differently so I leave the subject of this issue and await @mariusgreuel's response to whether PR https://github.com/avrdudes/avrdude/pull/1246 significantly changes the experienced issues.

I agree with you here as well. I will see if I can reproduce the issue with ATTiny167 under Windows/Linux.

mcuee, 2022-12-28:
@mariusgreuel would like to work on #1186 as well. 
* #1186 

@stefanrueger, @MCUdude and @dl8dtl 
So we may have to wait a bit for the final 7.1 release to resolve #1245 and #1186.

mcuee, 2022-12-29:
> I now can reproduce the issue in Linux with the FTDI USB/serial converter. Turns out the read timeout of 10 ms is cutting it too fine for the FTDI. See PR #1246

@stefanrueger 
Interestingly I can not reproduce the issue with a Nano ATmega328P clone with FT232RL USB to Serial chip on board, even with 16ms latency timer.

But you can see the performance difference between 16ms latency timer and 1ms latency timer from the below testing results. Using 1ms latency timer saves 50% of the flash reading time in this test.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM7 -p m328p -U flash:r:nano_ft232rl_blink_readback_16ms_latency.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: reading flash memory ...

Reading | ################################################## | 100% 7.68 s

avrdude: writing output file nano_ft232rl_blink_readback_16ms_latency.hex

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM7 -p m328p -U flash:r:nano_ft232rl_blink_readback_1ms_latency.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: reading flash memory ...

Reading | ################################################## | 100% 3.61 s

avrdude: writing output file nano_ft232rl_blink_readback_1ms_latency.hex

avrdude done.  Thank you.
```

mcuee, 2022-12-29:
> @mariusgreuel
> 
> I now can reproduce the issue in Linux with the FTDI USB/serial converter. Turns out the read timeout of 10 ms is cutting it too fine for the FTDI. See PR #1246

@stefanrueger 

Even though I can not reproduce the issue under WIndows, I can reproduce the issue under Linux. And PR #1246 is good to fixx the issue.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c urclock -P /dev/ttyUSB0 -p m328p -U flash:r:nano_ft232rl_blink_readback_16ms_latency.hex:i

avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100

avrdude done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude_pr1246 -c urclock -P /dev/ttyUSB0 -p m328p -U flash:r:nano_ft232rl_blink_readback_16ms_latency.hex:i

avrdude_pr1246: AVR device initialized and ready to accept instructions
avrdude_pr1246: device signature = 0x1e950f (probably m328p)
avrdude_pr1246: reading flash memory ...

Reading | ################################################## | 100% 6.90 s 

avrdude_pr1246: writing output file nano_ft232rl_blink_readback_16ms_latency.hex

avrdude_pr1246 done.  Thank you.

```

mcuee, 2022-12-29:
@stefanrueger
Going back to the original report from @mariusgreuel, I am wondering two things.
1) Just wondering why you are not using the HW UART  (RX=AtmelPA0, TX=AtmelPA1) for this paticular bootloader.
2) Just wondering if 115200bps is too fast for Firmware UART?

@mariusgreuel 

Hopefully PR #1246 may help. If not, can you try the following two things.
1) Can you try the lower baudrate urboot firmware, like 19200bps to see if that helps?
2) The other thing is to see if you can build urboot firmware with HW UART (RX=AtmelPA0, TX=AtmelPA1)?
3) You may also want to try Optiboot FW from ATTinyCore for your FTDI converter to see if there is an issue or not. This is just for troubleshooting purpose. 
https://github.com/SpenceKonde/ATTinyCore

stefanrueger, 2022-12-29:
@mcuee 

1. The ATtiny167 has a somewhat non-standard UART which is not picked up by the U**S**ART interface in `urboot.c`.  (Need looking into at the urboot project.) So, alas, there is no UART support for this part.

2. I have tried the [urboot_digisparkpro_16mhz_115200bps_rxb6_txb3_led+b1_ur_vbl.hex](https://github.com/stefanrueger/urboot/blob/main/bootloaders/board_digispark-pro/fcpu_16mhz/115200_bps/urboot_digispark-pro_16mhz_115200bps_rxb6_txb3_led%2Bb1_ur_vbl.hex) with manual reset (release reset; start programming within one second) from this [directory](https://github.com/stefanrueger/urboot/tree/main/bootloaders/board_digispark-pro/fcpu_16mhz/115200_bps) and can confirm it workw fine under Linux with an FTDI USB/serial converter once PR #1246 was applied.

mariusgreuel, 2022-12-29:
I switched from Windows to a raspi/Ubuntu, and with the latest timing changes, I still have communication problems depending on the serial adapter I am using.

For instance, here is the output of a genuine FT232H adapter:

```console
pi@raspi:~/repos/avrdude/build_linux/src $ ./avrdude -c urclock -p t167 -P /dev/ttyUSB0 -vvvv

avrdude: Version 7.0-20221229 (0f956e9)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /home/pi/repos/avrdude/build_linux/src/avrdude.conf
         User configuration file is /home/pi/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

avrdude: libreadline is used; avrdude -t -c urclock should work interactively
         Using Port                    : /dev/ttyUSB0
         Using Programmer              : urclock
avrdude:  245 ms: enter urclock_getsync()
avrdude: send: 0 [30]   [20]
avrdude: ser_recv(): programmer is not responding
avrdude:  270 ms: sleeping for 32 ms
avrdude: send:   [20]   [20]
avrdude: ser_recv(): programmer is not responding
avrdude:  327 ms: sleeping for 64 ms
avrdude: send:   [20]   [20]
avrdude: recv: . [00] . [ea]
avrdude: send:   [20]   [20]
avrdude: recv: . [fc]   [20]
avrdude: send:   [20]   [20]
avrdude: recv: . [1d]   [20]
avrdude: send:   [20]   [20]
avrdude: recv: . [1d]   [20]
avrdude: send:   [20]
avrdude: recv: . [1d]
avrdude set_uP() [urclock.c:1054] warning: uP_table does not know mcuid 1306
avrdude:  459 ms: all good, ready to rock
         AVR Part                      : ATtiny167
         Chip Erase delay              : 15000 us
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    10     4    0 no        512    4      0  4000  4500 0xff 0xff
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                  65    10    64    0 yes     16384  128    128  4500  4500 0xff 0xff
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
[... 9 lines cut ...]
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : Urclock
         Description     : Urclock programmer for urboot bootloaders using urprotocol
         Urboot protocol for ATtiny167

avrdude: send: P [50]   [20]
avrdude: ser_recv(): programmer is not responding
avrdude urclock_recv() [urclock.c:1923] error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude main() [main.c:1246] error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude: send: Q [51]   [20]
avrdude: ser_recv(): programmer is not responding
avrdude urclock_recv() [urclock.c:1923] error: programmer is not responding; try -xstrict and/or vary -xdelay=100

avrdude done.  Thank you.
```

I produced the output using a manual-reset method. The urclock clock seems to be quite sensitive on when the bootloader starts. When I release the reset button while avrdude is already running, I can create a >50% failure rate.

mariusgreuel, 2022-12-29:
> Have you tried to set and vary -xdelay=100

I have both tried -xdelay and -xstrict, but I did not notice any differences

> Does your USB/serial chip reset the board when RTS/DTS is plucked?

I have a Arduino Nano clone with a m168p and a CH340 on it and that worked fine.

The t167 board I am currently resetting manually.

mariusgreuel, 2022-12-29:
> Maybe you can make that the default for avrdude-libftdi.

Good catch. I was aware of the 16ms latency, but I did not connect the dots. I tried it on my Windows machine, and it did not really seem to help.

As libftdi goes, I do not think the latency timer applies to bit-bang/MPSSE mode. There are some differences in the D2XX drivers compared to libftdi/libusb solution that I had to work around, however, it never occurred to me to change the latency timer. I will try to remember that the next time I play with that.

mariusgreuel, 2022-12-29:
> Just wondering why you are not using the HW UART (RX=AtmelPA0, TX=AtmelPA1) for this paticular bootloader. Is this because you want to save the HW UART for the user?

As the Digispark boards were mentioned on Stefans urboot page, I just grabbed the hex files from there. Plus, as some board require tweaking the fuses, I did not want to mess up my expensive Arduino boards, and turned to the cheap clones :-).

> Just wondering if 115200bps is too fast for Firmware UART?

Yes, 115k is on the edge of what a 16MHz AVR can do reliably. But then, these are the bootloaders linked from the README.

@stefanrueger 

BTW, I could not make the Digispark T85 clones work at all. They do not need different fuse settings, do they? Out of the box, they run at 16MHz off the PLL, I think (16.5MHz with the micronucleus bootloader active, I believe). Here, I think the 115k baud-rate combined with a PLL/RC oscillator may push it over the edge, not sure.

mcuee, 2022-12-29:
> > Maybe you can make that the default for avrdude-libftdi.
> 
> Good catch. I was aware of the 16ms latency, but I did not connect the dots. I tried it on my Windows machine, and it did not really seem to help.

This is kind of in-line with my expectation based on my testing of FT232RL based Nano Atmega328P Clone. The latency timer somehow does not cause the issue under Windows even though it causes issue under Linux before the merge of PR #1246.

> As libftdi goes, I do not think the latency timer applies to bit-bang/MPSSE mode. 

I believe that is correct.

> There are some differences in the D2XX drivers compared to libftdi/libusb solution that I had to work around, however, it never occurred to me to change the latency timer. I will try to remember that the next time I play with that.

That will be great. It is not urgent so take your time to work on avrdude-libftdi.

mcuee, 2022-12-29:
> As the Digispark boards were mentioned on Stefans urboot page, I just grabbed the hex files from there. Plus, as some board require tweaking the fuses, I did not want to mess up my expensive Arduino [...]

You can try other baud rate here, like 19200bps and 38400bps
https://github.com/stefanrueger/urboot/tree/main/bootloaders/board_digispark-pro/fcpu_16mhz

 > @stefanrueger
> BTW, I could not make the Digispark T85 clones work at all. They do not need different fuse settings, do they? Out of the box, they run at 16MHz off the PLL, I think (16.5MHz with the micronucleus [...]

Again you can try 19200bps and 38400bps urboot hex files here.
https://github.com/stefanrueger/urboot/tree/main/bootloaders/board_digispark/fcpu_16mhz

mcuee, 2022-12-29:
> I produced the output using a manual-reset method. The urclock clock seems to be quite sensitive on when the bootloader starts. When I release the reset button while avrdude is already running, I can create a >50% failure rate.

@stefanrueger and @mariusgreuel 

Unfortunately I have zero success with DigiSpark Pro ATtiny167 with either an FTDI FT232RL adapter or a CH340 adapter. I tried two boards and I tried 19200bps FW. I am using manually reset as well.

Both are cheap one bought from Taobao (China version of AliExpress) so maybe there are other issues. Still Micronucleus bootloader works fine.

mcuee, 2022-12-29:
Same thing under Linux. I have tried the urboot hex files and I also tried the following 4 hex files I built from source.

The board has 16MHz crystal.
```
mcuee@UbuntuSwift3:~/build/avr/urboot/src$ ls *.hex
attiny167_16mhz_115200bps_ev1f7_swio1_txPB3_rxPB6_ledPB1_wdto1s_ur.hex
attiny167_16mhz_115200bps_ev1f7_swio1_txPB3_rxPB6_ledPB1_wdto8s_ur.hex
attiny167_16mhz_19200bps_ev1f7_swio1_txPB3_rxPB6_ledPB1_wdto1s_ur.hex
attiny167_16mhz_19200bps_ev1f7_swio1_txPB3_rxPB6_ledPB1_wdto8s_ur.hex
```

stefanrueger, 2022-12-29:
> When I release the reset button while avrdude is already running, I can create a >50% failure rate.

@mariusgreuel I recommend to  *first* release reset; *then* press enter on the pre-typed command line `avrdude -c urclock ....` within the `WDTO` time.

>  I have tried the urboot hex files and I also tried the following 4 hex files I built from source

@mcuee The output you get is symptomatic of "dead" pins. Maybe change the RX/TX pins? If one of them is not led out on the board or blown that won't work. You mentioned that micronucleus works. Try using the RX/TX lines that are used in that bootloader?

@mcuee and @mariusgreuel 

I wonder whether a hardware-pullup on rx (FTDI-side) = tx (AVR side) helps? The [neatly documented runs](https://github.com/avrdudes/avrdude/issues/1245#issue-1512204136)  tell me that the comms is as it should be, just that avrdude receives some garbage initially from the AVR. 

| notice there is still a `ser_recv(): programmer is not responding`

`ser_recv()` timing out is part and parcel of the synchronisation comms and *is* expected. This is owing to potential autobaud gobbling up stuff and to `urclock.c`'s best attempt to deal with all sorts of shenannigans that different optiboot version from 4.4 to 8.3 have thrown avrdude's way over the years (eg, optiboot enabling the USART then 300 ms not listening to it (whilst flashing lights) can kill the first byte of AVRDUDE's 2-byte get_sync command; optiboot's WDT reset loops due to its desire to keep the reset cause in R2 as best as possible; etc).

Basically `-c urclock` expects (for this MCU) the response of `0x20` `0x1d` in quick succession that ID's the ATTiny167 plus a few bootloader proerties. In the failing case the comms is out of step by one byte and it receives `0x1d` `0x20`, but not in quick succession. Somehow draining/time out that is designed to get comms back in sync does not work here. I cannot tell whether this is a property of the FTDI, Windows, or a combination.

In the meantime I will reflect on whether a modification of the protocol might be able to deal with initial garbage and/or a Windows `ser_recv()` that tells me 2 bytes were received (but only one was) though that *is* a challenging task! (All of this should not be rushed, though, and I wouldn't want to change the tried and tested sync routine just before a release.)

stefanrueger, 2022-12-29:
> BTW, I could not make the Digispark T85 clones work at all. They do not need different fuse settings, do they? Out of the box, they run at 16MHz off the PLL, I think (16.5MHz with the micronucleus [...]

I had a digispark ATtiny85 running with v7.x urboot some time ago. I don't think I needed to change fuses. MHz are MHz. Is the problem that the PLL-generated F_CPU tends to have a higher F_CPU error than otherwise? If so, might try compiling with F_CPU=16500000L ...

Most of my (vvv limited) different AVRs run with 16 MHz. All run with 115200 bootloading, either USART or SW I/O (on parts without USART). My 8 MHz boards need SW I/O as the classic AVR USART baud rate divisor generates a too high quantisation error for 115,200 baud; SW I/O allows a finer granularity here.

mcuee, 2022-12-30:
> @mcuee The output you get is symptomatic of "dead" pins. Maybe change the RX/TX pins? If one of them is not led out on the board or blown that won't work. You mentioned that micronucleus works. Try using the RX/TX lines that are used in that bootloader?

@stefanrueger 

Thanks. I managed to get Optiboot working first on my board. And it is using PA1/PA0 (HW UART pin). So I rebuild urboot using SW UART with PA1/PA0. Now it works under Linux. I will test WIndows later.

```
mcuee@UbuntuSwift3:~/build/avr/urboot/src$ make MCU=attiny167 F_CPU=16000000L URPROTOCOL=1 SWIO=1 TX=AtmelPA1 RX=AtmelPA0 VBL=1 LED=AtmelPB1 FRILLS=7 WDTO=8S BAUD_RATE=115200
./avr-toolchain/5.4.0/bin/avr-gcc -DSTART=0x3e80UL -DRJMPWP=0xcfd6 -Wl,--section-start=.text=0x3e80 -Wl,--section-start=.version=0x3ffa -D_urboot_AVAILABLE=26 -g -Wundef -Wall -Os -fno-split-wide-types -mrelax -mmcu=attiny167 -DF_CPU=16000000L -Wno-clobbered -DWDTO=8S -DBAUD_RATE=115200 -DLED=AtmelPB1 -DDUAL=0 -DEEPROM=1 -DURPROTOCOL=1 -DFRILLS=7 -DVBL=1 -DSWIO=1 -DTX=AtmelPA1 -DRX=AtmelPA0 -Wl,--relax -nostartfiles -nostdlib -o attiny167_16mhz_115200bps_ev1f7_swio1_txPA1_rxPA0_ledPB1_wdto8s_ur.elf urboot.c
./avr-toolchain/5.4.0/bin/avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex attiny167_16mhz_115200bps_ev1f7_swio1_txPA1_rxPA0_ledPB1_wdto8s_ur.elf attiny167_16mhz_115200bps_ev1f7_swio1_txPA1_rxPA0_ledPB1_wdto8s_ur.hex
./avr-toolchain/5.4.0/bin/avr-objdump -h -S attiny167_16mhz_115200bps_ev1f7_swio1_txPA1_rxPA0_ledPB1_wdto8s_ur.elf > attiny167_16mhz_115200bps_ev1f7_swio1_txPA1_rxPA0_ledPB1_wdto8s_ur.lst
-- 372 384 u7.7 weu-jPr-c attiny167_16mhz_115200bps_ev1f7_swio1_txPA1_rxPA0_ledPB1_wdto8s_ur.hex

mcuee@UbuntuSwift3:~/build/avr/urboot/src$ avrdude -c usbasp -p t167 -U attiny167_16mhz_115200bps_ev1f7_swio1_txPA1_rxPA0_ledPB1_wdto8s_ur.hex -qq && echo OK
OK
mcuee@UbuntuSwift3:~/build/avr/urboot/src$ avrdude -c urclock -P /dev/ttyUSB0 -p t167 -xshowall

avrdude: AVR device initialized and ready to accept instructions
ffffffffffff 0000-00-00 00.00  application 0 store 0 meta 0 boot 384 u7.7 weu-jPr-c vector 16 (EE_RDY) ATtiny167
```

mcuee, 2022-12-30:
@stefanrueger 

And it actually works well under WIndows as well. 

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM7 -p t167  -xshowall

avrdude: AVR device initialized and ready to accept instructions
0000ffffffff 0000-00-00 00.00  application 0 store 0 meta 0 boot 384 u7.7 weu-jPr-c vector 16 (EE_RDY) ATtiny167

```

mcuee, 2022-12-30:
> I had a digispark ATtiny85 running with v7.x urboot some time ago. I don't think I needed to change fuses. MHz are MHz. Is the problem that the PLL-generated F_CPU tends to have a higher F_CPU error than otherwise? If so, might try compiling with F_CPU=16500000L ...

@stefanrueger 

Indeed this seems to work under Linux. There is no reset button so I just burn the bootloader and run with `-c urclock`. I will test Windows later.

```
mcuee@UbuntuSwift3:~/build/avr/urboot/src$ make MCU=attiny85 F_CPU=16500000L MCU=attiny85 URPROTOCOL=1 SWIO=1 TX=AtmelPB3 RX=AtmelPB4 VBL=1 FRILLS=7 WDTO=8S BAUD_RATE=19200
./avr-toolchain/5.4.0/bin/avr-gcc -DSTART=0x1e80UL -DRJMPWP=0xcfd7 -Wl,--section-start=.text=0x1e80 -Wl,--section-start=.version=0x1ffa -D_urboot_AVAILABLE=24 -g -Wundef -Wall -Os -fno-split-wide-types -mrelax -mmcu=attiny85 -DF_CPU=16500000L -Wno-clobbered -DWDTO=8S -DBAUD_RATE=19200 -DDUAL=0 -DEEPROM=1 -DURPROTOCOL=1 -DFRILLS=7 -DVBL=1 -DSWIO=1 -DTX=AtmelPB3 -DRX=AtmelPB4 -Wl,--relax -nostartfiles -nostdlib -o attiny85_16500khz_19200bps_ev1f7_swio1_txPB3_rxPB4_wdto8s_ur.elf urboot.c
./avr-toolchain/5.4.0/bin/avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex attiny85_16500khz_19200bps_ev1f7_swio1_txPB3_rxPB4_wdto8s_ur.elf attiny85_16500khz_19200bps_ev1f7_swio1_txPB3_rxPB4_wdto8s_ur.hex
./avr-toolchain/5.4.0/bin/avr-objdump -h -S attiny85_16500khz_19200bps_ev1f7_swio1_txPB3_rxPB4_wdto8s_ur.elf > attiny85_16500khz_19200bps_ev1f7_swio1_txPB3_rxPB4_wdto8s_ur.lst
-- 374 384 u7.7 weu-jPr-c attiny85_16500khz_19200bps_ev1f7_swio1_txPB3_rxPB4_wdto8s_ur.hex

mcuee@UbuntuSwift3:~/build/avr/urboot/src$ avrdude -c usbasp -p t85 -U attiny85_16500khz_19200bps_ev1f7_swio1_txPB3_rxPB4_wdto8s_ur.hex -qq && echo OK
OK
mcuee@UbuntuSwift3:~/build/avr/urboot/src$ avrdude -c urclock -p t85 -P /dev/ttyUSB1 -b 19200 -xshowall

avrdude: AVR device initialized and ready to accept instructions
ffffffffffff 0000-00-00 00.00  application 0 store 0 meta 0 boot 384 u7.7 weu-jPr-c vector 6 (EE_RDY) ATtiny85

```

mcuee, 2022-12-30:
And indeed it is okay under Windows as well.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t85 -U .\attiny85_16500khz_19200bps_ev1f7_swio1_txPB3_rxPB4_wdto8s_ur.hex -qq && echo OK
OK
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM7 -b 19200 -p t85 -xshowall

avrdude: AVR device initialized and ready to accept instructions
0000ffffffff 0000-00-00 00.00  application 0 store 0 meta 0 boot 384 u7.7 weu-jPr-c vector 6 (EE_RDY) ATtiny85

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t85 -U .\attiny85_16500khz_19200bps_ev1f7_swio1_txPB3_rxPB4_wdto8s_ur.hex -qq && echo OK
OK
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM7 -b 19200 -p t85 -U .\Blink.ino_t85.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e930b (probably t85)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink.ino_t85.hex for flash
         with 542 bytes in 1 section within [0, 0x21d]
         using 9 pages and 34 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 542 bytes flash ...

Writing | ################################################## | 100% 0.48 s

avrdude: 542 bytes of flash written
avrdude: verifying flash memory against .\Blink.ino_t85.hex

Reading | ################################################## | 100% 0.41 s

avrdude: 542 bytes of flash verified

avrdude done.  Thank you.

```

stefanrueger, 2022-12-30:
@mcuee Thanks for testing. Looks like the t85 board has a tricky frequency. Any way you can measure it? Like measuring the timing of a blink sketch with an osci (or another board with a halfway reliable quartz). If you use the measured *real* frequency as `F_CPU=....L` argument then 115200 baud `*should*` work.  At 16 MHz there are ~139 cycles per bit. The SW I/O in `urboot.c` tries to match the required cycles, so there is a quanitisation error of ~ 1/139 < 1%. That should be OK. *But* if the *real* frequency deviates from the compiled-for frequency by another 2% one might get into trouble...

mcuee, 2022-12-30:
> @mcuee Thanks for testing. Looks like the t85 board has a tricky frequency. Any way you can measure it? Like measuring the timing of a blink sketch with an osci (or another board with a halfway [...]

I had three weeks' of long vacation this December. I will try to measure the frequency next week when I am back to work.

mcuee, 2022-12-30:
> I will test 8MHz and 16MHz clock settings as well.

@stefanrueger 

Both do not work, tested with 19200bps and 115200bps. So probably the clock is the issue as you mentioned.

stefanrueger, 2022-12-30:
>  raspi/Ubuntu, and with the latest timing changes, I still have communication problems depending on the serial adapter

Thanks for reporting this. @mariusgreuel I'll try to think about what happens when avrdude reads arbitrary data before the board gets into the bootloader. Maybe `-c urclock` can do sth to be more robust against this kind of scenario... Knowing whether a HW pullup on the AVR TX helps would be good.

stefanrueger, 2022-12-30:
> probably the  clock is the issue

The overall comms error is compounded by sender quantisation error, sender freq error, receiver quantisation error, receiver freq error, then multiplied by 10 bits per byte. That shouldn't go beyond a quarter bit, so a 2% clock error on one side may well ruin the comms (irrespective of UART or SW I/O). 

So, what's the story of a t85 F_CPU? What kind of error can we expect? Even a terrible 1000 ppm ceramic resonator "only" has a clock error of 0.1% (negligible for serial errors).

mcuee, 2022-12-30:
> The overall comms error is compounded by sender quantisation error, sender freq error, receiver quantisation error, receiver freq error, then multiplied by 10 bits per byte. That shouldn't go beyond [...]

My paticular ATTiny85 board is using the internal RC oscillator.

The ATTiny167 board has 16MHz external crystal.

mcuee, 2022-12-30:
> It seems to me my board may runnling at a frequency closer to 16.5MHz than the norminal 16MHz.

@stefanrueger 
I believe this is true. NOw I can run 115200bps.
```
mcuee@UbuntuSwift3:~/build/avr/urboot/src$ avrdude -c usbasp -p t85 -U ./attiny85_16500khz_115200bps_ev1f7_swio1_txPB3_rxPB4_wdto8s_ur.hex -qq && echo OK
OK
mcuee@UbuntuSwift3:~/build/avr/urboot/src$ avrdude -c urclock -p t85 -P /dev/ttyUSB1 -xshowall

avrdude: AVR device initialized and ready to accept instructions
ffffffffffff 0000-00-00 00.00  application 0 store 0 meta 0 boot 384 u7.7 weu-jPr-c vector 6 (EE_RDY) ATtiny85

```

mcuee, 2022-12-30:
Tested under Windows as well.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t85 -U .\attiny85_16mhz_115200bps_ev1f7_swio1_txPB3_rxPB4_wdto8s_ur.hex -qq && echo OK
OK
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM7 -b 115200 -p t85 -U .\Blink.ino_t85.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e930b (probably t85)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink.ino_t85.hex for flash
         with 542 bytes in 1 section within [0, 0x21d]
         using 9 pages and 34 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 542 bytes flash ...

Writing | ################################################## | 100% 0.17 s

avrdude: 542 bytes of flash written
avrdude: verifying flash memory against .\Blink.ino_t85.hex

Reading | ################################################## | 100% 0.09 s

avrdude: 542 bytes of flash verified

avrdude done.  Thank you.
```

MCUdude, 2022-12-30:
> Looks like the t85 board has a tricky frequency. Any way you can measure it? Like measuring the timing of a blink sketch with an osci (or another board with a halfway reliable quartz).

A simple way of measuring the clock frequency is to enable the CKOUT fuse bit. Then the clock signal will be outputted on pin PB4, which can easily be measured using an oscilloscope or a frequency counter.

mcuee, 2022-12-30:
> A simple way of measuring the clock frequency is to enable the CKOUT fuse bit. Then the clock signal will be outputted on pin PB4, which can easily be measured using an oscilloscope or a frequency counter.

@MCUdude 

Nice tip.

Based on my latest test results, in the end the clock frequency is not the issue. I think my problems are mostly related to the use of manual reset. In the end I can not reproduce the issue reported by @mariusgreuel. 

I need to have better testing facility for urboot/optiboot. I need to find some components (10k resistors and 100nF capacitors) first and then a few USB-TTL converters with the DTR pin available so that I can have the auto-reset circuitries.

As of now I only have two FTDI adapters with the DTR pin exposed. I have quite a few  CH340 adapters but they do not have the exposed DTR pins. They usually have only 5V/TX/RX/GND or 5V/3V3/TX/RX/GND pins.  The single PL2303HX adapter does not have the DTR pin either. 

I was trying to reuse two damaged Uno Clones with the ATmega328P SMD chips removed. They have the RESET circuitry and RXD/TXD pins. Somehow they do not work. I have not figured out why they do not work yet.

Can you share your test setup for the ATtiny chips?

mcuee, 2022-12-30:
@stefanrueger and @mariusgreuel 

I think in the end, I can not reproduce the issue under Linux/Windows in the end (other than the issue of FT232 adapter under Linux which was addressed by PR #1246).

We will have to wait for more results from @mariusgreuel.

stefanrueger, 2022-12-30:
@mariusgreuel Could you try what happens when the serial interface is drained periodically (rather than at the start only):
``` diff
diff --git a/src/urclock.c b/src/urclock.c
index 7f32d88..e461947 100644
--- a/src/urclock.c
+++ b/src/urclock.c
@@ -1985,6 +1985,7 @@ static int urclock_getsync(const PROGRAMMER *pgm) {
       if(!ur.gs.seen || iob[0] != ur.gs.stk_insync || iob[1] != ur.gs.stk_ok || iob[0] == iob[1]) {
         ur.gs.stk_insync = iob[0];
         ur.gs.stk_ok = iob[1];
+        serial_drain(&pgm->fd, 0); // Drain periodically to guard against initial line noise
         ur.gs.seen = 1;
       } else
         break;
@@ -2247,9 +2248,7 @@ static int urclock_open(PROGRAMMER *pgm, const char *port) {
     usleep((125+ur.delay)*1000); // Wait until board starts up accommodating effective drain time
 #endif
 
-  // Drain any extraneous input
   serial_drain_timeout = 20;    // ms
-  serial_drain(&pgm->fd, 0);
 
   pmsg_debug("%4ld ms: enter urclock_getsync()\n", avr_mstimestamp());
   if(urclock_getsync(pgm) < 0)
```

It is these three bytes that seem to jeopardise the connection. They might be some kind of line noise/garbage/FTDI intialisation artifact, but let's see whether above intervention gets rid of the problem.

[image]

mariusgreuel, 2022-12-30:
> Could you try what happens when the serial interface is drained periodically

@stefanrueger Im the meantime, I came to the same conclusion. I think what happens during a manual-reset communication failure is that there is one extra byte in the RX buffer that enters after you drain the buffer. So the response is [1d][20] instead of [20][1d], with one remaining RX byte not read.

I fixed that by adding this line:

```console
diff --git a/src/urclock.c b/src/urclock.c
index 7f32d88e..49bc9809 100644
--- a/src/urclock.c
+++ b/src/urclock.c
@@ -1978,6 +1978,7 @@ static int urclock_getsync(const PROGRAMMER *pgm) {
      * that are strict about the protocol, eg, picoboot, the presence of -xstrict implies that
      * comms should use Cmnd_STK_GET_SYNC for getting in sync.
      */
+    serial_drain(&pgm->fd, 0); // Drain periodically to guard against initial line noise
     iob[0] = attempt == 0? autobaud_sync: ur.strict? Cmnd_STK_GET_SYNC: Sync_CRC_EOP;
     iob[1] = Sync_CRC_EOP;
     urclock_send(pgm, iob, 2);
```

With that line in place I have not seen this kind of error anymore, using the t167 board and various serial converters.

BTW, when I use set up the RTS auto-reset mechanism, the error disappeared also.

I have two more suggestions:

1) Change this error to a warning:

    ```console
    avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
    ```

2) Print the warning only when run in terminal mode:

    ```console
    avrdude warning: compiled without readline library, cannot use avrdude -t -c urclock interactively
    ```

I am probably going to spend some more time with urboot. I have a project where I hacked some commercial device using a m88p, and there was just not enough space to fit a bootloader. That you managed to squeeze urboot into some 250 bytes is really amazing!

stefanrueger, 2022-12-30:
@mariusgreuel Thanks for testing and confirming my hunch. Could you see whether my suggestion also works? I know it sounds counter-intuitive to drop the initial drain and then to periodically drain *after* receiving (and evaluating!) responses rather than *before every* request is sent, but 
 - Draining is not normally needed *and* each drain costs 20 ms, so that adds up
 - urclock_getsync() needs two consistent responses so can afford evaluating initial line noise
 - @mcuee and I spent a lot of calibration work to ensure `-c urclock` works specifically with `optiboot v8.3` out of the box without the need to add `-xdelay=...`; if my suggestion works that would make it easier to calibrate again

Your other suggestions are great - let's see how I can implement them

stefanrueger, 2022-12-30:
> RTS auto-reset mechanism, the error disappeared also

Interesting. i wonder whether the initial drain sees to that. If so it might be that the error also disappears if avrdude is started *after* manually releasing reset.

mariusgreuel, 2022-12-30:
@stefanrueger I tried your patch and it appears to work just as well. Tried that on the raspi, too.

BTW:

When I am varying the reset relative to the start of avrdude , I sometimes get an output like this:

```
avrdude warning: compiled without readline library, cannot use avrdude -t -c urclock interactively
        but it is still possible to pipe: echo "d fl 0 32; quit" | tr \; \\n | avrdude -t -curclock
avrdude warning: attempt 1 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 2 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 3 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 4 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 5 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 6 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 7 of 10: not in sync
avrdude error: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude warning: attempt 8 of 10: not in sync
avrdude warning: attempt 9 of 10: not in sync
avrdude warning: attempt 10 of 10: not in sync
avrdude error: unable to open programmer urclock on port COM3

avrdude done.  Thank you.
```

In this example, communication was established at attempt 9, however, as it takes two rounds to sync, it fails. This might be confusing to the user since there is 'something happening', yet it still fails. Perhaps we should just make the 'not in sync' message a debug message?

stefanrueger, 2022-12-30:
@mariusgreuel Brilliant. So I'll create a PR along these lines. 

I'd suggest keeping the "not in sync" messages as warning messages but remove the associated "not responding" messages bar the last one to hint at things that might help:

```
$ avrdude -tc urclock
avrdude warning: attempt 1 of 10: not in sync
avrdude warning: attempt 2 of 10: not in sync
avrdude warning: attempt 3 of 10: not in sync
avrdude warning: attempt 4 of 10: not in sync
avrdude warning: attempt 5 of 10: not in sync
avrdude warning: attempt 6 of 10: not in sync
avrdude warning: attempt 7 of 10: not in sync
avrdude warning: attempt 8 of 10: not in sync
avrdude warning: attempt 9 of 10: not in sync
avrdude warning: attempt 10 of 10: not in sync
avrdude warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude error: unable to open programmer urclock on port /dev/ttyUSB0

avrdude done.  Thank you.
```

Turns out it's difficult to suppress the warning `compiled without readline library` as the place where it is issued does not know about the local terminal variable in `main` ... If only we knew someone who has the skills to change cmake in a way that it ensures the readline library is always used. :wink: PR #1186 ...

stefanrueger, 2022-12-31:
@mariusgreuel I am pretty chuffed with the changes in PR #1251; my tests show the synchronisation is now *much* more resilient in manual reset. Fun fact 1: I could even press reset when in terminal mode and it would catch up.  So, thanks for raising the issue and for debugging.

@mcuee Fun fact 2: I have just found two different versions of the digispark pro board (one of which by digistump)  that have the following difference: `PB6` and `PB3` are broken out to Pins 3 and 4, while it is Pins 4 and 3 on the other version. Following the wrong doc will mix up TX and RX of the pre-compiled urboot bootloaders :(

mcuee, 2022-12-31:
@stefanrueger 

Indeed there are different versions of DigiSpark Pro clone boards as I learned from the Internet.

Actually there are different versions for the DigiSpark ATtiny85 clone boards as well but the pin assignment seems to be the same.

mcuee, 2022-12-31:
> Turns out it's difficult to suppress the warning `compiled without readline library` as the place where it is issued does not know about the local terminal variable in `main` ... If only we knew someone who has the skills to change cmake in a way that it ensures the readline library is always used. 😉 Issue #1186 ...

@stefanrueger and @mariusgreuel 

#1186 is of course good, especially when avrdude official Windows binary release will use MSVC build to give the users best out-of-box user experiences with the use of avrdude-libusb and avrdude-libftdi. Hopefully @mariusgreuel will work out the solution soon.

On the other hand, can we use the same trick of #1207 to differentiate terminal mode and normal mode?
* #1207

closed 2022-12-31 by stefanrueger

mcuee, 2023-01-01:
Just a sanity check on the latest git main (PR #1251 merged).

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ avrdude -c usbasp -p t88 
-U ./attiny88_16mhz_115200bps_ef7_swio1_txPD6_rxPD7_ledPD0_wdto8s_ur.hex -qq && echo OK
OK

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ avrdude -c urclock -P /dev/ttyUSB0 -p t88 -U Blink_t88.ino.hex 

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9311 (probably t88)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file Blink_t88.ino.hex for flash
         with 468 bytes in 1 section within [0, 0x1d3]
         using 8 pages and 44 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 468 bytes flash ...

Writing | ################################################## | 100% 0.14 s 

avrdude: 468 bytes of flash written
avrdude: verifying flash memory against Blink_t88.ino.hex

Reading | ################################################## | 100% 0.07 s 

avrdude: 468 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2023-01-01:
But I have a strangt issue that Arduino ATtinyCore Optiboot upload will have the right blinking frequency.

If I use urboot and urclock, then the blinking frequency is too fast.

Problem -- 16MHz is out of spec for external clock as per the datasheet.
https://ww1.microchip.com/downloads/en/DeviceDoc/doc8008.pdf

[4 more replies]