avrdudes/avrdude #1250: Running avrdude remotely

merlinblack, 2022-12-30:
https://github.com/merlinblack/RemoteAVRDUDE

I made this in the last few days while on holiday (from coding for work :rofl: ) - mostly for fun and to play with c++17 features - however, it works well for what I use it for.

Basically a proxy for avrdude that understands some of the options enough to copy the needed files back and forth.

Perhaps other uses of avrdude will find it useful too.

mcuee, 2022-12-30:
Thanks for the contribution. I tend to think this will not be included in the avrdude project.

merlinblack, 2022-12-31:
"I tend to think this will not be included in the avrdude project." - Quite right - it's outside the scope of "do one thing and do it well". :smiley: