avrdudes/avrdude #1253: ftdi_syncbb: very slow D2XX speed compared to libftdi

mcuee, 2023-01-01:
Somehow the speed is very slow when using the FTDI driver (D2XX) with the MSVC build. I have set the latency timer for the serial port to 1s in this case but I think that does not matter here.

```
 PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git_msvc  -c arduino-ft232r -p m328p -U .\hex\Blink.ino.with_bootloader.standard.hex -v

avrdude_git_msvc: Version 7.0-20221231 (b2c03f5)
                  Copyright the AVRDUDE authors;
                  see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                  System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

                  Using Port                    : usb
                  Using Programmer              : arduino-ft232r
avrdude_git_msvc: ft245r_open(): no device identifier in portname, using default
avrdude_git_msvc: input file .\hex\Blink.ino.with_bootloader.standard.hex auto detected as Intel Hex
                  AVR Part                      : ATmega328P
                  Chip Erase delay              : 9000 us
                  PAGEL                         : PD7
                  BS2                           : PC2
                  RESET disposition             : possible i/o
                  RETRY pulse                   : SCK
                  Serial program mode           : yes
                  Parallel program mode         : yes
                  Timeout                       : 200
                  StabDelay                     : 100
                  CmdexeDelay                   : 25
                  SyncLoops                     : 32
                  PollIndex                     : 3
                  PollValue                     : 0x53
                  Memory Detail                 :

                                                    Block Poll               Page                       Polled
                    Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                    ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                    eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
                    flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
                    lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                    hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                    efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                    lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                    signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                    calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

                  Programmer Type : ftdi_syncbb
                  Description     : Arduino: FT232R connected to ISP
                  Pin assignment  : 0..7 = DBUS0..7
                    VCC     =  (not used)
                    BUFF    =  (not used)
                    RESET   =  7
                    SCK     =  5
                    SDO     =  6
                    SDI     =  3
                    ERR LED =  (not used)
                    RDY LED =  (not used)
                    PGM LED =  (not used)
                    VFY LED =  (not used)

avrdude_git_msvc: AVR device initialized and ready to accept instructions
avrdude_git_msvc: device signature = 0x1e950f (probably m328p)
avrdude_git_msvc: Note: flash memory has been specified, an erase cycle will be performed.
                  To disable this feature, specify the -D option.
avrdude_git_msvc: erasing chip
avrdude_git_msvc: reading input file .\hex\Blink.ino.with_bootloader.standard.hex for flash
                  with 1426 bytes in 3 sections within [0, 0x7fff]
                  using 12 pages and 110 pad bytes
avrdude_git_msvc: writing 1426 bytes flash ...

Writing | ################################################## | 100% 76.42 s

avrdude_git_msvc: 1426 bytes of flash written
avrdude_git_msvc: verifying flash memory against .\hex\Blink.ino.with_bootloader.standard.hex

Reading | ################################################## | 100% 57.97 s

avrdude_git_msvc: 1426 bytes of flash verified

avrdude_git_msvc done.  Thank you.
```

If I switch the driver to WInUSB and use libftdi-1.0 (MSYS2 build) will get much better speed.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git  -c arduino-ft232r -p m328p -U .\hex\Blink.ino.with_bootloader.standard.hex -v

avrdude_git: Version 7.0-20221231 (b2c03f5)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : usb
             Using Programmer              : arduino-ft232r
avrdude_git: ft245r_open(): no device identifier in portname, using default
avrdude_git: input file .\hex\Blink.ino.with_bootloader.standard.hex auto detected as Intel Hex
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : possible i/o
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

             Programmer Type : ftdi_syncbb
             Description     : Arduino: FT232R connected to ISP
             Pin assignment  : 0..7 = DBUS0..7
               VCC     =  (not used)
               BUFF    =  (not used)
               RESET   =  7
               SCK     =  5
               SDO     =  6
               SDI     =  3
               ERR LED =  (not used)
               RDY LED =  (not used)
               PGM LED =  (not used)
               VFY LED =  (not used)

avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e950f (probably m328p)
avrdude_git: Note: flash memory has been specified, an erase cycle will be performed.
             To disable this feature, specify the -D option.
avrdude_git: erasing chip
avrdude_git: reading input file .\hex\Blink.ino.with_bootloader.standard.hex for flash
             with 1426 bytes in 3 sections within [0, 0x7fff]
             using 12 pages and 110 pad bytes
avrdude_git: writing 1426 bytes flash ...

Writing | ################################################## | 100% 0.89 s

avrdude_git: 1426 bytes of flash written
avrdude_git: verifying flash memory against .\hex\Blink.ino.with_bootloader.standard.hex

Reading | ################################################## | 100% 0.65 s

avrdude_git: 1426 bytes of flash verified

avrdude_git done.  Thank you.
```

mcuee, 2023-01-01:
@mariusgreuel 

Please take a look when you got some time.

mariusgreuel, 2023-01-02:
@mcuee Thanks for the report. I have assigned this issue to me.

One thing I am aware of is a difference between the way read timeouts work, i.e. when you request more bytes to read as are actually being received, and the operation times out with a partial read. With libftdi/libusb it seems (though that is only a guess) that the read implementation waits at least one USB poll interval and then returns (which introduces a millisecond delay), whereas the D2XX drivers read implementation returns immediately with whatever in in the PCs read buffer. As a result, the D2XX drivers may return zero bytes, whereas the libftdi/libusb implementation may return one or more bytes. Unfortunately, the AVRDUDE implementation relies on the particular way libftdi/libusb works, instead of retrying when zero bytes are received and more are expected.

I simply implemented a [hack in the D2XX port](https://github.com/avrdudes/libftdi/blob/windows/src/ftdi.cpp#L347) that waits for a long time (1000ms) when there are no bytes received. Perhaps this is the reason, I have to check that out.

It would have been better to change the AVRDUDE code instead, but at the time I did not want to make changes to the AVRDUDE codebase, so that my Windows fork was easy to sync with the upstream. Now that the relevant code has been merged upstream, things have changed.

mcuee, 2023-01-03:
@mariusgreuel 

Thanks for the detailed explanation. I am okay to postpone the issue resolution after 7.1 release if the fix needs some time, as there is a work-around (to use MSY2 build which will use libftdi1 library).

mcuee, 2023-04-09:
> It would have been better to change the AVRDUDE code instead, but at the time I did not want to make changes to the AVRDUDE codebase, so that my Windows fork was easy to sync with the upstream. Now that the relevant code has been merged upstream, things have changed.

@mariusgreuel 

Now that avrdude 7.1 has been released for a while, please take a look at this when you got some time. Thanks.

raspB975, 2024-01-26:
Hi! I'm having the same issue with writing/reading SLOW speed via FTDI on v7.2. 
On v5.9 All seem to be working fine.
How are you solving the problem?
on v7.2:
```
         Programmer Type : ftdi_syncbb
         Description     : FT232R Synchronous BitBang
         Pin assignment  : 0..7 = DBUS0..7
           VCC     =  (not used)
           BUFF    = ~0
           RESET   =  1
           SCK     =  2
           SDO     =  7
           SDI     =  4
           ERR LED =  (not used)
           RDY LED =  (not used)
           PGM LED =  (not used)
           VFY LED =  (not used)
avrdude: AVR device initialized and ready to accept instructions
Reading | ################################################## | 100% 0.07 s
avrdude: device signature = 0x1e930b (probably t85)

avrdude: processing -U flash:r:test_out.hex:i
avrdude: reading flash memory ...
Reading | ################################################## | 100% 217.79 s
avrdude: writing output file test_out.hex
```

MCUdude, 2024-01-26:
It may be that the bitbang clock speed needs to be adjusted.
I have not used the FTDI for bit banging before, but I think the delay/clock speed can be specified using -B, -b or -i.
I think the Avrdude documents this properly.

raspB975, 2024-01-26:
I think it's so much better!

```
avrdude: input file test_out.hex contains 4492 bytes
Reading | ################################################## | 100% 3.49 s
avrdude: 4492 bytes of flash verified
```
Thanks a lot.

impnet, 2024-01-26:
Same problem mentioned here: https://github.com/mariusgreuel/avrdude/issues/13

Last working (fast) version was v6.3.1.1. Any higher version are so slow.

mcuee, 2024-01-27:
> Same problem mentioned here: [mariusgreuel/avrdude#13](https://github.com/mariusgreuel/avrdude/issues/13)
> 
> Last working (fast) version was v6.3.1.1. Any higher version are so slow.

For avrdude 7.0/7.1/7.2 and git main, you can always use the mingw version to have fast speed. The catch is that you have to use Zadig to switch to WinUSB driver.

stefanrueger, 2024-01-27:
@mariusgreuel Do you want to have a look at this issue? Last year you gave some [good insight](https://github.com/avrdudes/avrdude/issues/1253#issuecomment-1369184499) into it and you seem to know what needs doing. We are getting increasingly more reports for this issue, and it would be great to solve it! Thanks

impnet, 2024-01-28:
> > Same problem mentioned here: [mariusgreuel/avrdude#13](https://github.com/mariusgreuel/avrdude/issues/13)
> > Last working (fast) version was v6.3.1.1. Any higher version are so slow.
> 
> Take note that version is not from official avrdude project, but rather the fork from @mariusgreuel. So it is not supported by avrdude project. https://github.com/mariusgreuel/avrdude/releases
> 
> For avrdude 7.0/7.1/7.2 and git main, as a work-around, you can use the mingw version to have fast speed. The catch is that you have to use Zadig to switch to WinUSB driver.

Not a very usefull workaround. Image how skilless users will be forced to install via third party software another drivers for USB device. All this just for update code in any AVR.
Not every users will undesrstand how to correctly instal WinUSB driver. And of course, almost every user will not trust Zadig software (it looks like blotware or any kind of scam software).
You muset undesrantd that it is not issue only of few users, you must think globally.

mcuee, 2024-01-29:
@impnet 

We understand that it is better to fix the issue with the MSVC build. Unfortunately @mariusgreuel is busy and has not been able to come back to this issue for a while.

Until someone steps in to work on this issue, the recommended workaround from avrdude project is to use the MinGW version and switch driver to WinUSB using Zadig,

mcuee, 2024-01-29:
And it is the same situation for the following two issues.
* #1271 (not so critical as of now since we use GNU Readline under MinGW which mostly works)
* #968 (a real bug but not that critical as the known affected programmer is USBASP with alternative FW. Workaround is to use other FW without the extra HID interface, or use the MinGW version.

mcuee, 2024-01-30:
To make this issue more prominent I have added the `bug` label.

mcuee, 2024-03-17:
Remove this from avrdude 7.4 milestone since there is no active development on this topic.

stefanrueger, 2025-06-02:
Cool. Thanks for documenting the workaround. Owing to that and seeing the team doesn't have the capacity to fix the issue we can close as not planned.

mcuee, 2025-06-02:
We can always reopen the issue later when someone works on it.

closed 2025-06-02 by mcuee