avrdudes/avrdude #1254: stk500v2.c dumps core on open

dl8dtl, 2023-01-01:
The culprit is commit c991ae97 from 2022-12-21:

```
  // Get USB serial number
  pgm->usbsn = serial_serno();
```

`serial_serno` is a macro defined as `#define serial_serno (serdev->serno)`.
So, `serdev->serno` is a simple number, but is called as a function.

What was the idea? Just omit the parenthesis in the above line?

dl8dtl, 2023-01-01:
What is supposed to happen (as in my case, on a real STK500) if the device is not attached through USB?

dl8dtl, 2023-01-01:
Seems it's not a number but supposed to be a string?

dl8dtl, 2023-01-01:
Trying to understand the code … it seems it is supposed to be a function returning the serial number from a cache.
In that case, either ser_posix (and probably also the Windows counterpart) needs to implement a `.serno` method, *or* the user of the serial number function needs to take into account that such a method probably doesn't exist, so there is simply no serial number available.

dl8dtl, 2023-01-01:
There is another inconsistency with this, in jtag3.c:
```
  // Use serial number pulled from the USB driver. If not present, query the programmer
  if (pgm->usbsn && *pgm->usbsn)
    sn = pgm->usbsn;
  else {
```
`pgm->usbsn` here results from a call to the driver's `.serno` method, so it is unlikely `NULL` at that point. If the driver has no serial number found, it is more likely to return an empty string. That one would not trigger the `else` part though.

MCUdude, 2023-01-01:
Happy new year!

I'm very sorry that a critical issue like this slipped through. I only tested with STK500v2 compatible programmers with native USB, so that's why I never caught this. `serial_serno` is only set to point to a function in usb_hidapi.c and usb_libusb.c, so when using a serial device, `serial_serno` is just function pointer that points to NULL. However, I'm able to reproduce the issue when using my STK500 board.

https://github.com/avrdudes/avrdude/blob/b2c03f5d4094335544e2becaba157fa4689006a6/src/libavrdude.h#L649-L652

https://github.com/avrdudes/avrdude/blob/b2c03f5d4094335544e2becaba157fa4689006a6/src/usb_hidapi.c#L320-L322
https://github.com/avrdudes/avrdude/blob/b2c03f5d4094335544e2becaba157fa4689006a6/src/usb_hidapi.c#L47

https://github.com/avrdudes/avrdude/blob/b2c03f5d4094335544e2becaba157fa4689006a6/src/usb_libusb.c#L596-L599
https://github.com/avrdudes/avrdude/blob/b2c03f5d4094335544e2becaba157fa4689006a6/src/usb_libusb.c#L63

As for stk500v2.c, this should fix the issue:

```c
  // Get USB serial number
  if (serial_serno)
    pgm->usbsn = serial_serno();
```

```
  // Use serial number pulled from the USB driver. If not present, query the programmer
  if (pgm->usbsn && *pgm->usbsn)
    sn = pgm->usbsn;
  else {
```

> pgm->usbsn here results from a call to the driver's .serno method, so it is unlikely NULL at that point. If the driver has no serial number found, it is more likely to return an empty string. That one would not trigger the else part though.

There are AFAIK no native serial JTAG3 programmers that I'm aware of, so this won't really be an issue. However, it should be handled correctly regardless.

This _should_ be possible to fix with the same code as for stk500v2.c:

```c
  // Get USB serial number
  if (serial_serno)
    pgm->usbsn = serial_serno();
```

This ensures that `pgm->usbsn` always points to a function.

dl8dtl, 2023-01-01:
I'm just wondering what `pgm->usbsn` is being used for later, and whether all their users take into account it might possibly be `NULL`.
Ah yes, hyggeligt nyt år!

closed 2023-01-01 by dl8dtl

MCUdude, 2023-01-01:
`pgm->usbsn` holds a pointer to a character array that either sits in usb_hidapi.c or usb_libusb.c. If the programmer is not using, usb_hidapi.c or usb_libusb.c, there's no `serial_serno`, and therefore no serial number char array that `pgm->usbsn` can point to.

https://github.com/avrdudes/avrdude/blob/b2c03f5d4094335544e2becaba157fa4689006a6/src/usb_hidapi.c#L45-L49

https://github.com/avrdudes/avrdude/blob/b2c03f5d4094335544e2becaba157fa4689006a6/src/usb_libusb.c#L61-L65

dl8dtl, 2023-01-01:
And it only seems to be used within some printouts, where the non-NULL was properly checked.
I just pushed your fix (directly, without a PR).

MCUdude, 2023-01-01:
Thanks!

BTW, perhaps the comment should also be changed for it to make a little more sense.

How about something like this instead?

```c
  // Get USB serial number function if function pointer is present
  if (serial_serno)
    pgm->usbsn = serial_serno();
```

dl8dtl, 2023-01-01:
Makes sense, yes. That's what probably confused me in the first place (in particular, as the serdev method is also called `.serno`).

stefanrueger, 2023-01-01:
Hmm, the stickler in me thinks that the original comment was correct, and the new one is ... mildly confusing. The code actually gets the USB serial number and puts a reference to it into pgm->usbsn. (pgm->usbn is printed, not called).

The if clause checks whether the function pointer serial_serno has been initialised, which is standard practice. I overlooked that sanity check when reviewing, and am sorry about that.

mcuee, 2023-01-02:
Ah, I did not test the normal serial based STK500v2 programmer so I did not catch this either.

I can reproduce the issue and I can confirm latest git fixed the issue.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git_b2c03f5 -c stk500v2 -P COM6 -p m328p -v

avrdude_git_b2c03f5: Version 7.0-20221231 (b2c03f5)
                     Copyright the AVRDUDE authors;
                     see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                     System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

                     Using Port                    : COM6
                     Using Programmer              : stk500v2
                     (crashed)
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c stk500v2 -P COM6 -p m328p

avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e950f (probably m328p)

avrdude_git done.  Thank you.

```

mcuee, 2023-01-02:
BTW, I have different STK500v2 clones.

1) More proper clone with the same Firmware as the original (using ATmega8535 as the main CPU).
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c stk500v2 -P COM6 -p m328p -v

avrdude_git: Version 7.0-20230101 (b6d50ef)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : COM6
             Using Programmer              : stk500v2
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : possible i/o
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

             Programmer Type : STK500V2
             Description     : Atmel STK500 version 2.x firmware
             Programmer Model: STK500
             Hardware Version: 2
             Firmware Version Controller : 2.10
             Topcard         : Unknown
             Vtarget         : 5.1 V
             SCK period      : 2.2 us
             Varef           : 5.0 V
             Oscillator      : 3.686 MHz

avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e950f (probably m328p)

avrdude_git done.  Thank you.
```

Clones with their own implementation using 8051 based CPU.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c stk500v2 -P COM6 -p m328p -v

avrdude_git: Version 7.0-20230101 (b6d50ef)
             Copyright the AVRDUDE authors;
             see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : COM6
[... 20 lines cut ...]
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

             Programmer Type : STK500V2
             Description     : Atmel STK500 version 2.x firmware
             Programmer Model: STK500
             Hardware Version: 18
             Firmware Version Controller : 2.10
             Topcard         : Unknown
             Vtarget         : 5.2 V
             SCK period      : 2.2 us
             Varef           : 0.0 V
             Oscillator      : Off

avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e950f (probably m328p)

avrdude_git done.  Thank you.
```

dl8dtl, 2023-01-02:
After thinking a bit more about it:
What I don't really like is storing the USB serial number in a static variable. It prevents the library function(s) from being used more than once, e.g. by a customized toplevel tool that tries to handle multiple programmers in parallel.
Why not just store the serial number itself in `struct serdev` (using `strdup()`), instead of complicating things by having to call a function that retrieves it?

dl8dtl, 2023-01-02:
Accepted. But, while this certainly would need to be fixed, I don't see the point of adding even more static variables – in particular, since I think it's completely unnecessary to have an access function here. Just storing the serial number string in the serdev object would have done it – and if we really (some day ;-) turn the serdev object into one dynamically assigned, we don't have another extra thing to clean up.

stefanrueger, 2023-01-02:
| unnecessary to have an access function here. Just storing the serial number string in the serdev object would have done it

OK, I'll check and raise a PR if the serno access function is not needed. I can make usbsn a component with, say 128, bytes. IIRC @MCUdude mentioned 127 is the max length of a serial number.

dl8dtl, 2023-01-02:
It's not even an effort to make it completely dynamic (using `strdup`); we have a close function that can subsequently free up the space.

stefanrueger, 2023-01-02:
@dl8dtl I don't think it is as easy as you made it sound: The code tries to put the USB device's serial number into `pgm->usbsn`. The hurdle is that the function `serial_open()` that actually reads the serial number at runtime has no access to `pgm`, let alone `pgm->usbsn`. Therefore the affected `serial_open()` functions do the second-best thing: storing a static pointer `usbsn` (used in src/usb_libusb.c and src/usb_hidapi.c) to a properly allocated space as intermediate storage, so that microseconds later, once `serial_open()` returns, `pgm->open()`  can copy the static `usbsn` pointer to `pgm->usbsn` where it belongs. It does so through an access funtion called `serno()`.

So, one could get rid of the access function pointer serno by 
  1. Placing the pointer to usbsn into struct serial_device directly (rather than the pointer to the access function) but funnily enough that needs serial_open() to know the serial_device, which I don't think it does except through another(!) global variable `serdev`  
  2. Changing the API for serial_open() to add a pointer `PROGRAMMER *pgm` as first argument (a *lot* of work, possibly upsetting a few people who have been using libavrdude for their projects)

The "correct" solution is 2, and that better waits until 8.0 when we have a better defence for changing APIs. 

I suggest to leave that be for now.

dl8dtl, 2023-01-02:
I thought of (each) `serial_open()` storing the pointer to the string in `struct serdev`, rather than putting an access function there that eventually retrieves the string from a static variable. Later on, the respective programmer code can copy the pointer into its `pgm->usbsn`.
But I agree it's nothing urgent that needs to be changed now.

stefanrueger, 2023-01-02:
OK, that's solution 1 from above; it's easy enough, so will create a PR. The difference is whether one thinks C++ (access function) or C (accessing the component). But it moves the variable from global static to the structures (though that's also a global static)

commit 06e999b8 (2023-01-01): Check whether serial_serno is not NULL before calling it.

For devices connected through a real serial device rather than
USB, there is no transport-layer serial number available.

stk500v2.c (and jtag3.c) tried to always call this function.

Closes #1254