avrdudes/avrdude #1261: `ENABLE_PARPORT` does not enable the respective snippet in `avrdude.conf`

dl8dtl, 2023-01-03:
Tried to run CMake with `-DENABLE_PARPORT=YES`. While this includes the respective code pieces, rewriting `avrdude.conf` from `avrdude.conf.in` still lacks the parport-related programmer definitions.
Didn't check, but the same might apply to other conditional entries in `avrdude.conf.in` (like linuxgpio).

mcuee, 2023-01-03:
@dl8dtl 

For linuxgpio and linuxspi, we have a work-around already. I have updated the Wiki and `build.sh` to address the issue.

Please refer to Issue #1107 and my workaround in PR #1240.
* #1107 
* https://github.com/avrdudes/avrdude/pull/1240

dl8dtl, 2023-01-03:
OK, yes, using "1" seems to work.
I think, in addition to mention it in the Wiki, we might add commented-out examples for `extra_enable` in `build.sh`.

dl8dtl, 2023-01-03:
I see that linuxgpio and linuxspi are already automatically enabled on ARM Linux systems. I think that doesn't need more examples.
I added an example for enabling parport.

closed 2023-01-03 by dl8dtl