avrdudes/avrdude #1263: Official .hex from Atmel doesn't program correctly

cinderblock, 2023-01-03:
I'm programming an ATmega32u4 with `avrdude version 7.0-20221229 (0f956e9)` via `linuxspi` on a Raspberry Pi Zero W:

I'm trying to flash it with the official Atmel DFU bootloader: [`ATMega32U4-usbdevice_dfu-1_0_0.hex`](https://ww1.microchip.com/downloads/aemDocuments/documents/OTH/ProductDocuments/SoftwareLibraries/Firmware/megaUSB_DFU_Bootloaders.zip)

```bash
avrdude -c linuxspi -p atmega32u4 -U flash:w:ATMega32U4-usbdevice_dfu-1_0_0.hex
```

I *always* get a verification mismatch at `0x7090` (device has `0xff`, `.hex` has `0x7f`). At first I assumed it was a bad chip, but then I was able to reproduce on a bunch of devices.

I was able to flash and verify with an older version of `avrdude` (6.3-20171130) with a different programmer.

Once programmed, I'm able to read back the flash from the device and created a "[works.hex](https://github.com/avrdudes/avrdude/files/10340333/works.hex.txt)" _(remove the `.txt` file extension)_.

- The resulting `.hex` file seems to be identical to the official `.hex` (after `hex2bin`)
- The recent version of `avrdude` can verify the device's flash against the `ATMega32U4-usbdevice_dfu-1_0_0.hex`
- The recent version of `avrdude` is able to program and verify correctly with `works.hex`

Curiously, the problematic `.hex` has a bunch of different "blocks" of data. It's clear that one block of data ends after `0x7090`, right where the verification has trouble.

Looking at the raw hex on the device after verification fails, it's written `0xffff` to the device instead of `0xfff7`. So, the verification is working as expected. It's something in the writing process that isn't working as expected.

I suspect this is an issue related to #1068 but this is an even number of bytes... Maybe it's because they started at an even address of `0x702F`? Maybe the fix #1188 caused a regression?

----

I'm setting up a Pi + AVR to enable continuous testing and coverage on/of `dfu-programmer`. I'd be happy to make it available here.

[image]

mcuee, 2023-01-04:
Does this only affect linuxspi or linuxgpio? Does it affect other programmers when you use the later version of avrdude with other programmers?

Maybe your issue is the same as the following.
* https://github.com/avrdudes/avrdude/issues/455

cinderblock, 2023-01-04:
The first 3 lines of the `.hex` are also shorter than `10` but don't seem to cause the issue.

I will try.

mcuee, 2023-01-04:
@stefanrueger 

Please help to review the comments from @cinderblock. Thanks.

mcuee, 2023-01-04:
@cinderblock 

Thanks for the detailed analysis. I have added the `bug` label. 

Just wondering if you only encounter the issue with linuxspi/linuxgpio, or you also encounter the issue with other programmers?

stefanrueger, 2023-01-04:
BTW, your CI integration with the RPi looks *just great*. Vvv cool. @cinderblock  When you get a minute check out the PR #1265 - I'm hoping this solves this problem and quite possibly #455 too.

mcuee, 2023-01-05:
@stefanrueger 

Great. Thanks for the detailed analysis and fast fix. I can confirm that PR #1265 is good to fix both this issue #1263 and the long standing issue #455.

cinderblock, 2023-01-05:
Thanks for the quick fix.

@stefanrueger regarding the CI tool, when I get it running in dfu-programmer, I'll see if I can make it work here too.

One issue right now is that this is a Pi Zero W, the ARMv6 based cpu. The GitHub Action Runner won't run on this hardware because .NET doesn't work on ARMv6, only ARMv7. I'm debating biting the bullet and swapping to a Pi 4 I have... For now just running some simple scripts.

closed 2023-01-05 by cinderblock

mcuee, 2023-01-05:
> @stefanrueger regarding the CI tool, when I get it running in dfu-programmer, I'll see if I can make it work here too.
>
> One issue right now is that this is a Pi Zero W, the ARMv6 based cpu. The [...]

I have two Raspberry Pi 400 and one of them has the extension ribbon cable plus T-shape I/O expansion board setup to test linuxspi/linuxgpio. The ribbon cable setup is not that robust though. It has the USB ports to test other programmers as well.

I also have two Raspberry Pi 3B+ and one of them has a I/O expansion board to test  linuxspi/linuxgpio. It has the USB ports to test other programmers as well.

[image]

cinderblock, 2023-01-10:
I decided I wanted to remake this onto a newer Pi to support GitHub Actions. This turned into a full on hat.

v0 fabrication has started.

[image]

I haven't posed the designs yet, but they will be at https://github.com/cinderblock/avr-test-hat once I clean up and test a couple things.