avrdudes/avrdude #1271: To adopt PR #1264 approach of using Readline replacement for MinGW build

mcuee, 2023-01-07:
PR #1264 has been merged and it performs much better than using GNU Readline under Windows. It is good to adopt the same approach for MinGW.

mcuee, 2023-01-07:
@mariusgreuel 
When you got time, please help to take a look at this issue. Thanks.

mcuee, 2023-03-27:
@mariusgreuel 

I add this as part of the AVRDUDE 7.2 milestone. But you can remove this if you feel that this is of lower priority. Thanks.

mcuee, 2023-10-26:
PR https://github.com/avrdudes/avrdude/pull/1513 (Add GNU Readline to mingw github action) brings in undesired dependency of libwinpthread-1.dll to the mingw32/mingw64 binaries. I have to revert the PR.

It is very strange that there is no such side effect under my own MSYS2 ming64 build.
https://github.com/avrdudes/avrdude/pull/1516#issuecomment-1760768341

Anyway, I think it is better to use the approach used by @mariusgreuel for MSVC to the MinGW build as well since GNU Readline is not that good under Windows.

mcuee, 2023-11-01:
Since @mariusgreuel is busy, I will try to see if I can do something about this one.
* https://github.com/mcuee/avrdude/tree/issue_1271 (WIP)

mcuee, 2023-11-01:
@stefanrueger 

Need your help here as I am stuck at this error. I have zero knowledge on flex/bison.
https://github.com/mcuee/avrdude/actions/runs/6719800195/job/18262102949

```
[ 90%] Building C object src/CMakeFiles/libavrdude.dir/__/lexer.c.obj
lexer.l:31:10: fatal error: avrdude.h: No such file or directory
compilation terminated.
make[2]: *** [src/CMakeFiles/libavrdude.dir/build.make:941: src/CMakeFiles/libavrdude.dir/__/lexer.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:128: src/CMakeFiles/libavrdude.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
Error: Process completed with exit code 2.
```

I believe my changes are quite similar to PR#1264.
https://github.com/mcuee/avrdude/commit/a79daf79946fd58ce3088e8b4c2dd305e23df944
* #1264

stefanrueger, 2023-11-01:
It's a mystery why make/the compiler does not find avrdude.h for lexer.c (that will have been generated from lexer.l by flex). Virtually all other .c sources require avrdude.h too. Maybe sprinkle in a -Isrc or -Isrc/CMakeFiles/libavrdude.dir/all or similar?

mcuee, 2023-11-02:
The complete verbose build log for reference.

```
MINGW64 /c/work/avr/avrdude_test/avrdude_issue1271
$ diff -u build.sh build_verbose.sh
--- build.sh    2023-11-01 20:23:34.098124700 +0800
+++ build_verbose.sh    2023-11-02 08:00:42.764442800 +0800
@@ -37,7 +37,7 @@

 ostype=$(uname | tr '[A-Z]' '[a-z]')

-build_flags=""
+build_flags="-D CMAKE_VERBOSE_MAKEFILE=1 -D DEBUG_CMAKE=1"

 while getopts :hf: OPT; do
   case "$OPT" in
```

<details>
<summary>verbose build log </summary>

```
$ ./build_verbose.sh
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Windows-10.0.22621
-- CMAKE_SYSTEM: Windows-10.0.22621
-- CMAKE_FIND_ROOT_PATH:
-- CMAKE_C_COMPILER: C:/msys64/mingw64/bin/cc.exe
-- CONFIG_DIR: C:/Program Files (x86)/avrdude/etc
-- AVRDUDE_FULL_VERSION: 7.2-20231101 (1ecc5e3e)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: C:/msys64/mingw64/lib/libelf.a
-- HAVE_LIBUSB: C:/msys64/mingw64/lib/libusb.a
-- HAVE_LIBUSB_1_0: C:/msys64/mingw64/lib/libusb-1.0.a
-- HAVE_LIBUSB_WIN32: C:/msys64/mingw64/lib/libusb0.dll.a
-- HAVE_LIBHIDAPI: C:/msys64/mingw64/lib/libhidapi.a
-- HAVE_LIBFTDI: HAVE_LIBFTDI-NOTFOUND
-- HAVE_LIBFTDI1: C:/msys64/mingw64/lib/libftdi1.a
-- HAVE_LIBREADLINE: C:/msys64/mingw64/lib/libreadline.a
-- HAVE_LIBSERIALPORT: 1
-- HAVE_LIBELF_H: 1
-- HAVE_LIBELF_LIBELF_H: 1
-- HAVE_USB_H: 1
-- HAVE_LUSB0_USB_H: 1
-- HAVE_LIBUSB_H:
-- HAVE_LIBUSB_1_0_LIBUSB_H: 1
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR: C:/msys64/mingw64/include/libusb-compat
-- HAVE_LIBGPIOD:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DO HAVE    libserialport
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done (0.3s)
-- Generating done (0.0s)
-- Build files have been written to: C:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621
Change Dir: 'C:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621'

[... 58 lines cut ...]
FAILED: src/CMakeFiles/libavrdude.dir/__/lexer.c.obj
C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_issue1271 -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621 -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_issue1271/src/mingw -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/__/lexer.c.obj -MF src\CMakeFiles\libavrdude.dir\__\lexer.c.obj.d -o src/CMakeFiles/libavrdude.dir/__/lexer.c.obj -c C:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/lexer.c
lexer.l:31:10: fatal error: avrdude.h: No such file or directory
compilation terminated.
[58/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_issue1271 -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621 -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_issue1271/src/mingw -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/__/config_gram.c.obj -MF src\CMakeFiles\libavrdude.dir\__\config_gram.c.obj.d -o src/CMakeFiles/libavrdude.dir/__/config_gram.c.obj -c C:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/config_gram.c
FAILED: src/CMakeFiles/libavrdude.dir/__/config_gram.c.obj
C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_issue1271 -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621 -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_issue1271/src/mingw -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/__/config_gram.c.obj -MF src\CMakeFiles\libavrdude.dir\__\config_gram.c.obj.d -o src/CMakeFiles/libavrdude.dir/__/config_gram.c.obj -c C:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/config_gram.c
C:/work/avr/avrdude_test/avrdude_issue1271/src/config_gram.y:30:10: fatal error: avrdude.h: No such file or directory
   30 | #include "avrdude.h"
      |          ^~~~~~~~~~~
compilation terminated.
[59/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_issue1271 -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621 -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_issue1271/src/mingw -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/usb_libusb.c.obj -MF src\CMakeFiles\libavrdude.dir\usb_libusb.c.obj.d -o src/CMakeFiles/libavrdude.dir/usb_libusb.c.obj -c C:/work/avr/avrdude_test/avrdude_issue1271/src/usb_libusb.c
[60/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_issue1271 -IC:/work/ [... cut ...] IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621 -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_issue1271/src/mingw -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/term.c.obj -MF src\CMakeFiles\libavrdude.dir\term.c.obj.d -o src/CMakeFiles/libavrdude.dir/term.c.obj -c C:/work/avr/avrdude_test/avrdude_issue1271/src/term.c
ninja: build stopped: subcommand failed.

Build failed.
```

</detials>

stefanrueger, 2023-11-02:
Compilation of `lexer.c` misses `-IC:/work/avr/avrdude_test/avrdude_issue1271/src/`
The same for compilation of `config_gram.c`

These `.c` files are created by flex and bison and written to a *different* directory than the one where all other sources are. Hence, on inclusion of "avrdude.h" the compiler won't find it's location.

mcuee, 2023-11-02:
git main verbose build log for comparison.

<details>
<summary> git main verbose build log </summary>

```
MINGW64 /c/work/avr/avrdude_test/avrdude_main
$ ./build_verbose.sh
-- Building for: Ninja
-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/mingw64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/msys64/usr/bin/git.exe (found version "2.42.0")
-- Found FLEX: C:/msys64/usr/bin/flex.exe (found version "2.6.4")
-- Found BISON: C:/msys64/usr/bin/bison.exe (found version "3.8.2")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Windows-10.0.22621
-- CMAKE_SYSTEM: Windows-10.0.22621
-- CMAKE_FIND_ROOT_PATH:
-- CMAKE_C_COMPILER: C:/msys64/mingw64/bin/cc.exe
-- CONFIG_DIR: C:/Program Files (x86)/avrdude/etc
-- AVRDUDE_FULL_VERSION: 7.2-20231030 (2bb36b9b)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: C:/msys64/mingw64/lib/libelf.a
-- HAVE_LIBUSB: C:/msys64/mingw64/lib/libusb.a
-- HAVE_LIBUSB_1_0: C:/msys64/mingw64/lib/libusb-1.0.a
-- HAVE_LIBUSB_WIN32: C:/msys64/mingw64/lib/libusb0.dll.a
-- HAVE_LIBHIDAPI: C:/msys64/mingw64/lib/libhidapi.a
-- HAVE_LIBFTDI: HAVE_LIBFTDI-NOTFOUND
-- HAVE_LIBFTDI1: C:/msys64/mingw64/lib/libftdi1.a
-- HAVE_LIBREADLINE: C:/msys64/mingw64/lib/libreadline.a
-- HAVE_LIBSERIALPORT: 1
-- HAVE_LIBELF_H: 1
-- HAVE_LIBELF_LIBELF_H: 1
-- HAVE_USB_H: 1
-- HAVE_LUSB0_USB_H: 1
-- HAVE_LIBUSB_H:
-- HAVE_LIBUSB_1_0_LIBUSB_H: 1
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR: C:/msys64/mingw64/include/libusb-compat
-- HAVE_LIBGPIOD:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
[... 70 lines cut ...]
[59/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_main -IC:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22621 -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_main/src -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/xbee.c.obj -MF src\CMakeFiles\libavrdude.dir\xbee.c.obj.d -o src/CMakeFiles/libavrdude.dir/xbee.c.obj -c C:/work/avr/avrdude_test/avrdude_main/src/xbee.c
[60/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_main -IC:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22621 -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_main/src -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/__/lexer.c.obj -MF src\CMakeFiles\libavrdude.dir\__\lexer.c.obj.d -o src/CMakeFiles/libavrdude.dir/__/lexer.c.obj -c C:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22621/lexer.c
[61/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_main -IC:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22621 -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_main/src -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/__/config_gram.c.obj -MF src\CMakeFiles\libavrdude.dir\__\config_gram.c.obj.d -o src/CMakeFiles/libavrdude.dir/__/config_gram.c.obj -c C:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22621/config_gram.c
[62/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_main -IC:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22621 -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_main/src -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/usbasp.c.obj -MF src\CMakeFiles\libavrdude.dir\usbasp.c.obj.d -o src/CMakeFiles/libavrdude.dir/usbasp.c.obj -c C:/work/avr/avrdude_test/avrdude_main/src/usbasp.c
[63/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_main -IC:/work/avr/avrdude_test/avrdude_main/build_mingw64_nt-10.0-22621 -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_main/src -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/wiring.c.obj -MF src\CMakeFiles\libavrdude.dir\wiring.c.obj.d -o src/CMa [... cut ...] C:/msys64/mingw64/lib/libusb.a  C:/msys64/mingw64/lib/libusb-1.0.a  C:/msys64/mingw64/lib/libhidapi.a  C:/msys64/mingw64/lib/libftdi1.a  C:/msys64/mingw64/lib/libreadline.a  C:/msys64/mingw64/lib/libserialport.a  C:/msys64/mingw64/lib/libncurses.a  -lsetupapi  -lhid  -lws2_32  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."

Build succeeded.

Run

sudo cmake --build build_mingw64_nt-10.0-22621 --target install

to install.
```

</details>

stefanrueger, 2023-11-02:
Yes, git main has the needed `-IC:/work/avr/avrdude_test/avrdude_main/src` (and it works)

mcuee, 2023-11-02:
> Yes, git main has the needed `-IC:/work/avr/avrdude_test/avrdude_main/src` (and it works)

Thanks for finding the root cause. Now I need to see how to fix it. Probably the CMake scripts need to be updated.

mcuee, 2023-11-02:
**Known Windows specific issues.**

**Bug**
* https://github.com/avrdudes/avrdude/issues/968 (MSVC build only)
* https://github.com/avrdudes/avrdude/issues/1414

**Lack of features compared to Linux and macOS**
MSVC build
* https://github.com/avrdudes/avrdude/issues/1552
* https://github.com/avrdudes/avrdude/discussions/1440

MinGW build
* https://github.com/avrdudes/avrdude/issues/1271

Youw, 2023-11-02:
Short description "why" (why it doesn't work):
`avrdude` CMake scripts is a a **_mess_** :)
It works in main by accident.

Full description:

The target uses `"${EXTRA_WINDOWS_INCLUDES}"` (with quotes) as part of its include directories, and by default (in main) it resolves to `""` which is an empty string and a _separate argument_ to the `target_include_directories` function which is later resolved by CMake as `"${CMAKE_CURRENT_SOURCE_DIR}"` (effectively).

Otherwise `"${CMAKE_CURRENT_SOURCE_DIR}"` (the `<root>/src`) is not added to include path explicitly, so when `EXTRA_WINDOWS_INCLUDES` is set to anything non-empty - `<root>/src` is no longer used.

---
@mcuee if you grant me write permission to your fork - I'll push the fix directly into your branch, I already have it locally and don't want to create another fork of my own.

mcuee, 2023-11-02:
@Youw 

Thanks a lot. I have added you as a collabortor for my avrdude fork. You should be able to push to my fork once you accept the invitation.
https://github.com/mcuee/avrdude

Youw, 2023-11-02:
So the missing includes issue is fixed but now we have another one :)

I don't have mingw locally to try/debug this.
@mcuee  can you run it with CMake verbosity and show the output?

mcuee, 2023-11-02:
@Youw 

Your change is good on my local MSYS2 mingw installation.

<details>
<summary> local build log </summary>

```
MINGW64 /c/work/avr/avrdude_test/avrdude_issue1271
$ ./build_verbose.sh
-- Building for: Ninja
-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/mingw64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/msys64/usr/bin/git.exe (found version "2.42.0")
-- Found FLEX: C:/msys64/usr/bin/flex.exe (found version "2.6.4")
-- Found BISON: C:/msys64/usr/bin/bison.exe (found version "3.8.2")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Windows-10.0.22621
-- CMAKE_SYSTEM: Windows-10.0.22621
-- CMAKE_FIND_ROOT_PATH:
-- CMAKE_C_COMPILER: C:/msys64/mingw64/bin/cc.exe
-- CONFIG_DIR: C:/Program Files (x86)/avrdude/etc
-- AVRDUDE_FULL_VERSION: 7.2-20231102 (6ec6a7b4)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: C:/msys64/mingw64/lib/libelf.a
-- HAVE_LIBUSB: C:/msys64/mingw64/lib/libusb.a
-- HAVE_LIBUSB_1_0: C:/msys64/mingw64/lib/libusb-1.0.a
-- HAVE_LIBUSB_WIN32: C:/msys64/mingw64/lib/libusb0.dll.a
-- HAVE_LIBHIDAPI: C:/msys64/mingw64/lib/libhidapi.a
-- HAVE_LIBFTDI: HAVE_LIBFTDI-NOTFOUND
-- HAVE_LIBFTDI1: C:/msys64/mingw64/lib/libftdi1.a
-- HAVE_LIBREADLINE: C:/msys64/mingw64/lib/libreadline.a
-- HAVE_LIBSERIALPORT: 1
-- HAVE_LIBELF_H: 1
-- HAVE_LIBELF_LIBELF_H: 1
-- HAVE_USB_H: 1
-- HAVE_LUSB0_USB_H: 1
-- HAVE_LIBUSB_H:
-- HAVE_LIBUSB_1_0_LIBUSB_H: 1
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR: C:/msys64/mingw64/include/libusb-compat
-- HAVE_LIBGPIOD:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
[... 72 lines cut ...]
[59/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_issue1271 -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621 -IC:/work/avr/avrdude_test/avrdude_issue1271/src -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_issue1271/src/mingw -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/usb_libusb.c.obj -MF src\CMakeFiles\libavrdude.dir\usb_libusb.c.obj.d -o src/CMakeFiles/libavrdude.dir/usb_libusb.c.obj -c C:/work/avr/avrdude_test/avrdude_issue1271/src/usb_libusb.c
[60/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_issue1271 -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621 -IC:/work/avr/avrdude_test/avrdude_issue1271/src -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_issue1271/src/mingw -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/xbee.c.obj -MF src\CMakeFiles\libavrdude.dir\xbee.c.obj.d -o src/CMakeFiles/libavrdude.dir/xbee.c.obj -c C:/work/avr/avrdude_test/avrdude_issue1271/src/xbee.c
[61/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_issue1271 -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621 -IC:/work/avr/avrdude_test/avrdude_issue1271/src -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_issue1271/src/mingw -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/usbasp.c.obj -MF src\CMakeFiles\libavrdude.dir\usbasp.c.obj.d -o src/CMakeFiles/libavrdude.dir/usbasp.c.obj -c C:/work/avr/avrdude_test/avrdude_issue1271/src/usbasp.c
[62/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_issue1271 -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621 -IC:/work/avr/avrdude_test/avrdude_issue1271/src -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_issue1271/src/mingw -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/__/config_gram.c.obj -MF src\CMakeFiles\libavrdude.dir\__\config_gram.c.obj.d -o src/CMakeFiles/libavrdude.dir/__/config_gram.c.obj -c C:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/config_gram.c
[63/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_issue1271 -IC:/work/avr/avrdude_test/avrdude_issue12 [... cut ...] C:/msys64/mingw64/lib/libusb.a  C:/msys64/mingw64/lib/libusb-1.0.a  C:/msys64/mingw64/lib/libhidapi.a  C:/msys64/mingw64/lib/libftdi1.a  C:/msys64/mingw64/lib/libreadline.a  C:/msys64/mingw64/lib/libserialport.a  C:/msys64/mingw64/lib/libncurses.a  -lsetupapi  -lhid  -lws2_32  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."

Build succeeded.

Run

sudo cmake --build build_mingw64_nt-10.0-22621 --target install

to install.
```

</details>

mcuee, 2023-11-02:
@Youw 
But my build links to `C:/msys64/mingw64/lib/libreadline.a`. let me remove libreadline on my local machine and try again.

Youw, 2023-11-02:
Do you even want the possibility to long against system-provided libreadline?
Maybe use only embedded msvc/mingw implementation for corresponding systems?

mcuee, 2023-11-02:
Here is the verbose log.

<details>
<summary> failed build log </summary>

```
MINGW64 /c/work/avr/avrdude_test/avrdude_issue1271
$ ./build_verbose.sh
-- Building for: Ninja
-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/mingw64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/msys64/usr/bin/git.exe (found version "2.42.0")
-- Found FLEX: C:/msys64/usr/bin/flex.exe (found version "2.6.4")
-- Found BISON: C:/msys64/usr/bin/bison.exe (found version "3.8.2")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Windows-10.0.22621
-- CMAKE_SYSTEM: Windows-10.0.22621
-- CMAKE_FIND_ROOT_PATH:
-- CMAKE_C_COMPILER: C:/msys64/mingw64/bin/cc.exe
-- CONFIG_DIR: C:/Program Files (x86)/avrdude/etc
-- AVRDUDE_FULL_VERSION: 7.2-20231102 (1c5898dd)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: C:/msys64/mingw64/lib/libelf.a
-- HAVE_LIBUSB: C:/msys64/mingw64/lib/libusb.a
-- HAVE_LIBUSB_1_0: C:/msys64/mingw64/lib/libusb-1.0.a
-- HAVE_LIBUSB_WIN32: C:/msys64/mingw64/lib/libusb0.dll.a
-- HAVE_LIBHIDAPI: C:/msys64/mingw64/lib/libhidapi.a
-- HAVE_LIBFTDI: HAVE_LIBFTDI-NOTFOUND
-- HAVE_LIBFTDI1: C:/msys64/mingw64/lib/libftdi1.a
-- HAVE_LIBREADLINE: Embedded MINGW
-- HAVE_LIBSERIALPORT: 1
-- HAVE_LIBELF_H: 1
-- HAVE_LIBELF_LIBELF_H: 1
-- HAVE_USB_H: 1
-- HAVE_LUSB0_USB_H: 1
-- HAVE_LIBUSB_H:
-- HAVE_LIBUSB_1_0_LIBUSB_H: 1
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR: C:/msys64/mingw64/include/libusb-compat
-- HAVE_LIBGPIOD:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
[... 80 lines cut ...]
[69/72] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621/src -IC:/work/avr/avrdude_test/avrdude_issue1271 -IC:/work/avr/avrdude_test/avrdude_issue1271/build_mingw64_nt-10.0-22621 -IC:/work/avr/avrdude_test/avrdude_issue1271/src -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_issue1271/src/mingw -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/libavrdude.dir/term.c.obj -MF src\CMakeFiles\libavrdude.dir\term.c.obj.d -o src/CMakeFiles/libavrdude.dir/term.c.obj -c C:/work/avr/avrdude_test/avrdude_issue1271/src/term.c
[70/72] cmd.exe /C "cd . && C:\msys64\mingw64\bin\cmake.exe -E rm -f src\libavrdude.a && C:\msys64\mingw64\bin\ar.exe qc src\libavrdude.a  src/CMakeFiles/libavrdude.dir/arduino.c.obj src/CMakeFiles/libavrdude.dir/avr.c.obj src/CMakeFiles/libavrdude.dir/avr910.c.obj src/CMakeFiles/libavrdude.dir/avrcache.c.obj src/CMakeFiles/libavrdude.dir/avrftdi.c.obj src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.obj src/CMakeFiles/libavrdude.dir/avrintel.c.obj src/CMakeFiles/libavrdude.dir/avrpart.c.obj src/CMakeFiles/libavrdude.dir/bitbang.c.obj src/CMakeFiles/libavrdude.dir/buspirate.c.obj src/CMakeFiles/libavrdude.dir/butterfly.c.obj src/CMakeFiles/libavrdude.dir/ch341a.c.obj src/CMakeFiles/libavrdude.dir/config.c.obj src/CMakeFiles/libavrdude.dir/confwin.c.obj src/CMakeFiles/libavrdude.dir/crc16.c.obj src/CMakeFiles/libavrdude.dir/dfu.c.obj src/CMakeFiles/libavrdude.dir/dryrun.c.obj src/CMakeFiles/libavrdude.dir/fileio.c.obj src/CMakeFiles/libavrdude.dir/flip1.c.obj src/CMakeFiles/libavrdude.dir/flip2.c.obj src/CMakeFiles/libavrdude.dir/ft245r.c.obj src/CMakeFiles/libavrdude.dir/jtagmkI.c.obj src/CMakeFiles/libavrdude.dir/jtagmkII.c.obj src/CMakeFiles/libavrdude.dir/jtag3.c.obj src/CMakeFiles/libavrdude.dir/leds.c.obj src/CMakeFiles/libavrdude.dir/linuxgpio.c.obj src/CMakeFiles/libavrdude.dir/linuxspi.c.obj src/CMakeFiles/libavrdude.dir/lists.c.obj src/CMakeFiles/libavrdude.dir/micronucleus.c.obj src/CMakeFiles/libavrdude.dir/par.c.obj src/CMakeFiles/libavrdude.dir/pgm.c.obj src/CMakeFiles/libavrdude.dir/pgm_type.c.obj src/CMakeFiles/libavrdude.dir/pickit2.c.obj src/CMakeFiles/libavrdude.dir/pindefs.c.obj src/CMakeFiles/libavrdude.dir/ppi.c.obj src/CMakeFiles/libavrdude.dir/ppiwin.c.obj src/CMakeFiles/libavrdude.dir/serbb_posix.c.obj src/CMakeFiles/libavrdude.dir/serbb_win32.c.obj src/CMakeFiles/libavrdude.dir/ser_avrdoper.c.obj src/CMakeFiles/libavrdude.dir/ser_posix.c.obj src/CMakeFiles/libavrdude.dir/ser_win32.c.obj src/CMakeFiles/libavrdude.dir/serialadapter.c.obj src/CMakeFiles/libavrdude.dir/serialupdi.c.obj src/CMakeFiles/libavrdude.dir/stk500.c.obj src/CMakeFiles/libavrdude.dir/stk500v2.c.obj src/CMakeFiles/libavrdude.dir/stk500generic.c.obj src/CMakeFiles/libavrdude.dir/strutil.c.obj src/CMakeFiles/libavrdude.dir/teensy.c.obj src/CMakeFiles/libavrdude.dir/term.c.obj src/CMakeFiles/libavrdude.dir/updi_link.c.obj src/CMakeFiles/libavrdude.dir/updi_nvm.c.obj src/CMakeFiles/libavrdude.dir/updi_readwri [... cut ...]
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/work/avr/avrdude_test/avrdude_issue1271/src/term.c:2163:(.text+0x7b65): undefined reference to `rl_callback_handler_remove'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/libavrdude.a(term.c.obj):term.c:(.rdata$.refptr.rl_readline_version[.refptr.rl_readline_version]+0x0): undefined reference to `rl_readline_version'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

Build failed.
```

</details>

mcuee, 2023-11-02:
We are almost there. I think we may need to change some ifdef in the file `term.c`.

mcuee, 2023-11-02:
Just one more verbose log for reference (MSYS2 clang64 build)

<details>
<summary> verbose build log for MSYS2 clang64 </summary>

```
CLANG64 /c/work/avr/avrdude_test/avrdude_issue1271
$ ./build_verbose.sh
-- Building for: Ninja
-- The C compiler identification is Clang 17.0.2
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/clang64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/msys64/usr/bin/git.exe (found version "2.42.0")
-- Found FLEX: C:/msys64/usr/bin/flex.exe (found version "2.6.4")
-- Found BISON: C:/msys64/usr/bin/bison.exe (found version "3.8.2")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Windows-6.2.9200
-- CMAKE_SYSTEM: Windows-6.2.9200
-- CMAKE_FIND_ROOT_PATH:
-- CMAKE_C_COMPILER: C:/msys64/clang64/bin/cc.exe
-- CONFIG_DIR: C:/Program Files (x86)/avrdude/etc
-- AVRDUDE_FULL_VERSION: 7.2-20231102 (1c5898dd)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: C:/msys64/clang64/lib/libelf.a
-- HAVE_LIBUSB: C:/msys64/clang64/lib/libusb.a
-- HAVE_LIBUSB_1_0: C:/msys64/clang64/lib/libusb-1.0.a
-- HAVE_LIBUSB_WIN32: C:/msys64/clang64/lib/libusb0.dll.a
-- HAVE_LIBHIDAPI: C:/msys64/clang64/lib/libhidapi.a
-- HAVE_LIBFTDI: HAVE_LIBFTDI-NOTFOUND
-- HAVE_LIBFTDI1: C:/msys64/clang64/lib/libftdi1.a
-- HAVE_LIBREADLINE: Embedded MINGW
-- HAVE_LIBSERIALPORT: 1
-- HAVE_LIBELF_H: 1
-- HAVE_LIBELF_LIBELF_H: 1
-- HAVE_USB_H: 1
-- HAVE_LUSB0_USB_H: 1
-- HAVE_LIBUSB_H:
-- HAVE_LIBUSB_1_0_LIBUSB_H: 1
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR: C:/msys64/clang64/include/libusb-compat
-- HAVE_LIBGPIOD:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
[... 103 lines cut ...]
>>>               libavrdude.a(term.c.obj):(term_gotline)

ld.lld: error: undefined symbol: rl_callback_read_char
>>> referenced by C:/work/avr/avrdude_test/avrdude_issue1271/src/term.c:2182
>>>               libavrdude.a(term.c.obj):(terminal_mode)

ld.lld: error: undefined symbol: add_history
>>> referenced by C:/work/avr/avrdude_test/avrdude_issue1271/src/term.c:2139
>>>               libavrdude.a(term.c.obj):(term_gotline)

ld.lld: error: undefined symbol: rl_callback_handler_remove
>>> referenced by C:/work/avr/avrdude_test/avrdude_issue1271/src/term.c:2153
>>>               libavrdude.a(term.c.obj):(term_gotline)
>>> referenced by C:/work/avr/avrdude_test/avrdude_issue1271/src/term.c:2163
>>>               libavrdude.a(term.c.obj):(term_gotline)

ld.lld: error: undefined symbol: rl_readline_version
>>> referenced by libavrdude.a(term.c.obj):(.refptr.rl_readline_version)
cc: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

Build failed.
```

</details>

Youw, 2023-11-02:
No, not that.
`mingw/readline.cpp` wasn't built at all. I was woudering [why](https://github.com/avrdudes/avrdude/commit/c5583cb990229a28a035ebde3af353e43b3e0368)...

mcuee, 2023-11-02:
> > I think we may need to change some ifdefs in the file term.c
> 
> No, not that. `mingw/readline.cpp` wasn't built at all. I was woudering [why](https://github.com/avrdudes/avrdude/commit/c5583cb990229a28a035ebde3af353e43b3e0368)...

Wonderful. Now it is a success.

Youw, 2023-11-02:
I'l unsubscribing from this conversation now.
If anything - tag me explicitly again.

mcuee, 2023-11-03:
@stefanrueger and @Youw 

One potential side effect of using tje C++ `readline.cpp` is that this may bring more dependancies (GCC C++ related run time files) to MinGW build. It is not a real problem for MSVC build but it may be not so desired with MinGW.

Just wondering how difficult it is to rewrite `readline.cpp` to `readline.c`. I am for sure not up to the task but I think maybe you can sort it out fast.

```
MINGW64 /c/work/avr/avrdude_test/avrdude_bin/github_bin/pr1553/avrdude-mingw-x86_64
$ ldd avrdude.exe
        ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ff8f8590000)
        KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ff8f62f0000)
        KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ff8f5ec0000)
        ADVAPI32.dll => /c/WINDOWS/System32/ADVAPI32.dll (0x7ff8f68e0000)
        msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ff8f69b0000)
        sechost.dll => /c/WINDOWS/System32/sechost.dll (0x7ff8f7f60000)
        RPCRT4.dll => /c/WINDOWS/System32/RPCRT4.dll (0x7ff8f8430000)
        SETUPAPI.dll => /c/WINDOWS/System32/SETUPAPI.dll (0x7ff8f77f0000)
        WS2_32.dll => /c/WINDOWS/System32/WS2_32.dll (0x7ff8f6850000)
        HID.DLL => /c/WINDOWS/SYSTEM32/HID.DLL (0x7ff8f4120000)
        libgcc_s_seh-1.dll => /mingw64/bin/libgcc_s_seh-1.dll (0x7ff8dd950000)
        libwinpthread-1.dll => /mingw64/bin/libwinpthread-1.dll (0x1b97c320000)
        libwinpthread-1.dll => /mingw64/bin/libwinpthread-1.dll (0x7ff8dd930000)
        libstdc++-6.dll => /mingw64/bin/libstdc++-6.dll (0x7ff87e500000)
        cfgmgr32.DLL => /c/WINDOWS/SYSTEM32/cfgmgr32.DLL (0x7ff8f5580000)
        ucrtbase.dll => /c/WINDOWS/System32/ucrtbase.dll (0x7ff8f5da0000)
```

https://github.com/avrdudes/avrdude/blob/35f735c3124dc6aa7631323e6f79892c8a399bb1/src/mingw/readline.cpp
```
//
// readline.cpp
// Copyright (C) 2022 Marius Greuel
// SPDX-License-Identifier: GPL-2.0-or-later
//

#include <string.h>
#include <iostream>
#include <memory>
#include <mutex>
#include <string>
#include <thread>
#include "readline/readline.h"
#include "readline/history.h"

int rl_readline_version = 0x0502;

static rl_vcpfunc_t* rl_handler;
static std::unique_ptr<std::thread> rl_thread;
static std::mutex rl_mutex;
static std::string rl_line;
static bool rl_has_line = false;

static void get_line_thread()
{
    std::string line;
    std::getline(std::cin, line);

    const std::lock_guard<std::mutex> lock(rl_mutex);
    rl_line = line;
    rl_has_line = true;
}

static void call_handler(const char* string)
{
    if (rl_thread)
    {
        rl_thread->join();
        rl_thread = nullptr;
    }

    if (rl_handler != nullptr)
    {
        if (string == nullptr)
        {
            rl_handler(nullptr);
        }
        else
        {
            rl_handler(_strdup(string));
        }
    }
}

int rl_input_available(void)
{
    return 1;
}

void rl_callback_read_char(void)
{
    if (std::cin.eof())
    {
        call_handler(nullptr);
    }
    else if (!rl_thread)
    {
        rl_thread = std::make_unique<std::thread>(get_line_thread);
    }
    else
    {
        const std::lock_guard<std::mutex> lock(rl_mutex);
        if (rl_has_line)
        {
            rl_has_line = false;
            call_handler(rl_line.c_str());
        }
    }
}

void rl_callback_handler_install(char* prompt, rl_vcpfunc_t* handler)
{
    rl_handler = handler;

    std::cout << prompt;
}

void rl_callback_handler_remove(void)
{
    rl_handler = nullptr;
}

void add_history(const char*)
{
}
```

Youw, 2023-11-03:
Lets discuss this as part fo #1553

mcuee, 2024-01-29:
After the merging of #1576, this issue becomes less important since GNU Readline will work in most cases for the MinGW build. It is still nice to have but can be of lower priority.

mcuee, 2024-03-03:
Close this one as not planned until someone is actibely working on this issue.

This issue is also linked here.
https://github.com/avrdudes/avrdude/discussions/1710

closed 2024-03-03 by mcuee