avrdudes/avrdude #1287: Add useful `-xhelp` to all programmers that has extended options

MCUdude, 2023-01-14:
At the moment, there's no way to know which programmer support which extended options without closely reading the manual.
It would be _very_ useful with a `-xhelp` message to list all available extended options for a programmer with a little description, just like with `-curclock`:

```
$ ./avrdude -curclock -xhelp
avrdude -c urclock extended options:
  -xshowall         Show all info for connected part and exit
  -xshowid          Show Urclock ID and exit
  -xshowdate        Show last-modified date of flash application and exit
  -xshowfilename    Show filename of last uploaded application and exit
  -xshowapp         Show application size and exit
  -xshowstore       Show store size and exit
  -xshowmeta        Show metadata size and exit
  -xshowboot        Show bootloader size and exit
  -xshowversion     Show bootloader version and capabilities and exit
  -xshowvector      Show vector bootloader vector # and name and exit
  -xid=<arg>        Location of Urclock ID, eg, F.12345.6
  -xtitle=<arg>     Title stored and shown in lieu of a filename
  -xbootsize=<arg>  Override/set bootloader size
  -xvectornum=<arg> Treat bootloader as vector b/loader using this vector
  -xeepromrw        Assert bootloader EEPROM read/write capability
  -xemulate_ce      Emulate chip erase
  -xrestore         Restore a flash backup as is trimming the bootloader
  -xinitstore       Fill store with 0xff on writing to flash
  -xnofilename      Do not store filename on writing to flash
  -xnodate          Do not store application filename and no date either
  -xnometadata      Do not store metadata at all (ie, no store support)
  -xdelay=<arg>     Add delay [ms] after reset, can be negative
  -xstrict          Use strict synchronisation protocol
  -xhelp            Show this help menu and exit
```

stefanrueger, 2023-01-15:
Urclock.c uses a [table](https://github.com/avrdudes/avrdude/blob/5ec72e240b63874fd72b556f804ad834d420f08f/src/urclock.c#L2442-L2470) for parsing. When I have some time, I could review that approach and perhaps farm out a library function that 
 - Takes a similar, more descriptive table with valid -x parameters and a short description as input,
 - Parses the extended parameter list in always the same, expected AVRDUDE-style way
 - Gives a systematic help info

Then someone with knowledge of programmers (@MCUdude? @mcuee?) could draft the parameter tables for use with that generic parser function, rewrite the extended-parameter parsing functions to use that library function.

Deal?

MCUdude, 2023-01-15:
> Then someone with knowledge of programmers (@MCUdude? @mcuee?) could draft the parameter tables for use with that generic parser function, rewrite the extended-parameter parsing functions to use that library function.

With a little help and review from you, I think this should be doable. My time is a bit limited at the moment, but I have some evenings available that I can spend on this.

MCUdude, 2023-03-30:
> Then someone with knowledge of programmers (@MCUdude? @mcuee?) could draft the parameter tables for use with that generic parser function, rewrite the extended-parameter parsing functions to use that library function.

@stefanrueger I've just pushed a branch where I've added `-xhelp` to most official AVR programmers. It is no way near as elegant as your -xhelp Urclock implementation, but it is a start. Maybe this can be used as a base for a more general approach.

https://github.com/MCUdude/avrdude/tree/xhelp

stefanrueger, 2023-04-02:
Neat! I have had a (brief!) look. I like that you have chosen the most difficult cases first: ones where there are different derivative programmers with different options and help texts. I might have thought about how to implement parsing (including a generic `-xhelp`) through tables and some data-driven, flexible processing, but looking at all the special cases that you have uncovered here I am no longer sure that my initial way of approaching this is better (or appropriate or even possible for that matter). Your approach does the job, and does the job well (disclaimer: I have not had time for a thorough review). So, yes, seems like you get this problem sorted and this is good.

MCUdude, 2023-04-03:
OK, I'll continue working on the xhelp implementation and will submit a PR once I think it's ready.
Maybe I'll even add str.c/str.h to make all the string-checking neater.

MCUdude, 2023-04-03:
> Maybe I'll even add str.c/str.h to make all the string-checking neater.

Or perhaps util(s).c/h, so we can have all "utility functions" stored in one place?

MCUdude, 2023-04-03:
I've pushed a few commits to my `xhelp` branch where I've added `strutil.c/h`. All programmers that support extended parameters now have `-xhelp` support, except for the Bus Pirate. Will add this very soon

MCUdude, 2023-04-04:
I've now added `-xhelp` to the Bus Pirate as well. I'll have to wait until #1331 is merged to update the docs and submit a PR.

stefanrueger, 2023-04-04:
> update the docs

Great plan!

What should be done about programmers without `-x` parameters, ie, how would a user know a programmer has parameters and therefore can enquire `-xhelp`?

If you put `strutils.c` into libavrdude (which you should), please remember to put the function declarations into `libavrdude.h` as is customary so users of that library only have to include one header `libavrdude.h`. (And as I am sure you will anyway, double the new files are in desired format without trailing spaces etc).

MCUdude, 2023-04-04:
Do you mean removing strutil.h, and moving the function declarations into libavrdude.h?
The function definitions needs to be stored somewhere, for instance in strutil.c.

stefanrueger, 2023-04-04:
> moving the function declarations into `libavrdude.h`

Yes, `strutils.h` is not needed.

MCUdude, 2023-04-04:
I've just moved the strutil declarations into libavrdude.h and deleted strutil.h, but for some reason, Avrdude doesn't build on Windows anymore. Not sure how I can have broken this...

MCUdude, 2023-04-04:
Maybe @mariusgreuel knows why my `xhelp` branch fails when building for Windows?
https://github.com/MCUdude/avrdude/actions/runs/4610084893/jobs/8148110051

stefanrueger, 2023-04-05:
> doesn't build on Windows anymore

Try to `#include "ac_cfg.h"` first thing in `strutils.c`

closed 2023-04-17 by stefanrueger